intel/ipu6-drivers #5: intel_ipu6 fails to load with "CSE boot_load failed"

lsuchocki, 2022-01-31:
After attempting to get the IPU6 driver working on a Gigabyte/GigaIPC Jasper Lake board iTXL-4500A (aka MJPLNAT-SI),
We've run into an issue where the driver fails to authenticate the firmware with "CSE boot_load failed".

The IPC response is 0x820, which does not indicate too much, considering there is no documentation of the registers outside of the driver.

The driver is also unable set the IPU to "non-secure mode" (with option "secure_mode_enable=N"), so the assumption is either the BIOS or an eFuse has disabled this functionality.

We HAVE been able to successfully get the driver working, with camera video, on a different system (a "Designed for Windows" Asus BR1100 notebook, Same N4500 SoC), after patching ACPI to create the proper MIPI port/endpoint links, using the same kernel, driver and firmware files.

Seeing as we had to ask Gigabyte to re-spin the BIOS to enable the IPU as a discoverable PCI device, is there something that may still missing (such as a firmware certificate perhaps?) which the BIOS must set before an OS can successfully initialize the device?

Linux kernel:
5.16.0-rc7 (latest at the time, from Kernel.org)

Driver:
"JSL penguin peak" branch from:
https://github.com/intel/ipu6-drivers/tree/jsl_penguin_peak

GigaIPC board information:
https://www.gigaipc.com/index.php?action=products3&cid=1&tid=3&pid=141&utm_source=gigaipc&utm_medium=product&utm_campaign=iTXL-4500A
https://www.gigaipc.com/filedown.php?file=iTXL-4500A_20211112.pdf
https://www.gigaipc.com/filedown.php?file=Manual-iTXL-5105A_4500A_20211124.pdf

dmesg output:
intel-ipu6 intel-ipu: Device 0x4e19 (rev: 0x0)
intel-ipu6 intel-ipu: physical base address 0x6000000000
intel-ipu6 intel-ipu: mapped as: 0x00000000a33195d7
intel_ipu6:ipu_pci_probe:440: intel-ipu6 intel-ipu: isys_base: 0xffffbc2683000000
intel_ipu6:ipu_pci_probe:441: intel-ipu6 intel-ipu: psys_base: 0xffffbc2683000000
intel-ipu6 intel-ipu: IPU in secure mode
intel_ipu6:ipu_buttress_ipc_reset:111: intel-ipu6 intel-ipu: ipu_buttress_ipc_reset:IPC_PEER_COMP_ACTIONS_RST_PHASE1
intel_ipu6:ipu_buttress_ipc_reset:95: intel-ipu6 intel-ipu: ipu_buttress_ipc_reset:IPC_PEER_COMP_ACTIONS_RST_PHASE1 & IPC_PEER_COMP_ACTIONS_RST_PHASE2
intel_ipu6:ipu_buttress_ipc_reset:126: intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: IPC_PEER_COMP_ACTIONS_RST_PHASE2
intel-ipu6 intel-ipu: IPC reset done
intel-ipu6 intel-ipu: cpd file name: intel/ipu6se_fw.bin
intel-ipu6 intel-ipu: FW version: 20210414
intel_ipu6:get_dummy_page:136: intel-ipu6 intel-ipu: get_dummy_page get_zeroed_page() == 000000000224262f
intel_ipu6:alloc_dummy_l2_pt:171: intel-ipu6 intel-ipu: alloc_dummy_l2_pt get_zeroed_page() == 00000000fc27c7d6
intel_ipu6:alloc_l1_pt:209: intel-ipu6 intel-ipu: alloc_l1_pt get_zeroed_page() == 00000000ccd35858
intel_ipu6:alloc_l1_pt:221: intel-ipu6 intel-ipu: l1 pt 00000000ccd35858 mapped at 1144000
intel_ipu6:ipu_mmu_alloc:550: intel-ipu6 intel-ipu: domain initialised
intel_ipu6:alloc_dma_mapping:596: intel-ipu6 intel-ipu: alloc mapping
intel_ipu6:bus_pm_runtime_resume:51: intel-ipu6-bus intel-ipu6-isys0: bus_pm_runtime_resume: buttress power up 0
intel_ipu6:bus_pm_runtime_resume:56: intel-ipu6-bus intel-ipu6-isys0: bus_pm_runtime_resume: resume 0
intel_ipu6:get_dummy_page:136: intel-ipu6 intel-ipu: get_dummy_page get_zeroed_page() == 0000000052fe013a
intel_ipu6:alloc_dummy_l2_pt:171: intel-ipu6 intel-ipu: alloc_dummy_l2_pt get_zeroed_page() == 00000000bd019e0d
intel_ipu6:alloc_l1_pt:209: intel-ipu6 intel-ipu: alloc_l1_pt get_zeroed_page() == 000000005102976f
intel_ipu6:alloc_l1_pt:221: intel-ipu6 intel-ipu: l1 pt 000000005102976f mapped at 1143000
intel_ipu6:ipu_mmu_alloc:550: intel-ipu6 intel-ipu: domain initialised
intel_ipu6:bus_pm_runtime_suspend:31: intel-ipu6-bus intel-ipu6-isys0: bus_pm_runtime_suspend: buttress power down 0
intel_ipu6:alloc_dma_mapping:596: intel-ipu6 intel-ipu: alloc mapping
intel_ipu6:bus_pm_runtime_resume:51: intel-ipu6-bus intel-ipu6-isys0: bus_pm_runtime_resume: buttress power up 0
intel_ipu6:bus_pm_runtime_resume:56: intel-ipu6-bus intel-ipu6-isys0: bus_pm_runtime_resume: resume 0
intel_ipu6:bus_pm_runtime_resume:51: intel-ipu6-bus intel-ipu6-psys0: bus_pm_runtime_resume: buttress power up 0
intel_ipu6:bus_pm_runtime_resume:56: intel-ipu6-bus intel-ipu6-psys0: bus_pm_runtime_resume: resume 0
intel_ipu6:ipu_mmu_hw_init:442: intel-ipu6-bus intel-ipu6-psys0: mmu hw init
intel_ipu6:alloc_l2_pt:238: intel-ipu6 intel-ipu: alloc_l2_pt get_zeroed_page() == 00000000c8a1b540
intel_ipu6:l2_map:284: intel-ipu6 intel-ipu: page for l1_idx 1022 00000000c8a1b540 allocated
intel_ipu6:alloc_l2_pt:238: intel-ipu6 intel-ipu: alloc_l2_pt get_zeroed_page() == 00000000f50de5a8
intel_ipu6:l2_map:284: intel-ipu6 intel-ipu: page for l1_idx 1023 00000000f50de5a8 allocated
intel_ipu6:allocate_trash_buffer:422: intel-ipu6-bus intel-ipu6-psys0: iova trash buffer for MMUID: 0 is 4286578688
intel_ipu6:ipu_dma_map_sg:319: intel-ipu6-bus intel-ipu6-psys0: dmamap: mapping sg 14 entries, 93 pages
intel_ipu6:ipu_dma_map_sg:326: intel-ipu6-bus intel-ipu6-psys0: dmamap: iova low pfn 1046435, high pfn 1046527
intel_ipu6:alloc_l2_pt:238: intel-ipu6 intel-ipu: alloc_l2_pt get_zeroed_page() == 0000000086728e3e
intel_ipu6:l2_map:284: intel-ipu6 intel-ipu: page for l1_idx 1021 0000000086728e3e allocated
intel-ipu6 intel-ipu: Sending BOOT_LOAD to CSE
intel_ipu6:ipu_buttress_ipc_send_bulk:261: intel-ipu6 intel-ipu: bulk IPC command: 0x1
intel_ipu6:ipu_buttress_isr:425: intel-ipu6 intel-ipu: BUTTRESS_ISR_IPC_EXEC_DONE_BY_CSE
intel_ipu6:ipu_buttress_isr:411: intel-ipu6 intel-ipu: BUTTRESS_ISR_IPC_FROM_CSE_IS_WAITING
intel-ipu6 intel-ipu: **expected resp: 0x1, IPC response: 0x820**
intel-ipu6 intel-ipu: **CSE boot_load failed**
intel-ipu6 intel-ipu: FW authentication failed(-5)
intel_ipu6:bus_pm_runtime_suspend:31: intel-ipu6-bus intel-ipu6-psys0: bus_pm_runtime_suspend: buttress power down 0
intel_ipu6:bus_pm_runtime_suspend:31: intel-ipu6-bus intel-ipu6-isys0: bus_pm_runtime_suspend: buttress power down 0
intel-ipu6: probe of intel-ipu failed with error -5

lsuchocki, 2022-02-03:
Turns out I was indeed missing the IPU bootloader firmware within the motherboard's firmware flash.
After using Intel's Flash Image Tool (FIT) to add the "ipu6se_bootloader.bin" file (found in the "chrome_jsl_release" branch of ipu6-camera-bins repo) to the "IUnit Sub-Partition" of the flash image, the driver was able to successfully be loaded and authenticate the main firmware.

closed 2022-02-03 by lsuchocki

junocomp, 2022-11-21:
@lsuchocki were you able to get the camera working?

lsuchocki, 2022-11-21:
@junocomp Somewhat. I was finally receiving data from the sensor/IPU that looked like it could have been an image (displayed pixels changed when you wave in front of the sensor), but was pulled off the project before I had a chance to convert it to something that looks "nice" on the screen.  Either way, it was a milestone considering the lack of implementation documentation and Intel refusing to support us with Linux not being on the embedded roadmap for this CPU/image processor combo.

junocomp, 2022-11-21:
@lsuchocki mmm... anyway you can share how you got that far?

lsuchocki, 2022-11-21:
Sure, but this might not be the best place to do that; I've sent you a message via your company's 'Contact Us' form on your website. (I think.)