intel/ipu6-drivers #13: build failure: missing linux/vsc.h

vicamo, 2022-04-27:
```
$ make -j EXTERNAL_BUILD=1
make -C /lib/modules/`uname -r`/build M=/work/vicamo/canonical/dkms/ipu6-drivers/ipu6-drivers modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-25-generic'
  CC [M]  /work/vicamo/canonical/dkms/ipu6-drivers/ipu6-drivers/drivers/media/i2c/ov01a10.o
  CC [M]  /work/vicamo/canonical/dkms/ipu6-drivers/ipu6-drivers/drivers/media/i2c/ov02c10.o
/work/vicamo/canonical/dkms/ipu6-drivers/ipu6-drivers/drivers/media/i2c/ov02c10.c:14:10: fatal error: linux/vsc.h: No such file or directory
   14 | #include <linux/vsc.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.build:285: /work/vicamo/canonical/dkms/ipu6-drivers/ipu6-drivers/drivers/media/i2c/ov02c10.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/work/vicamo/canonical/dkms/ipu6-drivers/ipu6-drivers/drivers/media/i2c/ov01a10.c:14:10: fatal error: linux/vsc.h: No such file or directory
   14 | #include <linux/vsc.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.build:285: /work/vicamo/canonical/dkms/ipu6-drivers/ipu6-drivers/drivers/media/i2c/ov01a10.o] Error 1
make[2]: *** [scripts/Makefile.build:548: /work/vicamo/canonical/dkms/ipu6-drivers/ipu6-drivers/drivers/media/i2c] Error 2
make[1]: *** [Makefile:1875: /work/vicamo/canonical/dkms/ipu6-drivers/ipu6-drivers] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-25-generic'
make: *** [Makefile:22: all] Error 2
```

vicamo, 2022-04-27:
How about moving ov01a10 and ov02c10 into ivsc-drivers? I didn't find there is a directly dependency to ipu6-drivers but ivsc.

kfechter, 2022-04-28:
I added the vsc,h from the ivsc-drivers repo, which gets the build past that error. however, there is another error right after

` LD [M]  /home/kfechter/Downloads/ipu6-drivers-master/drivers/media/pci/intel/ipu6/intel-ipu6.o
  LD [M]  /home/kfechter/Downloads/ipu6-drivers-master/drivers/media/pci/intel/ipu6/intel-ipu6-isys.o
  LD [M]  /home/kfechter/Downloads/ipu6-drivers-master/drivers/media/pci/intel/ipu6/intel-ipu6-psys.o
  MODPOST /home/kfechter/Downloads/ipu6-drivers-master/Module.symvers
ERROR: modpost: "vsc_release_camera_sensor" [/home/kfechter/Downloads/ipu6-drivers-master/drivers/media/i2c/ov02c10.ko] undefined!
ERROR: modpost: "vsc_acquire_camera_sensor" [/home/kfechter/Downloads/ipu6-drivers-master/drivers/media/i2c/ov02c10.ko] undefined!
ERROR: modpost: "vsc_release_camera_sensor" [/home/kfechter/Downloads/ipu6-drivers-master/drivers/media/i2c/ov01a10.ko] undefined!
ERROR: modpost: "vsc_acquire_camera_sensor" [/home/kfechter/Downloads/ipu6-drivers-master/drivers/media/i2c/ov01a10.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:134: /home/kfechter/Downloads/ipu6-drivers-master/Module.symvers] Error 1
make[2]: *** Deleting file '/home/kfechter/Downloads/ipu6-drivers-master/Module.symvers'
make[1]: *** [Makefile:1750: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.17.0-1003-oem'
make: *** [Makefile:22: all] Error 2`

cjechlitschek, 2022-04-28:
At this moment, the vsc driver needs to be built as part of the kernel so that the sensor drivers can link against it.

vicamo, 2022-04-29:
> At this moment, the vsc driver needs to be built as part of the kernel so that the sensor drivers can link against it.

I'm afraid that's not gonna happen any more as Intel has no interest in upstreaming this any time soon. It has to be a standaline dkms. That's all.

So this, missing `client.h`/`hbm.h`/... problem, can be fixed by carrying those header files inside ivsc-drivers, then that's what has to be done. And since now ivsc [fails to build against v5.18-rc3 or newer](https://github.com/intel/ivsc-driver/issues/10) due to DMA API changes, you'll need some kind of version guards like #12 as well.

vicamo, 2022-04-29:
> > At this moment, the vsc driver needs to be built as part of the kernel so that the sensor drivers can link against it.
> 
> I'm afraid that's not gonna happen any more as Intel has no interest in upstreaming this any time soon. It has to be a standaline dkms. That's all.

And please note we are expecting your commitment into CI tests for _both_ ipu6-drivers and ivsc-drivers.

wentongwu, 2022-05-04:
> And please note we are expecting your commitment into CI tests for _both_ ipu6-drivers and ivsc-drivers.

what the CI tests will be? Can you please add some extra commands to clone/wget the missing header files from https://github.com/torvalds/linux/tree/master/drivers/misc/mei after git clone ivsc driver code?

wentongwu, 2022-05-04:
@vicamo if just copying these missing header files into ivsc driver repo, we have to take much effort to maintain these header files  which are not developed by us as kernel upgrades.

vicamo, 2022-05-04:
> @vicamo if just copying these missing header files into ivsc driver repo, we have to take much effort to maintain these header files which are not developed by us as kernel upgrades.

Then you probably now understand how downstream kernel vendors struggled to maintain your ivsc driver between kernel versions. Again, we're expecting them being built as dkms.

wentongwu, 2022-05-04:
> Then you probably now understand how downstream kernel vendors struggled to maintain your ivsc driver between kernel versions. Again, we're expecting them being built as dkms.

please note copying these header files may break other modules if vendors still follow the current building way.

vicamo, 2022-05-04:
> @vicamo if just copying these missing header files into ivsc driver repo, we have to take much effort to maintain these header files which are not developed by us as kernel upgrades.
>
>
> Then you [...]

See https://github.com/intel/ivsc-driver/pull/13. Next is to move the two affected sensors here into ivsc either in your repo or in our debian package.

> please note copying these header files into ivsc driver repo may break other modules if vendors still follow the current building way. Is that ok for you to make another branch for dkms?

No.

wentongwu, 2022-05-04:
> 

Ok, but actually I'm trying to take out the structure and function declaration needed into a separate header file. Thanks for your PR.

bk2204, 2022-06-21:
Is there a good way to get the modpost error fixed for those of us using the DKMS build (in my case, using Debian sid's 5.18.0-2-amd64)?  I've built and installed the ivsc-driver already with DKMS.

zebra-ok, 2022-07-16:
Did anyone get past these MODPOST errors? Trying to build this driver on Fedora 36.

vicamo, 2022-07-20:
> Did anyone get past these MODPOST errors? Trying to build this driver on Fedora 36.

https://launchpad.net/~vicamo/+archive/ubuntu/ppa-1972109
https://launchpad.net/~vicamo/+archive/ubuntu/ppa-1972106
Source code branches in the PPA description.

zebra-ok, 2022-07-21:
Thanks! I cloned the repos, applied the patches in `debian/patches`, and built sucessfully with `dkms`. Camera works now!

thepiwo, 2022-07-22:
@zebra-ok did you just install the dkms modules for those both projects or anything else? Building and installing worked well, but a camera device is not showing up for me.

netllama, 2022-07-24:
@zebra-ok I'm also trying to get this working on Fedora36, but am struggling to make any real progress.  The patches don't apply cleanly for me, and I can't get this repo or the 'hal' repo to build successfully currently.

Any chance you could provide more details on what order you built and patched everything to get the camera working?

Thesola10, 2022-07-28:
so there's no way for a DKMS module to depend on another DKMS module? Would an approach based on `kprobe` like in xcellerator/linux_kernel_hacking#3 work?

paulmenzel, 2022-08-11:
Just for the record, the ChromiumOS Linux kernel, has the driver integrated in some of their Linux version branches – at least up to 5.15 – for example [CHROMIUM: media: intel-ipu6: copy IPU6 driver from v5.4 branch](https://chromium-review.googlesource.com/c/chromiumos/third_party/kernel/+/3034663) and follow-up change-sets/commits.

thepiwo, 2022-08-11:
@Thesola10 have you managed building and getting the camera to run manually, if so with what steps?

Thesola10, 2022-08-11:
@thepiwo disabling modules that require IVSC basically makes the module useless on Alder Lake, but I did get a PKGBUILD working for non-IVSC controllers. Could not test though, I don't own a Tiger Lake machine

Thesola10, 2022-08-22:
Update: Got a pure-DKMS version of the whole ipu6 driver stack working over on AUR: [intel-ipu6-dkms-git](https://aur.archlinux.org/packages/intel-ipu6-dkms-git)

paulmenzel, 2022-08-22:
> Update: Got a pure-DKMS version of the whole ipu6 driver stack working over on AUR: [`intel-ipu6-dkms-git`](https://aur.archlinux.org/packages/intel-ipu6-dkms-git)

Nice. What Linux versions are supported?

Thesola10, 2022-08-22:
Likely everything that this repo supports, as long as `CONFIG_KPROBES` is enabled. I'm using the same workaround as @choff on [anbox-modules](https://github.com/choff/anbox-modules)

closed 2022-10-09 by onceforall