intel/ipu6-drivers #19: failed to enable platforms with ov01a1s or hm11b1 sensors on TGL

vicamo, 2022-05-20:
Differences from `media-ctl -p` output of oem-5.14 kernel to oem-5.17 gives:
```
--- media-ctl.5.14.0-1036-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.log       2022-05-18 00:49:15.849439919 +0800
+++ media-ctl.5.17.0-9004-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.log       2022-05-18 00:49:15.869440701 +0800
@@ -1,4 +1,4 @@
-Media controller API version 5.14.21
+Media controller API version 5.17.3
 
 Media device information
 ------------------------
@@ -7,7 +7,7 @@
 serial          
 bus info        pci:pci0000:00
 hw revision     0x0
-driver version  5.14.21
+driver version  5.17.3
 
 Device topology
 - entity 1: Intel IPU6 CSI-2 0 (2 pads, 1 link)
@@ -19,12 +19,11 @@
                [fmt:Y10_1X10/4096x3072 field:none]
                -> "Intel IPU6 CSI2 BE SOC 0":0 []
 
-- entity 4: Intel IPU6 CSI-2 1 (2 pads, 2 links)
+- entity 4: Intel IPU6 CSI-2 1 (2 pads, 1 link)
             type V4L2 subdev subtype Unknown flags 0
             device node name /dev/v4l-subdev1
        pad0: Sink
                [fmt:Y10_1X10/4096x3072 field:none]
-               <- "ov01a1s 20-0036":0 []
        pad1: Source
                [fmt:Y10_1X10/4096x3072 field:none]
                -> "Intel IPU6 CSI2 BE SOC 0":0 []
@@ -107,10 +106,3 @@
        pad0: Sink
                <- "Intel IPU6 CSI2 BE SOC 0":1 [DYNAMIC]
 
-- entity 68: ov01a1s 20-0036 (1 pad, 1 link)
-             type V4L2 subdev subtype Sensor flags 0
-             device node name /dev/v4l-subdev9
-       pad0: Source
-               [fmt:SGRBG10_1X10/1296x798 field:none]
-               -> "Intel IPU6 CSI-2 1":0 []
-
```
It seems some device is no longer probed/created.

The following logs were captured using ipu6/ivsc DKMS from [vicamo/ppa-1972106](https://launchpad.net/~vicamo/+archive/ubuntu/ppa-1972106) and [vicamo/ppa-1972109](https://launchpad.net/~vicamo/+archive/ubuntu/ppa-1972109).
* ov01a1s
  - [cheese.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.bad.log](https://github.com/intel/ipu6-drivers/files/8737173/cheese.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.bad.log)
  - [dmesg.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.log](https://github.com/intel/ipu6-drivers/files/8737174/dmesg.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.log)
  - [media-ctl.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.log](https://github.com/intel/ipu6-drivers/files/8737175/media-ctl.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.log)
* hm11b1
  - [cheese.5.17.0-9004-oem.ANDW-DVT2-C2.Vicamo-hm11b1.bad.log](https://github.com/intel/ipu6-drivers/files/8737182/cheese.5.17.0-9004-oem.ANDW-DVT2-C2.Vicamo-hm11b1.bad.log)
  - [dmesg.5.17.0-9004-oem.ANDW-DVT2-C2.Vicamo-hm11b1.log](https://github.com/intel/ipu6-drivers/files/8737183/dmesg.5.17.0-9004-oem.ANDW-DVT2-C2.Vicamo-hm11b1.log)
  - [media-ctl.5.17.0-9004-oem.ANDW-DVT2-C2.Vicamo-hm11b1.log](https://github.com/intel/ipu6-drivers/files/8737184/media-ctl.5.17.0-9004-oem.ANDW-DVT2-C2.Vicamo-hm11b1.log)

vicamo, 2022-05-26:
ANDW-DVT2-C3: (TGL, ov01a1s)
* 5.15.0-9032-generic:
  - [cheese.5.15.0-9032-generic.ANDW-DVT2-C3.KC-TGL-ov01a1s.bad.log](https://github.com/intel/ipu6-drivers/files/8776646/cheese.5.15.0-9032-generic.ANDW-DVT2-C3.KC-TGL-ov01a1s.bad.log)
  - [dmesg.5.15.0-9032-generic.ANDW-DVT2-C3.KC-TGL-ov01a1s.log](https://github.com/intel/ipu6-drivers/files/8776647/dmesg.5.15.0-9032-generic.ANDW-DVT2-C3.KC-TGL-ov01a1s.log)
  - [media-ctl.5.15.0-9032-generic.ANDW-DVT2-C3.KC-TGL-ov01a1s.log](https://github.com/intel/ipu6-drivers/files/8776648/media-ctl.5.15.0-9032-generic.ANDW-DVT2-C3.KC-TGL-ov01a1s.log)
* 5.15.12-for-ubuntu:
  - [cheese.5.15.12-for-ubuntu.ANDW-DVT2-C3.KC-TGL-ov01a1s.bad.log](https://github.com/intel/ipu6-drivers/files/8776666/cheese.5.15.12-for-ubuntu.ANDW-DVT2-C3.KC-TGL-ov01a1s.bad.log)
  - [dmesg.5.15.12-for-ubuntu.ANDW-DVT2-C3.KC-TGL-ov01a1s.log](https://github.com/intel/ipu6-drivers/files/8776667/dmesg.5.15.12-for-ubuntu.ANDW-DVT2-C3.KC-TGL-ov01a1s.log)
  - [media-ctl.5.15.12-for-ubuntu.ANDW-DVT2-C3.KC-TGL-ov01a1s.log](https://github.com/intel/ipu6-drivers/files/8776668/media-ctl.5.15.12-for-ubuntu.ANDW-DVT2-C3.KC-TGL-ov01a1s.log)
* 5.17.0-9004-oem:
  - [cheese.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.bad.log](https://github.com/intel/ipu6-drivers/files/8776672/cheese.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.bad.log)
  - [dmesg.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.log](https://github.com/intel/ipu6-drivers/files/8776673/dmesg.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.log)
  - [media-ctl.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.log](https://github.com/intel/ipu6-drivers/files/8776674/media-ctl.5.17.0-9004-oem.ANDW-DVT2-C3.KC-TGL-ov01a1s.log)

hao-yao, 2022-05-26:
> ANDW-DVT2-C3: (TGL, ov01a1s)
>
> * 5.15.0-9032-generic:
>
>   * [...]

Seems kerner driver is OK. We need more info to locate the issue.
@vicamo , could you stop the v4l2-relayd service, try this command and post the log?
sudo -E cameraDebug=3 gst-launch-1.0 icamerasrc device-name=ov01a1s-uf ! video/x-raw,format=YUY2,width=1280,height=720 ! videoconvert ! ximagesink

vicamo, 2022-05-26:
> Seems kerner driver is OK. We need more info to locate the issue. @vicamo , could you stop the v4l2-relayd service, try this command and post the log? sudo -E cameraDebug=3 gst-launch-1.0 icamerasrc device-name=ov01a1s-uf ! video/x-raw,format=YUY2,width=1280,height=720 ! videoconvert ! ximagesink

- [camera-debug.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777192/camera-debug.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
- [cheese.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777193/cheese.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
- [dmesg.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777194/dmesg.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
- [media-ctl.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777195/media-ctl.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)

vicamo, 2022-05-26:
> > * [camera-debug.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777192/camera-debug.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
> > * [cheese.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777193/cheese.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
> > * [dmesg.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777194/dmesg.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
> > * [media-ctl.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777195/media-ctl.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
> 
> In this log, seems kernel driver failed again. Would you try to create `/etc/modprobe.d/blacklist-int3472.conf` and add: `blacklist int3472-discrete` then attach a dmesg log of the same device again?

There is no such module 'int3472-discrete', do you mean `intel-skl-int3472`?

- [camera-debug.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.Blacklist-intel-skl-int3472.log](https://github.com/intel/ipu6-drivers/files/8778943/camera-debug.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.Blacklist-intel-skl-int3472.log)
- [dmesg.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.Blacklist-intel-skl-int3472.log](https://github.com/intel/ipu6-drivers/files/8778944/dmesg.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.Blacklist-intel-skl-int3472.log)

hao-yao, 2022-05-27:
> There is no such module 'int3472-discrete', do you mean `intel-skl-int3472`?
>
> * [...]
Seems i2c transfer failed. It is weird that 5.14.0-oem is OK but 5.15-generic fails while 5.15-generic OK on my device.

ccgjimlai, 2022-05-27:
Hi Vicamo,
Could you also provide the oem5.14 debs which could work on your device?

hao-yao, 2022-05-30:
Hi @vicamo , could you try to get the MCU firmware version on the failed device?
`cat /sys/bus/usb/devices/usb3/3-6/3-6:1.0/version`
My device is 0.8.0.8.

vicamo, 2022-05-30:
> Hi @vicamo , could you try to get the MCU firmware version on the failed device? `cat /sys/bus/usb/devices/usb3/3-6/3-6:1.0/version` My device is 0.8.0.8.

* ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0: 0.6.0.42
* ANDW-DVT2-C3.KC-TGL-ov01a1s.BIOS-1.4.3: 0.8.0.16

hao-yao, 2022-05-30:
> * ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0: 0.6.0.42
> * ANDW-DVT2-C3.KC-TGL-ov01a1s.BIOS-1.4.3: 0.8.0.16

According to the logs here, we can see on KC's DVT2-C3 ipu6 kernel driver works on 5.15, but fails when using cheese. On Vicamo's DVT2-C1, kernel driver failed at i2c reading with kernel 5.15, but OK on 5.14.

I think this should be 2 different issues. On KC's DVT, I think we can stop the `v4l2-relayd` service try try this command `sudo -E cameraDebug=3 gst-launch-1.0 icamerasrc device-name=ov01a1s-uf ! video/x-raw,format=YUY2,width=1280,height=720 ! videoconvert ! ximagesink` and save the log. And it is better to open a new issue in `ipu6-camera-hal` repo.

As for Vicamo's DVT kernel issue, I noticed that 5.14.0-1036-oem is OK but 5.15.12-for-ubuntu fails on the same device, while both build are OK on my side. So far I think this could be caused by hardware or firmware issue, especially MCU firmware. I will check this with MCU team.

@vicamo , could you help to check if anything wrong?

ccgjimlai, 2022-05-31:
> Hi Vicamo, Could you also provide the oem5.14 debs which could work on your device? and also, could you let us know your test result when you update your BIOS to 1.1.2?

@vicamo  do you have any feedback for these two?

vicamo, 2022-05-31:
> > Hi Vicamo, Could you also provide the oem5.14 debs which could work on your device?

You may find experimental kernels for oem projects in https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel. Add that PPA with:
```bash
$ sudo add-apt-repository ppa:canonical-hwe-team/intermediate-kernel
```
Click the link [View package details](https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+packages), click the triangle next to **linux-oem-5.14** in the new page, then it will give all the links to the built packages. The ones you'll need are:
* https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+files/linux-image-unsigned-5.14.0-9040-oem_5.14.0-9040.44+exp.74_amd64.deb
* https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+files/linux-modules-5.14.0-9040-oem_5.14.0-9040.44+exp.74_amd64.deb

> and also, could you let us know your test result when you update your BIOS to 1.1.2?

I don't see a reason to try an outdated BIOS version. We have failures with v1.4.3 and v1.10.0 already.

ccgjimlai, 2022-05-31:
> > > Hi Vicamo, Could you also provide the oem5.14 debs which could work on your device?
> 
> You may find experimental kernels for oem projects in https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel. Add that PPA with:
> 
> ```shell
> $ sudo add-apt-repository ppa:canonical-hwe-team/intermediate-kernel
> ```
> 
> Click the link [View package details](https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+packages), click the triangle next to **linux-oem-5.14** in the new page, then it will give all the links to the built packages. The ones you'll need are:
> 
> * https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+files/linux-image-unsigned-5.14.0-9040-oem_5.14.0-9040.44+exp.74_amd64.deb
> * https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+files/linux-modules-5.14.0-9040-oem_5.14.0-9040.44+exp.74_amd64.deb
> 
> > and also, could you let us know your test result when you update your BIOS to 1.1.2?
> 
> I don't see a reason to try an outdated BIOS version. We have failures with v1.4.3 and v1.10.0 already.

Hi @vicamo  
The information is now confusing, and the error symptom is difference between KC's and your device.

Let's focus on your device now.  In our previous sync meeting, you have mentioned that you were using BIOS 0.9, do you mean that you have already upgraded your device to BIOS 1.10?

Please use that device and do the following test
1. install 5.14 debs
2. open camera
3. save dmesg logs
4. print MCU fw version
5. dump dsdt
6. install 5.15 deb and repeat steps 2-5

vicamo, 2022-05-31:
> Hi @vicamo The information is now confusing, and the error symptom is difference between KC's and your device.
> 
> Let's focus on your device now. In our previous sync meeting, you have mentioned that you were using BIOS 0.9, do you mean that you have already upgraded your device to BIOS 1.10?

https://github.com/intel/ipu6-drivers/issues/19#issuecomment-1138253646

> Please use that device and do the following test
> 
> 1. install 5.14 debs
> 2. open camera
> 3. save dmesg logs
> 4. print MCU fw version

https://github.com/intel/ipu6-drivers/issues/19#issuecomment-1140825769

> 5. dump dsdt
> 6. install 5.15 deb and repeat steps 2-5

* 5.14.0-1034-oem:
  - [camera-debug.5.14.0-1034-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8804587/camera-debug.5.14.0-1034-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
  - [cheese.5.14.0-1034-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.good.log](https://github.com/intel/ipu6-drivers/files/8804588/cheese.5.14.0-1034-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.good.log)
  - [dmesg.5.14.0-1034-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8804589/dmesg.5.14.0-1034-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
  - [dsdt.5.14.0-1034-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8804591/dsdt.5.14.0-1034-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
  - [media-ctl.5.14.0-1034-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8804593/media-ctl.5.14.0-1034-oem.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
* 5.15.0-9034-generic
  - [camera-debug.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777192/camera-debug.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
  - [cheese.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777193/cheese.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
  - [dmesg.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777194/dmesg.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
  - [dsdt.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8804658/dsdt.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)
  - [media-ctl.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log](https://github.com/intel/ipu6-drivers/files/8777195/media-ctl.5.15.0-9034-generic.ANDW-DVT2-C1.Vicamo-TGL-ov01a1s.BIOS-1.10.0.log)

vicamo, 2022-05-31:
How do I upgrade MCU fw?

hao-yao, 2022-05-31:
Hi @vicamo , would you try the 5.15.12-for-ubuntu kernel image we sent you before? And also the devices info at `/sys/bus/acpi/devices` and `/sys/bus/i2c/devices`.

> How do I upgrade MCU fw?

We will try to send you a MCU FW and flash guide if you can make sure version 0.6.0.42 is on your device (BIOS=1.10). It is weird that the latest BIOS contains an very old FW.

vicamo, 2022-05-31:
> Hi @vicamo , would you try the 5.15.12-for-ubuntu kernel image we sent you before? And also the devices info at `/sys/bus/acpi/devices` and `/sys/bus/i2c/devices`.

See https://github.com/intel/ipu6-drivers/issues/19#issuecomment-1132696705 and https://github.com/intel/ipu6-drivers/issues/19#issuecomment-1138188192. **5.15.12-for-ubuntu** is your kernel.

> > How do I upgrade MCU fw?
> 
> We will try to send you a MCU FW and flash guide if you can make sure version 0.6.0.42 is on your device (BIOS=1.10). It is weird that the latest BIOS contains an very old FW.

Maybe MCU fw is not included in BIOS updates.

We're also tracking an intermittent failure to probe ov01a1s on last few revisions of oem-5.14 kernels also on both ANDW-DVT2-C1 and C3. It appears the sensor may sometimes not be probed, and following message is missing from dmesg:
```
intel-ipu6-isys intel-ipu6-isys0: bind ov01a1s 20-0036 nlanes is 1 port is 1
```
Last known good is 5.14.0-1034-oem, and roughly 50% failure rate in 1035, and 100% in 1038 and 1040.

hao-yao, 2022-06-01:
> See [#19 (comment)](https://github.com/intel/ipu6-drivers/issues/19#issuecomment-1132696705) and [#19 (comment)](https://github.com/intel/ipu6-drivers/issues/19#issuecomment-1138188192). **5.15.12-for-ubuntu** is your kernel.
> 

@vicamo , I noticed that your log from 5.15.12 kernel shows BIOS=89.5.23. Is this the same device as you tried 5.15.0-oem and 5.14.0-oem kernel on? Or you did a BIOS upgrade?

vicamo, 2022-06-01:
> @vicamo , I noticed that your log from 5.15.12 kernel shows BIOS=89.5.23. Is this the same device as you tried 5.15.0-oem and 5.14.0-oem kernel on? Or you did a BIOS upgrade?

BIOS upgrade. Both devices.

ccgjimlai, 2022-06-01:
> How do I upgrade MCU fw?

Hi @vicamo please refer to the attached file and the steps below for upgrading to 0.8.0.9 fw

## d21 fw download
1. install dfu-util
   `sudo apt install dfu-util`

2. update d21 fw
```
echo dfu > /sys/bus/usb/drivers/ljca/1-5\:1.0/cmd
dfu-util -v -R -D usb_bridge_dev_fw.dfu
```

I have also included fw version 0.6.0.42, which is what you have in your device now

[d21_fw.zip](https://github.com/intel/ipu6-drivers/files/8811553/d21_fw.zip)

vicamo, 2022-06-01:
> Hi @vicamo please refer to the attached file and the steps below for upgrading to 0.8.0.9 fw
>
> ## d21 fw download
> 1. install dfu-util
>    `sudo apt install dfu-util`
> 2. update d21 fw
>
> ```
> echo dfu > [...]

I don't have `/sys/bus/usb/drivers/ljca/1-5:1.0` but `/sys/bus/usb/drivers/ljca/3-6:1.0`. After flashed fw 0.8.0.9 on ANDW-DVT2-C1(Vicamo-TGL-ov01a1s), which had 0.6.0.42 previously, now doesn't load ljca driver on both oem-5.14 and 5.15 kernels. The usb device become:
```
Bus 003 Device 002: ID 04d8:0037 Microchip Technology, Inc. USB DFU
```
While on ANDW-DVT2-C3(KC-TGL-ov01a1s), it has:
```
Bus 003 Device 003: ID 8086:0b63 Intel Corp. USB Bridge
```
And ljca driver matches only [8086:0b63](https://github.com/intel/ivsc-driver/blob/main/drivers/mfd/ljca.c#L1165).

ccgjimlai, 2022-06-02:
Hi @vicamo  it means that FW flash has failed
you could try to flash it back to the original FW

vicamo, 2022-06-02:
> original

Nope. The ljca driver now never loads. There is no such `.../ljca/.../cmd` node anymore.

hao-yao, 2022-06-02:
> Nope. The ljca driver now never loads. There is no such `.../ljca/.../cmd` node anymore.

Just use dfu-utils command is OK, no need to echo to cmd. The dfu device should still be there.

ccgjimlai, 2022-06-02:
When you reboot your device, will it exit dfu mode?
if the device is still in dfu mode, it means that the fw was not flashed successfully.
Otherwise, if the device exits dfu mode after reboot, then ljca will be loaded.

vicamo, 2022-06-06:
> When you reboot your device, will it exit dfu mode? if the device is still in dfu mode, it means that the fw was not flashed successfully. Otherwise, if the device exits dfu mode after reboot, then ljca will be loaded.

it stays as is even after warm/cold reboots.

ccgjimlai, 2022-06-06:
If you still couldn't update the MCU FW on your device,
You could send it to Intel Office,
We will see if we could update it for you.

As for KC's device,
The issue seems to be in user space.
We have tested the following combination, and camera works.
Could you install the image below to KC's device and provide us the logs if it fails

OS image
https://people.canonical.com/~vicamo/temp/dell-bto-focal-fossa-mewtwo-X105-20220531-56.iso

Kernel
Canonical kernel and deb
* 5.15:
- https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+sourcepub/13636501/+listing-archive-extra
1. https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+files/linux-image-unsigned-5.15.0-9034-generic_5.15.0-9034.35+exp.30_amd64.deb
2. https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+files/linux-modules-5.15.0-9034-generic_5.15.0-9034.35+exp.30_amd64.deb
3. https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+files/linux-modules-extra-5.15.0-9034-generic_5.15.0-9034.35+exp.30_amd64.deb
4. https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+files/linux-headers-5.15.0-9034-generic_5.15.0-9034.35+exp.30_amd64.deb
(optional)
5. https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel/+files/linux-headers-5.15.0-9034_5.15.0-9034.35+exp.30_all.deb
(optional)

[image]

ccgjimlai, 2022-06-09:
After updating the MCU fw on Vicamo's device, the camera can work now,
SW is untouched (supposedly 5.15 kernel)

[image]

fourdollars, 2022-06-14:
I can not upgrade the fw.
```
$ cat /sys/bus/usb/drivers/ljca/*/version
0.6.0.42
u@u:~/d21_fw/0.8.0.9$ sudo dfu-util -v -R -D usb_bridge_dell_fw.dfu
[sudo] password for u: 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix version 100
Match vendor ID from file: 04d8
Match product ID from file: 0037
dfu-util: No DFU capable USB device available
u@u:~/d21_fw/0.8.0.9$ uname -a
Linux u 5.15.0-9036-generic #37+exp.32-Ubuntu SMP Thu Jun 9 08:34:09 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
u@u:~/d21_fw/0.8.0.9$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 8086:0b63 Intel Corp. USB Bridge
Bus 003 Device 003: ID 8087:0032 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

fourdollars, 2022-06-14:
There is no /sys/bus/usb/drivers/ljca/*/version now.
```
$ cat /sys/bus/usb/drivers/ljca/*/version
cat: '/sys/bus/usb/drivers/ljca/*/version': No such file or directory

$ sudo dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [04d8:0037] ver=0401, devnum=2, cfg=1, intf=0, path="3-6", alt=0, name="UNKNOWN", serial="MCHP-DF-005"

$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 04d8:0037 Microchip Technology, Inc. USB DFU
Bus 003 Device 003: ID 8087:0032 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

ccgjimlai, 2022-06-15:
[d21_fw_0.8.0.16.zip](https://github.com/intel/ipu6-drivers/files/8905248/d21_fw_0.8.0.16.zip)

Hi $4,
Could you try again with the attached FW?

fourdollars, 2022-06-15:
@ccgjimlai It is also trapped in DFU state now.
So I can not flash the fw again.

ccgjimlai, 2022-06-15:
Hi @fourdollars 
Just use dfu-utils command is OK, no need to echo to cmd. The dfu device should still be there, so that you could still perform fw flash

fourdollars, 2022-06-15:
```
u@u:~/0.8.0.16$ sudo dfu-util -v -R -D usb_bridge_dell_fw_production.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix version 100
Match vendor ID from file: 04d8
Match product ID from file: 0037
Opening DFU capable USB device...
ID 04d8:0037
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 64
Copying data from PC to DFU device
Download	[=========================] 100%        32752 bytes
Download done.
Sent a total of 32752 bytes
state(2) = dfuIDLE, status(0) = No error condition is present
Done!
dfu-util: can't detach
Resetting USB to switch back to runtime mode
```

fourdollars, 2022-06-15:
```
[  737.413699] usb 3-6: reset full-speed USB device number 2 using xhci_hcd
[  741.117481] usb 3-6: device descriptor read/64, error -71
[  741.373748] usb 3-6: device firmware changed
[  741.374034] usb 3-6: USB disconnect, device number 2
[  741.501406] usb 3-6: new full-speed USB device number 4 using xhci_hcd
[  741.650814] usb 3-6: New USB device found, idVendor=8086, idProduct=0b63, bcdDevice= 8.16
[  741.650823] usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  741.650826] usb 3-6: Product: USB Bridge
[  741.650829] usb 3-6: Manufacturer: MCHP
[  742.249265] ljca 3-6:1.0: acked sem wait timed out ret:0 timeout:20 ack:0
[  742.249277] ljca 3-6:1.0: MNG_ENUM_SPI failed ret:-110 len:7 num:0
[  742.249517] ljca 3-6:1.0: LJCA USB device init success
[  742.249602] usbcore: registered new interface driver ljca
[  742.271237] int3472-discrete INT3472:06: error -EBUSY: Failed to map GPIO to clock
[  742.271472] int3472-discrete: probe of INT3472:06 failed with error -16
[  742.274546] intel-ipu6-isys intel-ipu6-isys0: bind ov01a1s 20-0036 nlanes is 1 port is 1
[  742.274655] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
```
```
u@u:~/0.8.0.16$ cat /sys/bus/usb/drivers/ljca/*/version
0.8.0.16
```

fourdollars, 2022-06-15:
@ccgjimlai Thx. The camera works on 5.15.0-9036-generic #37+exp.32-Ubuntu.

vicamo, 2022-08-26:
Closing this as resolved for me.

closed 2022-08-26 by vicamo