intel/ipu6-drivers #23: Add support for OV8856 sensor

prof-dr-pepper, 2022-06-23:
In Dell Latitude 7320 Detachable (with Tiger Lake processor) the rear camera is an OV8856 sensor. For this sensor the official Linux kernel already contains an i2c driver (see https://github.com/torvalds/linux/blob/master/drivers/media/i2c/ov8856.c) but this driver does not work together with this IPU6 driver, does it?

Help would be very appreciated. I am a C/C++ software developer myself but with no knowledge about Linux kernel development. So what would have to be done in order to make this work?

Thank you,
Marcus

PhilDevProg, 2023-05-27:
Doesn't this repository already include a [driver](https://github.com/intel/ipu6-drivers/blob/master/drivers/media/i2c/ov8856.c)  for the OV8856?

andyyeh75, 2024-07-29:
It's Andy from Intel camera driver team. Although sensor driver exists, integration and validation are still required. We didn't receive PC vendor's request to enable the camera on this device yet. Suggest to get contact with Dell customer service. Hope to hear good news from Dell.