intel/ipu6-drivers #24: Add support for OV5678 sensor

prof-dr-pepper, 2022-06-23:
In Dell Latitude 7320 Detachable (with Tiger Lake processor) the front camera is an OV5678 sensor. For this sensor no i2c driver seems to exist for Linux but a driver for OV01A1S exists (see https://github.com/intel/ipu6-drivers/blob/master/drivers/media/i2c/ov01a1s.c) which seems to have the same output format (Bayer 4x4 RGB-Ir) and I assume that adapting this driver should be doable.

Help would be very appreciated. I am a C/C++ software developer myself but with no knowledge about Linux kernel development. So what would have to be done in order to make this work?

Thank you,
Marcus

Kissycat, 2022-09-13:
Please..

eliska-povolna, 2023-03-13:
I would also appreciate the driver very much

pauLuque, 2023-09-13:
Me too, I would appreciate this driver

krjones91, 2024-05-03:
I would also like to see this module added.

elghfz, 2025-07-11:
I would like it too, for exactly the same device