intel/ipu6-drivers #33: Add support for OV01A1B infrared sensor

Thesola10, 2022-08-25:
I looked into the Windows driver package for the Dell Latitude 9430 and noticed several `*.aiqb` and graph files that aren't present in the open source stack, as well as mentions of `ov01a1b`. Please add support for those sensors as well.

hao-yao, 2022-09-26:
> I looked into the Windows driver package for the Dell Latitude 9430 and noticed several `*.aiqb` and graph files that aren't present in the open source stack, as well as mentions of `ov01a1b`. Please add support for those sensors as well.

Do we have any applications for IR camera?

Thesola10, 2022-09-26:
Face authentication with the likes of [howdy](https://github.com/boltgolt/howdy), and it would seem the rest of the sensor array on the laptop also seems to control the proximity sensor?

ctrlcctrlv, 2023-03-11:
Even if we had no applications, it's a chicken and egg problem. If Intel won't support its hardware, then who will write the applications?

striczkof, 2023-06-12:
Up, same IR sensors for the Dell XPS 9320 need love too.