intel/ipu6-drivers #34: CONFIG_INTEL_SKL_INT3472 vs. CONFIG_POWER_CTRL_LOGIC in Engineer release 2022-08-10

vicamo, 2022-08-26:
In https://github.com/intel/ipu6-drivers/blob/master/drivers/media/i2c/ov01a1s.c#L334:
```
#if IS_ENABLED(CONFIG_INTEL_SKL_INT3472)
...
#elif IS_ENABLED(CONFIG_POWER_CTRL_LOGIC)
...
#endif
```
This is going to shadow **CONFIG_POWER_CTRL_LOGIC** when **CONFIG_INTEL_SKL_INT3472** is enabled, and actually the latter is always enabled by host kernel on recent (>= 5.14) Ubuntu kernels. So:
1. Even one tries to comment-out the `export CONFIG_INTEL_SKL_INT3472=m` line in the Makefile, it will still be enabled anyway as it's enabled by host kernel.
2. And since CONFIG_INTEL_SKL_INT3472 is always enabled by host kernel, the preprocessor macros CONFIG_INTEL_SKL_INT3472 and CONFIG_INTEL_SKL_INT3472_MODULE will also be defined in host generated `autoconf.h`. The cflags appending in the Makefile is useless.
3. As my experiments, The TGL/ov01a1s platforms still need **CONFIG_POWER_CTRL_LOGIC** to drive the gpio correctly. Are you sure CONFIG_INTEL_SKL_INT3472 can replace CONFIG_POWER_CTRL_LOGIC in this sensor? We need to keep backward compatibility as always.

hao-yao, 2022-09-26:
Hi Vicamo, we can use the INT3472 patch to let ov01a1s work. Once we have the patch, power_ctrl_logic can be deprecated. I think we can use power_ctrl_logic for kernel without intel_skl_int3472. For kernel > 5.14, we can use intel_skl_int3472 and apply the patch instead of old power_ctrl_logic.

vicamo, 2022-09-27:
> Hi Vicamo, we can use the INT3472 patch to let ov01a1s work. Once we have the patch, power_ctrl_logic can be deprecated. I think we can use power_ctrl_logic for kernel without intel_skl_int3472. For kernel > 5.14, we can use intel_skl_int3472 and apply the patch instead of old power_ctrl_logic.

It's not about what to use. It's about the way you write the conditional guards.
```
$ grep -nr INT3472 /usr/src/linux-headers-5.14.0-1052-oem/
/usr/src/linux-headers-5.14.0-1052-oem/include/config/auto.conf:4619:CONFIG_INTEL_SKL_INT3472=m
/usr/src/linux-headers-5.14.0-1052-oem/include/generated/autoconf.h:4621:#define CONFIG_INTEL_SKL_INT3472_MODULE 1
/usr/src/linux-headers-5.14.0-1052-oem/.config:8935:CONFIG_INTEL_SKL_INT3472=m
```
And since the `linux/kconfig.h` has:
```
#define IS_ENABLED(option) __or(IS_BUILTIN(option), IS_MODULE(option))
```
It follows `IS_ENABLED(CONFIG_INTEL_SKL_INT3472)` will be always true on Ubuntu kernels >= 5.14, no matter you enable or disable CONFIG_INTEL_SKL_INT3472 in the root Makefile of this repository.

hao-yao, 2022-09-28:
We meant that if int3472 is disabled in a different config, we can still enable power_ctrl_logic to enable ov01a1s. If we want ov01a1s, hm11b1 and ov2740 work with int3472, we need [this kernel patch](https://github.com/intel/ipu6-drivers/blob/master/patch/int3472-support-independent-clock-and-LED-gpios-5.17%2B.patch). I hope I explained it clearly, and we can also choose to change the code in 3 ways:

- Swtich macro order of power_ctrl_logic and int3472
- Use int3472 on kernel 5.14+, and power_ctrl_logic in kernel before 5.14
- Deprecate power_ctrl_logic and use int3472 only

@vicamo which do you suggest?

vicamo, 2022-09-29:
> We meant that if int3472 is disabled in a different config, we can still enable power_ctrl_logic to enable ov01a1s. If we want ov01a1s, hm11b1 and ov2740 work with int3472, we need [this kernel [...]

Hi, the role of this repository can be a dkms, so I think it's reasonable to achieve the ability to detect the target kernel and configure itself correspondingly. Your options should be based on two more questions that only Intel may answer:
1. whether it's possible to switch to INT3472 for all the kernel versions promised. If so, obsoleting power_logic_ctrl is then possible, which maps to your option 3.
2. based on 1., whether to depend on INT3472 for all kernel versions promised. If so, coexistence of int3472 and power_logic_ctrl is not necessary. This maps to your option 2 and 3.

And finally, I don't think option 1 is a good idea as it:
1. not documented. Even you don't know it's going to bring issues to platforms with INT3472 previously enabled,
2. takes extra modification to the source as being integrated to kernel tree is possible, but not for a standalone dkms that is meant to be compiled against multiple kernel variants, versions, etc.

closed 2023-08-04 by vicamo