intel/ipu6-drivers #36: Kernel trap ov01a10 - kallsyms_lookup_name+0x4/0xf0

Delorien84, 2022-09-01:
Hello, I am using Majaro Linux with `5.19.1-3-MANJARO` kernel. I was able to compile this modules using https://aur.archlinux.org/packages/intel-ipu6-dkms-git AUR package. But get kernel trap during system boot.

```
[   15.831157] traps: Missing ENDBR: kallsyms_lookup_name+0x4/0xf0
[   15.831164] ------------[ cut here ]------------
[   15.831165] kernel BUG at arch/x86/kernel/traps.c:253!
[   15.831172] invalid opcode: 0000 [#1] PREEMPT SMP NOPTI
[   15.831176] CPU: 8 PID: 444 Comm: systemd-udevd Tainted: G           OE     5.19.1-3-MANJARO #1 04cffc0a920d6f0accb28b85e9c8488dd72ba600
[   15.831180] Hardware name: Dell Inc. XPS 9320/0CW9KM, BIOS 1.4.0 05/13/2022
[   15.831181] RIP: 0010:exc_control_protection+0xc2/0xd0
[   15.831187] Code: 8b 93 80 00 00 00 be fa 00 00 00 48 c7 c7 b6 b5 88 92 e8 21 ad 4a ff e9 72 ff ff ff 48 c7 c7 9d b5 88 92 e8 a6 06 fb ff 0f 0b <0f> 0b 66 66 2e 0f 1f 84 00 00 00 00 00 90 66 0f 1f 00 55 53 48 89
[   15.831188] RSP: 0018:ffffbf3c823ff6b8 EFLAGS: 00010002
[   15.831191] RAX: 0000000000000033 RBX: ffffbf3c823ff6d8 RCX: 0000000000000000
[   15.831192] RDX: 0000000000000000 RSI: ffff98426f821660 RDI: ffff98426f821660
[   15.831194] RBP: 0000000000000003 R08: 0000000000000000 R09: ffffbf3c823ff550
[   15.831195] R10: 0000000000000003 R11: ffffffff930cb428 R12: 0000000000000000
[   15.831195] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[   15.831197] FS:  00007fa381137200(0000) GS:ffff98426f800000(0000) knlGS:0000000000000000
[   15.831199] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   15.831200] CR2: 000055cac07f2000 CR3: 000000010244e004 CR4: 0000000000f70ee0
[   15.831201] PKRU: 55555554
[   15.831209] Call Trace:
[   15.831212]  <TASK>
[   15.831214]  asm_exc_control_protection+0x26/0x30
[   15.831217] RIP: 0010:kallsyms_lookup_name+0x4/0xf0
[   15.831219] Code: ff 0f 1f 44 00 00 f3 0f 1e fa 0f 1f 44 00 00 45 31 c0 31 c9 31 d2 e9 3b fe ff ff 66 66 2e 0f 1f 84 00 00 00 00 00 66 0f 1f 00 <0f> 1f 44 00 00 41 56 b9 10 00 00 00 41 55 49 89 fd 41 54 55 53 48
[   15.831221] RSP: 0018:ffffbf3c823ff788 EFLAGS: 00010246
[   15.831222] RAX: ffffffff9136a664 RBX: ffff983b44ef2020 RCX: 0000000000000002
[   15.831223] RDX: 0000000000000001 RSI: 0000000055555554 RDI: ffffffffc0ed70db
[   15.831224] RBP: ffffbf3c823ff810 R08: 0000000000032630 R09: 0000000000000000
[   15.831225] R10: 0000000000000001 R11: 0000000000000008 R12: ffff983b44ef2000
[   15.831226] R13: ffff983b44ef2000 R14: ffff983b44ef2020 R15: ffff983b02d4f030
[   15.831227]  ? kallsyms_lookup_name+0x4/0xf0
[   15.831231]  ov01a10_probe+0x438/0x490 [ov01a10 9de964567655b25584cd0978272930f272590611]
[   15.831236]  ? kmem_cache_alloc_trace+0x16b/0x310
[   15.831240]  ? i2c_device_probe+0x11f/0x2b0
[   15.831245]  i2c_device_probe+0x11f/0x2b0
[   15.831248]  really_probe+0x1bf/0x390
[   15.831251]  __driver_probe_device+0xff/0x170
[   15.831253]  driver_probe_device+0x1f/0x90
[   15.831254]  __device_attach_driver+0x76/0xe0
[   15.831256]  ? driver_allows_async_probing+0x70/0x70
[   15.831258]  bus_for_each_drv+0x8c/0xe0
[   15.831260]  __device_attach+0xb2/0x1e0
[   15.831261]  bus_probe_device+0x92/0xb0
[   15.831263]  device_add+0x422/0x9a0
[   15.831266]  i2c_new_client_device+0x19c/0x2c0
[   15.831268]  i2c_acpi_add_device+0x97/0xb0
[   15.831270]  acpi_ns_walk_namespace+0x16d/0x2db
[   15.831275]  ? i2c_acpi_resource_count+0x30/0x30
[   15.831276]  ? i2c_acpi_resource_count+0x30/0x30
[   15.831277]  acpi_walk_namespace+0xf5/0x141
[   15.831280]  i2c_acpi_register_devices+0x48/0x80
[   15.831281]  i2c_register_adapter+0x225/0x6a0
[   15.831284]  platform_probe+0x45/0x90
[   15.831286]  really_probe+0x1bf/0x390
[   15.831288]  __driver_probe_device+0xff/0x170
[   15.831289]  driver_probe_device+0x1f/0x90
[   15.831291]  __driver_attach+0xc2/0x1b0
[   15.831292]  ? __device_attach_driver+0xe0/0xe0
[   15.831294]  bus_for_each_dev+0x88/0xd0
[   15.831295]  bus_add_driver+0x164/0x220
[   15.831297]  driver_register+0x8d/0xe0
[   15.831299]  ? 0xffffffffc1075000
[   15.831300]  do_one_initcall+0x5a/0x220
[   15.831305]  do_init_module+0x4a/0x1e0
[   15.831307]  __do_sys_init_module+0x138/0x1b0
[   15.831309]  do_syscall_64+0x5c/0x90
[   15.831312]  ? vfs_read+0x148/0x190
[   15.831314]  ? syscall_exit_to_user_mode+0x1b/0x40
[   15.831316]  ? do_syscall_64+0x6b/0x90
[   15.831318]  ? do_user_addr_fault+0x1db/0x690
[   15.831320]  ? exc_page_fault+0x74/0x170
[   15.831322]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
```

Thesola10, 2022-09-27:
Ah, I'm the maintainer for this package, the issue is due to one of my patches. I'm working to deprecate them by using the new feature they added to disable IVSC integration

Thesola10, 2022-09-27:
Annnd... update's ready! You can install it from AUR.

Delorien84, 2022-09-27:
Thank you, works like charm.

closed 2022-09-27 by Delorien84