intel/ipu6-drivers #37: Fails to build on Intel Jasper Lake

junocomp, 2022-09-09:
I am currently running Kernel 5.19 (Debian) and I pulled the jsl_penguin_peak branch. I am not sure if the errors are related to my kernel or my board, but this is the error I am getting.

```
$ make -j8
make -C /lib/modules/5.19.0-1-amd64/build M=/home/mobian/ipu6-drivers modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-1-amd64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.3.0-5) 11.3.0
  You are using:           gcc-11 (Debian 11.3.0-6) 11.3.0
  CC [M]  /home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.o
  CC [M]  /home/mobian/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu.o
  CC [M]  /home/mobian/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.o
  CC [M]  /home/mobian/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-dma.o
  CC [M]  /home/mobian/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-mmu.o
  CC [M]  /home/mobian/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-buttress.o
  CC [M]  /home/mobian/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-trace.o
  CC [M]  /home/mobian/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-cpd.o
/home/mobian/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.c:137:19: error: initialization of ‘void (*)(struct device *)’ from incompatible pointer type ‘int (*)(struct device *)’ [-Werror=incompatible-pointer-types]
  137 |         .remove = ipu_bus_remove,
      |                   ^~~~~~~~~~~~~~
/home/mobian/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.c:137:19: note: (near initialization for ‘ipu_bus.remove’)
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-5.19.0-1-common/scripts/Makefile.build:254: /home/mobian/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.o] Error 1
make[4]: *** Waiting for unfinished jobs....
In file included from /usr/src/linux-headers-5.19.0-1-common/include/media/v4l2-device.h:13,
                 from /home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:9:
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c: In function ‘ov13858_open’:
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1383:75: error: ‘struct v4l2_subdev_fh’ has no member named ‘pad’
 1383 |                                                                         fh->pad,
      |                                                                           ^~
/usr/src/linux-headers-5.19.0-1-common/include/media/v4l2-subdev.h:1095:40: note: in definition of macro ‘v4l2_subdev_get_try_format’
 1095 |         v4l2_subdev_get_pad_format(sd, state, pad)
      |                                        ^~~~~
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c: In function ‘ov13858_do_get_pad_format’:
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1565:59: error: passing argument 2 of ‘v4l2_subdev_get_pad_format’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 1565 |                 framefmt = v4l2_subdev_get_try_format(sd, cfg, fmt->pad);
      |                                                           ^~~
      |                                                           |
      |                                                           struct v4l2_subdev_pad_config *
/usr/src/linux-headers-5.19.0-1-common/include/media/v4l2-subdev.h:1095:40: note: in definition of macro ‘v4l2_subdev_get_try_format’
 1095 |         v4l2_subdev_get_pad_format(sd, state, pad)
      |                                        ^~~~~
/usr/src/linux-headers-5.19.0-1-common/include/media/v4l2-subdev.h:1046:54: note: expected ‘struct v4l2_subdev_state *’ but argument is of type ‘struct v4l2_subdev_pad_config *’
 1046 |                            struct v4l2_subdev_state *state,
      |                            ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c: In function ‘ov13858_set_pad_format’:
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1614:59: error: passing argument 2 of ‘v4l2_subdev_get_pad_format’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 1614 |                 framefmt = v4l2_subdev_get_try_format(sd, cfg, fmt->pad);
      |                                                           ^~~
      |                                                           |
      |                                                           struct v4l2_subdev_pad_config *
/usr/src/linux-headers-5.19.0-1-common/include/media/v4l2-subdev.h:1095:40: note: in definition of macro ‘v4l2_subdev_get_try_format’
 1095 |         v4l2_subdev_get_pad_format(sd, state, pad)
      |                                        ^~~~~
/usr/src/linux-headers-5.19.0-1-common/include/media/v4l2-subdev.h:1046:54: note: expected ‘struct v4l2_subdev_state *’ but argument is of type ‘struct v4l2_subdev_pad_config *’
 1046 |                            struct v4l2_subdev_state *state,
      |                            ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c: At top level:
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1805:27: error: initialization of ‘int (*)(struct v4l2_subdev *, struct v4l2_subdev_state *, struct v4l2_subdev_mbus_code_enum *)’ from incompatible pointer type ‘int (*)(struct v4l2_subdev *, struct v4l2_subdev_pad_config *, struct v4l2_subdev_mbus_code_enum *)’ [-Werror=incompatible-pointer-types]
 1805 |         .enum_mbus_code = ov13858_enum_mbus_code,
      |                           ^~~~~~~~~~~~~~~~~~~~~~
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1805:27: note: (near initialization for ‘ov13858_pad_ops.enum_mbus_code’)
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1806:20: error: initialization of ‘int (*)(struct v4l2_subdev *, struct v4l2_subdev_state *, struct v4l2_subdev_format *)’ from incompatible pointer type ‘int (*)(struct v4l2_subdev *, struct v4l2_subdev_pad_config *, struct v4l2_subdev_format *)’ [-Werror=incompatible-pointer-types]
 1806 |         .get_fmt = ov13858_get_pad_format,
      |                    ^~~~~~~~~~~~~~~~~~~~~~
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1806:20: note: (near initialization for ‘ov13858_pad_ops.get_fmt’)
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1807:20: error: initialization of ‘int (*)(struct v4l2_subdev *, struct v4l2_subdev_state *, struct v4l2_subdev_format *)’ from incompatible pointer type ‘int (*)(struct v4l2_subdev *, struct v4l2_subdev_pad_config *, struct v4l2_subdev_format *)’ [-Werror=incompatible-pointer-types]
 1807 |         .set_fmt = ov13858_set_pad_format,
      |                    ^~~~~~~~~~~~~~~~~~~~~~
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1807:20: note: (near initialization for ‘ov13858_pad_ops.set_fmt’)
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1808:28: error: initialization of ‘int (*)(struct v4l2_subdev *, struct v4l2_subdev_state *, struct v4l2_subdev_frame_size_enum *)’ from incompatible pointer type ‘int (*)(struct v4l2_subdev *, struct v4l2_subdev_pad_config *, struct v4l2_subdev_frame_size_enum *)’ [-Werror=incompatible-pointer-types]
 1808 |         .enum_frame_size = ov13858_enum_frame_size,
      |                            ^~~~~~~~~~~~~~~~~~~~~~~
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1808:28: note: (near initialization for ‘ov13858_pad_ops.enum_frame_size’)
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c: In function ‘ov13858_probe’:
/home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.c:1992:15: error: implicit declaration of function ‘v4l2_async_register_subdev_sensor_common’; did you mean ‘v4l2_async_register_subdev_sensor’? [-Werror=implicit-function-declaration]
 1992 |         ret = v4l2_async_register_subdev_sensor_common(&ov13858->sd);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |               v4l2_async_register_subdev_sensor
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.19.0-1-common/scripts/Makefile.build:254: /home/mobian/ipu6-drivers/drivers/media/i2c/ov13858_intel.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.19.0-1-common/scripts/Makefile.build:471: /home/mobian/ipu6-drivers/drivers/media/i2c] Error 2
make[2]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-5.19.0-1-common/scripts/Makefile.build:471: /home/mobian/ipu6-drivers/drivers/media/pci/intel/ipu6] Error 2
make[2]: *** [/usr/src/linux-headers-5.19.0-1-common/scripts/Makefile.build:471: /home/mobian/ipu6-drivers/drivers/media/pci/intel] Error 2
make[1]: *** [/usr/src/linux-headers-5.19.0-1-common/Makefile:1871: /home/mobian/ipu6-drivers] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-1-amd64'
make: *** [Makefile:17: all] Error 2
```

junocomp, 2022-09-14:
@hao-yao you think you can help me with this?

junocomp, 2022-09-15:
I replaced `v4l2_async_register_subdev_sensor_common` to `v4l2_async_register_subdev_sensor` line 1992 and not it almost builds everything. I am getting one tiny last error

```
make
make -C /lib/modules/5.19.8-1-MANJARO/build M=/home/juno/ipu6-drivers modules
make[1]: Entering directory '/usr/lib/modules/5.19.8-1-MANJARO/build'
  CC [M]  /home/juno/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu.o
  CC [M]  /home/juno/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.o
/home/juno/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.c:137:19: error: initialization of ‘void (*)(struct device *)’ from incompatible pointer type ‘int (*)(struct device *)’ [-Werror=incompatible-pointer-types]
  137 |         .remove = ipu_bus_remove,
      |                   ^~~~~~~~~~~~~~
/home/juno/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.c:137:19: note: (near initialization for ‘ipu_bus.remove’)
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:249: /home/juno/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.o] Error 1
make[3]: *** [scripts/Makefile.build:466: /home/juno/ipu6-drivers/drivers/media/pci/intel/ipu6] Error 2
make[2]: *** [scripts/Makefile.build:466: /home/juno/ipu6-drivers/drivers/media/pci/intel] Error 2
make[1]: *** [Makefile:1847: /home/juno/ipu6-drivers] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.19.8-1-MANJARO/build'
make: *** [Makefile:17: all] Error 2

```

junocomp, 2022-09-17:
Is `ov13858` supported on the master branch?

ccgjimlai, 2022-10-26:
> Is `ov13858` supported on the master branch?

OV13858 is currently not supported, as we have not validate it.

junocomp, 2022-11-09:
OV13858 is now part of Ubuntu's and Manjaro's kernel. I can add it to my modprobe list and it seems to find it when launching icamerasrc. I still get an error.

```
sudo -E gst-launch-1.0 icamerasrc device-name=ov13858-uf af-mode=2 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! waylandsink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.002479277
Setting pipeline to NULL ...
Freeing pipeline ...

```

junocomp, 2022-11-09:
I dont know what this means.

`ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
`

junocomp, 2022-11-15:
@brianmercer I didn't, but I will try your solution

brianmercer, 2022-11-15:
It's not a solution. I just didn't get as far as you. I was asking if you'd made any changes.

I get this:

```
sudo -E gst-launch-1.0 icamerasrc device-name=ov13858-uf af-mode=2 '!' video/x-raw,format=NV12,width=1280,height=720 '!' videoconvert '!' waylandsink
WARNING: erroneous pipeline: could not set property "device-name" in element "camerasrc0" to "ov13858-uf"
```

`gst-inspect-1.0 icamerasrc` doesn't show our camera as a valid value for device-name.

I haven't tried adding it yet.

junocomp, 2022-11-15:
I am trying right now, I will get back soon

brianmercer, 2022-11-15:
Those icamera values look like they come from https://github.com/intel/ipu6-camera-hal/tree/2a7c9c263d2cb20335a2b3445d2b6e9f47f3dd10/config/linux/ipu6ep/sensors

junocomp, 2022-11-15:
I just tried now and I am still getting the same error.

junocomp, 2022-11-19:
@brianmercer did you managed to get it working yet on jasper lake?

brianmercer, 2022-11-20:
Sorry, no.

It would be nice if we had some word that they plan to support jasper lake and ov13858.

junocomp, 2022-11-20:
Maybe @hao-yao can help with this.

junocomp, 2022-11-23:
I am getting pretty close to build the `ipu6-drivers` from the jsl branch on `kernel 6.1`. 

I replaced the `ov13858_intel.c` content from the kernel `6.1 branch `

https://github.com/torvalds/linux/blob/master/drivers/media/i2c/ov13858.c

It now partially builds but it can't build `ipu-bus.c`. I also noticed this file **is not** on the master branch. 

If anyone know how to fix this, please let me know and I can continue to try to build it for jasperlake.

```
$ make -j4
make -C /lib/modules/6.1.0-0-amd64/build M=/home/caligaris/ipu6-drivers modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-0-amd64'
  CC [M]  /home/caligaris/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu.o
  CC [M]  /home/caligaris/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.o
  CC [M]  /home/caligaris/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-dma.o
  CC [M]  /home/caligaris/ipu6-drivers/drivers/media/pci/intel/ipu6/ipu6.o
/home/caligaris/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.c:137:19: error: initialization of ‘void (*)(struct device *)’ from incompatible pointer type ‘int (*)(struct device *)’ [-Werror=incompatible-pointer-types]
  137 |         .remove = ipu_bus_remove,
      |                   ^~~~~~~~~~~~~~
/home/caligaris/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.c:137:19: note: (near initialization for ‘ipu_bus.remove’)
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-6.1.0-0-common/scripts/Makefile.build:255: /home/caligaris/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-6.1.0-0-common/scripts/Makefile.build:505: /home/caligaris/ipu6-drivers/drivers/media/pci/intel/ipu6] Error 2
make[2]: *** [/usr/src/linux-headers-6.1.0-0-common/scripts/Makefile.build:505: /home/caligaris/ipu6-drivers/drivers/media/pci/intel] Error 2
make[1]: *** [/usr/src/linux-headers-6.1.0-0-common/Makefile:2017: /home/caligaris/ipu6-drivers] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-0-amd64'
make: *** [Makefile:17: all] Error 2
```

junocomp, 2023-03-01:
Any updates on jasper lake, did anyone managed to get it working?