intel/ipu6-drivers #38: Can't load firmware

junocomp, 2022-09-09:
I installed ipu6 and ivsc from this PPA. Everything installs fine but it won't load the firmware.

https://launchpad.net/~vicamo/+archive/ubuntu/mipi-ipu6/+packages

```
$ sudo dmesg | grep ipu
[    3.949259] intel-ipu6 intel-ipu: enabling device (0000 -> 0002)
[    3.949469] intel-ipu6 intel-ipu: Device 0x4e19 (rev: 0x0)
[    3.949499] intel-ipu6 intel-ipu: physical base address 0x6000000000
[    3.949501] intel-ipu6 intel-ipu: mapped as: 0x000000001158a7df
[    3.949732] intel-ipu6 intel-ipu: IPU in secure mode
[    3.951076] intel-ipu6 intel-ipu: IPC reset done
[    3.951083] intel-ipu6 intel-ipu: cpd file name: intel/ipu6se_fw.bin
[    3.955227] intel-ipu6 intel-ipu: Direct firmware load for intel/ipu6se_fw.bin failed with error -2
[    3.955238] intel-ipu6 intel-ipu: Requesting signed firmware failed
[    3.955263] intel-ipu6: probe of intel-ipu failed with error -2

```

junocomp, 2022-09-12:
Managed to load the jsl firmware. Closing

closed 2022-09-12 by junocomp

Sorunome, 2022-11-19:
> Managed to load the jsl firmware. Closing

ok, and how did you do it?

junocomp, 2022-11-19:
@Sorunome 

```
git clone https://github.com/intel/ipu6-camera-bins/tree/jsl_penguin_peak
cd ipu6-camera-bins
git checkout jsl_penguin_peak
sudo cp -rP include/* /usr/include/
sudo cp -rP lib/* /usr/lib/
```

Restart after and check dmesg, it should show that it has loaded.

MrTrukiny, 2023-08-14:
The link to the repo mentioned here is no longer working!