intel/ipu6-drivers #40: Failed to enable ov01a10 on Dell XPS 9320: int3472: GPIO type 0x02 unknown

vicamo, 2022-09-22:
[dmesg.Dell-XPS9320.ADL-P.ov01a10.unknown-gpio-type-0x02.txt](https://github.com/intel/ipu6-drivers/files/9626276/dmesg.Dell-XPS9320.ADL-P.ov01a10.unknown-gpio-type-0x02.txt)
```
[    3.817346] int3472-discrete INT3472:01: INT3472 seems to have no dependents.
[    3.821642] int3472-discrete INT3472:06: sensor name: 0BF122N3
[    3.821967] int3472-discrete INT3472:06: GPIO type 0x02 unknown; the sensor may not work
```
Additional patch 
[dmesg.Dell-XPS9320.ADL-P.ov01a10.unknown-gpio-type-0x02.txt](https://github.com/intel/ipu6-drivers/files/9626284/dmesg.Dell-XPS9320.ADL-P.ov01a10.unknown-gpio-type-0x02.txt) has been applied to match sensor config.

vicamo, 2022-09-27:
Turns out the major problem is INTEL_IOMMU being disabled by default in https://bugs.launchpad.net/bugs/1971699.

closed 2022-09-27 by vicamo

ccgjimlai, 2022-10-26:
Hi Vicamo,
Since OV01A is working now on your device, please close this issue

vicamo, 2022-10-26:
> Hi Vicamo, Since OV01A is working now on your device, please close this issue

That's already closed a month ago.