intel/ipu6-drivers #41: Failed to enable ov02c10 on Dell Latitude 7640(Oasis): int3472: No supply name defined

vicamo, 2022-09-22:
[dmesg.Dell-Latitude7640.Oasis.ov02c10.no-supply-name-defined.txt](https://github.com/intel/ipu6-drivers/files/9626682/dmesg.Dell-Latitude7640.Oasis.ov02c10.no-supply-name-defined.txt)
```
[    1.418430] intel_pmc_core INT33A1:00:  initialized
[    4.495069] int3472-discrete INT3472:01: INT3472 seems to have no dependents.
[    4.501975] int3472-discrete INT3472:06: sensor name: LM3643
[    4.502906] int3472-discrete INT3472:06: No supply name defined
[    4.502909] int3472-discrete INT3472:06: error -ENODEV: Failed to map regulator to sensor
```
An additional patch
[patch](https://github.com/intel/ipu6-drivers/files/9626709/0001-UBUNTU-SAUCE-platform-x86-int3472-support-ov01a10-ov02c10-sensors.patch.txt) is applied in advance to match sensor config.

vicamo, 2022-09-27:
```
$ sudo dmesg|grep ov0
[   19.363273] ov02c10 i2c-OVTI02C1:00: Acquire VSC failed
[   19.363687] ov02c10: probe of i2c-OVTI02C1:00 failed with error -110
$ lsmod|grep vsc
mei_vsc                40960  2
mei                   139264  10 mei_ace,mei_csi,mei_hdcp,mei_pxp,mei_vsc,mei_me
intel_vsc              16384  3 mei_ace,mei_csi,ov02c10
$ sudo dmesg|grep vsc
...
[   14.014241] vsc_ace spi-INTC1094:00-5db76cf6-0a68-4ed6-9b78-0361635e2447: recv firmware id
[   19.255230] vsc_ace spi-INTC1094:00-5db76cf6-0a68-4ed6-9b78-0361635e2447: command 19 notify reply timeout
[   23.979836] mei_vsc spi-INTC1094:00: spi_validate_packet crc error cmd 13 0x0 0x489b8c1d
[   23.980059] mei_vsc spi-INTC1094:00: corrupted message header 0x00000000
[   23.980063] mei_vsc spi-INTC1094:00: mei_irq_read_handler ret = -74.
[   23.980076] mei_vsc spi-INTC1094:00: unexpected reset: dev_state = ENABLED fw status = 
[   33.666568] vsc_ace spi-INTC1094:00-5db76cf6-0a68-4ed6-9b78-0361635e2447: recv firmware id
```

vicamo, 2022-09-27:
And even the sensor is bound, still get following error when starting camera streaming:
```
[   14.056843] vsc_ace spi-INTC1094:00-5db76cf6-0a68-4ed6-9b78-0361635e2447: recv firmware id
[   19.259293] vsc_ace spi-INTC1094:00-5db76cf6-0a68-4ed6-9b78-0361635e2447: command 19 notify reply timeout
[   24.015983] mei_vsc spi-INTC1094:00: spi_validate_packet crc error cmd 13 0x0 0x489b8c1d
[   24.016132] mei_vsc spi-INTC1094:00: corrupted message header 0x00000000
[   24.016135] mei_vsc spi-INTC1094:00: mei_irq_read_handler ret = -74.
[   24.016146] mei_vsc spi-INTC1094:00: unexpected reset: dev_state = ENABLED fw status = 
[   33.301898] vsc_ace spi-INTC1094:00-5db76cf6-0a68-4ed6-9b78-0361635e2447: recv firmware id
```

vicamo, 2022-09-29:
Same error for another OV02C10 equipped model Oasis 14:
```
[   13.361356] vsc_ace spi-INTC1009:00-5db76cf6-0a68-4ed6-9b78-0361635e2447: recv firmware id
[   18.500359] vsc_ace spi-INTC1009:00-5db76cf6-0a68-4ed6-9b78-0361635e2447: command 19 notify reply timeout
[   23.330185] mei_vsc spi-INTC1009:00: spi_validate_packet crc error cmd 13 0x0 0x489b8c1d
[   23.330347] mei_vsc spi-INTC1009:00: corrupted message header 0x00000000
[   23.330348] mei_vsc spi-INTC1009:00: mei_irq_read_handler ret = -74.
[   23.330352] mei_vsc spi-INTC1009:00: unexpected reset: dev_state = ENABLED fw status = 
[   32.314544] vsc_ace spi-INTC1009:00-5db76cf6-0a68-4ed6-9b78-0361635e2447: recv firmware id
```
It seems that command 19 is **IPU_OWN_CAMERA**. Is there any prerequisites to meet or any other way to diagnose that timeout?

hao-yao, 2022-10-08:
int3472 driver is for the cameras without CVF, so I think you can remove that patch and have a try.

vicamo, 2022-10-12:
[dmesg.6.0.0-9005-oem.Oasis16.ov02c10.ubsan-error.txt](https://github.com/intel/ipu6-drivers/files/9761666/dmesg.6.0.0-9005-oem.Oasis16.ov02c10.ubsan-error.txt)
Works after a few tweaks (reinstall system, bump to upstream HEAD, use kprobe, ...) except an UBSAN error filed as #52.

Revision info:
* ipu6-driver:
  - upstream commit: https://github.com/intel/ipu6-drivers/commit/4ef83f208511678a417dc1546d460d111919271a
  - dkms source: https://git.launchpad.net/~vicamo/+git/intel-ipu6-dkms/commit/?h=ubuntu/devel&id=c1fdd8acfe2b68e68efc6257589f79157103e2aa
  - dkms PPA: https://launchpad.net/~vicamo/+archive/ubuntu/ppa-1972109
* ivsc-driver:
  - upstream commit: https://github.com/intel/ivsc-driver/commit/99a53bd7160452bbbcc26bcb662cad33c13b36d1
  - dkms source: https://git.launchpad.net/~vicamo/+git/intel-vsc-dkms/commit/?h=ubuntu/devel&id=11800d9ed487a7f9f19c16675fddad796a00ed5d
  - dkms PPA: https://launchpad.net/~vicamo/+archive/ubuntu/ppa-1972106
* kernel:
  - upstream revision: v6.0
  - source: https://git.launchpad.net/~canonical-hwe-team/hwe-next/+git/intermediate-kernel/commit/?h=linux-oem-6.0/jammy/experimental&id=d38678632c2b66e6d959bd2d3aac7962733e1815
  - PPA: https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel
* software:
  - libipu6ep: 0~git202209280632.d96c8fe-1~ubuntu22.04.1
  - libcamhal-common, libcamhal: 0~git202210110204.8bad42c~ubuntu22.04.1
  - gstreamer1.0-icamera, libgsticamerainterface-1.0-1: 0~git202209280632.3b7cdb9~ubuntu22.04.14
  - linux-firmware: 20220329.git681281e4-0ubuntu3.6

closed 2022-10-12 by vicamo