intel/ipu6-drivers #48: Error when loading intel-ipu6 module

randomehro, 2022-10-01:
I am running 5.19.12-1-default on OpenSUSE Tumbleweed. My hardware is LENOVO 21CBCTO1WW/21CBCTO1WW, BIOS N3AET65W (1.30 ). I got an error when loading ipu6 module during boot:

```
[    6.746457] intel-ipu6 intel-ipu: enabling device (0000 -> 0002)
[    6.748265] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x4)
[    6.748301] intel-ipu6 intel-ipu: physical base address 0x603c000000
[    6.748305] intel-ipu6 intel-ipu: mapped as: 0x00000000496d1881
[    6.748422] intel-ipu6 intel-ipu: IPU in secure mode
[    6.748426] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[    6.748428] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[    6.754013] proc_thermal_pci 0000:00:04.0: enabling device (0000 -> 0002)
[    6.759019] intel_rapl_common: Found RAPL domain package
[    6.762885] intel-ipu6 intel-ipu: IPC reset done
[    6.762891] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[    6.763627] intel-ipu6 intel-ipu: FW version: 20220510
[    6.763955] BUG: kernel NULL pointer dereference, address: 0000000000000304
[    6.763962] #PF: supervisor read access in kernel mode
[    6.763966] #PF: error_code(0x0000) - not-present page
[    6.763969] PGD 0 P4D 0 
[    6.763975] Oops: 0000 [#1] PREEMPT SMP NOPTI
[    6.763981] CPU: 5 PID: 665 Comm: systemd-udevd Tainted: G     U  W  OE     5.19.12-1-default #1 openSUSE Tumbleweed 665b142383228b747d71fcd0f7280693267f8cd7
[    6.763988] Hardware name: LENOVO 21CBCTO1WW/21CBCTO1WW, BIOS N3AET65W (1.30 ) 08/02/2022
[    6.763991] RIP: 0010:ipu_pci_probe.cold+0x3f0/0x90e [intel_ipu6]
[    6.764023] Code: 48 89 fa 48 85 ff 0f 84 b8 00 00 00 48 8b bf 10 03 00 00 48 85 ff 0f 84 ba 00 00 00 e8 56 b6 ff ff 48 8b 43 18 48 85 c0 74 11 <48> 8b b8 10 03 00 00 48 85 ff 74 05 e8 3c b6 ff ff 48 8b 53 20 48
[    6.764028] RSP: 0018:ffffb71440747b58 EFLAGS: 00010282
[    6.764033] RAX: fffffffffffffff4 RBX: ffff88b6c6789828 RCX: 0000000000000000
[    6.764036] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[    6.764039] RBP: ffff88b9e81bb000 R08: ffffb71440747af0 R09: ffff88b6c3f425a0
[    6.764043] R10: 0000000000000001 R11: 0000000000000001 R12: ffff88b9e81bb0d0
[    6.764046] R13: ffffb71445000000 R14: 00000000fffffff4 R15: ffffb71445000000
[    6.764049] FS:  00007f2330946b00(0000) GS:ffff88ba0f540000(0000) knlGS:0000000000000000
[    6.764054] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    6.764057] CR2: 0000000000000304 CR3: 000000010b3e4002 CR4: 0000000000770ee0
[    6.764062] PKRU: 55555554
[    6.764064] Call Trace:
[    6.764069]  <TASK>
[    6.764076]  local_pci_probe+0x3e/0x80
[    6.764087]  pci_device_probe+0xc3/0x230
[    6.764094]  really_probe+0x1bf/0x390
[    6.764101]  __driver_probe_device+0xff/0x170
[    6.764106]  driver_probe_device+0x1f/0x90
[    6.764111]  __driver_attach+0xcf/0x1d0
[    6.764116]  ? __device_attach_driver+0x110/0x110
[    6.764121]  bus_for_each_dev+0x73/0xc0
[    6.764126]  bus_add_driver+0x160/0x210
[    6.764131]  driver_register+0x89/0xe0
[    6.764136]  ? 0xffffffffc0f83000
[    6.764140]  ipu_init+0x3b/0x1000 [intel_ipu6 47b5614a0b9b9e937a29eb4630298c70c28ff96d]
[    6.764161]  do_one_initcall+0x41/0x200
[    6.764169]  ? kmem_cache_alloc_trace+0x177/0x360
[    6.764177]  do_init_module+0x4a/0x1f0
[    6.764185]  __do_sys_finit_module+0xac/0x120
[    6.764192]  do_syscall_64+0x58/0x80
[    6.764200]  ? do_syscall_64+0x67/0x80
[    6.764206]  ? exit_to_user_mode_prepare+0x190/0x1f0
[    6.764211]  ? syscall_exit_to_user_mode+0x17/0x40
[    6.764216]  ? do_syscall_64+0x67/0x80
[    6.764221]  ? syscall_exit_to_user_mode+0x17/0x40
[    6.764225]  ? do_syscall_64+0x67/0x80
[    6.764230]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
[    6.764237] RIP: 0033:0x7f23313128fd
[    6.764241] Code: 5d c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d e3 74 0d 00 f7 d8 64 89 01 48
[    6.764245] RSP: 002b:00007fff1e93fa38 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    6.764250] RAX: ffffffffffffffda RBX: 0000557837186240 RCX: 00007f23313128fd
[    6.764253] RDX: 0000000000000000 RSI: 00007f233187e05d RDI: 0000000000000006
[    6.764256] RBP: 00007f233187e05d R08: 0000000000000000 R09: 0000557836fb3cd0
[    6.764259] R10: 0000000000000006 R11: 0000000000000246 R12: 0000000000020000
[    6.764262] R13: 0000000000000000 R14: 000055783718af30 R15: 0000000000000000
[    6.764266]  </TASK>
[    6.764268] Modules linked in: processor_thermal_device_pci(+) processor_thermal_device processor_thermal_rfim processor_thermal_mbox processor_thermal_rapl fjes(-) acpi_cpufreq(-) intel_vsec pcc_cpufreq(-) intel_rapl_common intel_ipu6(OE+) i2c_dev igen6_edac thermal tiny_power_button button fan thinkpad_acpi ledtrig_audio platform_profile snd mei_vsc(OE) mei soundcore int3403_thermal int340x_thermal_zone soc_button_array ac ov2740(OE) v4l2_fwnode v4l2_async videodev mc intel_skl_int3472_tps68470 tps68470_regulator intel_skl_int3472_discrete intel_hid sparse_keymap int3400_thermal acpi_thermal_rel intel_pmc_core acpi_tad acpi_pad nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat nf_tables ip6table_nat ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_raw vboxnetadp(O) iptable_security vboxnetflt(O)
[    6.764334]  nfnetlink rfkill ip6table_filter ip6_tables iptable_filter bpfilter vboxdrv(O) qrtr dmi_sysfs nls_iso8859_1 nls_cp437 vfat fat fuse configfs ip_tables x_tables ext4 mbcache jbd2 hid_sensor_custom hid_sensor_hub hid_generic intel_ishtp_hid i915 drm_buddy drm_display_helper crct10dif_pclmul xhci_pci crc32_pclmul xhci_pci_renesas crc32c_intel xhci_hcd cec ucsi_acpi nvme rc_core typec_ucsi intel_ish_ipc i2c_hid_acpi ghash_clmulni_intel aesni_intel crypto_simd cryptd serio_raw roles usbcore intel_ishtp nvme_core typec ttm wmi i2c_hid battery video pinctrl_tigerlake coretemp sg dm_multipath dm_mod scsi_dh_rdac scsi_dh_emc scsi_dh_alua msr efivarfs
[    6.764392] Unloaded tainted modules: acpi_cpufreq():1 fjes():1 pcc_cpufreq():1 acpi_cpufreq():1 fjes():1 pcc_cpufreq():1 acpi_cpufreq():1 fjes():1 pcc_cpufreq():1 acpi_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 fjes():1 acpi_cpufreq():1 pcc_cpufreq():1 fjes():1 pcc_cpufreq():1 acpi_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1
[    6.764460]  pcc_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 pcc_cpufreq():1 acpi_cpufreq():1 acpi_cpufreq():1
[    6.764479] CR2: 0000000000000304
[    6.764483] ---[ end trace 0000000000000000 ]---
[    6.764486] RIP: 0010:ipu_pci_probe.cold+0x3f0/0x90e [intel_ipu6]
[    6.764509] Code: 48 89 fa 48 85 ff 0f 84 b8 00 00 00 48 8b bf 10 03 00 00 48 85 ff 0f 84 ba 00 00 00 e8 56 b6 ff ff 48 8b 43 18 48 85 c0 74 11 <48> 8b b8 10 03 00 00 48 85 ff 74 05 e8 3c b6 ff ff 48 8b 53 20 48
[    6.764513] RSP: 0018:ffffb71440747b58 EFLAGS: 00010282
[    6.764516] RAX: fffffffffffffff4 RBX: ffff88b6c6789828 RCX: 0000000000000000
[    6.764519] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[    6.764522] RBP: ffff88b9e81bb000 R08: ffffb71440747af0 R09: ffff88b6c3f425a0
[    6.764525] R10: 0000000000000001 R11: 0000000000000001 R12: ffff88b9e81bb0d0
[    6.764527] R13: ffffb71445000000 R14: 00000000fffffff4 R15: ffffb71445000000
[    6.764531] FS:  00007f2330946b00(0000) GS:ffff88ba0f540000(0000) knlGS:0000000000000000
[    6.764535] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    6.764538] CR2: 0000000000000304 CR3: 000000010b3e4002 CR4: 0000000000770ee0
[    6.764541] PKRU: 55555554
```

drogenlied, 2022-10-06:
Hardware name: LENOVO 21CDCTO1WW/21CDCTO1WW, BIOS N3AET65W (1.30 ) 08/02/2022

I can confirm the null pointer dereference  in `ipu_pci_probe` as well, same BIOS, slightly different Laptop configuration, on Debian Bookworm/Sid, happens on kernels 5.18, 5.19 and custom compiled 6.0-rc.

norbusan, 2022-10-10:
My wild guess is that it has to do with the patch int3472-support-independent-clock-and-LED-gpios-5.17+ included in this repo is not applied with dkms builds, and furthermore, it does not contain entries for the X1 Carbon:
```
--- a/drivers/platform/x86/intel/int3472/discrete.c
+++ b/drivers/platform/x86/intel/int3472/discrete.c
@@ -57,11 +57,17 @@ static const struct int3472_gpio_function_remap ov2680_gpio_function_remaps[] =
 
 static const struct int3472_sensor_config int3472_sensor_configs[] = {
        /* Lenovo Miix 510-12ISK - OV2680, Front */
-       { "GNDF140809R", { 0 }, ov2680_gpio_function_remaps },
+       { "GNDF140809R", { 0 }, ov2680_gpio_function_remaps, false },
        /* Lenovo Miix 510-12ISK - OV5648, Rear */
-       { "GEFF150023R", REGULATOR_SUPPLY("avdd", NULL), NULL },
+       { "GEFF150023R", REGULATOR_SUPPLY("avdd", NULL), NULL, false },
        /* Surface Go 1&2 - OV5693, Front */
-       { "YHCU", REGULATOR_SUPPLY("avdd", NULL), NULL },
+       { "YHCU", REGULATOR_SUPPLY("avdd", NULL), NULL, false },
+       /* Dell Latitude 9420 - OV01A1S, Front */
+       { "0BF111N3", { 0 }, NULL, true },
+       /* Dell Latitude 9420 - HM11B1, Front */
+       { "9BF123N3", { 0 }, NULL, true },
+       /* Lenovo X1 Yoga - OV2740, Front */
+       { "CJFLE23", { 0 }, NULL, true },
 };
```

I guess there some entry for the X1 Carbon (which I also have) is necessary!?

fhloston, 2022-10-10:
Seing this null pointer dereference with the patch int3472-support-independent-clock-and-LED-gpios-5.17+ included.

norbusan, 2022-10-10:
@fhloston On which hardware?

fhloston, 2022-10-10:
@norbusan 
LENOVO 21CDS04A00/21CDS04A00, BIOS N3AET66W (1.31 ) 09/09/2022
X1 Yoga Gen7

fhloston, 2022-10-11:
```

fhloston, 2022-10-11:
```
[   29.499244] platform INT3472:05: deferred probe pending
[  485.120306] intel-ipu6 intel-ipu: enabling device (0000 -> 0002)
[  485.120500] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[  485.120528] intel-ipu6 intel-ipu: physical base address 0x603c000000
[  485.120529] intel-ipu6 intel-ipu: mapped as: 0x000000004348259f
[  485.120532] intel-ipu6 intel-ipu: isys_base: 0xffffb82d49000000
[  485.120533] intel-ipu6 intel-ipu: psys_base: 0xffffb82d49000000
[  485.120654] intel-ipu6 intel-ipu: IPU in secure mode
[  485.120655] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[  485.120656] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[  485.121165] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: unexpected CSR 0x0
[  485.121669] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: unexpected CSR 0x0
[  485.122184] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: unexpected CSR 0x0
[  485.122700] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: unexpected CSR 0x0
[  485.123216] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: unexpected CSR 0x0
[  485.123732] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: unexpected CSR 0x0
[  485.124248] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: unexpected CSR 0x0
[  485.124764] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: unexpected CSR 0x0
[  485.125272] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: unexpected CSR 0x0
[  485.125785] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: unexpected CSR 0x0
[  485.134480] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset:IPC_PEER_COMP_ACTIONS_RST_PHASE1 & IPC_PEER_COMP_ACTIONS_RST_PHASE2
[  485.136022] intel-ipu6 intel-ipu: ipu_buttress_ipc_reset: IPC_PEER_COMP_ACTIONS_RST_PHASE2
[  485.136340] intel-ipu6 intel-ipu: IPC reset done
[  485.136355] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[  485.147637] intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
[  485.147663] intel-ipu6 intel-ipu: FW version: 20220510
[  485.148154] BUG: kernel NULL pointer dereference, address: 00000000000002fc
[  485.148159] #PF: supervisor read access in kernel mode
[  485.148162] #PF: error_code(0x0000) - not-present page
[  485.148166] PGD 0 P4D 0 
[  485.148173] Oops: 0000 [#1] PREEMPT SMP NOPTI
[  485.148179] CPU: 0 PID: 3779 Comm: modprobe Tainted: P     U  W  OE      6.0.0 #1 
[  485.148186] Hardware name: LENOVO 21CDS04A00/21CDS04A00, BIOS N3AET66W (1.31 ) 09/09/2022
[  485.148189] RIP: 0010:ipu_pci_probe.cold+0x3fe/0x91c [intel_ipu6]
[  485.148230] Code: 48 89 fa 48 85 ff 0f 84 b8 00 00 00 48 8b bf 08 03 00 00 48 85 ff 0f 84 ba 00 00 00 e8 c6 b5 ff ff 48 8b 43 18 48 85 c0 74 11 <48> 8b b8 08 03 00 00 48 85 ff 74 05 e8 ac b5 ff ff 48 8b 53 20 48
[  485.148235] RSP: 0018:ffffb82d481bbbf0 EFLAGS: 00010282
[  485.148240] RAX: fffffffffffffff4 RBX: ffff95d50832f428 RCX: 0000000000000000
[  485.148244] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[  485.148246] RBP: ffff95d501700000 R08: ffff95d501700370 R09: 0000000000000024
[  485.148249] R10: 0000000000000001 R11: 0000000000000000 R12: ffff95d5017000d0
[  485.148252] R13: ffffb82d49000000 R14: 00000000fffffff4 R15: ffffb82d49000000
[  485.148256] FS:  00007f080dc00480(0000) GS:ffff95dc3f400000(0000) knlGS:0000000000000000
[  485.148260] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  485.148264] CR2: 00000000000002fc CR3: 000000014f318003 CR4: 0000000000770ef0
[  485.148268] PKRU: 55555554
[  485.148271] Call Trace:
[  485.148276]  <TASK>
[  485.148285]  local_pci_probe+0x3e/0x80
[  485.148300]  pci_device_probe+0xc3/0x230
[  485.148309]  really_probe+0xdb/0x380
[  485.148316]  ? pm_runtime_barrier+0x50/0x90
[  485.148322]  __driver_probe_device+0x78/0x170
[  485.148327]  driver_probe_device+0x1f/0x90
[  485.148332]  __driver_attach+0xd1/0x1d0
[  485.148337]  ? __device_attach_driver+0x110/0x110
[  485.148341]  bus_for_each_dev+0x84/0xd0
[  485.148351]  bus_add_driver+0x1ae/0x200
[  485.148356]  driver_register+0x89/0xe0
[  485.148361]  ? 0xffffffffc09a4000
[  485.148365]  ipu_init+0x3b/0x1000 [intel_ipu6]
[  485.148387]  ? 0xffffffffc09a4000
[  485.148389]  do_one_initcall+0x56/0x220
[  485.148400]  do_init_module+0x4a/0x200
[  485.148408]  __do_sys_finit_module+0xac/0x120
[  485.148417]  do_syscall_64+0x37/0xc0
[  485.148427]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
[  485.148435] RIP: 0033:0x7f080d303859
[  485.148440] Code: 08 44 89 e0 5b 41 5c c3 66 0f 1f 84 00 00 00 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 87 05 0f 00 f7 d8 64 89 01 48
[  485.148444] RSP: 002b:00007ffda1018d08 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[  485.148449] RAX: ffffffffffffffda RBX: 0000562ec4d89ff0 RCX: 00007f080d303859
[  485.148452] RDX: 0000000000000000 RSI: 0000562ec4d8a3c0 RDI: 0000000000000003
[  485.148455] RBP: 0000562ec4d8a3c0 R08: 0000000000000000 R09: 0000000000000014
[  485.148457] R10: 0000000000000003 R11: 0000000000000246 R12: 0000000000040000
[  485.148460] R13: 0000000000000000 R14: 0000562ec4d89f80 R15: 0000000000000015
[  485.148465]  </TASK>
[  485.148467] Modules linked in: intel_ipu6(OE+) nvme_fabrics ctr ccm rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device qrtr cmac algif_hash algif_skcipher af_alg bnep snd_ctl_led snd_soc_skl_hda_dsp snd_soc_intel_hda_dsp_common snd_sof_probes snd_soc_hdac_hdmi snd_soc_dmic mei_hdcp intel_rapl_msr x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass rapl intel_cstate intel_uncore pcspkr serio_raw efi_ [... cut ...]
[  485.888660] RBP: ffff95d501700000 R08: ffff95d501700370 R09: 0000000000000024
[  485.888674] R10: 0000000000000001 R11: 0000000000000000 R12: ffff95d5017000d0
[  485.888674] R13: ffffb82d49000000 R14: 00000000fffffff4 R15: ffffb82d49000000
[  485.888675] FS:  00007f080dc00480(0000) GS:ffff95dc3f400000(0000) knlGS:0000000000000000
[  485.888677] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  485.888694] CR2: 00000000000002fc CR3: 000000014f318003 CR4: 0000000000770ef0
[  485.888695] PKRU: 55555554
```

jeffreyfq, 2022-10-12:
Similar on HP Spectre x360 16-f1010TX (i7-1260P) @ 5.19.0-15-generic (K)Ubuntu 20.10:

[ 3594.980687] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[ 3594.980793] intel-ipu6 intel-ipu: physical base address 0x612c000000
[ 3594.980796] intel-ipu6 intel-ipu: mapped as: 0x000000001aa49395
[ 3594.980957] intel-ipu6 intel-ipu: IPU in secure mode
[ 3594.980962] intel-ipu6 intel-ipu: IPU secure touch = 0x80000000
[ 3594.980965] intel-ipu6 intel-ipu: IPU camera mask = 0x0
[ 3594.998356] intel-ipu6 intel-ipu: IPC reset done
[ 3594.998369] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[ 3595.030262] intel-ipu6 intel-ipu: FW version: 20211011
[ 3595.030954] BUG: kernel NULL pointer dereference, address: 00000000000002fc
[ 3595.030967] #PF: supervisor read access in kernel mode
[ 3595.030973] #PF: error_code(0x0000) - not-present page
[ 3595.030980] PGD 0 P4D 0 
[ 3595.030991] Oops: 0000 [#1] PREEMPT SMP NOPTI
[ 3595.031003] CPU: 8 PID: 52603 Comm: modprobe Tainted: G     U  W  OE     5.19.0-15-generic #15-Ubuntu
[ 3595.031016] Hardware name: HP HP Spectre x360 2-in-1 Laptop 16-f1xxx/891D, BIOS F.05 03/23/2022
[ 3595.031022] RIP: 0010:ipu_pci_probe.cold+0x3fd/0x8e7 [intel_ipu6]
[ 3595.031083] Code: 48 89 fa 48 85 ff 0f 84 b8 00 00 00 48 8b bf 08 03 00 00 48 85 ff 0f 84 ba 00 00 00 e8 b7 ac ff ff 49 8b 46 18 48 85 c0 74 11 <48> 8b b8 08 03 00 00 48 85 ff 74 05 e8 9d ac ff ff 49 8b 56 20 48
[ 3595.031092] RSP: 0018:ffffbc9e09cdb9b8 EFLAGS: 00010282
[ 3595.031102] RAX: fffffffffffffff4 RBX: ffff9e01c2d47000 RCX: 0000000000000000
[ 3595.031109] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[ 3595.031115] RBP: ffffbc9e09cdba00 R08: ffffbc9e09cdb950 R09: ffff9e01d2e4ea80
[ 3595.031121] R10: 0000000000000001 R11: 0000000000000000 R12: ffff9e01c2d470d0
[ 3595.031127] R13: ffff9e01d2e4eaa8 R14: ffff9e076ac35428 R15: 00000000fffffff4
[ 3595.031134] FS:  00007f0b64866040(0000) GS:ffff9e094fa00000(0000) knlGS:0000000000000000
[ 3595.031142] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 3595.031148] CR2: 00000000000002fc CR3: 0000000204c34003 CR4: 0000000000770ee0
[ 3595.031156] PKRU: 55555554
[ 3595.031161] Call Trace:
[ 3595.031167]  <TASK>
[ 3595.031182]  local_pci_probe+0x44/0x80
[ 3595.031200]  pci_call_probe+0x55/0x180
[ 3595.031213]  pci_device_probe+0x84/0x110
[ 3595.031223]  really_probe+0x1ce/0x3a0
[ 3595.031238]  __driver_probe_device+0x11b/0x190
[ 3595.031250]  driver_probe_device+0x24/0xd0
[ 3595.031262]  __driver_attach+0xd8/0x200
[ 3595.031274]  ? __device_attach_driver+0x120/0x120
[ 3595.031286]  bus_for_each_dev+0x79/0xc0
[ 3595.031298]  driver_attach+0x1e/0x30
[ 3595.031309]  bus_add_driver+0x178/0x220
[ 3595.031320]  driver_register+0x8f/0xf0
[ 3595.031333]  ? 0xffffffffc0a8f000
[ 3595.031340]  __pci_register_driver+0x62/0x70
[ 3595.031348]  ipu_init+0x3f/0x1000 [intel_ipu6]
[ 3595.031382]  ? 0xffffffffc0a8f000
[ 3595.031386]  do_one_initcall+0x47/0x210
[ 3595.031398]  ? kmem_cache_alloc_trace+0x181/0x300
[ 3595.031408]  ? do_init_module+0x27/0x200
[ 3595.031418]  do_init_module+0x50/0x200
....

vicamo, 2022-10-21:
> Similar on HP Spectre x360 16-f1010TX (i7-1260P) @ 5.19.0-15-generic and (K)Ubuntu 20.10:

Hi, wondering if https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel works for you:
```
$ sudo add-apt-repository ppa:canonical-hwe-team/intermediate-kernel
$ sudo apt install linux-image-generic linux-modules-ipu6-generic linux-modules-ivsc-generic
```

jeffreyfq, 2022-10-21:
> Hi, wondering if https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel works for you:
>
> ```
> $ sudo add-apt-repository ppa:canonical-hwe-team/intermediate-kernel
> $ sudo apt [...]

Thanks for your suggestion.
I tried it out, but seems to generate the same error.
This is the dmesg output, in case it can be useful:
[5.19.0-9021-generic_dmesg.txt](https://github.com/intel/ipu6-drivers/files/9839804/5.19.0-9021-generic_dmesg.txt)

------------------------------

The point is, I tried something crazy, I merged the driver with the current `next` branch (development branch).
It was a little bit tricky, as the interface changed even more (it seems media entities had previously only one pipeline. But in the new kernel they have multiple pods that can have a pipeline each. A compatibility function exists that goes through all pods and returns the first pipeline that it can find. As it is assumes that usually there is only one pipeline involved. I changed the code to call this utility function. Also I looped through the pods to find the pipeline during closing/deinitalization. But I only close only the first as in the utility function. As I assume that the camera driver only uses one pipeline, I have no idea if this is ok?) 
This is the full diff between kernel branch `next-20221019` and the integrated/modified driver in my local branch:
[gitdiff-kernel.txt](https://github.com/intel/ipu6-drivers/files/9839788/gitdiff-kernel.txt)
Kernel compiles but still has the same error!
To regenerate the next-kernel something like this should work:
```
git clone https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git  
cd linux  
git remote add linux-next https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git
git fetch linux-next
git fetch --tags linux-next
git checkout master
git remote update
git checkout -b hp_spectre_i7-1260p next-20221019
git apply <insert ~/Download/ or where the file is>gitdiff-kernel.txt
```
Then you have to add the `.config` file, mine was this: [config.txt](https://github.com/intel/ipu6-drivers/files/9840078/config.txt)
Then, to build the kernel:
```
make -j 12
make deb-pkg LOCALVERSION=-ipu6mod -j 12
```
After this you can install the kernel:
```
cd ..
sudo dpkg -i linux-libc-dev_6.1.0-rc1-next-20221019-ipu6mod-1_amd64.deb
sudo dpkg -i linux-image-6.1.0-rc1-next-20221019-ipu6mod_6.1.0-rc1-next-20221019-ipu6mod-1_amd64.deb 
```

But I was looking through the diffs and saw that the camera driver files are most probably not in the latest version of the ipu6 repository, i will try to change this and see what happens. (not sure if the ipu driver was already loading the camera driver while it got stuck with the error, or if it was still in an earlier state). 

In case I find something out I will post it. 
Actually, I am not familiar with kernel development and the media subsystem in particular, I actually not even have a clue what the pipeline (image stream from senor to memory, I assume) & pods are, so I am not certain if my modifications for the `next-20221019` are ok, but maybe it can be useful for someone.

jwrdegoede, 2022-10-22:
The specific crash mentioned here is fixed by: https://github.com/intel/ipu6-drivers/pull/58/commits/5a36712499e83cdf1ec6b89e13df573e09ebe565

But fixing it just results in somewhat related crashes further down the road, note I have fixes for those too, see #58 in general. But you also need some other changes to get things to actually work on Alderlake Lenovo Thinkpads let me copy and paste my comment from #55 about this:

I have been working on making the ipu6-driver stack work on a Lenovo Thinkpad X1 Yoga gen 7 (alderlake) with a 6.0 kernel and using the INT3472 mainline kernel driver for power-control.

I have submitted the following fixes for this, I believe these should also fix things not working on the X1 Nano Gen2:

https://github.com/intel/ipu6-drivers/pull/58
https://github.com/intel/ivsc-driver/pull/24
https://lore.kernel.org/linux-acpi/20221022150722.31787-1-hdegoede@redhat.com/

Note this relies on the mainline kernel itself also being build with the 2 patches from ipu6-drivers/patch. And if you look at the int3472-support-independent-clock-and-LED-gpios-5.17+.patch you will see it adds an entry for the Yoga Gen 7 to the `int3472_sensor_configs[]` table. You may need to add a copy of that entry with the sensor-module-name used for the main camera on your model. Note the IR camera is not supported yet.

fhloston, 2022-10-23:
I can confirm that applying the before mentioned fixes i can see video from the webcam for the first time!

Thank you, great job @jwrdegoede 

`sudo -E gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink`

Sadly v4l2-relayd does not work for me yet.

@jwrdegoede how do you then use the camera?

jwrdegoede, 2022-10-23:
> @jwrdegoede how do you then use the camera?

I have not gotten around to trying v4l2-relayd yet. I was happy to have gst-launch work after a marathon coding session involving all the kernel fixes and then called it a day :) I might give v4l2-relayd a shot coming Tuesday.

jwrdegoede, 2022-10-23:
BTW did you have to add an entry to the `int3472_sensor_configs[]` table ?

jeffreyfq, 2022-10-24:
> BTW did you have to add an entry to the `int3472_sensor_configs[]` table ?

I looks like that I do not need to add an entry to `int3472_sensor_configs[]` for the  HP Spectre x360 16-f1010TX (i7-1260P)

I compiled the kernel now with all patches applied,
seems that I am almost there :)

This looks to me as if there are some HAL-driver related files missing, but it seems that the device is available.

```
sudo -E gst-launch-1.0 icamerasrc device-name=ov13b10-wf af-mode=2 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
[10-24 14:42:37.962] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-24 14:42:37.962] CamHAL[ERR] there is no aiqb file:ov13b10
[10-24 14:42:37.962] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-24 14:42:37.962] CamHAL[ERR] there is no aiqb file:ov13b10
[10-24 14:42:37.962] CamHAL[INF] aiqb file name ov8856.aiqb
[10-24 14:42:37.962] CamHAL[ERR] there is no aiqb file:ov8856
[10-24 14:42:37.962] CamHAL[INF] aiqb file name ov8856.aiqb
[10-24 14:42:37.962] CamHAL[ERR] there is no aiqb file:ov8856
[10-24 14:42:37.962] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-24 14:42:37.962] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-24 14:42:37.962] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-24 14:42:37.962] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-24 14:42:37.962] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-24 14:42:37.963] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-24 14:42:37.963] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[10-24 14:42:37.963] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[10-24 14:42:37.963] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-24 14:42:37.963] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-24 14:42:37.963] CamHAL[INF] aiqb file name hi556.aiqb
[10-24 14:42:37.963] CamHAL[INF] aiqb file name hi556.aiqb
[10-24 14:42:37.963] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-24 14:42:37.963] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-24 14:42:37.963] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
[10-24 14:42:37.965] CamHAL[ERR] MediaControl init failed
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
```

same for the other command:
`sudo -E gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesin`

Btw, is there any easy way to list the acpi ensor-module-names in the console?

fhloston, 2022-10-24:
> BTW did you have to add an entry to the `int3472_sensor_configs[]` table ?

No, I just applied your mentioned patches plus an unrelated i915 patch. Also X1 Yoga gen7 here.

drogenlied, 2022-10-24:
@fhloston was that i915 patch useful?

@jeffreyfq There are multiple names that are necessary for sensor setup.

The _HID and _DDN  from ACPI can be obtained easily and are used in the camera sensor driver.
On my X1 Yoga Gen 7, the cameras have HIDs like INT3474 and that name can be found in `drivers/media/i2c/ov2740.c`.
Script to find these (needs acpica-tools for iasl):
```
cat /sys/firmware/acpi/tables/DSDT > dsdt.dat
iasl -tc dsdt.dsl
grep -A9 "Device (CAM" dsdt.dsl
```

Then there is a model string from ACPI _DSM() of some device that is used by int3472 and compared to the list in `int3472_sensor_configs[]`.

Maybe it is contained in the DSDT, but not verbatim, so I just added 
`	dev_info(int3472->dev, "Sensor _DSM: %s\n", obj->string.pointer);` to `skl_int3472_get_sensor_module_config(...)` and saved the work of figuring out where that string ultimately comes from.

As a sort of long shot I added this to `int3472_sensor_configs[]`
```
	/* Lenovo ThinkPad X1 Yoga Gen7 - OV9234 monochromatic sensor? */
	{ "6BF115T2", { 0 }, NULL, true },
```
and changed `ov9734_acpi_ids[]` in `drivers/media/i2c/ov9734.c` to
```
	{ "OVTI9234", },
	{ "OVTI9734", },
	{ "INT3471", },
```
but the ov9734 sensor driver does not like its monochrome sibling variant and fails with -EINVAL during probing. No crash though.

jeffreyfq, 2022-10-25:
> @jeffreyfq There are multiple names that are necessary for sensor setup.

Thank you for your help and the detailed instructions, I was able to read out the ACPI data:
```
        Device (CAM0)
        {
            Name (_HID, "INT3471")  // _HID: Hardware ID
            Name (_DDN, "IMX135-CRDG2")  // _DDN: DOS Device Name
            Name (_UID, "0")  // _UID: Unique ID
            Name (_DEP, Package (0x01)  // _DEP: Dependencies
            {
                PMIC
            })
            Name (_PLD, Package (0x01)  // _PLD: Physical Location of Device
--
        Device (CAM1)
        {
            Name (_HID, "INT3474")  // _HID: Hardware ID
            Name (_DDN, "OV2740-CRDG2")  // _DDN: DOS Device Name
            Name (_UID, "0")  // _UID: Unique ID
            Name (_DEP, Package (0x01)  // _DEP: Dependencies
            {
                ^^I2C2.PMIC
            })
            Name (_PLD, Package (0x01)  // _PLD: Physical Location of Device
```
So I have the device INT3474 with OV2740-CRDG2
and a device INT3471 with sensor IMX135-CRDG2.

It seems that OV2740 and IMX135 (for imx135 there is seemingly no driver available in drivers/media/i2c/), both are RGB sensors, kind of strange, I thought the IR sensor would be grayscale.

Then I checked,
the driver should be fine, there is my `_HID` name INT3474 in the driver structure `ov2740_acpi_ids` already.

So next I will compile the kernel module with some print statements to get the `_DSM` name to add it to the `int3472_sensor_configs[]`

I still have the `_DDN` name, do I have to add this name also somewhere? 
Or is it just relevant to figure out which driver/sensor type it is (e.g. in my case OV2740)?

jeffreyfq, 2022-10-26:
@fhloston I just tried now, but results in the same error. But considering that my hardware is so similar to the X1 Yoga Gen7, like in your case, I guess it has to work for me too. I guess to be on the safe side, I have to start over and patch the kernel/modules from a fresh checkout, could be possible that I messed up something.

this is the output (does not cause new dmesg errors):
```
Setting pipeline to PAUSED ...
[10-26 23:22:03.361] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-26 23:22:03.361] CamHAL[ERR] there is no aiqb file:ov13b10
[10-26 23:22:03.361] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-26 23:22:03.361] CamHAL[ERR] there is no aiqb file:ov13b10
[10-26 23:22:03.361] CamHAL[INF] aiqb file name ov8856.aiqb
[10-26 23:22:03.361] CamHAL[ERR] there is no aiqb file:ov8856
[10-26 23:22:03.361] CamHAL[INF] aiqb file name ov8856.aiqb
[10-26 23:22:03.361] CamHAL[ERR] there is no aiqb file:ov8856
[10-26 23:22:03.361] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-26 23:22:03.362] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-26 23:22:03.362] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-26 23:22:03.362] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-26 23:22:03.362] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-26 23:22:03.362] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-26 23:22:03.362] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[10-26 23:22:03.362] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[10-26 23:22:03.362] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-26 23:22:03.363] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-26 23:22:03.363] CamHAL[INF] aiqb file name hi556.aiqb
[10-26 23:22:03.363] CamHAL[INF] aiqb file name hi556.aiqb
[10-26 23:22:03.363] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-26 23:22:03.363] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-26 23:22:03.363] CamHAL[ERR] Failed to find DevName for cameraId: 7, get video node: ov2740 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[10-26 23:22:03.370] CamHAL[ERR] Get entity fail for calling getEntityById
[10-26 23:22:03.370] CamHAL[ERR] Get entity fail for calling getEntityById
[10-26 23:22:03.370] CamHAL[ERR] setup Link ov2740  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[10-26 23:22:03.370] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[10-26 23:22:03.370] CamHAL[ERR] set up mediaCtl failed
[10-26 23:22:03.370] CamHAL[ERR] @configure Device Configure failed
[10-26 23:22:03.370] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.004552897
Setting pipeline to NULL ...
Freeing pipeline ...

```

polair, 2022-10-26:
@jeffreyfq This is the same error I have with the OV2740 in the Nano X1 Gen2 that I've filed a bug report on ( #55 )

I haven't been able to get further than you and like you I'm a little fatigued by it. I do have one tidbit of info though. Launching your gstreamer command with GST_DEBUG env var set can get you a lot more information. Something like:

```
GST_DEBUG=4 sudo -E gst-launch-1.0 icamerasrc buffer-count=7 device-name=ov2740-uf ! video/x-raw,format=YUY2,width=1280,height=720 ! v4l2sink device=/dev/video1
```

jwrdegoede, 2022-10-27:
> You seem to have another secondary sensor in your X1 Yoga Gen7 than me

Note that the presence of the sensor in the DSDT does not say everything, there are multiple sensor entries and only some actually are "active", this is governed by the _STA method of the ACPI device returning 0xf (present) or 0 (not present).

To really see which sensors you have, first build the main kernel with the int3472-gpio + https://lore.kernel.org/linux-acpi/20221022150722.31787-1-hdegoede@redhat.com/ patches. After this the sensors should show up under in `ls /sys/bus/i2c/devices` for me the 2 sensors are `i2c-INT3474:01` which is the ov2740 normal sensor and `i2c-OVTI9234:00` which is an ov9234 monochrome sensor which is used as ir-sensor here.

fhloston, 2022-10-27:
> @fhloston What does `dmesg | grep vsc` say for you when it's working? Can you give a set of outputs, the v4l2 camera information and `dmesg | grep ipu` and `lsmod | grep 'ipu|mei|vsc'`. Would be very helpful <3

```
# dmesg | grep vsc
# dmesg | grep ipu
[   28.450481] intel-ipu6 intel-ipu: enabling device (0000 -> 0002)
[   28.453087] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[   28.454991] intel-ipu6 intel-ipu: physical base address 0x603c000000
[   28.454993] intel-ipu6 intel-ipu: mapped as: 0x0000000061c5c228
[   28.460159] intel-ipu6 intel-ipu: IPU in secure mode
[   28.463390] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[   28.466871] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[   28.500533] intel-ipu6 intel-ipu: IPC reset done
[   28.504951] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[   28.508879] intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
[   28.513018] intel-ipu6 intel-ipu: FW version: 20220510
[   28.517294] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[   28.519859] intel-ipu6 intel-ipu: physical base address 0x603c000000
[   28.523983] intel-ipu6 intel-ipu: mapped as: 0x00000000c4fc3a40
[   28.527636] intel-ipu6 intel-ipu: IPU in secure mode
[   28.529708] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[   28.533369] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[   28.538099] intel-ipu6 intel-ipu: IPC reset done
[   28.541327] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[   28.543325] intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
[   28.545132] intel-ipu6 intel-ipu: FW version: 20220510
[   28.547798] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[   28.549641] intel-ipu6 intel-ipu: physical base address 0x603c000000
[   28.551409] intel-ipu6 intel-ipu: mapped as: 0x0000000061c5c228
[   28.553672] intel-ipu6 intel-ipu: IPU in secure mode
[   28.555461] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[   28.557230] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[   28.560820] intel-ipu6 intel-ipu: IPC reset done
[   28.562526] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[   28.564282] intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
[   28.566386] intel-ipu6 intel-ipu: FW version: 20220510
[   28.595815] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[   28.599498] intel-ipu6 intel-ipu: physical base address 0x603c000000
[   28.603484] intel-ipu6 intel-ipu: mapped as: 0x0000000061c5c228
[   28.606541] intel-ipu6 intel-ipu: IPU in secure mode
[   28.608326] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[   28.610013] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[   28.613515] intel-ipu6 intel-ipu: IPC reset done
[   28.615179] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[   28.616958] intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
[   28.618649] intel-ipu6 intel-ipu: FW version: 20220510
[   28.661257] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[   28.664261] intel-ipu6 intel-ipu: physical base address 0x603c000000
[   28.669602] intel-ipu6 intel-ipu: mapped as: 0x0000000061c5c228
[   28.673366] intel-ipu6 intel-ipu: IPU in secure mode
[   28.676600] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[   28.680153] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[   28.700766] intel-ipu6 intel-ipu: IPC reset done
[   28.702947] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[   28.704808] intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
[   28.707069] intel-ipu6 intel-ipu: FW version: 20220510
[   28.742875] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[   28.746019] intel-ipu6 intel-ipu: physical base address 0x603c000000
[   28.747950] intel-ipu6 intel-ipu: mapped as: 0x0000000057b928b5
[   28.749869] intel-ipu6 intel-ipu: IPU in secure mode
[   28.751633] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[   28.753385] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[   28.757028] intel-ipu6 intel-ipu: IPC reset done
[   28.758741] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[   28.760510] intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
[   28.762220] intel-ipu6 intel-ipu: FW version: 20220510
[   28.843266] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[   28.847175] intel-ipu6 intel-ipu: physical base address 0x603c000000
[   28.855199] intel-ipu6 intel-ipu: mapped as: 0x0000000057b928b5
[   28.863577] intel-ipu6 intel-ipu: IPU in secure mode
[... 113 lines cut ...]
[  197.862215] intel-ipu6-isys intel-ipu6-isys0: csi2-1 error: Incomplete long packet detected
[  197.862221] intel-ipu6-isys intel-ipu6-isys0: csi2-1 error: Line sync error
[  197.862227] intel-ipu6-isys intel-ipu6-isys0: csi2-1 error: Inter-frame short packet discarded
[  197.862234] intel-ipu6-isys intel-ipu6-isys0: csi2-1 error: Inter-frame long packet discarded
[  538.518245] intel-ipu6-isys intel-ipu6-isys0: stream off ov2740 19-0036
[  830.215654] intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 19-0036
[  881.638036] intel-ipu6-isys intel-ipu6-isys0: stream off ov2740 19-0036
# lsmod | grep 'ipu\|mei\|vsc'
mei_hdcp               24576  0
mei_me                 45056  1
intel_ipu6_isys       135168  0
videobuf2_dma_contig    24576  1 intel_ipu6_isys
videobuf2_v4l2         36864  1 intel_ipu6_isys
videobuf2_common       73728  4 videobuf2_dma_contig,videobuf2_v4l2,intel_ipu6_isys,videobuf2_memops
intel_ipu6_psys        98304  0
[... cut ...]
v4l2_async             32768  3 v4l2_fwnode,intel_ipu6_isys,ov2740
videodev              286720  6 v4l2_async,videobuf2_v4l2,v4l2loopback,intel_ipu6_isys,videobuf2_common,ov2740
mc                     69632  6 v4l2_async,videodev,videobuf2_v4l2,intel_ipu6_isys,videobuf2_common,ov2740
```

I did however `blacklist intel-ipu6` in `/etc/modprobe.d/ipu6.conf` because in the boots after installation the system would crash. To my astonishment it then works even though blacklisted. So I assume there still is an issue with module loading order.

fhloston, 2022-10-27:
> To really see which sensors you have, first build the main kernel with the int3472-gpio + https://lore.kernel.org/linux-acpi/20221022150722.31787-1-hdegoede@redhat.com/ patches. After this the [...]

yes, indeed! thanks!

```
lrwxrwxrwx 1 root root 0 Oct 23 16:44 i2c-ELAN067C:00 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ELAN067C:00
lrwxrwxrwx 1 root root 0 Oct 23 18:06 i2c-INT3474:01 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/ljca-i2c.31.auto/i2c-19/i2c-INT3474:01
lrwxrwxrwx 1 root root 0 Oct 23 18:06 i2c-INT347D:00 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/ljca-i2c.30.auto/i2c-18/i2c-INT347D:00
lrwxrwxrwx 1 root root 0 Oct 23 18:06 i2c-LATT2021:00 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/ljca-i2c.30.auto/i2c-18/i2c-LATT2021:00
lrwxrwxrwx 1 root root 0 Oct 23 16:44 i2c-NXP1001:00 -> ../../../devices/pci0000:00/0000:00:19.0/i2c_designware.2/i2c-3/i2c-NXP1001:00
lrwxrwxrwx 1 root root 0 Oct 23 18:06 i2c-OVTI9234:00 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/ljca-i2c.30.auto/i2c-18/i2c-OVTI9234:00
lrwxrwxrwx 1 root root 0 Oct 23 16:44 i2c-WACF2200:00 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-WACF2200:00```

fhloston, 2022-10-27:
How I installed kernel and drivers:

```
cd /usr/src/
apt source linux-image-6.0.0-2-amd64-unsigned
cd linux-6.0.3 
cp /boot/config-6.0.0-2-amd64 .config
make oldconfig
# make menuconfig and enable CONFIG_INTEL_SKL_INT3472=m
patch -p1 <../ipu6-drivers-0.0.0/patch/int3472-support-independent-clock-and-LED-gpios-5.17+.patch 
patch -p1 <../latt2021.diff 
nice make -j`nproc` bindeb-pkg
```

```
cd /usr/src/
git clone https://github.com/intel/ipu6-drivers.git ipu6-drivers-0.0.0 
cd ipu6-drivers-0.0.0/
git pull origin pull/58/head
git clone https://github.com/intel/ivsc-driver.git
cd ivsc-driver/
git pull origin pull/24/head
cd ..
cp -r ivsc-driver/backport-include ivsc-driver/drivers ivsc-driver/include .
rm -rf ivsc-driver
dkms add ipu6-drivers/0.0.0
dkms install ipu6-drivers/0.0.0
```

```
cd /usr/src/
git clone https://github.com/intel/ivsc-driver.git ivsc-driver-1.0.0
cd ivsc-driver-1.0.0/
git pull origin pull/24/head
dkms add ivsc-driver/1.0.0
dkms install ivsc-driver/1.0.0
```

polair, 2022-10-27:
@fhloston thank you for taking the time to write these out, very helpful. I'll get to this sometime today.

jeffreyfq, 2022-10-28:
Just something I noted while applying the kernel patch of @jwrdegoede ,
in drivers/acpi/scan.c, there is an ignore and honor list:
```
/* List of HIDs for which we honor deps of matching ACPI devs, when checking _DEP lists. */
static const char * const acpi_honor_dep_ids[] = {
	"INT3472", /* Camera sensor PMIC / clk and regulator info */
	NULL
};
```
My devices are named `INT3471` and `INT3474`, so I am currently compiling the kernel with those two device names added to the list. Not sure if this is necessary, but later in the code:
```
bool acpi_dev_ready_for_enumeration(const struct acpi_device *device)
{
	if (device->flags.honor_deps && device->dep_unmet)
		return false;

	return acpi_device_is_present(device);
}
```
maybe the driver is never loaded if the device name is not in this honor list....

jeffreyfq, 2022-10-28:
As I obviously developed an obsession of compiling my kernel, I tried another round.

This time, I used kernel 6.0.5, downloaded from kernel.org.
I applied all the patches in the order @fhloston  mentioned in the previous post....
..except... :)
I applied additionally `IOMMU-passthrough-for-intel-ipu.diff`
I added my ids to the `int3472_sensor_configs`, like so:
```
/* HP Spectre x360 16-f1xxx */
{ "CJFKE26", REGULATOR_SUPPLY("avdd", NULL), NULL, false },
```
Actually I tried true and false as the last parameter, and false seems to work better somehow^^
I also added my _HID IDs to the struct `acpi_honor_dep_ids`.

Then it seems to be crucial to install the v4l2loopback kernel module.

Somewhere in the internet, someone mentioned `v4l2-relayd`, so I installed it... just in case it does something useful.

-------------------------

As it turns out, I am getting same error messages as before on my self-modified 6.2-rc2/next kernel.

The only related dmesg errors show this:
```
1.388599] intel vsc not ready
[   16.714727] hp_wmi: Unknown event_id - 131073 - 0x0 <-- this is just the keyboard
[   18.036145] intel vsc not ready
[   18.037633] i2c i2c-INT3537:00: deferred probe pending
```

I am not sure, but I had a look into the source, the `intel vsc not ready`
orignates in `ipu6-drivers-0.0.0/drivers/misc/ivsc/intel_vsc.c` and the related source is:
```
	ret = check_component_ready();
	if (ret < 0) {
		pr_info("intel vsc not ready\n");
		return -EAGAIN;
	}
```
and the `check_component_ready` function is doing something like this:
```
...
if (vsc.ace_registerred && vsc.csi_registerred)
		ret = 0;
...
````
so it will not be ready until `mei_csi.c` and `mei_ace.c` call their respective functions `vsc_register_csi(csi, &csi_ops);` and `vsc_register_ace(ace, &ace_ops);` that are located in the module probe functions.

I even put some additional print-outs there, and it turns out those modules are listed in lsmod, but their probe function is never called!! Not sure how this is possible, I thought when I modprobe the module the probe function is called... but my guess is that there is some fancy function in the kernel which relates to the
`i2c i2c-INT3537:00: deferred probe pending` message in the dmesg (not sure).`, maybe some module probe is pending?

One thing that seems strange to me is that there is a *hi556* kernel module loaded,
even though I dont have this sensor in my device (according to this acpi information)!
This could be related to why it is not working. There is no OV2740 module.
```
lsmod | grep intel
snd_soc_intel_hda_dsp_common    20480  1 snd_soc_skl_hda_dsp
snd_sof_pci_intel_tgl    16384  0
snd_sof_intel_hda_common   143360  1 snd_sof_pci_intel_tgl
soundwire_intel        40960  1 snd_sof_intel_hda_common
soundwire_generic_allocation    16384  1 soundwire_intel
soundwire_cadence      40960  1 soundwire_intel
snd_sof_intel_hda      20480  1 snd_sof_intel_hda_common
snd_sof_pci            24576  2 snd_sof_intel_hda_common,snd_sof_pci_intel_tgl
snd_sof_xtensa_dsp     16384  1 snd_sof_intel_hda_common
snd_sof               233472  3 snd_sof_pci,snd_sof_intel_hda_common,snd_sof_probes
snd_soc_hdac_hda       24576  1 snd_sof_intel_hda_common
snd_hda_ext_core       36864  4 snd_sof_intel_hda_common,snd_soc_hdac_hdmi,snd_soc_hdac_hda,snd_sof_intel_hda
snd_soc_acpi_intel_match    69632  2 snd_sof_intel_hda_common,snd_sof_pci_intel_tgl
snd_soc_acpi           16384  2 snd_soc_acpi_intel_match,snd_sof_intel_hda_common
[... 27 lines cut ...]
intel_rapl_common      40960  2 intel_rapl_msr,processor_thermal_rapl
intel_ipu6            106496  2 intel_ipu6_isys,intel_ipu6_psys
intel_skl_int3472_tps68470    20480  0
intel_vsc              16384  3 mei_ace,mei_csi,hi556
v4l2_fwnode            32768  2 hi556,intel_ipu6_isys
v4l2_async             28672  3 v4l2_fwnode,hi556,intel_ipu6_isys
intel_skl_int3472_discrete    16384  0
intel_hid              24576  0
sparse_keymap          16384  2 hp_wmi,intel_hid
videodev              274432  7 v4l2_async,hi556,videobuf2_v4l2,v4l2loopback,intel_ipu6_isys,videobuf2_common
mc                     65536  6 v4l2_async,videodev,hi556,videobuf2_v4l2,intel_ipu6_isys,videobuf2_common
hid_sensor_hub         28672  9 hid_sensor_gyro_3d,hid_sensor_trigger,hid_sensor_iio_common,hid_sensor_custom_intel_hinge,hid_sensor_incl_3d,hid_sensor_als,hid_sensor_accel_3d,hid_sensor_rotation,hid_sensor_custom
intel_ishtp_hid        28672  0
ghash_clmulni_intel    16384  0
spi_intel_pci          16384  0
aesni_intel           393216  8
crypto_simd            16384  1 aesni_intel
cryptd                 24576  3 crypto_simd,ghash_clmulni_intel
spi_intel              28672  1 spi_intel_pci
intel_ish_ipc          28672  0
intel_lpss_pci         28672  0
intel_lpss             16384  1 intel_lpss_pci
intel_ishtp            57344  2 intel_ishtp_hid,intel_ish_ipc
hid                   163840  6 i2c_hid,hid_multitouch,hid_sensor_hub,intel_ishtp_hid,hid_generic,uhid
```

polair, 2022-10-28:
@jeffreyfq checkout #55 and read a bit. There's some helpful things in there, does `/dev/ipu*` exist? 

What is in `/sys/bus/i2c/devices/`??

Did you actually apply the "LATT2021" patch from the kernel mailing list? Copied manually?

What happens when you modprobe ov2740?

Here is what a working GPIO/i2c dev tree looks like for the ov2740 sensor on a nano gen2:

```
» ll /sys/bus/i2c/devices                                                                           Fri 28 Oct 2022 09:36:46 AM MDT
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-3 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-3
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-10 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/i2c-10
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-1 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-1
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-8 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-8
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-6 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-6
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-13 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-4/i2c-13
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-4 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-4
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-11 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/i2c-11
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-2 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-2
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-0 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-0
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-9 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/i2c-9
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-7 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-7
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-5 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-5
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-12 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-3/i2c-12
lrwxrwxrwx 0 root 28 Oct 09:20 i2c-15 -> ../../../devices/pci0000:00/0000:00:1f.4/i2c-15
lrwxrwxrwx 0 root 28 Oct 09:36 i2c-OVTI9234:00 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/ljca-i2c.2.auto/i2c-16/i2c-OVTI9234:00
lrwxrwxrwx 0 root 28 Oct 09:36 i2c-17 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/ljca-i2c.3.auto/i2c-17
lrwxrwxrwx 0 root 28 Oct 09:36 i2c-ELAN0670:00 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-14/i2c-ELAN0670:00
lrwxrwxrwx 0 root 28 Oct 09:36 i2c-INT347D:00 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/ljca-i2c.2.auto/i2c-16/i2c-INT347D:00
lrwxrwxrwx 0 root 28 Oct 09:36 i2c-INT3474:01 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/ljca-i2c.3.auto/i2c-17/i2c-INT3474:01
lrwxrwxrwx 0 root 28 Oct 09:36 i2c-LATT2021:00 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/ljca-i2c.2.auto/i2c-16/i2c-LATT2021:00
lrwxrwxrwx 0 root 28 Oct 09:36 i2c-16 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/ljca-i2c.2.auto/i2c-16
lrwxrwxrwx 0 root 28 Oct 09:36 i2c-14 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-14
```

jwrdegoede, 2022-10-28:
@jeffreyfq  you should not `acpi_honor_dep_ids` to make the i2c devices for the sensors show up under `/sys/bus/i2c/devices` you need 2 changes:

1. My patch which adds "LATT2021" to `acpi_ignore_dep_ids[]`
2. https://github.com/intel/ivsc-driver/pull/24/commits/c5aa0b65a6fb3cca7ee3b6fae093cfadfdd3a5cf

For details for how this all actually works (it has to do with ensuring that things are probed in a certain order) see: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=9d9bcae47fd5a

polair, 2022-10-28:
@jwrdegoede That's exactly the details I wanted. Very great reading/learning material. Thanks.

drogenlied, 2022-10-29:
@fhloston @jwrdegoede If you have trouble with v4l-relayd not working, try using v4l2loopback from Ubuntu (https://git.launchpad.net/ubuntu/+source/v4l2loopback), it contains patches that are crucial to make v4l-relayd work, mostly relating to V4L2_EVENT_PRI_CLIENT_USAGE, aka v4l events needed to start the stream when a client opens the loopback device.

I just got that to sort of work today.

jwrdegoede, 2022-10-30:
@drogenlied a thanks, that probably explains why it is not working for me.

drogenlied, 2022-10-30:
Another tip: applying https://github.com/umlaeute/v4l2loopback/pull/435 to v4l2loopback will allow some ffmpeg based applications to use the device, otherwise they complain about buffer sizes.
Maybe this just fixes symptoms and the root cause is somewhere in gstreamer, it helps nonetheless.

jeffreyfq, 2022-10-31:
Thanks @jwrdegoede and @polair for following up on this,

@jwrdegoede, I can confirm, 
- I now removed the the IDs I added to the `acpi_honor_dep_ids` struct.
- The patch which adds `LATT2021` to `acpi_ignore_dep_ids[]` was applied-> one liner + "LATT2021", /* Lattice FW Update Client Driver */
(But I dont have this `LATT2021` device in my system)
- I applied https://github.com/intel/ivsc-driver/pull/24 /  (probe order fixes) and also the #58
- I2C devices seem to be available (listing below; /sys/bus/i2c/devices/)
- Of course, I also applied the gpio and IOMMU patches

@polair:
> does /dev/ipu* exist?

Yes

> Did you actually apply the "LATT2021" patch from the kernel mailing list? Copied manually?

Yes, I patched this, the one-liner, but I dont have that device in my system.
I also applied the IOMMU, gpio, #58 and #24 patches

> What is in /sys/bus/i2c/devices/??
```
jfq@jeffreyhp:~/src/kernel/ipu6-drivers-0.0.0$ ll /sys/bus/i2c/devices
total 0
drwxr-xr-x 2 root root 0 Oct 31 15:46 ./
drwxr-xr-x 4 root root 0 Oct 31 15:46 ../
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-0 -> ../../../devices/pci0000:00/0000:00:1f.4/i2c-0/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-1 -> ../../../devices/pci0000:00/0000:00:10.0/i2c_designware.0/i2c-1/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-10 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-10/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-11 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-11/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-12 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-12/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-13 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-13/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-14 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-14/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-15 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/i2c-15/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-16 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/i2c-16/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-17 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/i2c-17/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-18 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-3/i2c-18/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-19 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-4/i2c-19/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-2 -> ../../../devices/pci0000:00/0000:00:10.1/i2c_designware.1/i2c-2/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-20 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-5/i2c-20/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-3 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.2/i2c-3/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-4 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.3/i2c-4/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-5 -> ../../../devices/pci0000:00/0000:00:19.0/i2c_designware.4/i2c-5/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-6 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-6/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-7 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-7/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-8 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-8/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-9 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-9/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-ELAN2513:00 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.3/i2c-4/i2c-ELAN2513:00/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-INT3537:00 -> ../../../devices/pci0000:00/0000:00:19.0/i2c_designware.4/i2c-5/i2c-INT3537:00/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-OVTI00AB:00 -> ../../../devices/pci0000:00/0000:00:10.1/i2c_designware.1/i2c-2/i2c-OVTI00AB:00/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-SYNA32C0:00 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.2/i2c-3/i2c-SYNA32C0:00/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-TXNW3643:01 -> ../../../devices/pci0000:00/0000:00:10.1/i2c_designware.1/i2c-2/i2c-TXNW3643:01/
```

>What happens when you modprobe ov2740?

Nothing, no dmesg entry, no error, I can not notice any difference after loading the module by hand.
```
jfq@jeffreyhp:~$ modinfo ov2740 
filename:       /lib/modules/6.0.5/updates/dkms/ov2740.ko
license:        GPL v2
description:    OmniVision OV2740 sensor driver
author:         Bingbu Cao <bingbu.cao@intel.com>
author:         Shawn Tu <shawnx.tu@intel.com>
author:         Qiu, Tianshu <tian.shu.qiu@intel.com>
srcversion:     A93395D93ACFD7D88D82462
alias:          acpi*:INT3474:*
depends:        videodev,v4l2-async,v4l2-fwnode,mc
retpoline:      Y
name:           ov2740
[... 34 lines cut ...]
            Name (_HID, "INT3474")  // _HID: Hardware ID
            Name (_DDN, "OV2740-CRDG2")  // _DDN: DOS Device Name
            Name (_UID, "0")  // _UID: Unique ID
            Name (_DEP, Package (0x01)  // _DEP: Dependencies
            {
                ^^I2C2.PMIC
            })
            Name (_PLD, Package (0x01)  // _PLD: Physical Location of Device

```

Also there is no LATT2021 device, checked  `grep -A9 "LATT2021" dsdt.dsl`,
can it be that the device has another name in my system, is there a way to figure this out?
I guess the suspicious devices are:
```
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-ELAN2513:00 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.3/i2c-4/i2c-ELAN2513:00/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-INT3537:00 -> ../../../devices/pci0000:00/0000:00:19.0/i2c_designware.4/i2c-5/i2c-INT3537:00/
[... cut ...]
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-SYNA32C0:00 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.2/i2c-3/i2c-SYNA32C0:00/
lrwxrwxrwx 1 root root 0 Oct 31 15:46 i2c-TXNW3643:01 -> ../../../devices/pci0000:00/0000:00:10.1/i2c_designware.1/i2c-2/i2c-TXNW3643:01/
```

But even then, I seems strange to me that the hi556 module loads without a device present in my system.

Right after boot, I can see many messages `intel vsc not ready` and one message `i2c i2c-INT3537:00: deferred probe pending` in the dmesg output.

jwrdegoede, 2022-10-31:
@jeffreyfq the ACPI tables unfortunately are not always written in such a way that they are easily readable. E.g. the ACPI hw-id (HID/CID) may come out of a variable set by the BIOS.

E.g. you say there is no INT3537 in your DSDT (did you also check the SSDTs?) but since you do have a `i2c-INT3537:00` device you clearly have an ACPI device in your tables with a INT3537 HID and this is also why the hi556 module is loading.

It is also interesting to see if the driver is actually binding to the device. Can you do:

`ls -l /sys/bus/i2c/devices/INT3537:00/`

And see if there is a driver symlink in there ? And the same for the other i2c-devices.

Note that it might be tricky to get things to work on laptop models which do not have an officially supported Linux OS option (typically an Ubuntu OEM install) . On these laptop models some extra work / fixes may be necessary and it may be quite tricky to figure out what changes are necessary.

Can you attach a full dmesg output from after your latest attempt here? And also please run `sudo acpidump -o acpidump.txt` and attach the acpidump.txt file here. Or send them both to me directly at  hdegoede@redhat.com.
Note I'm not making any promises here, but I can at least take a quick look. Also I'm quite busy this week so it may be a while before I get back to you.

jwrdegoede, 2022-10-31:
@jeffreyfq Also please run: `cat /sys/bus/i2c/devices/INT3537:00/firmware_node/path` and let us know what the output is? This will tell us which "Device (NAME0000) {}" in the ACPI tables corresponds to the hi556 sensor. Typically a "Device (LNK0) {}" entry in the tables belongs to the main sensor.

norbusan, 2022-11-01:
Additional information: With this repo: https://github.com/stefanpartheym/archlinux-ipu6-webcam and in particular the v4l2loopback and v4l2-relayd packages installed and loade using
```
modprobe v4l2loopback exclusive_caps=1
```
I get video output in MS Edge (Linux) in Jit.si and Google Meet.

Incredible!

jeffreyfq, 2022-11-02:
> @jeffreyfq the ACPI tables unfortunately are not always written in such a way that they are easily readable. E.g. the ACPI hw-id (HID/CID) may come out of a variable set by the BIOS.
> 
> E.g. you say there is no INT3537 in your DSDT (did you also check the SSDTs?) but since you do have a `i2c-INT3537:00` device you clearly have an ACPI device in your tables with a INT3537 HID and this is also why the hi556 module is loading.
> 
> It is also interesting to see if the driver is actually binding to the device. Can you do:
> 
> `ls -l /sys/bus/i2c/devices/INT3537:00/`
> 
> And see if there is a driver symlink in there ? And the same for the other i2c-devices.
> 
> Note that it might be tricky to get things to work on laptop models which do not have an officially supported Linux OS option (typically an Ubuntu OEM install) . On these laptop models some extra work / fixes may be necessary and it may be quite tricky to figure out what changes are necessary.
> 
> Can you attach a full dmesg output from after your latest attempt here? And also please run `sudo acpidump -o acpidump.txt` and attach the acpidump.txt file here. Or send them both to me directly at [hdegoede@redhat.com](mailto:hdegoede@redhat.com). Note I'm not making any promises here, but I can at least take a quick look. Also I'm quite busy this week so it may be a while before I get back to you.

@jwrdegoede Thanks, of course I understand if you don't have time for this. But in case you can have a quick look, and maybe find something I could check, it would be great.

---------------------------------------------

> E.g. you say there is no INT3537 in your DSDT (did you also check the SSDTs?) but since you do have a `i2c-INT3537:00` device you clearly have an ACPI device in your tables with a INT3537 HID and this is also why the hi556 module is loading.

I now figured out, I don't have the ov2740 sensor.
The camera is advertised as 5Mp and ov2740 is below that, I also booted windows and looked in the device manager.
I have the `hi556` sensor in my device!

This means that the loading of the kernel module is indeed ok. The entries shown by doing a `grep -A9 "INT3537" dsdt.dsl`
 on the dst table are obviously misleading and are not activated.

------------------------------------------------------

> `ls -l /sys/bus/i2c/devices/INT3537:00/`
> 
> And see if there is a driver symlink in there ? And the same for the other i2c-devices.

There is a symlink:
```
jfq@jeffreyhp:~$ ls -l /sys/bus/i2c/devices/i2c-INT3537\:00
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-INT3537:00 -> ../../../devices/pci0000:00/0000:00:19.0/i2c_designware.4/i2c-5/i2c-INT3537:00
```

These are the symlinks of the other devices:
```
jfq@jeffreyhp:~$ ls -l /sys/bus/i2c/devices/i2c-*
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-0 -> ../../../devices/pci0000:00/0000:00:1f.4/i2c-0
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-1 -> ../../../devices/pci0000:00/0000:00:10.0/i2c_designware.0/i2c-1
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-10 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-10
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-11 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-11
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-12 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-12
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-13 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-13
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-14 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-14
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-15 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/i2c-15
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-16 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/i2c-16
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-17 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/i2c-17
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-18 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-3/i2c-18
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-19 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-4/i2c-19
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-2 -> ../../../devices/pci0000:00/0000:00:10.1/i2c_designware.1/i2c-2
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-20 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-5/i2c-20
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-3 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.2/i2c-3
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-4 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.3/i2c-4
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-5 -> ../../../devices/pci0000:00/0000:00:19.0/i2c_designware.4/i2c-5
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-6 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-6
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-7 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-7
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-8 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-8
lrwxrwxrwx 1 root root 0 Nov  2 10:05 /sys/bus/i2c/devices/i2c-9 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-9
[... cut ...]
I confirmed that `INT3472_GPIO_TYPE_RESET`, `INT3472_GPIO_TYPE_POWER_ENABLE`, `INT3472_GPIO_TYPE_PRIVACY_LED` are set. But `No clk GPIO. The privacy LED won't work` occurs as well.

My suspicion is that the new/unknown GPIO type 0x12 is maybe the physical lid of the sensor that can be closed opened by an electric mechanism (there is a key on the keyboard that physically closes/opens a lid in front of the camera lens). This is maybe why it is not yet defined in the driver. But it is just a guess for now.

fhloston, 2022-11-05:
I still fail to use /dev/video0 with vlc or firefox with v4l2loopback and v4l2-relayd

With vlc and trying to playback /dev/video0 at least the camera led turns on, but i see no picture. I see vlc detecting YUV format - even though i have set the format to NV12.

Any ideas what I could be holding wrong?

drogenlied, 2022-11-06:
@fhloston as a workaround for the NV12 issue you can edit the v4l-relayd.service file, check my comment https://github.com/intel/ipu6-drivers/issues/48#issuecomment-1296241668

csdvrx, 2023-02-23:
> Maybe it is contained in the DSDT, but not verbatim, so I just added ` dev_info(int3472->dev, "Sensor _DSM: %s\n", obj->string.pointer);` to `skl_int3472_get_sensor_module_config(...)` and saved the work of figuring out where that string ultimately comes from.
> (...)
> and changed `ov9734_acpi_ids[]` in `drivers/media/i2c/ov9734.c`  

That's a very practical way to do it, but the solution may not scalable as you will only find the _DSM for the precise version of the camera you're using.

For example, the Nano has 2 different _DSM for each camera, depending if the screen is touchscreen enabled or not

> 	/* Lenovo ThinkPad X1 Yoga Gen7 - OV9234 monochromatic sensor? */
> 	{ "6BF115T2", { 0 }, NULL, true },

If you have access to the Windows partition, an alternative is to check the .inf files like:

/w11/Windows/System32/DriverStore/FileRepository/ov2740.inf_amd64_* and  /w11/Windows/System32/DriverStore/FileRepository/ov9234.inf_amd64_*

You will find there the names without much need to experiment:

OV2740:
 - 4SF259T2
 - CJFLE23

OV9234:
 - 0BG101N3
 - 6BF115T2

> but the ov9734 sensor driver does not like its monochrome sibling variant and fails with -EINVAL during probing. No crash though.

I think you may be requesting a resolution or a format that's not supported.

Check /win/Windows/System32/drivers/graph_settings_OV9234_6BF115T2_ADL.xml for supported formats, ex:

```
grep main.format /win/Windows/System32/drivers/graph_settings_OV9234_6BF115T2_ADL.xml
    <main format="YUV420_10_PL" width="640" height="384" />
    <main format="YUV420_8_SP" width="640" height="360" peer="video0" />
    <main format="YUV420_10_PL" width="640" height="384" />
    <main format="YUV420_8_SP" width="640" height="360" peer="video0" />
    <main format="YUV420_10_PL" width="640" height="384" />
    <main format="YUV420_8_SP" width="640" height="360" peer="video0" />
    <main format="YUV420_10_PL" width="640" height="384" />
    <main format="YUV420_8_SP" width="640" height="360" peer="video0" />
    <main format="YUV420_10_PL" width="640" height="384" />
    <main format="YUV420_8_SP" width="640" height="360" peer="still0" />
    <main format="YUV420_10_PL" width="640" height="384" />
    <main format="YUV420_8_SP" width="640" height="360" peer="still0" />
    <main format="YUV420_10_PL" width="640" height="384" />
    <main format="YUV420_8_SP" width="640" height="360" peer="still0" />
    <main format="YUV420_10_PL" width="640" height="384" />
    <main format="YUV420_8_SP" width="640" height="360" peer="still0" />
```

ctrlcctrlv, 2023-03-11:
I'm now here after opening #106.

Thanks for helpful comments.

But I see:

```
Scope (_SB.PC00.I2C2)
    {
        Device (CAM0)
        {
            Name (_ADR, Zero)  // _ADR: Address
            Name (_HID, "INT3471")  // _HID: Hardware ID
            Name (_CID, "INT3481")  // _CID: Compatible ID
            Name (_DDN, "IMX135-CRDG2")  // _DDN: DOS Device Name
            Name (_UID, "0")  // _UID: Unique ID
            Name (_DEP, Package (0x01)  // _DEP: Dependencies
            {
                PMIC
            })
            Name (_PLD, Package (0x01)  // _PLD: Physical Location of Device
            {
                ToPLD (
                    PLD_Revision           = 0x2,
                    PLD_IgnoreColor        = 0x1,
                    PLD_Red                = 0x0,
                    PLD_Green              = 0x0,
                    PLD_Blue               = 0x0,
                    PLD_Width              = 0x0,
                    PLD_Height             = 0x0,
                    PLD_UserVisible        = 0x1,
                    PLD_Dock               = 0x0,
                    PLD_Lid                = 0x0,
                    PLD_Panel              = "BACK",
                    PLD_VerticalPosition   = "CENTER",
                    PLD_HorizontalPosition = "RIGHT",
                    PLD_Shape              = "VERTICALRECTANGLE",
                    PLD_GroupOrientation   = 0x0,
                    PLD_GroupToken         = 0x0,
                    PLD_GroupPosition      = 0x0,
                    PLD_Bay                = 0x0,
                    PLD_Ejectable          = 0x1,
                    PLD_EjectRequired      = 0x1,
                    PLD_CabinetNumber      = 0x0,
                    PLD_CardCageNumber     = 0x0,
                    PLD_Reference          = 0x0,
                    PLD_Rotation           = 0x0,
                    PLD_Order              = 0x0,
                    PLD_VerticalOffset     = 0xFFFF,
                    PLD_HorizontalOffset   = 0xFFFF)

            })
[...]
Scope (_SB.PC00.I2C4)
    {
        Device (CAM1)
        {
            Name (_ADR, Zero)  // _ADR: Address
            Name (_HID, "INT3474")  // _HID: Hardware ID
            Name (_CID, "INT3484")  // _CID: Compatible ID
            Name (_DDN, "OV2740-CRDG2")  // _DDN: DOS Device Name
            Name (_UID, "0")  // _UID: Unique ID
            Name (_DEP, Package (0x01)  // _DEP: Dependencies
            {
                ^^I2C2.PMIC
            })
            Name (_PLD, Package (0x01)  // _PLD: Physical Location of Device
            {
                ToPLD (
                    PLD_Revision           = 0x2,
                    PLD_IgnoreColor        = 0x1,
                    PLD_Red                = 0x0,
                    PLD_Green              = 0x0,
                    PLD_Blue               = 0x0,
                    PLD_Width              = 0x0,
                    PLD_Height             = 0x0,
                    PLD_UserVisible        = 0x1,
                    PLD_Dock               = 0x0,
                    PLD_Lid                = 0x0,
                    PLD_Panel              = "FRONT",
                    PLD_VerticalPosition   = "CENTER",
                    PLD_HorizontalPosition = "RIGHT",
                    PLD_Shape              = "VERTICALRECTANGLE",
                    PLD_GroupOrientation   = 0x0,
                    PLD_GroupToken         = 0x0,
                    PLD_GroupPosition      = 0x0,
                    PLD_Bay                = 0x0,
                    PLD_Ejectable          = 0x1,
                    PLD_EjectRequired      = 0x1,
                    PLD_CabinetNumber      = 0x0,
                    PLD_CardCageNumber     = 0x0,
                    PLD_Reference          = 0x0,
                    PLD_Rotation           = 0x0,
                    PLD_Order              = 0x0,
                    PLD_VerticalOffset     = 0xFFFF,
                    PLD_HorizontalOffset   = 0xFFFF)

            })
```

So I have what an Arducam IMX135?? lol

And OV2740-CRDG2 is Omnivision OV2740?

yorik, 2023-08-14:
I also have a problem with Lenovo Thinkpad X1 carbon gen 11: it loads, but there is no usable camera, even with kernel 6.4.4 (also for some reason `CONFIG_INTEL_SKL_INT3472` is disabled in Debian, so I have to rebuild kernel).

<details>
  <summary>$ grep -A4 '(CAM' dsdt.dsl</summary>

  ```
        Device (CAM0)
        {
            Name (_HID, "INT3471")  // _HID: Hardware ID
            Name (_DDN, "IMX135-CRDG2")  // _DDN: DOS Device Name
            Name (_UID, "0")  // _UID: Unique ID
--
        Device (CAM1)
        {
            Name (_HID, "INT3474")  // _HID: Hardware ID
            Name (_DDN, "OV2740-CRDG2")  // _DDN: DOS Device Name
            Name (_UID, "0")  // _UID: Unique ID
  ```
</details>

<details>
  <summary>$ lsmod | grep "ivsc\|ipu\|ov"</summary>

```
overlay               184320  0
intel_ipu6_isys       155648  0
videobuf2_dma_contig    24576  1 intel_ipu6_isys
videobuf2_v4l2         36864  2 intel_ipu6_isys,uvcvideo
videobuf2_common       73728  6 videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,intel_ipu6_isys,uvcvideo,videobuf2_memops
intel_ipu6_psys       110592  0
intel_ipu6            118784  2 intel_ipu6_isys,intel_ipu6_psys
ov2740                 24576  1
v4l2_fwnode            32768  2 intel_ipu6_isys,ov2740
v4l2_async             28672  3 v4l2_fwnode,intel_ipu6_isys,ov2740
videodev              327680  7 v4l2_async,v4l2_fwnode,videobuf2_v4l2,v4l2loopback,intel_ipu6_isys,uvcvideo,ov2740
mc                     86016  8 v4l2_async,videodev,snd_usb_audio,videobuf2_v4l2,intel_ipu6_isys,uvcvideo,videobuf2_common,ov2740
async_raid6_recov      20480  1 raid456
async_memcpy           16384  2 raid456,async_raid6_recov
async_pq               16384  2 raid456,async_raid6_recov
async_xor              16384  3 async_pq,raid456,async_raid6_recov
async_tx               16384  5 async_pq,async_memcpy,async_xor,raid456,async_raid6_recov
raid6_pq              122880  3 async_pq,raid456,async_raid6_recov
  ```
</details>

<details>
  <summary>$ v4l2-ctl --list-devices </summary>

```
ipu6 (PCI:pci:pci0000:00):
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8

ipu6 (pci:pci0000:00):
	/dev/media0

Intel MIPI Camera (platform:v4l2loopback-000):
	/dev/video0
  ```
</details>

<details>
  <summary>$ gst-launch-1.0 icamerasrc device-name=ov2740-uf buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink</summary>

[... 225 lines cut ...]
[   25.550234] intel-ipu6 0000:00:05.0: IPC reset done
[   25.550239] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[   25.550466] intel-ipu6 0000:00:05.0: firmware: direct-loading firmware intel/ipu6ep_fw.bin
[   25.554137] intel-ipu6 0000:00:05.0: FW version: 20230102
[   25.571127] intel-ipu6 0000:00:05.0: Found supported sensor INT3474:01
[   25.571338] intel-ipu6 0000:00:05.0: Connected 1 cameras
[   25.572705] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   25.581312] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   25.650805] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   25.650846] intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[   25.652115] intel-ipu6-isys intel-ipu6-isys0: bind ov2740 12-0036 nlanes is 2 port is 1
[   25.652835] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
[   25.678582] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[   25.678780] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[  114.546678] intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
 
  ```
</details>

```
$ sha256sum /lib/firmware/intel/ipu6ep_fw.bin
4603c7b261889a544c1110b8f5044a1ae39196b3f1730dd35c684530ea46cc70  /lib/firmware/intel/ipu6ep_fw.bin
```

`ipu6-drivers` is built on c3b762b275f6cbba473458859736b76548a80568 using DKMS instruction.

fhloston, 2023-08-15:
Have you also enabled  `CONFIG_VIDEO_V4L2_SUBDEV_API=y`?

The wishlist bug for including `CONFIG_INTEL_SKL_INT3472=m` already exists:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1038385

yorik, 2023-08-15:
@fhloston, thank you for the idea! It finally somewhat works!
I can see picture via `gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! xvimagesink`

But `/usr/bin/v4l2-relayd -i "icamerasrc buffer-count=7" -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! v4l2sink name=v4l2sink device=/dev/video0"` crashes with segmentation fault:
```
Aug 15 13:05:37 kernel: appsrc:src[78466]: segfault at 6b ip 00007f6d46fc85cd sp 00007f6d45c50728 error 4 in libgstreamer-1.0.so.0.2204.0[7f6d46f74000+b7000] likely on CPU 2 (core 4, socket 0)
Aug 15 13:05:37 kernel: Code: 06 00 e8 b6 cc fa ff 31 c0 eb ae 66 90 48 85 ff 74 23 48 85 f6 74 07 48 8b 47 60 48 89 06 48 85 d2 74 07 48 8b 47 50 48 89 02 <48> 8b 47 68 c3 66 0f 1f 44 00 00 48 83 ec 08 48 8d 15 88 7a 06 00
```

fhloston, 2023-08-15:
v4l2-relayd enchantment looks a bit different here:

`/usr/bin/v4l2-relayd -i icamerasrc buffer-count=7 -o appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink name=v4l2sink device=/dev/video0`

yorik, 2023-08-15:
Without putting args for `-i` and `-o` into quotes it starts ok, but relay no video. With the quotes it crashes. Unit file runs it with quotes.

I tried both options with `! video/x-raw,format=YUY2` part and without, it makes no difference (at least for now).

fhloston, 2023-08-15:
I also used 6.4.4 for a while but have reverted back to 6.3.7...

yorik, 2023-08-27:
After playing I managed to fix sigfault, but I have no idea how... My best guess is that enabling debug in module `v4l2loopback` fix it...

So `gst-launch-1.0 -v icamerasrc ! video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink device=/dev/video0` works wonderfully in chrome or mpv, but the camera is always on. I guess I can use it for now, manually switch it on and off when needed...

Unfortunately `v4l2-relayd -d -i icamerasrc -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink name=v4l2sink device=/dev/video0"` never enables camera and only gives black box  as result.

Debugging shows nothing useful:(

P.S. I was surprised how bad video quality of this camera, I'd expect something amazing...

yorik, 2023-08-28:
And it finally works! The problem was introduced in https://github.com/umlaeute/v4l2loopback/commit/c7a5cd4c06b41370790eec3a2a9884b764330074, I had to patch `v4l2-relayd` correspondingly, I'll send it to upstream soon. 

So now it works on Lenovo ThinkPad X1 gen 11, Debian/Sid, kernel 6.4.11

fhloston, 2023-09-04:
> So now it works on Lenovo ThinkPad X1 gen 11, Debian/Sid, kernel 6.4.11

Distro kernel without recompile? I just noticed, the debian wishlist bug with inclusion of `CONFIG_INTEL_SKL_INT3472=m` was closed successfully.

So it should be possible to use ipu6-dkms as is.

ChocolateLoverRaj, 2023-10-18:
@fhloston I also get this error `[   16.503942] i2c i2c-INT3537:00: deferred probe pending`. What are the exact steps (including which OS?) u used to get your webcam to work? I want to try the same steps to see if it works on my laptop.