intel/ipu6-drivers #49: Kernel 6.0, ov01a10 cannot find sensor anymore

madsl, 2022-10-06:
Dell XPS 9320

System was working with 5.18 kernel, but after updating kernel to 6.0 and update to newest git on ipu6-drivers and ivsc-driver today, ov01a10 cannot find the sensor anymore. I've applied [IOMMU-passthrough-for-intel-ipu.diff](https://github.com/intel/ipu6-drivers/blob/master/patch/IOMMU-passthrough-for-intel-ipu.diff) and [int3472-support-independent-clock-and-LED-gpios-5.17](https://github.com/intel/ipu6-drivers/blob/master/patch/int3472-support-independent-clock-and-LED-gpios-5.17%2B.patch) to the kernel, but it made no difference (maybe because the sensor is not an OV01A1S?)

Tried both with power_ctrl_logic enabled and disabled in the ipu6-drivers build, but maybe that module cannot be loaded at the same time as intel_skl_int3472* ?

```
[   33.902599] vsc_ace spi-INTC1094:00-5db76cf6-0a68-4ed6-9b78-0361635e2447: recv firmware id
[   33.931912] intel-ipu6 intel-ipu: enabling device (0000 -> 0002)
[   33.932043] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[   33.932088] intel-ipu6 intel-ipu: physical base address 0x603c000000
[   33.932089] intel-ipu6 intel-ipu: mapped as: 0x00000000b02d7b40
[   33.932135] intel-ipu6 intel-ipu: Unable to set secure mode
[   33.932136] intel-ipu6 intel-ipu: IPU in non-secure mode
[   33.932137] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[   33.932138] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[   33.932143] intel-ipu6 intel-ipu: Skip ipc reset for non-secure mode
[   33.932144] intel-ipu6 intel-ipu: IPC reset done
[   33.932145] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[   33.932147] Loading firmware: intel/ipu6ep_fw.bin
[   33.933156] intel-ipu6 intel-ipu: FW version: 20220510
[   33.933967] intel-ipu6 intel-ipu: IPU6-v3 driver version 1.0
[   34.019511] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[   34.019757] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[   34.022296] i2c i2c-12: i2c start failed len:-1 id:0 0
[   34.022299] ov01a10 i2c-OVTI01A0:00: failed to find sensor: -5
[   34.022382] ov01a10: probe of i2c-OVTI01A0:00 failed with error -5

```

hao-yao, 2022-10-09:
@madsl Could you try `dmesg | grep -i vsc`? if no output, this PR may resolve it: https://github.com/intel/ipu6-drivers/pull/50

madsl, 2022-10-10:
You're right, I get no output. I'll try the new instructions.

madsl, 2022-10-11:
You're correct, it works when merging with ivsc-driver.