intel/ipu6-drivers #65: Getting it working with rpm-ostree

Sorunome, 2022-11-19:
On fedora kinoite/silverblue you have `rpm-ostree` and there is [no support for dkms yet](https://github.com/coreos/rpm-ostree/issues/1091). Soru had tried to build with the `Build outside kernel source tree` method in a toolbox and that worked, however, after insmod'ing the module, this is the dmesg output:

```
[  121.407554] intel_ipu6: loading out-of-tree module taints kernel.
[  121.407712] intel_ipu6: module verification failed: signature and/or required key missing - tainting kernel
[  121.410343] intel-ipu6 intel-ipu: enabling device (0000 -> 0002)
[  121.410608] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[  121.410711] intel-ipu6 intel-ipu: physical base address 0x603c000000
[  121.410714] intel-ipu6 intel-ipu: mapped as: 0x000000004e3f88d7
[  121.412625] intel-ipu6 intel-ipu: IPU in secure mode
[  121.412632] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[  121.412633] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[  121.428923] intel-ipu6 intel-ipu: IPC reset done
[  121.428939] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[  121.429156] intel-ipu6 intel-ipu: Direct firmware load for intel/ipu6ep_fw.bin failed with error -2
[  121.429167] intel-ipu6 intel-ipu: Requesting signed firmware failed
[  121.429173] intel-ipu6: probe of intel-ipu failed with error -2
```

Any help would be greatly appreciated!i

nathansamson, 2022-11-19:
I get the same on kernel 6.0.8 on ('normal') fedora 37 with dkms install method

i am not certain but I think these might have to do with it https://github.com/intel/ipu6-drivers/pull/58

(note: what hw are you on, I am trying this out on carbon x1 gen 10)

Sorunome, 2022-11-19:
soru is using kernel 6.0.6 on a lenovo thinkpad x1 yoga gen 7

nathansamson, 2022-11-19:
Yes I am pretty sure you need the fixes from [jwrdegoede](https://github.com/jwrdegoede) but one of them is a patch in the kernel (https://lore.kernel.org/linux-acpi/20221022150722.31787-1-hdegoede@redhat.com/)

So unless thats merged (or backported to fedora) you would need a full build

Sorunome, 2022-11-20:
#38 is probably related, too, however `/usr/lib` is read-only with rpm-ostree