intel/ipu6-drivers #66: probe of intel-ipu failed with error -12

norbusan, 2022-11-20:
Lenovo X1 Carbon Gen10
Kernel 6.0.9 with following patches:
- int3472-support-independent-clock-and-LED-gpios-5.17+.patch
- IOMMU-passthrough-for-intel-ipu.diff
- https://lore.kernel.org/linux-acpi/20221022150722.31787-1-hdegoede@redhat.com/
(self built)
OS: Arch Linux

ipu6-drivers as of today, commit a1e6f0494bcc85

I get the following error on boot:
```
kernel: intel-ipu6 intel-ipu: enabling device (0000 -> 0002)
kernel: intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
kernel: intel-ipu6 intel-ipu: physical base address 0x603c000000
kernel: intel-ipu6 intel-ipu: mapped as: 0x000000007174b0a7
kernel: intel-ipu6 intel-ipu: IPU in secure mode
kernel: intel-ipu6 intel-ipu: IPU secure touch = 0x0
kernel: intel-ipu6 intel-ipu: IPU camera mask = 0xff
kernel: proc_thermal_pci 0000:00:04.0: enabling device (0000 -> 0002)
kernel: intel-ipu6 intel-ipu: IPC reset done
kernel: intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
kernel: intel_rapl_common: Found RAPL domain package
kernel: intel-ipu6 intel-ipu: FW version: 20220510
kernel: intel-ipu6: probe of intel-ipu failed with error -12
```

What am I missing? 

Thanks for any suggestions.

karolszk, 2023-02-11:
Hi @norbusan , have you got camera working eventually on ThinkPad Carbon Gen10?

On Debian I have installed  kernel driver as well, then camera firmware binaries, hal (the latest checkout from github ipu6) but camera still doesn't work. 
Boot logs are similar :

lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: enabling device (0000 -> 0002)
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: physical base address 0x603c000000
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: mapped as: 0x00000000d50191a0
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU in secure mode
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU secure touch = 0x0
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU camera mask = 0xff
lut 10 21:58:45 karol kernel: alg: No test for fips(ansi_cprng) (fips_ansi_cprng)
lut 10 21:58:45 karol kernel: proc_thermal_pci 0000:00:04.0: enabling device (0000 -> 0002)
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPC reset done
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: FW version: 20220510
lut 10 21:58:45 karol kernel: intel_rapl_common: Found RAPL domain package
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: physical base address 0x603c000000
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: mapped as: 0x000000005b6777b7
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU in secure mode
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU secure touch = 0x0
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU camera mask = 0xff
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPC reset done
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: FW version: 20220510
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: physical base address 0x603c000000
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: mapped as: 0x00000000e4a986ad
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU in secure mode
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU secure touch = 0x0
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU camera mask = 0xff
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPC reset done
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: FW version: 20220510
lut 10 21:58:45 karol mtp-probe[897]: checking bus 1, device 3: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8"
lut 10 21:58:45 karol mtp-probe[897]: bus: 1, device: 3 was not an MTP device
lut 10 21:58:45 karol kernel: ljca 1-8:1.0: LJCA USB device init success
lut 10 21:58:45 karol kernel: usbcore: registered new interface driver ljca
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: physical base address 0x603c000000
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: mapped as: 0x00000000d50191a0
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU in secure mode
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU secure touch = 0x0
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU camera mask = 0xff
lut 10 21:58:45 karol kernel: mei_me 0000:00:16.0: enabling device (0000 -> 0002)
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPC reset done
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: FW version: 20220510
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: physical base address 0x603c000000
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: mapped as: 0x00000000d50191a0
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU in secure mode
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU secure touch = 0x0
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPU camera mask = 0xff
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: IPC reset done
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: firmware: direct-loading firmware intel/ipu6ep_fw.bin
lut 10 21:58:45 karol kernel: intel-ipu6 intel-ipu: FW version: 20220510

any ideas?

norbusan, 2023-02-12:
> have you got camera working eventually on ThinkPad Carbon Gen10?

Hi @karolszk 

depending on your definition of "working". I got a video stream displayed from gst-streamer, so yes, it worked in some way. But trying to get it work in browser was a failure. Tried with v4l2-relayd and v4l2-loopback etc etc, but bsides the gst-streamer output (which is useless by itself) I cannot use it.

So I have stopped looking at it for the last months, lots of tinkering for no gain ATM. Will revisit it maybe in a year.