intel/ipu6-drivers #79: Doesn't work with CONFIG_X86_KERNEL_IBT=y (missing ENDBR)

twouters, 2023-01-10:
Got the following error while booting Arch Linux kernel version 6.1.x and camera doesn't work, laptop doesn't power down after shutdown and freezes during suspend.  
When I set `ibt=warn` the warning is thrown a few times until eventually the sensor registration completes and camera works (shutdown and suspend seem to work properly too).

```
Jan 10 11:08:15 kernel: traps: Missing ENDBR: kallsyms_lookup_name+0x4/0xf0
Jan 10 11:08:15 kernel: ------------[ cut here ]------------
Jan 10 11:08:15 kernel: WARNING: CPU: 8 PID: 467 at arch/x86/kernel/traps.c:252 kallsyms_lookup_name+0x4/0xf0
Jan 10 11:08:15 kernel: Modules linked in: mei_hdcp spi_ljca(OE) i2c_ljca(OE+) gpio_ljca(OE) dell_laptop intel_rapl_msr intel_tcc_cooling x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass rapl intel_cstate intel_uncore pcspkr psmouse dell_wmi_sysman firmware_attributes_class dell_wmi dell_smbios dcdbas ledtrig_audio dell_wmi_descriptor wmi_bmof snd_hda_codec_hdmi snd_sof_pci_intel_tgl snd_sof_intel_hda_common soundwire_intel soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_hdac_hda snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi soundwire_bus snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec iwlmvm snd_hda_core snd_hwdep mac80211 snd_pcm libarc4 spi_nor i2c_i801 snd_timer mtd i2c_smbus snd soundcore btusb iwlwifi btrtl btbcm intel_lpss_pci mei_me btintel intel_lpss btmtk cfg80211 idma64 ljca(OE) intel_ipu6_isys(OE) bluetooth
Jan 10 11:08:15 kernel:  videobuf2_dma_contig processor_thermal_device_pci videobuf2_memops ecdh_generic processor_thermal_device videobuf2_v4l2 rfkill crc16 videobuf2_common processor_thermal_rfim processor_thermal_mbox intel_ish_ipc intel_ipu6_psys(OE) processor_thermal_rapl intel_ishtp thunderbolt intel_ipu6(OE) intel_rapl_common ucsi_acpi typec_ucsi typec igen6_edac roles i2c_hid_acpi i2c_hid mei_vsc(OE) int3403_thermal int340x_thermal_zone ov01a10(OE) mei_csi(OE) mei_ace(OE) mei intel_vsc(OE) v4l2_fwnode v4l2_async intel_skl_int3472_tps68470 tps68470_regulator clk_tps68470 intel_hid intel_skl_int3472_discrete sparse_keymap int3400_thermal acpi_thermal_rel acpi_tad acpi_pad vfat fat mac_hid vboxnetflt(OE) vboxnetadp(OE) vboxdrv(OE) v4l2loopback(OE) videodev mc dm_multipath sg crypto_user nfsd auth_rpcgss nfs_acl lockd grace sunrpc fuse ip_tables x_tables btrfs blake2b_generic libcrc32c crc32c_generic xor raid6_pq dm_crypt cbc encrypted_keys trusted asn1_encoder tee usbhid dm_mod
Jan 10 11:08:15 kernel:  crct10dif_pclmul crc32_pclmul crc32c_intel polyval_clmulni serio_raw polyval_generic gf128mul atkbd ghash_clmulni_intel nvme libps2 vivaldi_fmap sha512_ssse3 aesni_intel nvme_core spi_intel_pci crypto_simd i8042 xhci_pci cryptd spi_intel nvme_common xhci_pci_renesas serio i915 drm_buddy intel_gtt video wmi drm_display_helper cec ttm
Jan 10 11:08:15 kernel: CPU: 8 PID: 467 Comm: systemd-udevd Tainted: G           OE      6.1.4-arch1-1 #1 b56a0be67d6a5f69f99015da4a908cae98ee5acc
Jan 10 11:08:15 kernel: Hardware name: Dell Inc. XPS 9315/00KRKP, BIOS 1.6.1 11/22/2022
Jan 10 11:08:15 kernel: RIP: 0010:kallsyms_lookup_name+0x4/0xf0
Jan 10 11:08:15 kernel: Code: 0b 00 48 8b 54 24 28 65 48 2b 14 25 28 00 00 00 75 0e 48 83 c4 30 5b c3 cc cc cc cc 31 c0 eb e2 e8 61 0c ab 00 90 66 0f 1f 00 <0f> 1f 44 00 00 41 56 b9 40 00 00 00 41 55 49 89 fd 41 54 55 53 48
Jan 10 11:08:15 kernel: RSP: 0018:ffffa85e00ed7618 EFLAGS: 00010246
Jan 10 11:08:15 kernel: RAX: ffffa85e00ed7630 RBX: 0000000000000000 RCX: 0000000000000002
Jan 10 11:08:15 kernel: RDX: 0000000000000001 RSI: 0000000055555554 RDI: ffffffffc14540b1
Jan 10 11:08:15 kernel: RBP: ffffa85e00ed76d0 R08: 0000000000032790 R09: 0000000000000000
Jan 10 11:08:15 kernel: R10: 0000000000000008 R11: 0000000000000000 R12: ffffa85e00ed7620
Jan 10 11:08:15 kernel: R13: ffffffff89d6fc34 R14: ffff8a630d885800 R15: ffff8a6301f61030
Jan 10 11:08:15 kernel: FS:  00007f4a85645080(0000) GS:ffff8a6a6f800000(0000) knlGS:0000000000000000
Jan 10 11:08:15 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan 10 11:08:15 kernel: CR2: 000055c092bd1000 CR3: 0000000109eb0004 CR4: 0000000000f70ee0
Jan 10 11:08:15 kernel: PKRU: 55555554
Jan 10 11:08:15 kernel: Call Trace:
Jan 10 11:08:15 kernel:  <TASK>
Jan 10 11:08:15 kernel:  init_vsc_symbols+0x68/0xb0 [ov01a10 4e3dfa54a67381019516620897f0731b3644f571]
Jan 10 11:08:15 kernel:  ? kallsyms_lookup_name+0x4/0xf0
Jan 10 11:08:15 kernel:  ov01a10_probe+0x57/0x470 [ov01a10 4e3dfa54a67381019516620897f0731b3644f571]
Jan 10 11:08:15 kernel:  ? devres_open_group+0x45/0x100
Jan 10 11:08:15 kernel:  ? i2c_device_probe+0x135/0x330
Jan 10 11:08:15 kernel:  i2c_device_probe+0x135/0x330
Jan 10 11:08:15 kernel:  really_probe+0xdb/0x380
Jan 10 11:08:15 kernel:  ? pm_runtime_barrier+0x54/0x90
Jan 10 11:08:15 kernel:  __driver_probe_device+0x78/0x170
Jan 10 11:08:15 kernel:  driver_probe_device+0x1f/0x90
Jan 10 11:08:15 kernel:  __device_attach_driver+0x89/0x110
Jan 10 11:08:15 kernel:  ? driver_allows_async_probing+0x70/0x70
Jan 10 11:08:15 kernel:  bus_for_each_drv+0x8c/0xe0
Jan 10 11:08:15 kernel:  __device_attach+0xb2/0x1e0
Jan 10 11:08:15 kernel:  bus_probe_device+0x92/0xb0
Jan 10 11:08:15 kernel:  device_add+0x422/0x9a0
Jan 10 11:08:15 kernel:  i2c_new_client_device+0x197/0x2c0
Jan 10 11:08:15 kernel:  i2c_acpi_add_device+0x94/0xb0
Jan 10 11:08:15 kernel:  acpi_ns_walk_namespace+0x16d/0x2db
Jan 10 11:08:15 kernel:  ? i2c_acpi_resource_count+0x30/0x30
Jan 10 11:08:15 kernel:  ? i2c_acpi_resource_count+0x30/0x30
Jan 10 11:08:15 kernel:  acpi_walk_namespace+0xf5/0x141
Jan 10 11:08:15 kernel:  i2c_acpi_register_devices+0x48/0x80
Jan 10 11:08:15 kernel:  i2c_register_adapter+0x225/0x6a0
Jan 10 11:08:15 kernel:  ljca_i2c_probe+0x1fe/0x270 [i2c_ljca 0b3b7c670dc09a589f6bdd3732ba788e08199e00]
Jan 10 11:08:15 kernel:  platform_probe+0x45/0x90
Jan 10 11:08:15 kernel:  really_probe+0xdb/0x380
Jan 10 11:08:15 kernel:  ? pm_runtime_barrier+0x54/0x90
Jan 10 11:08:15 kernel:  __driver_probe_device+0x78/0x170
Jan 10 11:08:15 kernel:  driver_probe_device+0x1f/0x90
Jan 10 11:08:15 kernel:  __driver_attach+0xd2/0x1c0
Jan 10 11:08:15 kernel:  ? __device_attach_driver+0x110/0x110
Jan 10 11:08:15 kernel:  bus_for_each_dev+0x88/0xd0
Jan 10 11:08:15 kernel:  bus_add_driver+0x1b2/0x200
Jan 10 11:08:15 kernel:  driver_register+0x8d/0xe0
Jan 10 11:08:15 kernel:  ? 0xffffffffc062b000
Jan 10 11:08:15 kernel:  do_one_initcall+0x5a/0x220
Jan 10 11:08:15 kernel:  do_init_module+0x4a/0x1e0
Jan 10 11:08:15 kernel:  __do_sys_init_module+0x17f/0x1b0
Jan 10 11:08:15 kernel:  do_syscall_64+0x5c/0x90
Jan 10 11:08:15 kernel:  ? syscall_exit_to_user_mode+0x1b/0x40
Jan 10 11:08:15 kernel:  ? do_syscall_64+0x6b/0x90
Jan 10 11:08:15 kernel:  ? do_user_addr_fault+0x1e0/0x6a0
Jan 10 11:08:15 kernel:  ? syscall_exit_to_user_mode+0x1b/0x40
Jan 10 11:08:15 kernel:  ? exc_page_fault+0x74/0x170
Jan 10 11:08:15 kernel:  entry_SYSCALL_64_after_hwframe+0x63/0xcd
Jan 10 11:08:15 kernel: RIP: 0033:0x7f4a86121eae
Jan 10 11:08:15 kernel: Code: 48 8b 0d dd ee 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d aa ee 0c 00 f7 d8 64 89 01 48
Jan 10 11:08:15 kernel: RSP: 002b:00007ffc314f9048 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
Jan 10 11:08:15 kernel: RAX: ffffffffffffffda RBX: 000055c092b497c0 RCX: 00007f4a86121eae
Jan 10 11:08:15 kernel: RDX: 00007f4a86608343 RSI: 000000000000505d RDI: 000055c092bb2800
Jan 10 11:08:15 kernel: RBP: 00007f4a86608343 R08: 27d4eb2f165667c5 R09: 85ebca77c2b2ae63
Jan 10 11:08:15 kernel: R10: 00000000000035f1 R11: 0000000000000246 R12: 0000000000020000
Jan 10 11:08:15 kernel: R13: 000055c092b43d90 R14: 000055c092b497c0 R15: 000055c092aa93e0
Jan 10 11:08:15 kernel:  </TASK>
Jan 10 11:08:15 kernel: ---[ end trace 0000000000000000 ]---
```

Thesola10, 2023-02-20:
I see `kallsyms_lookup_name`, so I presume you are using my AUR package? I recently updated it with the official build process which does not require this hack anymore.

twouters, 2023-02-22:
Closing per https://github.com/intel/ipu6-drivers/issues/79#issuecomment-1436726479

closed 2023-02-22 by twouters