intel/ipu6-drivers #82: Is `dkms add .` still possible?

realestninja, 2023-01-25:
When following the instructions to [build with dkms](https://github.com/intel/ipu6-drivers/#3-build-with-dkms) I receive an error message:

```
$ sudo dkms add .

Error! Arguments <module> and <module-version> are not specified.
Usage: add <module>/<module-version> or
       add -m <module>/<module-version> or
       add -m <module> -v <module-version>

```

Additional info:
```
$ dkms --version
dkms-3.0.10
```

What do I need to put as <module> and <module-version>?

realestninja, 2023-01-30:
nvm, I just saw that the [dkms.conf](https://github.com/intel/ipu6-drivers/blob/master/dkms.conf) file is provided but isn't copied when following the instructions. After adding that file, it works.

closed 2023-01-30 by realestninja

gustavgeissler, 2023-01-30:
Could you please elaborate?
After copying dkms.conf and doing `dkms add .` I get 

> /home/[user]/ipu6-drivers/dkms.conf: line 8: syntax error near unexpected token 'newline'
/home/[user]/ipu6-drivers/dkms.conf: line 8: `<!DOCTYPE html>'
dkms.conf: Error! No 'DEST_MODULE_LOCATION' directive specified.
dkms.conf: Error! No 'PACKAGE_NAME' directive specified.
dkms.conf: Error! No 'PACKAGE_VERSION' directive specified.
Error! Bad conf file.
File:  does not represent a valid dkms.conf file.

jimafisk, 2024-04-11:
I got the same error as @gustavgeissler when I tried to `wget https://github.com/intel/ipu6-drivers/blob/master/dkms.conf` in my project root, this was copying the HTML from the page and not just the raw contents of the file. Copying just the file contents manually into a `dkms.conf` file in my project root worked!