intel/ipu6-drivers #85: `dkms autoinstall` fails (6.1.7 kernel)

realestninja, 2023-01-30:
I followed the instructions from the README section 3. I copied the dkms.conf aswell.

Here is a log:
```
$ uname -r
6.1.7-1-MANJARO

$ sudo dkms autoinstall ipu6-drivers/0.0.0

Sign command: /usr/lib/modules/6.1.7-1-MANJARO/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...(bad exit status: 2)
make -j16 KERNELRELEASE=6.1.7-1-MANJARO KERNELRELEASE=6.1.7-1-MANJARO KERNEL_SRC=/usr/lib/modules/6.1.7-1-MANJARO/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.7-1-MANJARO (x86_64)
Consult /var/lib/dkms/ipu6-drivers/0.0.0/build/make.log for more information.
Sign command: /usr/lib/modules/6.1.7-1-MANJARO/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...(bad exit status: 2)
make -j16 KERNELRELEASE=6.1.7-1-MANJARO KERNELRELEASE=6.1.7-1-MANJARO KERNEL_SRC=/usr/lib/modules/6.1.7-1-MANJARO/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.7-1-MANJARO (x86_64)
Consult /var/lib/dkms/ivsc-driver/1.0.0/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
```

Here is the log for the `make.log` file:
```
DKMS make.log for ivsc-driver-1.0.0 for kernel 6.1.7-1-MANJARO (x86_64)
Mon 30 Jan 2023 17:55:52 CET
make: *** No targets specified and no makefile found.  Stop.
```

I have the `linux61-headers` package from the AUR installed. Anything else I require?

hao-yao, 2023-02-14:
Would you try directly make without dkms, or upload the `/var/lib/dkms/ivsc-driver/1.0.0/build/make.log`?

realestninja, 2023-02-14:
```
DKMS make.log for ivsc-driver-1.0.0 for kernel 6.1.9-1-MANJARO (x86_64)
Fri 03 Feb 2023 09:35:29 CET
make: *** No targets specified and no makefile found.  Stop.
```

c-sprinks, 2025-02-13:
was this ever solved?