intel/ipu6-drivers #94: two xps 9320 laptops with identical software, one has working ov01a10, one fails

kervel, 2023-02-22:
Hello,

due to touchpad issues my dell xps13 9320 is getting replaced by an identical one (the identical one could be a newer hardware revision, since the original one is from last august and the replacement from february). they both run the same os (ubuntu 2204) and firmware, but the old one has working webcam while the new one hasn't. the old laptop (with working webcam) will be shipped back to dell.

i was updating this issue https://github.com/intel/icamerasrc/issues/26 but i'm moving it here as that repo seems to be dead.

the output is identical on the working and non-working xps13 plus. I still have the nonworking one for two weeks before dell picks it up, i'm going to do some more tests to find exactly what is different between the two:

* both systems run ubuntu 22.04 with kernel 5.19.0-32-generic 

this is the dmesg output for the working system:

```
[    4.958108] intel-ipu6 intel-ipu: enabling device (0000 -> 0002)
[    4.958253] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[    4.958443] intel-ipu6 intel-ipu: physical base address 0x603c000000
[    4.958445] intel-ipu6 intel-ipu: mapped as: 0x000000005e4a6637
[    4.958549] intel-ipu6 intel-ipu: Unable to set secure mode
[    4.958551] intel-ipu6 intel-ipu: IPU in non-secure mode
[    4.958552] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[    4.958553] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[    4.958591] intel-ipu6 intel-ipu: Skip ipc reset for non-secure mode
[    4.958592] intel-ipu6 intel-ipu: IPC reset done
[    4.958593] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[    4.959949] intel-ipu6 intel-ipu: FW version: 20220510
[    4.962623] intel-ipu6 intel-ipu: IPU6-v3 driver version 1.0
[    5.006069] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[    5.006260] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[   15.558343] intel-ipu6-isys intel-ipu6-isys0: bind ov01a10 17-0036 nlanes is 1 port is 2
[   15.558528] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
```

this is the dmesg output for the non-working system:

```
[   70.816402] intel-ipu6 intel-ipu: enabling device (0000 -> 0002)
[   70.816759] intel-ipu6 intel-ipu: Device 0x465d (rev: 0x2)
[   70.816776] intel-ipu6 intel-ipu: physical base address 0x603c000000
[   70.816778] intel-ipu6 intel-ipu: mapped as: 0x00000000cb7fe56f
[   70.816820] intel-ipu6 intel-ipu: Unable to set secure mode
[   70.816822] intel-ipu6 intel-ipu: IPU in non-secure mode
[   70.816823] intel-ipu6 intel-ipu: IPU secure touch = 0x0
[   70.816823] intel-ipu6 intel-ipu: IPU camera mask = 0xff
[   70.816961] intel-ipu6 intel-ipu: Skip ipc reset for non-secure mode
[   70.816964] intel-ipu6 intel-ipu: IPC reset done
[   70.816966] intel-ipu6 intel-ipu: cpd file name: intel/ipu6ep_fw.bin
[   70.818655] intel-ipu6 intel-ipu: FW version: 20220510
[   70.822366] intel-ipu6 intel-ipu: IPU6-v3 driver version 1.0
[   70.863378] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[   70.863485] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[ 1002.585603] intel-ipu6 intel-ipu: IPU in non-secure mode
[ 1002.585611] intel-ipu6 intel-ipu: Skip ipc reset for non-secure mode
[12336.171476] intel-ipu6 intel-ipu: IPU in non-secure mode
[12336.171483] intel-ipu6 intel-ipu: Skip ipc reset for non-secure mode
```

both have userspace 0~git202302070245.a359c1e~ubuntu22.04.2 libcamhal-ipu6ep0
they both have the same kernel modules.
i verified the correct working of both cameras in windows to exclude hardware problems.

the dmesg on the non-working system doesn't name ov01a10, so maybe its a different camera (is there a way to figure this out?)

the userspace component logs this:

```
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:48.88] CamHAL[INF] aiqb file name ov01a1s.aiqb
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:48.88] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:48.92] CamHAL[ERR] Get entity fail for calling getEntityById
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:48.92] CamHAL[ERR] Get entity fail for calling getEntityById
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:48.92] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:48.92] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:48.92] CamHAL[ERR] set up mediaCtl failed
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:48.92] CamHAL[ERR] @configure Device Configure failed
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:48.92] CamHAL[ERR] failed to config streams.
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:57.85] CamHAL[INF] aiqb file name ov13b10.aiqb
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:57.85] CamHAL[ERR] there is no aiqb file:ov13b10
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:57.85] CamHAL[INF] aiqb file name ov13b10.aiqb
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:57.85] CamHAL[ERR] there is no aiqb file:ov13b10
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:57.85] CamHAL[INF] aiqb file name ov8856.aiqb
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:57.85] CamHAL[ERR] there is no aiqb file:ov8856
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:57.85] CamHAL[INF] aiqb file name ov8856.aiqb
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:57.85] CamHAL[ERR] there is no aiqb file:ov8856
feb 21 11:25:57 rhea sh[1412]: [02-21 11:25:57.85] CamHAL[INF] aiqb file name ov01a10.aiqb

```

csdvrx, 2023-02-23:
The `Failed to find DevName` problem means it's likely you need to hardcode a different identifier in the module.

> the dmesg on the non-working system doesn't name ov01a10, so maybe its a different camera (is there a way to figure this out?)

The simplest way is to look at the .inf files from either laptop to extract all the possible matching names; I commented on that on #48

kervel, 2023-02-27:
Hello,

on my laptop, according to the dsdt, i should have an ov2740 (INT3474, OV2740,CRDG2). however, i could not find the .inf files you described for ov2740. Instead, i found files named ov01a.inf_amd64_26fa9596f59c64c3 and 3 similar ones.
they mention CJFLE13, CJFLE21 and 1BG101N3 and 0BF118N3 in the [SourceDisksFiles] section.

so i now don't know if i have an ov2740 or an ov01a but i guess ov01a (since other dell 9320 https://github.com/intel/ipu6-drivers/issues/49 also mentions this)

greetings,
Frank

kervel, 2023-02-27:
For completeness, i discovered that gst-launch output is very different on the working laptop vs the broken one:

this is the command:
```
sudo  gst-launch-1.0 icamerasrc  device-name=ov01a10-uf buffer-count=7 ! \
 video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink 
```

working:
```
[02-27 21:57:57.551] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 21:57:57.551] CamHAL[ERR] Parse AE eExposure time range failed
[02-27 21:57:57.551] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 21:57:57.551] CamHAL[ERR] Parse AE gain range failed
Setting pipeline to PAUSED ...
[02-27 21:57:57.574] CamHAL[INF] aiqb file name ov01a10.aiqb
[02-27 21:57:57.574] CamHAL[INF] aiqb file name ov01a10.aiqb
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
```

broken:
```
[02-27 22:00:51.413] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.413] CamHAL[ERR] Parse AE eExposure time range failed
[02-27 22:00:51.413] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.413] CamHAL[ERR] Parse AE gain range failed
[02-27 22:00:51.413] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.413] CamHAL[ERR] Parse AE eExposure time range failed
[02-27 22:00:51.413] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.413] CamHAL[ERR] Parse AE gain range failed
[02-27 22:00:51.413] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.413] CamHAL[ERR] Parse AE eExposure time range failed
[02-27 22:00:51.413] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.413] CamHAL[ERR] Parse AE gain range failed
[02-27 22:00:51.413] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.413] CamHAL[ERR] Parse AE eExposure time range failed
[02-27 22:00:51.413] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.413] CamHAL[ERR] Parse AE gain range failed
[02-27 22:00:51.413] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.413] CamHAL[ERR] Parse AE eExposure time range failed
[02-27 22:00:51.413] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.413] CamHAL[ERR] Parse AE gain range failed
[02-27 22:00:51.414] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.414] CamHAL[ERR] Parse AE eExposure time range failed
[02-27 22:00:51.414] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.414] CamHAL[ERR] Parse AE gain range failed
[02-27 22:00:51.414] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.414] CamHAL[ERR] Parse AE eExposure time range failed
[02-27 22:00:51.414] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.414] CamHAL[ERR] Parse AE gain range failed
[02-27 22:00:51.414] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.414] CamHAL[ERR] Parse AE eExposure time range failed
[02-27 22:00:51.414] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.414] CamHAL[ERR] Parse AE gain range failed
[02-27 22:00:51.414] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.414] CamHAL[ERR] Parse AE eExposure time range failed
[02-27 22:00:51.414] CamHAL[ERR] Malformed ET range in exposure time range configuration
[02-27 22:00:51.414] CamHAL[ERR] Parse AE gain range failed
Setting pipeline to PAUSED ...
[02-27 22:00:51.697] CamHAL[INF] aiqb file name ov13b10.aiqb
[02-27 22:00:51.697] CamHAL[ERR] there is no aiqb file:ov13b10
[02-27 22:00:51.697] CamHAL[INF] aiqb file name ov13b10.aiqb
[02-27 22:00:51.697] CamHAL[ERR] there is no aiqb file:ov13b10
[02-27 22:00:51.697] CamHAL[INF] aiqb file name ov8856.aiqb
[02-27 22:00:51.697] CamHAL[ERR] there is no aiqb file:ov8856
[02-27 22:00:51.697] CamHAL[INF] aiqb file name ov8856.aiqb
[02-27 22:00:51.697] CamHAL[ERR] there is no aiqb file:ov8856
[... 22 lines cut ...]
[02-27 22:00:51.699] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[02-27 22:00:51.699] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[02-27 22:00:51.699] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[02-27 22:00:51.699] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[02-27 22:00:51.699] CamHAL[INF] aiqb file name ov01a1s.aiqb
[02-27 22:00:51.699] CamHAL[INF] aiqb file name ov01a1s.aiqb
[02-27 22:00:51.699] CamHAL[ERR] Failed to find DevName for cameraId: 4, get video node: ov01a10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[02-27 22:00:51.706] CamHAL[ERR] Get entity fail for calling getEntityById
[02-27 22:00:51.706] CamHAL[ERR] Get entity fail for calling getEntityById
[02-27 22:00:51.706] CamHAL[ERR] setup Link ov01a10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[02-27 22:00:51.706] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[02-27 22:00:51.706] CamHAL[ERR] set up mediaCtl failed
[02-27 22:00:51.706] CamHAL[ERR] @configure Device Configure failed
[02-27 22:00:51.706] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.004845432
Setting pipeline to NULL ...
```

kervel, 2023-02-27:
a kernel upgrade to 6.1.0 later and now i have a working webcam! I verified the kernel version on the "working" laptop to check if i missed something, but i didn't. There, the webcam is working fine with the older kernel. Leaving this here for documentation purposes.

closed 2023-02-27 by kervel

alesrebec, 2023-03-04:
Hi @kervel 
I noticed you have it working on kernel 6.1, I have 6.1.15 (latest) with ubuntu 22.04.2 on Lenovo X1 Carbon gen 10 and the error in gst is similar to yours.

```
[03-04 11:38:26.298] CamHAL[ERR] Malformed ET range in exposure time range configuration
[03-04 11:38:26.298] CamHAL[ERR] Parse AE eExposure time range failed
[03-04 11:38:26.298] CamHAL[ERR] Malformed ET range in exposure time range configuration
[03-04 11:38:26.298] CamHAL[ERR] Parse AE gain range failed
...
[03-04 11:38:26.299] CamHAL[ERR] Malformed ET range in exposure time range configuration
[03-04 11:38:26.299] CamHAL[ERR] Parse AE eExposure time range failed
[03-04 11:38:26.299] CamHAL[ERR] Malformed ET range in exposure time range configuration
[03-04 11:38:26.299] CamHAL[ERR] Parse AE gain range failed
Setting pipeline to PAUSED ...
[03-04 11:38:26.582] CamHAL[INF] aiqb file name ov13b10.aiqb
[03-04 11:38:26.582] CamHAL[ERR] there is no aiqb file:ov13b10
[03-04 11:38:26.582] CamHAL[INF] aiqb file name ov13b10.aiqb
[03-04 11:38:26.582] CamHAL[ERR] there is no aiqb file:ov13b10
[03-04 11:38:26.582] CamHAL[INF] aiqb file name ov8856.aiqb
[03-04 11:38:26.582] CamHAL[ERR] there is no aiqb file:ov8856
[03-04 11:38:26.582] CamHAL[INF] aiqb file name ov8856.aiqb
[03-04 11:38:26.582] CamHAL[ERR] there is no aiqb file:ov8856
[03-04 11:38:26.582] CamHAL[INF] aiqb file name ov01a10.aiqb
[03-04 11:38:26.582] CamHAL[INF] aiqb file name ov01a10.aiqb
[03-04 11:38:26.582] CamHAL[INF] aiqb file name ov01a10.aiqb
[03-04 11:38:26.582] CamHAL[INF] aiqb file name ov01a10.aiqb
[03-04 11:38:26.582] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[03-04 11:38:26.582] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[03-04 11:38:26.582] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[03-04 11:38:26.583] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[03-04 11:38:26.584] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[03-04 11:38:26.584] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[03-04 11:38:26.584] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[03-04 11:38:26.584] CamHAL[INF] aiqb file name ov01a1s.aiqb
[03-04 11:38:26.584] CamHAL[INF] aiqb file name ov01a1s.aiqb
[03-04 11:38:26.584] CamHAL[ERR] Failed to find DevName for cameraId: 12, get video node: ov2740 , devname: /dev/v4l-subdev1
[03-04 11:38:26.586] CamHAL[ERR] MediaControl init failed
Setting pipeline to NULL ...
Freeing pipeline ...
```
Do you have any clue what might it be?