intel/ipu6-drivers #105: Buil-in webcam HP TrueVision 5M does not work

constantine2nd, 2023-03-10:
`Webcam:` HP True Vision 5MP IR camera with camera shutter, temporal noise reduction and integrated dual array digital microphones

`System:`
Kernel: 5.15.0-67-generic x86_64 bits: 64 compiler: gcc v: 11.3.0 Desktop: Cinnamon 5.6.7
tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.1 Vera base: Ubuntu 22.04 jammy
Machine:
Type: Convertible System: HP product: HP Spectre x360 2-in-1 Laptop 16-f0xxx
v: Type1ProductConfigId serial: <superuser required> Chassis: type: 31
serial: <superuser required>
Mobo: HP model: 891A v: 92.45 serial: <superuser required> UEFI: Insyde v: F.14
date: 01/09/2023
Battery:
ID-1: BAT1 charge: 64.6 Wh (80.4%) condition: 80.3/83.0 Wh (96.7%) volts: 12.2 min: 11.6
model: Hewlett-Packard PABAS0241231 serial: <filter> status: Discharging
Device-1: hid-0018:04F3:2E0C.0001-battery model: ELAN2513:00 04F3:2E0C serial: N/A charge: N/A
status: N/A
CPU:
Info: quad core model: 11th Gen Intel Core i7-11390H bits: 64 type: MT MCP arch: Tiger Lake
rev: 2 cache: L1: 320 KiB L2: 5 MiB L3: 12 MiB
Speed (MHz): avg: 3297 high: 3703 min/max: 400/4800:5000 cores: 1: 2841 2: 2798 3: 3703
4: 3569 5: 2899 6: 3364 7: 3606 8: 3602 bogomips: 54681
Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
Device-1: Intel TigerLake-LP GT2 [Iris Xe Graphics] vendor: Hewlett-Packard driver: i915
v: kernel ports: active: eDP-1 empty: DP-1, DP-2, DP-3, DP-4, HDMI-A-1 bus-ID: 0000:00:02.0
chip-ID: 8086:9a49
Display: x11 server: X.Org v: 1.21.1.3 driver: X: loaded: modesetting unloaded: fbdev,vesa
gpu: i915 display-ID: :0 screens: 1
Screen-1: 0 s-res: 3072x1920 s-dpi: 96
Monitor-1: eDP-1 model: AU Optronics res: 3072x1920 dpi: 227 diag: 406mm (16")
OpenGL: renderer: Mesa Intel Xe Graphics (TGL GT2) v: 4.6 Mesa 22.2.5 direct render: Yes
Audio:
Device-1: Intel vendor: Hewlett-Packard driver: intel-ipu6 bus-ID: 0000:00:05.0
chip-ID: 8086:9a19
Device-2: Intel Tiger Lake-LP Smart Sound Audio vendor: Hewlett-Packard
driver: sof-audio-pci-intel-tgl bus-ID: 0000:00:1f.3 chip-ID: 8086:a0c8
Sound Server-1: ALSA v: k5.15.0-67-generic running: yes
Sound Server-2: PulseAudio v: 15.99.1 running: yes
Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
Device-1: Intel Wi-Fi 6 AX201 driver: iwlwifi v: kernel bus-ID: 0000:00:14.3 chip-ID: 8086:a0f0
IF: wlo1 state: up mac: <filter>
IF-ID-1: br-cbe742b0b3e5 state: down mac: <filter>
IF-ID-2: docker0 state: down mac: <filter>
Bluetooth:
Device-1: Intel AX201 Bluetooth type: USB driver: btusb v: 0.8 bus-ID: 3-10:3 chip-ID: 8087:0026
Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 3.0 lmp-v: 5.2
sub-v: 237e
RAID:
Hardware-1: Intel Volume Management Device NVMe RAID Controller driver: vmd v: 0.6
bus-ID: 0000:00:0e.0 chip-ID: 8086:9a0b
Drives:
Local Storage: total: 476.94 GiB used: 39.83 GiB (8.4%)
ID-1: /dev/nvme0n1 vendor: SK Hynix model: PC801 HFS512GEJ9X101N size: 476.94 GiB
speed: 63.2 Gb/s lanes: 4 serial: <filter> temp: 30.9 C
Partition:
ID-1: / size: 283.7 GiB used: 39.74 GiB (14.0%) fs: ext4 dev: /dev/nvme0n1p5
ID-2: /boot/efi size: 256 MiB used: 94.5 MiB (36.9%) fs: vfat dev: /dev/nvme0n1p1
Swap:
ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile
USB:
Hub-1: 1-0:1 info: Hi-speed hub with single TT ports: 1 rev: 2.0 speed: 480 Mb/s
chip-ID: 1d6b:0002
Hub-2: 2-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
Hub-3: 3-0:1 info: Hi-speed hub with single TT ports: 12 rev: 2.0 speed: 480 Mb/s
chip-ID: 1d6b:0002
Device-1: 3-6:2 info: Synaptics type: <vendor specific> driver: N/A rev: 2.0 speed: 12 Mb/s
chip-ID: 06cb:00ff
Device-2: 3-10:3 info: Intel AX201 Bluetooth type: Bluetooth driver: btusb rev: 2.0
speed: 12 Mb/s chip-ID: 8087:0026
Hub-4: 4-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
Sensors:
System Temperatures: cpu: 27.8 C mobo: 10.0 C
Fan Speeds (RPM): N/A
Repos:
Packages: 2285 apt: 2266 flatpak: 19
No active apt repos in: /etc/apt/sources.list
Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
1: deb http: //packages.linuxmint.com vera main upstream import backport
2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
Info:
Processes: 314 Uptime: 5m Memory: 15.38 GiB used: 2.77 GiB (18.0%) Init: systemd v: 249
runlevel: 5 Compilers: gcc: 11.3.0 alt: 11/12 Client: Unknown python3.10 client inxi: 3.3.13

dgttm, 2023-07-11:
yes, this affects the HP Spectre x360 2-in-1 Laptop 16 (e.g. 16-f2777ng (2023) 

OS: Ubuntu 22.04
Kernel 5.19.0-46-generic, also tested Kernel 6.4.0

no cam found at all

$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 06cb:00ff Synaptics, Inc. 
Bus 003 Device 003: ID 8087:0033 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I`d be super happy if a coder could take care of it! 

regards

Nicolas

dgttm, 2023-08-08:
not even with kernel 6.5.rc3

nico@lx3:~$ uname -a
Linux lx3 6.5.0-060500rc5-generic #202308062330 SMP PREEMPT_DYNAMIC Sun Aug  6 23:35:42 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
nico@lx3:~$ sudo lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 06cb:00ff Synaptics, Inc. 
Bus 003 Device 007: ID 2109:8888 VIA Labs, Inc. PD3.0 USB-C Device     
Bus 003 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 003: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Bus 003 Device 005: ID 8087:0033 Intel Corp. 
Bus 003 Device 002: ID 1532:0e03 Razer USA, Ltd Gaming Webcam [Kiyo]
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 002 Device 003: ID 05e3:0749 Genesys Logic, Inc. SD Card Reader and Writer
Bus 002 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub             
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

ilyasdc, 2023-08-26:
> Hi, if you familiar with driver development you can follow RealSense development and with small commitment you can get your webcam work. We have no alderlake laptops to work on as reference. https://github.com/dmipx/ipu6-drivers/tree/realsense_ipu6

Hello, 
I am willing to help. I recently got a HP ENVY x360 2-in-1 Laptop 13-bf0xxx with this camera.
Except using your fork of the drivers, what else can I do ?

I will build them and report.

ilyasdc, 2023-09-10:
Hi @dmipx,
The branch you refer (realsense_ipu6) doesn't build. Here are the logs 
[build.log](https://github.com/intel/ipu6-drivers/files/12569916/build.log)

ilyasdc, 2023-10-28:
> `Webcam:` HP True Vision 5MP IR camera with camera shutter, temporal noise reduction and integrated dual array digital microphones
> 
> `System:` Kernel: 5.15.0-67-generic x86_64 bits: 64 compiler: gcc v: 11.3.0 Desktop: Cinnamon 5.6.7 tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.1 Vera base: Ubuntu 22.04 jammy Machine: Type: Convertible System: HP product: HP Spectre x360 2-in-1 Laptop 16-f0xxx v: Type1ProductConfigId serial: Chassis: type: 31 serial: Mobo: HP model: 891A v: 92.45 serial: UEFI: Insyde v: F.14 date: 01/09/2023 Battery: ID-1: BAT1 charge: 64.6 Wh (80.4%) condition: 80.3/83.0 Wh (96.7%) volts: 12.2 min: 11.6 model: Hewlett-Packard PABAS0241231 serial: status: Discharging Device-1: hid-0018:04F3:2E0C.0001-battery model: ELAN2513:00 04F3:2E0C serial: N/A charge: N/A status: N/A CPU: Info: quad core model: 11th Gen Intel Core i7-11390H bits: 64 type: MT MCP arch: Tiger Lake rev: 2 cache: L1: 320 KiB L2: 5 MiB L3: 12 MiB Speed (MHz): avg: 3297 high: 3703 min/max: 400/4800:5000 cores: 1: 2841 2: 2798 3: 3703 4: 3569 5: 2899 6: 3364 7: 3606 8: 3602 bogomips: 54681 Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx Graphics: Device-1: Intel TigerLake-LP GT2 [Iris Xe Graphics] vendor: Hewlett-Packard driver: i915 v: kernel ports: active: eDP-1 empty: DP-1, DP-2, DP-3, DP-4, HDMI-A-1 bus-ID: 0000:00:02.0 chip-ID: 8086:9a49 Display: x11 server: X.Org v: 1.21.1.3 driver: X: loaded: modesetting unloaded: fbdev,vesa gpu: i915 display-ID: :0 screens: 1 Screen-1: 0 s-res: 3072x1920 s-dpi: 96 Monitor-1: eDP-1 model: AU Optronics res: 3072x1920 dpi: 227 diag: 406mm (16") OpenGL: renderer: Mesa Intel Xe Graphics (TGL GT2) v: 4.6 Mesa 22.2.5 direct render: Yes Audio: Device-1: Intel vendor: Hewlett-Packard driver: intel-ipu6 bus-ID: 0000:00:05.0 chip-ID: 8086:9a19 Device-2: Intel Tiger Lake-LP Smart Sound Audio vendor: Hewlett-Packard driver: sof-audio-pci-intel-tgl bus-ID: 0000:00:1f.3 chip-ID: 8086:a0c8 Sound Server-1: ALSA v: k5.15.0-67-generic running: yes Sound Server-2: PulseAudio v: 15.99.1 running: yes Sound Server-3: PipeWire v: 0.3.48 running: yes Network: Device-1: Intel Wi-Fi 6 AX201 driver: iwlwifi v: kernel bus-ID: 0000:00:14.3 chip-ID: 8086:a0f0 IF: wlo1 state: up mac: IF-ID-1: br-cbe742b0b3e5 state: down mac: IF-ID-2: docker0 state: down mac: Bluetooth: Device-1: Intel AX201 Bluetooth type: USB driver: btusb v: 0.8 bus-ID: 3-10:3 chip-ID: 8087:0026 Report: hciconfig ID: hci0 rfk-id: 0 state: up address: bt-v: 3.0 lmp-v: 5.2 sub-v: 237e RAID: Hardware-1: Intel Volume Management Device NVMe RAID Controller driver: vmd v: 0.6 bus-ID: 0000:00:0e.0 chip-ID: 8086:9a0b Drives: Local Storage: total: 476.94 GiB used: 39.83 GiB (8.4%) ID-1: /dev/nvme0n1 vendor: SK Hynix model: PC801 HFS512GEJ9X101N size: 476.94 GiB speed: 63.2 Gb/s lanes: 4 serial: temp: 30.9 C Partition: ID-1: / size: 283.7 GiB used: 39.74 GiB (14.0%) fs: ext4 dev: /dev/nvme0n1p5 ID-2: /boot/efi size: 256 MiB used: 94.5 MiB (36.9%) fs: vfat dev: /dev/nvme0n1p1 Swap: ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile USB: Hub-1: 1-0:1 info: Hi-speed hub with single TT ports: 1 rev: 2.0 speed: 480 Mb/s chip-ID: 1d6b:0002 Hub-2: 2-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003 Hub-3: 3-0:1 info: Hi-speed hub with single TT ports: 12 rev: 2.0 speed: 480 Mb/s chip-ID: 1d6b:0002 Device-1: 3-6:2 info: Synaptics type: driver: N/A rev: 2.0 speed: 12 Mb/s chip-ID: 06cb:00ff Device-2: 3-10:3 info: Intel AX201 Bluetooth type: Bluetooth driver: btusb rev: 2.0 speed: 12 Mb/s chip-ID: 8087:0026 Hub-4: 4-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003 Sensors: System Temperatures: cpu: 27.8 C mobo: 10.0 C Fan Speeds (RPM): N/A Repos: Packages: 2285 apt: 2266 flatpak: 19 No active apt repos in: /etc/apt/sources.list Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list 1: deb http: //packages.linuxmint.com vera main upstream import backport 2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse 3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse 4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse 5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse Info: Processes: 314 Uptime: 5m Memory: 15.38 GiB used: 2.77 GiB (18.0%) Init: systemd v: 249 runlevel: 5 Compilers: gcc: 11.3.0 alt: 11/12 Client: Unknown python3.10 client inxi: 3.3.13

This now works ! Install latest ipu6-drivers version and apply the fix from https://github.com/intel/ipu6-drivers/commit/3f813580116cc66099fb26e9a7ccf7e8f72424ec#commitcomment-131060757

Enriquepg013, 2026-01-30:
sorry i need this for windows 11, my camera doesn't work and is a HP Spectre x360 too. HELPP