intel/ipu6-drivers #106: Add drivers for ov7251 and ov5675

ctrlcctrlv, 2023-03-11:
Cf. https://github.com/intel/icamerasrc/issues/30

Required for Panasonic Toughbook FZ40.

striczkof, 2023-06-28:
LGTM

Just tested it and it installed smoothly on a clean Arch Linux installation

ctrlcctrlv, 2023-07-10:
Hello?

ctrlcctrlv, 2023-08-14:
Hello?

ctrlcctrlv, 2023-08-29:
你好？

ChocolateLoverRaj, 2023-09-22:
My camera also uses ov5675 and it does not work currently.

ctrlcctrlv, 2023-09-23:
**又过了一个月，由于英特尔公司的粗心大意和无底线的贪婪，我合法购买的网络摄像头仍然无法使用。**
_Another month has passed, and due to the carelessness and endless greed of Intel Corporation, the network camera that I bought fair and square still cannot be used._
Yòu guòle yīgè yuè, yóuyú yīngtè'ěr gōngsī de cūxīn dàyì hé wú dǐxiàn de tān lán, wǒ héfǎ gòumǎi de wǎngluò shèxiàngtóu réngrán wúfǎ shǐyòng.
ㄧㄡˋ ㄍㄨㄛˋ ㄌㄜ˙ ㄧㄧㄍㄜˋ ㄩㄝˋ，ㄧㄡˊ ㄩˊ ㄧㄥ ㄊㄜˋ ㄦˊ ㄍㄨㄥ ㄙ ㄉㄜ ㄘㄨ ㄒㄧㄣ ㄉㄚˋ ㄧˋ ㄏㄜˊ ㄨˊ ㄉㄧˇ ㄒㄧㄢˋ ㄉㄜ ㄊㄢ ㄍㄨㄢ，ㄨㄛˇ ㄏㄜˊ ㄈㄚˇ ㄍㄡˋ ㄇㄞˇ ㄉㄜ ㄨㄤˇ ㄌㄨㄛˋ ㄕㄜˋ ㄒㄧㄤˋ ㄊㄡˊ ㄖㄥˊ ㄖㄢˊ ㄨˊ ㄈㄚˇ ㄕˇ ㄧㄠˋ。

ChenYuYeh, 2023-09-23:
Hello,  Can you ask Panasonic to ask Intel to support your camera bug?

paulmenzel, 2023-09-23:
If you support vendors, that don’t support officially support GNU/Linux, it’s your own fault. From the [mailing list discussion](https://lore.kernel.org/all/YvUaEDMbZD70x+hD@kroah.com/):

> On Thu, Aug 11, 2022 at 04:54:53PM +0300, Laurent Pinchart wrote:
>> For the time being, I agree with your recommendation to not buy these
>> devices if you care about camera support.
> 
> I second this, don't buy these devices if the vendor is not willing to
> get their drivers upstreamed properly.

I recommend to return the device, and support vendors, that officially support GNU/Linux.

ctrlcctrlv, 2023-09-24:
> Hello, Can you ask Panasonic to ask Intel to support your camera bug?

Panasonic had no interest in my experience, I sent multiple emails in both English and Japanese. But if you'd like me to email another address you can write it here or send it to <FRED@MFEK.ORG>.

> it’s your own fault

I need a rugged laptop because I kept breaking my other laptops (I broke 2 from System 76 from clumsiness) but please do go on Paul about how my disability is my fault.

Voting with your wallet is GOP nonsense anyway. 

[image]