intel/ipu6-drivers #107: Suspend/hibernate issue on Thinkpad x1 carbon gen10

zemen, 2023-03-12:
I got the driver working(both gst-launch and v4l2-relayd) as per comment https://github.com/intel/ipu6-drivers/issues/48#issuecomment-1298596188 with arch linux on kernel 6.0.9.

Unfortunately laptop started to freeze when tried suspend/hibernate. dmesg/journalctl output is empty. After deleting drivers everything works normally.

I tried to compile newer kernels 6.1.7 and 6.2.2 and driver doesn't work(problems with mainline ov2740, I hope the fix is described here https://github.com/intel/ipu6-drivers/issues/55#issuecomment-1458574131 but haven't tried) but the suspend/hibernate issue persists even with camera not working. The same reproduces with mainline arch linux kernel without patches.

polair, 2023-06-16:
I don't have lockup on suspend but I *do* get a segfault of the ipu6-psys module on suspend/resume and have on every kernel that I've tried. I'm currently on 6.3.7 and the only way to restore operation is to reboot the system. Here is the kernel log+stack trace from the suspend request, it fails with "Modprobe Tainted":

```
[Thu Jun 15 18:09:45 2023] PM: suspend entry (deep)
[Thu Jun 15 18:09:49 2023] Filesystems sync: 3.175 seconds
[Thu Jun 15 18:09:49 2023] Freezing user space processes
[Thu Jun 15 18:09:49 2023] Freezing user space processes completed (elapsed 0.002 seconds)
[Thu Jun 15 18:09:49 2023] OOM killer disabled.
[Thu Jun 15 18:09:49 2023] Freezing remaining freezable tasks
[Thu Jun 15 18:09:49 2023] Freezing remaining freezable tasks completed (elapsed 0.001 seconds)
[Thu Jun 15 18:09:49 2023] printk: Suspending console(s) (use no_console_suspend to debug)
[Thu Jun 15 18:09:49 2023] ACPI: EC: interrupt blocked
[Thu Jun 15 18:09:49 2023] ACPI: PM: Preparing to enter system sleep state S3
[Thu Jun 15 18:09:49 2023] ACPI: EC: event blocked
[Thu Jun 15 18:09:49 2023] ACPI: EC: EC stopped
[Thu Jun 15 18:09:49 2023] ACPI: PM: Saving platform NVS memory
[Thu Jun 15 18:09:49 2023] Disabling non-boot CPUs ...
[Thu Jun 15 18:09:49 2023] smpboot: CPU 1 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 2 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 3 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 4 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 5 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 6 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 7 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 8 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 9 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 10 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 11 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 12 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 13 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 14 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 15 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 16 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 17 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 18 is now offline
[Thu Jun 15 18:09:49 2023] smpboot: CPU 19 is now offline
[Thu Jun 15 18:09:49 2023] ACPI: PM: Low-level resume complete
[Thu Jun 15 18:09:49 2023] ACPI: EC: EC started
[Thu Jun 15 18:09:49 2023] ACPI: PM: Restoring platform NVS memory
[Thu Jun 15 18:09:49 2023] Enabling non-boot CPUs ...
[Thu Jun 15 18:09:49 2023] x86: Booting SMP configuration:
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 1 APIC 0x1
[Thu Jun 15 18:09:49 2023] CPU1 is up
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 2 APIC 0x8
[Thu Jun 15 18:09:49 2023] CPU2 is up
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 3 APIC 0x9
[Thu Jun 15 18:09:49 2023] CPU3 is up
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 4 APIC 0x10
[Thu Jun 15 18:09:49 2023] CPU4 is up
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 5 APIC 0x11
[Thu Jun 15 18:09:49 2023] CPU5 is up
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 6 APIC 0x18
[Thu Jun 15 18:09:49 2023] CPU6 is up
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 7 APIC 0x19
[Thu Jun 15 18:09:49 2023] CPU7 is up
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 8 APIC 0x20
[Thu Jun 15 18:09:49 2023] CPU8 is up
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 9 APIC 0x21
[Thu Jun 15 18:09:49 2023] CPU9 is up
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 10 APIC 0x28
[Thu Jun 15 18:09:49 2023] CPU10 is up
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 11 APIC 0x29
[Thu Jun 15 18:09:49 2023] CPU11 is up
[Thu Jun 15 18:09:49 2023] smpboot: Booting Node 0 Processor 12 APIC 0x30
[Thu Jun 15 18:09:49 2023] core: cpu_atom PMU driver: PEBS-via-PT 
[Thu Jun 15 18:09:49 2023] ... version:                5
[Thu Jun 15 18:09:49 2023] ... bit width:              48
[Thu Jun 15 18:09:49 2023] ... generic registers:      6
[Thu Jun 15 18:09:49 2023] ... value mask:             0000ffffffffffff
[Thu Jun 15 18:09:49 2023] ... max period:             00007fffffffffff
[... 162 lines cut ...]
[Thu Jun 15 18:09:57 2023] R13: 000055a7a1c9b100 R14: 0000000000000000 R15: 000055a7a1c9c660
[Thu Jun 15 18:09:57 2023]  </TASK>
[Thu Jun 15 18:09:57 2023] Modules linked in: intel_ipu6_isys(OE+) intel_ipu6(OE) mei_csi(OE) mei_ace(OE) intel_vsc(OE) v4l2loopback(OE) ov2740(OE) rfcomm uhid cmac algif_hash algif_skcipher af_alg bnep snd_seq_dummy snd_hrtimer snd_seq snd_seq_device ccm wireguard curve25519_x86_64 libchacha20poly1305 chacha_x86_64 poly1305_x86_64 libcurve25519_generic libchacha ip6_udp_tunnel udp_tunnel bridge stp llc uinput btusb btrtl gpio_ljca(OE) spi_ljca(OE) i2c_ljca(OE) btbcm btintel btmtk bluetooth ecdh_generic ljca(OE) crc16 snd_ctl_led snd_soc_skl_hda_dsp snd_soc_intel_hda_dsp_common snd_soc_hdac_hdmi snd_sof_probes snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_soc_dmic snd_sof_pci_intel_tgl snd_sof_intel_hda_common soundwire_intel soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp intel_tcc_cooling snd_sof x86_pkg_temp_thermal intel_powerclamp snd_sof_utils coretemp snd_soc_hdac_hda snd_hda_ext_core snd_soc_acpi_intel_match kvm_intel snd_soc_acpi soundwire_bus snd_soc_core
[... cut ...]
[Thu Jun 15 18:09:57 2023] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Thu Jun 15 18:09:57 2023] CR2: 0000000000000000 CR3: 000000013e71e000 CR4: 0000000000f50ef0
[Thu Jun 15 18:09:57 2023] PKRU: 55555554
[Thu Jun 15 18:09:57 2023] note: modprobe[18312] exited with irqs disabled
[Thu Jun 15 18:09:57 2023] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[Thu Jun 15 18:09:57 2023] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
```
I am willing to help debug this if possible.

HuidaeCho, 2023-06-16:
[This issue](https://github.com/intel/ivsc-driver/issues/32#issuecomment-1589730406) might be related.

polair, 2023-06-16:
@HuidaeCho  I can confirm that they're likely related. I can re-create the suspend segfault behavior by unloading the i2c_ljca module, after having unloaded the relevant IPU modules. To be clear, the segfault does not occur and everything runs as expected if I remove the i2c_ljca lines. I run the following script to unload/reload the modules:

```
echo rmmod

sudo systemctl stop v4l2-relayd.service

sudo rmmod -f v4l2loopback
sudo rmmod -f intel_ipu6
sudo rmmod -f intel_ipu6_isys
sudo rmmod -f intel_ipu6_psys
sudo rmmod -f ov2740
sudo rmmod -f mei_vsc
sudo rmmod -f i2c_ljca

sleep 5

sudo modprobe i2c_ljca
sudo modprobe mei_vsc
sudo modprobe v4l2loopback video_nr=1 card_label='Camera' exclusive_caps=1 max_buffers=2
sudo modprobe ov2740
sudo modprobe intel_ipu6
sudo modprobe intel_ipu6_psys enable_suspend_resume=1 async_fw_init=1
sudo modprobe -f -v intel_ipu6_isys

```

Then in the dmesg log I have:

```
[Fri Jun 16 13:20:29 2023] intel-ipu6-isys intel-ipu6-isys0: removed
[Fri Jun 16 13:20:29 2023] intel-ipu6-isys intel-ipu6-isys0: unbind ov2740 17-0036
[Fri Jun 16 13:20:29 2023] intel-ipu6-psys intel-ipu6-psys0: removed
[Fri Jun 16 13:20:38 2023] v4l2loopback driver version 0.12.7 loaded
[Fri Jun 16 13:20:38 2023] intel-ipu6 0000:00:05.0: Device 0x465d (rev: 0x2)
[Fri Jun 16 13:20:38 2023] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[Fri Jun 16 13:20:38 2023] intel-ipu6 0000:00:05.0: mapped as: 0x0000000027ff5c58
[Fri Jun 16 13:20:38 2023] intel-ipu6 0000:00:05.0: IPU in secure mode
[Fri Jun 16 13:20:38 2023] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[Fri Jun 16 13:20:38 2023] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[Fri Jun 16 13:20:39 2023] intel-ipu6 0000:00:05.0: IPC reset done
[Fri Jun 16 13:20:39 2023] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[Fri Jun 16 13:20:39 2023] intel-ipu6 0000:00:05.0: FW version: 20220510
[Fri Jun 16 13:20:39 2023] intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[Fri Jun 16 13:20:39 2023] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[Fri Jun 16 13:20:39 2023] intel-ipu6-psys intel-ipu6-psys0: FW init done
[Fri Jun 16 13:20:39 2023] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[Fri Jun 16 13:20:39 2023] intel-ipu6-isys intel-ipu6-isys0: bind ov2740 17-0036 nlanes is 2 port is 1
[Fri Jun 16 13:20:39 2023] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
[Fri Jun 16 13:23:50 2023] intel-ipu6-isys intel-ipu6-isys0: removed
[Fri Jun 16 13:23:50 2023] intel-ipu6-isys intel-ipu6-isys0: unbind ov2740 17-0036
[Fri Jun 16 13:23:50 2023] intel-ipu6-psys intel-ipu6-psys0: removed
[Fri Jun 16 13:23:53 2023] ljca-i2c ljca-i2c.3.auto: hid INTC1097 uid 0 new uid1
[Fri Jun 16 13:23:53 2023] v4l2loopback driver version 0.12.7 loaded
[Fri Jun 16 13:23:53 2023] intel-ipu6 0000:00:05.0: Device 0x465d (rev: 0x2)
[Fri Jun 16 13:23:53 2023] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[Fri Jun 16 13:23:53 2023] intel-ipu6 0000:00:05.0: mapped as: 0x0000000052731752
[Fri Jun 16 13:23:53 2023] intel-ipu6 0000:00:05.0: IPU in secure mode
[Fri Jun 16 13:23:53 2023] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[Fri Jun 16 13:23:53 2023] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[Fri Jun 16 13:23:53 2023] intel-ipu6 0000:00:05.0: IPC reset done
[Fri Jun 16 13:23:53 2023] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[Fri Jun 16 13:23:53 2023] intel-ipu6 0000:00:05.0: FW version: 20220510
[Fri Jun 16 13:23:53 2023] intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[Fri Jun 16 13:23:53 2023] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[Fri Jun 16 13:23:53 2023] intel-ipu6-psys intel-ipu6-psys0: FW init done
[Fri Jun 16 13:23:53 2023] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[Fri Jun 16 13:23:53 2023] general protection fault, probably for non-canonical address 0x522e352a3d26d806: 0000 [#2] PREEMPT SMP NOPTI
[Fri Jun 16 13:23:53 2023] CPU: 0 PID: 22296 Comm: modprobe Tainted: PF R  UD    OE      6.3.7-1-ck #1 de36db804fbf5bd31fdc1f551f8a28a86103720b
[Fri Jun 16 13:23:53 2023] Hardware name: LENOVO 21E8xnano2/21E8xnano2, BIOS N3IET36W (1.16 ) 09/15/2022
[... 43 lines cut ...]
[Fri Jun 16 13:23:53 2023] RSP: 002b:00007ffe1afe2158 EFLAGS: 00000287 ORIG_RAX: 00000000000000af
[Fri Jun 16 13:23:53 2023] RAX: ffffffffffffffda RBX: 000055a930c3f470 RCX: 00007f80fc961f9e
[Fri Jun 16 13:23:53 2023] RDX: 000055a930a42cb2 RSI: 000000000006b06d RDI: 000055a930c43800
[Fri Jun 16 13:23:53 2023] RBP: 000055a930a42cb2 R08: 000055a930c43800 R09: 00007f80fc3edafe
[Fri Jun 16 13:23:53 2023] R10: 0000000000000000 R11: 0000000000000287 R12: 0000000000015aee
[Fri Jun 16 13:23:53 2023] R13: 000055a930c3f220 R14: 000055a930c3c470 R15: 000055a930c40780
[Fri Jun 16 13:23:53 2023]  </TASK>
[Fri Jun 16 13:23:53 2023] Modules linked in: intel_ipu6_isys(FOE+) intel_ipu6_psys(OE) intel_ipu6(OE) v4l2loopback(OE) mei_vsc(OE) ov2740(OE) i2c_ljca(OE) rfcomm uhid cmac algif_hash algif_skcipher af_alg bnep snd_seq_dummy snd_hrtimer snd_seq snd_seq_device ccm wireguard curve25519_x86_64 libchacha20poly1305 chacha_x86_64 poly1305_x86_64 libcurve25519_generic libchacha ip6_udp_tunnel udp_tunnel bridge stp llc uinput spi_ljca(OE) gpio_ljca(OE) btusb btrtl btbcm btintel btmtk bluetooth ecdh_generic ljca(OE) crc16 snd_ctl_led snd_soc_skl_hda_dsp snd_soc_intel_hda_dsp_common snd_sof_probes snd_soc_hdac_hdmi snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_soc_dmic snd_ [... cut ...]
[Fri Jun 16 13:23:53 2023] R10: 0000000000000001 R11: 0000000000000000 R12: ffff9cdf6cb88028
[Fri Jun 16 13:23:53 2023] R13: ffff9cdf6cb88344 R14: ffff9cde9ebae400 R15: 00000000000003e7
[Fri Jun 16 13:23:53 2023] FS:  00007f80fc851740(0000) GS:ffff9ce5bf200000(0000) knlGS:0000000000000000
[Fri Jun 16 13:23:53 2023] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Fri Jun 16 13:23:53 2023] CR2: 000055a930cad000 CR3: 000000013a546000 CR4: 0000000000f50ef0
[Fri Jun 16 13:23:53 2023] PKRU: 55555554
```

Zatvobor, 2024-05-16:
https://github.com/intel/ivsc-driver/issues/32#issuecomment-2115275991