intel/ipu6-drivers #124: driver failed to load because of unsigned firmware

ignaciohg, 2023-03-30:
I am getting

 [   23.390968] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[   23.391133] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[   23.391205] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[   23.391206] intel-ipu6 0000:00:05.0: mapped as: 0x0000000028e0d23b
[   23.391351] intel-ipu6 0000:00:05.0: IPU in secure mode
[   23.391354] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[   23.391356] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[   23.406513] intel-ipu6 0000:00:05.0: IPC reset done
[   23.406520] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[   23.406568] intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6_fw.bin failed with error -2
[   23.406571] intel-ipu6 0000:00:05.0: Requesting signed firmware failed
[   23.406573] intel-ipu6: probe of 0000:00:05.0 failed with error -2

When booting up my system, where are you supposed to get the firmware?

naps62, 2023-06-18:
Having the same error on a fresh Manjaro install

naps62, 2023-06-18:
Fixed  by what's following #38

dmipx, 2024-02-06:
I've seen such issue when we tried to enable IPU6 on RT kernels. Verify that you have CONFIG_PM enabled.