intel/ipu6-drivers #133: Ubuntu 23.04

shawn1thompson, 2023-04-22:
Are these being updated for use with Ubuntu 23.04?

PhilDevProg, 2023-05-28:
Does this comment fix it: https://github.com/intel/ipu6-drivers/commit/60004440e1a6a18a6698b1d557dcdfa67beba4d7?

almereyda, 2023-06-12:
With some hiccups around [the proper tag to use for the `icamerasrc`](https://github.com/intel/icamerasrc/issues/32#issuecomment-1586383112), I was able to build the stack [with some workaround](https://github.com/intel/icamerasrc/issues/28#issuecomment-1586399863) in a way that [the `ximagesink` shows the camera feed](https://github.com/intel/ipu6-drivers/issues/55#issuecomment-1586471764).

It appears some v4l2 loopback and relayd work is missing to get this to full fruition.