intel/ipu6-drivers #136: int3472 patch does not apply on v6.3-rcX

vicamo, 2023-05-03:
[Patch](https://github.com/intel/ipu6-drivers/blob/master/patch/int3472-support-independent-clock-and-LED-gpios-5.17%2B.patch) for int3472 conflicts with upstreamed implementation of privacy led. Is there an update to it already if necessary?

jwrdegoede, 2023-05-03:
@vicamo starting with 6.3 there no longer is any need to patch the "main" kernel. The ipu6-drivers can now work with a completely unmodified distro kernel build. See  #84  for the necessary changes.

Also see #137 for fixing ipu6-drivers to build against the 6.3 kernel.

With those 2 pull-reqs added to the latest master branch things should work with an unpatched 6.3 kernel.

vicamo, 2023-05-04:
> @vicamo starting with 6.3 there no longer is any need to patch the "main" kernel. The ipu6-drivers can now work with a completely unmodified distro kernel build. See #84 for the necessary changes.

Cool.

> Also see #137 for fixing ipu6-drivers to build against the 6.3 kernel.
> 
> With those 2 pull-reqs added to the latest master branch things should work with an unpatched 6.3 kernel.

Thank you. I've packaged 6.3 kernel with #137 in [ppa:canonical-hwe-team/experimental-unstable](https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/experimental-unstable). To be verified later.

closed 2023-05-04 by vicamo