intel/ipu6-drivers #153: Device recognized, no video

shawn1thompson, 2023-06-22:
Ubuntu 22.04
Installed via [this](https://www.dell.com/support/kbdoc/en-si/000203830/webcam-is-not-detected-on-xps-13-plus-9320-laptops-running-ubuntu-22-04)
Device is recognized on webcam test, but there is no video

githubtefo, 2023-06-24:
In case useful, I just found out that on Ubuntu 22.04.1 LTS updating from kernel linux-image-5.19.0-43-generic to linux-image-5.19.0-45-generic disrupts webcam recognition when oem-somerville-tentacool-meta is install. Fallback to previous kernel solved the issue, although I am getting tired of this faulty hardware (Dell's proposed solution was sending me a new motherboard and webcam).

shawn1thompson, 2023-07-06:
Is there another place I should be looking for support?

shawn1thompson, 2023-07-20:
[shawn1thompson](https://github.com/shawn1thompson) opened this issue last month · 2 comments (both from me)...sad

closed 2023-07-20 by shawn1thompson

githubtefo, 2023-07-24:
> In case useful, I just found out that on Ubuntu 22.04.1 LTS updating from kernel linux-image-5.19.0-43-generic to linux-image-5.19.0-45-generic disrupts webcam recognition when [...]

This was not accurate - in my case was an issue with the Secure Boot - Following the guidelines to "Enroll MOK" after upgrade solved it.