intel/ipu6-drivers #167: Fedora 38 on X1 carbon gen 10

angermcy, 2023-08-14:
Hi, 
i have a Lenovo X1 carbon gen 10 and installed fedora workstation 38 from scratch.
After installing the camera drivers as [shown here](https://hansdegoede.dreamwidth.org/27235.html) the camera is recognized but shows a black screen. 
The camera seems to work, when I use a smartphone torch and light directly into the camera. I can see a white spot. So it seems the camera is working but way too dark.

Fedora version is 6.4.10-200.fc38.x86_64.

v4l2-compliance shows up the following:
```
v4l2-compliance 1.22.1, 64 bits, 64-bit time_t

Compliance test for v4l2 loopback device /dev/video0:

Driver Info:
	Driver name      : v4l2 loopback
	Card type        : Intel MIPI Camera
	Bus info         : platform:v4l2loopback-000
	Driver version   : 6.4.10
	Capabilities     : 0x85200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x05200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format

Required ioctls:
	test VIDIOC_QUERYCAP: OK
	test invalid ioctls: OK

Allow for multiple opens:
	test second /dev/video0 open: OK
	test VIDIOC_QUERYCAP: OK
	test VIDIOC_G/S_PRIORITY: OK
		fail: v4l2-compliance.cpp(736): !ok
	test for unlimited opens: FAIL

Debug ioctls:
	test VIDIOC_DBG_G/S_REGISTER: OK (Not Supported)
	test VIDIOC_LOG_STATUS: OK (Not Supported)

Input ioctls:
	test VIDIOC_G/S_TUNER/ENUM_FREQ_BANDS: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_S_HW_FREQ_SEEK: OK (Not Supported)
	test VIDIOC_ENUMAUDIO: OK (Not Supported)
	test VIDIOC_G/S/ENUMINPUT: OK
	test VIDIOC_G/S_AUDIO: OK (Not Supported)
	Inputs: 1 Audio Inputs: 0 Tuners: 0

Output ioctls:
	test VIDIOC_G/S_MODULATOR: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_ENUMAUDOUT: OK (Not Supported)
	test VIDIOC_G/S/ENUMOUTPUT: OK (Not Supported)
	test VIDIOC_G/S_AUDOUT: OK (Not Supported)
	Outputs: 0 Audio Outputs: 0 Modulators: 0

Input/Output configuration ioctls:
	test VIDIOC_ENUM/G/S/QUERY_STD: OK (Not Supported)
	test VIDIOC_ENUM/G/S/QUERY_DV_TIMINGS: OK (Not Supported)
	test VIDIOC_DV_TIMINGS_CAP: OK (Not Supported)
	test VIDIOC_G/S_EDID: OK (Not Supported)

Control ioctls (Input 0):
		fail: v4l2-test-controls.cpp(101): non-zero min/max/step/def
		fail: v4l2-test-controls.cpp(201): invalid control 0098f903
	test VIDIOC_QUERY_EXT_CTRL/QUERYMENU: FAIL
		fail: v4l2-test-controls.cpp(351): node->controls.find(qctrl.id) == node->controls.end()
	test VIDIOC_QUERYCTRL: FAIL
	test VIDIOC_G/S_CTRL: OK
	test VIDIOC_G/S/TRY_EXT_CTRLS: OK
	test VIDIOC_(UN)SUBSCRIBE_EVENT/DQEVENT: OK
	test VIDIOC_G/S_JPEGCOMP: OK (Not Supported)
	Standard Controls: 1 Private Controls: 3

Format ioctls (Input 0):
		fail: v4l2-test-formats.cpp(327): Video Output cap not set, but Video Output formats defined
	test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: FAIL
		fail: v4l2-test-formats.cpp(1290): node->has_frmintervals && !cap->capability
	test VIDIOC_G/S_PARM: FAIL
	test VIDIOC_G_FBUF: OK (Not Supported)
		fail: v4l2-test-formats.cpp(623): Video Output cap not set, but Video Output formats defined
	test VIDIOC_G_FMT: FAIL
	test VIDIOC_TRY_FMT: OK
		fail: v4l2-test-formats.cpp(452): expected EINVAL, but got 16 when getting format for buftype 2
	test VIDIOC_S_FMT: FAIL
	test VIDIOC_G_SLICED_VBI_CAP: OK (Not Supported)
	test Cropping: OK (Not Supported)
	test Composing: OK (Not Supported)
	test Scaling: OK (Not Supported)

Codec ioctls (Input 0):
	test VIDIOC_(TRY_)ENCODER_CMD: OK (Not Supported)
	test VIDIOC_G_ENC_INDEX: OK (Not Supported)
	test VIDIOC_(TRY_)DECODER_CMD: OK (Not Supported)

Buffer ioctls (Input 0):
		fail: v4l2-test-buffers.cpp(456): g_timestamp().tv_sec || g_timestamp().tv_usec
		fail: v4l2-test-buffers.cpp(483): buf.check(Unqueued, i)
		fail: v4l2-test-buffers.cpp(612): testQueryBuf(node, i, q.g_buffers())
	test VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: FAIL
	test VIDIOC_EXPBUF: OK (Not Supported)
	test Requests: OK (Not Supported)

Total for v4l2 loopback device /dev/video0: 45, Succeeded: 37, Failed: 8, Warnings: 0
```

loicfavory, 2023-08-16:
Same issue with Fedora 38, same kernel & Dell XPS 13 Plus (9320)

drodre, 2023-08-17:
I have also the same kernel 6.4.10-200.fc38.x86_64 and same issue after an upgrade.

jwrdegoede, 2023-08-21:
Thank you all for reporting this and sorry for this regression in functionality.

The problem is the ipu6-camera-bins vs ipu6-camera-hal package versions have gotten out of sync.

You can fix this by running the following command:

```
sudo dnf update --enablerepo=rpmfusion-nonfree-updates-testing 'ipu6-camera-*'
```

For slightly more details of the problem see: https://hansdegoede.dreamwidth.org/27603.html

@angermcy, can you please close this issue? (assuming the fix works for you)

closed 2023-08-22 by angermcy

gutopardini, 2023-10-11:
I'm having this issue in Fedora 39 beta

`compliance test for v4l2 loopback device /dev/video0:

Driver Info:
	Driver name      : v4l2 loopback
	Card type        : Intel MIPI Camera
	Bus info         : platform:v4l2loopback-000
	Driver version   : 6.5.6
	Capabilities     : 0x85200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x05200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format

Required ioctls:
	test VIDIOC_QUERYCAP: OK
	test invalid ioctls: OK

Allow for multiple opens:
	test second /dev/video0 open: OK
	test VIDIOC_QUERYCAP: OK
	test VIDIOC_G/S_PRIORITY: OK
		fail: ../utils/v4l2-compliance/v4l2-compliance.cpp(753): !ok
	test for unlimited opens: FAIL

Debug ioctls:
	test VIDIOC_DBG_G/S_REGISTER: OK (Not Supported)
	test VIDIOC_LOG_STATUS: OK (Not Supported)

Input ioctls:
	test VIDIOC_G/S_TUNER/ENUM_FREQ_BANDS: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_S_HW_FREQ_SEEK: OK (Not Supported)
	test VIDIOC_ENUMAUDIO: OK (Not Supported)
	test VIDIOC_G/S/ENUMINPUT: OK
	test VIDIOC_G/S_AUDIO: OK (Not Supported)
	Inputs: 1 Audio Inputs: 0 Tuners: 0

Output ioctls:
	test VIDIOC_G/S_MODULATOR: OK (Not Supported)
	test VIDIOC_G/S_FREQUENCY: OK (Not Supported)
	test VIDIOC_ENUMAUDOUT: OK (Not Supported)
	test VIDIOC_G/S/ENUMOUTPUT: OK (Not Supported)
	test VIDIOC_G/S_AUDOUT: OK (Not Supported)
	Outputs: 0 Audio Outputs: 0 Modulators: 0

Input/Output configuration ioctls:
	test VIDIOC_ENUM/G/S/QUERY_STD: OK (Not Supported)
	test VIDIOC_ENUM/G/S/QUERY_DV_TIMINGS: OK (Not Supported)
	test VIDIOC_DV_TIMINGS_CAP: OK (Not Supported)
	test VIDIOC_G/S_EDID: OK (Not Supported)

Control ioctls (Input 0):
		fail: ../utils/v4l2-compliance/v4l2-test-controls.cpp(101): non-zero min/max/step/def
		fail: ../utils/v4l2-compliance/v4l2-test-controls.cpp(201): invalid control 0098f903
	test VIDIOC_QUERY_EXT_CTRL/QUERYMENU: FAIL
		fail: ../utils/v4l2-compliance/v4l2-test-controls.cpp(352): node->controls.find(qctrl.id) == node->controls.end()
	test VIDIOC_QUERYCTRL: FAIL
	test VIDIOC_G/S_CTRL: OK
	test VIDIOC_G/S/TRY_EXT_CTRLS: OK
	test VIDIOC_(UN)SUBSCRIBE_EVENT/DQEVENT: OK
	test VIDIOC_G/S_JPEGCOMP: OK (Not Supported)
	Standard Controls: 1 Private Controls: 3

Format ioctls (Input 0):
		fail: ../utils/v4l2-compliance/v4l2-test-formats.cpp(337): Video Output cap not set, but Video Output formats defined
	test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: FAIL
		fail: ../utils/v4l2-compliance/v4l2-test-formats.cpp(1300): node->has_frmintervals && !cap->capability
	test VIDIOC_G/S_PARM: FAIL
	test VIDIOC_G_FBUF: OK (Not Supported)
		fail: ../utils/v4l2-compliance/v4l2-test-formats.cpp(633): Video Output cap not set, but Video Output formats defined
	test VIDIOC_G_FMT: FAIL
	test VIDIOC_TRY_FMT: OK
		fail: ../utils/v4l2-compliance/v4l2-test-formats.cpp(462): expected EINVAL, but got 16 when getting format for buftype 1
	test VIDIOC_S_FMT: FAIL
	test VIDIOC_G_SLICED_VBI_CAP: OK (Not Supported)
	test Cropping: OK (Not Supported)
	test Composing: OK (Not Supported)
		fail: ../utils/v4l2-compliance/v4l2-test-formats.cpp(1824): doioctl(node, VIDIOC_S_FMT, &fmt)
		fail: ../utils/v4l2-compliance/v4l2-test-formats.cpp(1972): testBasicScaling(node, fmt)
	test Scaling: FAIL

Codec ioctls (Input 0):
	test VIDIOC_(TRY_)ENCODER_CMD: OK (Not Supported)
	test VIDIOC_G_ENC_INDEX: OK (Not Supported)
	test VIDIOC_(TRY_)DECODER_CMD: OK (Not Supported)

Buffer ioctls (Input 0):
		fail: ../utils/v4l2-compliance/v4l2-test-buffers.cpp(456): g_timestamp().tv_sec || g_timestamp().tv_usec
		fail: ../utils/v4l2-compliance/v4l2-test-buffers.cpp(483): buf.check(Unqueued, i)
		fail: ../utils/v4l2-compliance/v4l2-test-buffers.cpp(612): testQueryBuf(node, i, q.g_buffers())
	test VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: FAIL
	test VIDIOC_EXPBUF: OK (Not Supported)
	test Requests: OK (Not Supported)

Total for v4l2 loopback device /dev/video0: 45, Succeeded: 36, Failed: 9, Warnings: 0`