intel/ipu6-drivers #170: IPU6 driver does not work on XPS 9315

gaul, 2023-08-22:
I followed the instructions at https://hansdegoede.livejournal.com/27276.html but the camera still does not work on my XPS 9315.  @jwrdegoede

jwrdegoede, 2023-08-22:
Ok, lets see if we can get to the bottom of this.

1. Please reboot the laptop and then immediately after the reboot run:

```
dmesg > dmesg.txt
lsmod > lsmod.txt
rpm -qa | grep ipu6 > ipu6-rpms.txt
ls -l /sys/bus/i2c/devices > i2c-devices.txt
ls -l /dev/video* > video-devices.txt
```

And then attach the generated `dmesg.txt`,  `lsmod.txt`, `ipu6-rpms.txt`,  `i2c-devices.txt`  and `video-devices.txt` files here.

2. After collecting the files please try using the camera stack directly from gstreamer (instead of using the v4l2 emulation through v4l2loopback):

```
sudo systemctl stop v4l2-relayd.service
gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! xvimagesink
```

After this please:

2.1 copy and paste the output of the `gst-launch-1.0` command in a comment here
2.2 run `dmesg > dmesg2.txt` and attach the generated `dmesg2.txt` file here
2.3 Add a note here if the camera worked with the  `gst-launch-1.0` command  (or not) ?

3. Did you in the past maybe try to install the ipu6 drivers manually ?

gaul, 2023-08-22:
I attached the requested output:

[dmesg.txt](https://github.com/intel/ipu6-drivers/files/12413491/dmesg.txt)
[dmesg2.txt](https://github.com/intel/ipu6-drivers/files/12413492/dmesg2.txt)
[i2c-devices.txt](https://github.com/intel/ipu6-drivers/files/12413493/i2c-devices.txt)
[ipu6-rpms.txt](https://github.com/intel/ipu6-drivers/files/12413494/ipu6-rpms.txt)
[lsmod.txt](https://github.com/intel/ipu6-drivers/files/12413495/lsmod.txt)
[video-devices.txt](https://github.com/intel/ipu6-drivers/files/12413496/video-devices.txt)

Critically there do not appear to be any video devices:

```
ls: cannot access '/dev/video*': No such file or directory
```

The gstlaunch command also fails due to this:

```
$ gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! xvimagesink
Setting pipeline to PAUSED ...
[08-23 07:45:42.92] CamHAL[INF] aiqb file name ov13b10.aiqb
[08-23 07:45:42.92] CamHAL[ERR] there is no aiqb file:ov13b10
[08-23 07:45:42.92] CamHAL[INF] aiqb file name ov13b10.aiqb
[08-23 07:45:42.92] CamHAL[ERR] there is no aiqb file:ov13b10
[08-23 07:45:42.92] CamHAL[INF] aiqb file name ov8856.aiqb
[08-23 07:45:42.92] CamHAL[ERR] there is no aiqb file:ov8856
[08-23 07:45:42.92] CamHAL[INF] aiqb file name ov8856.aiqb
[08-23 07:45:42.92] CamHAL[ERR] there is no aiqb file:ov8856
[08-23 07:45:42.92] CamHAL[INF] aiqb file name ov01a10.aiqb
[08-23 07:45:42.101] CamHAL[INF] aiqb file name ov01a10.aiqb
[08-23 07:45:42.101] CamHAL[INF] aiqb file name ov01a10.aiqb
[08-23 07:45:42.102] CamHAL[INF] aiqb file name ov01a10.aiqb
[08-23 07:45:42.102] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[08-23 07:45:42.116] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[08-23 07:45:42.116] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[08-23 07:45:42.124] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[08-23 07:45:42.124] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[08-23 07:45:42.131] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[08-23 07:45:42.132] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[08-23 07:45:42.140] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[08-23 07:45:42.140] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[08-23 07:45:42.148] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[08-23 07:45:42.148] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[08-23 07:45:42.156] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[08-23 07:45:42.156] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[08-23 07:45:42.165] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[08-23 07:45:42.165] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[08-23 07:45:42.172] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[08-23 07:45:42.172] CamHAL[INF] aiqb file name ov01a1s.aiqb
[08-23 07:45:42.180] CamHAL[INF] aiqb file name ov01a1s.aiqb
[08-23 07:45:42.180] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
[08-23 07:45:42.188] CamHAL[ERR] MediaControl init failed
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
```

jwrdegoede, 2023-09-02:
@gaul, sorry for being a bit slow with getting back to you.

The issue seems to be that you are using secureboot and you have not generated + added a local certificate for signing locally build kernel modules like the IPU6 kernel modules:

```
[    5.750066] Loading of module with unavailable key is rejected
[    5.795896] Loading of module with unavailable key is rejected
etc.
```

There are ways to fix this:

1. Generate a local certificate and at this as a valid key to shim, see: https://rpmfusion.org/Howto/Secure%20Boot and then after this wait for the next Fedora kernel update, which will build modules which will now be signed. You can do: `dnf update 'kernel*' after following the instructions and with some luck a kernel update is already waiting.
2. Disable secureboot in your BIOS options.

Please give this a try and let me know if it helps.

gaul, 2023-09-03:
#2 works!  Thank you for your help!

closed 2023-09-03 by gaul