intel/ipu6-drivers #171: Support for Raptor Lake ( and OV02C10? ) Sensor module not seen by ipu6 modules.

sidepipe, 2023-08-23:
Firstly, I don't actually know if I'm trying something that isn't yet supported/working, so apologies if this is a little premature. However, I've been trying to make the camera work on a Raptor Lake platform ( it's actually a Samsung Galaxy Book 3 Pro 360. ) I'm also using Ubuntu, so there's that... though this doesn't appear to me to be OS specific, but does require a manual install ( unless there are packages somewhere that I'm unaware of. )

Essentially, all of the various modules appear to load correctly, but my guess is that the intel_ipu6* modules don't "see" the sensor module - this is based on the observation that the ov02c10 modules is showing as un-used and can be freely unloaded, and the fact that there are no corresponding /dev/video* ( or media* ) nodes created. The result of `lsmod | grep -E "ov|ipu6"` is:

```
intel_ipu6_isys       159744  0
videobuf2_dma_contig    24576  1 intel_ipu6_isys
videobuf2_v4l2         36864  1 intel_ipu6_isys
videobuf2_common       86016  4 videobuf2_dma_contig,videobuf2_v4l2,intel_ipu6_isys,videobuf2_memops
intel_ipu6_psys       110592  0
intel_ipu6            118784  2 intel_ipu6_isys,intel_ipu6_psys
ov02c10                28672  0
intel_vsc              16384  3 mei_ace,mei_csi,ov02c10
v4l2_fwnode            40960  2 ov02c10,intel_ipu6_isys
v4l2_async             28672  3 v4l2_fwnode,ov02c10,intel_ipu6_isys
videodev              319488  5 v4l2_async,videobuf2_v4l2,ov02c10,v4l2loopback,intel_ipu6_isys
mc                     81920  7 v4l2_async,videodev,snd_usb_audio,videobuf2_v4l2,ov02c10,intel_ipu6_isys,videobuf2_common
```

The boot log has:

```
[    7.554718] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    7.554921] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    7.554939] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    7.554940] intel-ipu6 0000:00:05.0: mapped as: 0x00000000d9f74103
[    7.554983] intel-ipu6 0000:00:05.0: Unable to set secure mode
[    7.554984] intel-ipu6 0000:00:05.0: IPU in non-secure mode
[    7.554985] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[    7.554987] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    7.555559] intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
[    7.555561] intel-ipu6 0000:00:05.0: IPC reset done
[    7.555561] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    7.557131] intel-ipu6 0000:00:05.0: FW version: 20220510
```

The lines after "enabling device" are then repeated a further 25 times, with only the "mapped as" address differing in some of them.

I won't include any more info yet because as I say, I'm not sure if this is even something that _should_ work yet, but hopefully it's enough info to see if it's worth looking at, or if there might be something obvious that I missed during the install process ( I compiled the modules using dkms, by the way. ) If it helps, I know my way around C and various bits of the kernel, but obviously know nothing about these particular drivers, but if there's anything specific I can try then let me know.

bastien8060, 2023-10-13:
I had same issue. You should have an int3472 error. I blacklisted the related kernel module. Now it does say it connected to 1 camera, but the following error ensues:

```
bastien@Unkown:~$ sudo gst-launch-1.0 icamerasrc ! autovideosink
[sudo] password for bastien: 
Setting pipeline to PAUSED ...
[10-13 01:40:28.128] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-13 01:40:28.128] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-13 01:40:28.129] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-13 01:40:28.129] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-13 01:40:28.129] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-13 01:40:28.130] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-13 01:40:28.130] CamHAL[INF] aiqb file name ov8856.aiqb
[10-13 01:40:28.130] CamHAL[INF] aiqb file name ov8856.aiqb
[10-13 01:40:28.130] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.130] CamHAL[INF] aiqb file name ov8856.aiqb
[10-13 01:40:28.130] CamHAL[INF] aiqb file name ov8856.aiqb
[10-13 01:40:28.130] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.130] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-13 01:40:28.131] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-13 01:40:28.131] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-13 01:40:28.131] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-13 01:40:28.131] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-13 01:40:28.132] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-13 01:40:28.132] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-13 01:40:28.132] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-13 01:40:28.133] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-13 01:40:28.133] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-13 01:40:28.133] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-13 01:40:28.133] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.133] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-13 01:40:28.133] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-13 01:40:28.133] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.133] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-13 01:40:28.133] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-13 01:40:28.134] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-13 01:40:28.134] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-13 01:40:28.134] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-13 01:40:28.135] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.135] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-13 01:40:28.135] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-13 01:40:28.135] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.135] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[10-13 01:40:28.135] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[10-13 01:40:28.135] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.135] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-13 01:40:28.135] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-13 01:40:28.136] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.136] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-13 01:40:28.136] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-13 01:40:28.136] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.136] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-13 01:40:28.136] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-13 01:40:28.136] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.136] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-13 01:40:28.136] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-13 01:40:28.137] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.137] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[10-13 01:40:28.137] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[10-13 01:40:28.137] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.137] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[10-13 01:40:28.137] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[10-13 01:40:28.137] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.137] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-13 01:40:28.137] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-13 01:40:28.137] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.137] CamHAL[INF] aiqb file name OV08A10_YHUT_ADL.aiqb
[10-13 01:40:28.138] CamHAL[INF] aiqb file name OV08A10_YHUT_ADL.aiqb
[10-13 01:40:28.138] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.138] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[... 61 lines cut ...]
[10-13 01:40:28.143] CamHAL[ERR] invalid media format, default value used.
[10-13 01:40:28.143] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-13 01:40:28.143] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-13 01:40:28.143] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-13 01:40:28.143] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[... cut ...]
[10-13 01:40:28.156] CamHAL[ERR] set up mediaCtl failed
[10-13 01:40:28.156] CamHAL[ERR] @configure Device Configure failed
[10-13 01:40:28.156] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.007769088
Setting pipeline to NULL ...
Freeing pipeline ...
```

hao-yao, 2023-10-30:
intel-ipu-isys loading need all sensors ready for ACPI enumeration. As for INT3472, it is essential for camera power controlling after Tiger Lake platform.
Not sure the hardware configuration of your device, but I guess that your device used INT3472 for power resource instead of ivsc.

Currently I only added ivsc-related power control code in ov02c10.c. But recently I changed hi556.c to support both ivsc and GPIO based power control (need INT3472 driver). If you are interested in it you can try to do similar changes and enable INT3472 then see if ov02c10 can be bind as IPU6 subdev. It should work on 6.2+ kernel.
https://github.com/hao-yao/ipu6-sensor-guide

sidepipe, 2023-11-15:
I've tried to make similar changes to the ov02c10 module ( with the caveat that I don't really know too much about any of it!! ) but am still getting exactly the same error that Bastien8060 reported above when using gst-launch-1.0. Currently I'm inserting the modules manually.

When inserting intel_skl_int3472_discrete ( which seems to be the necessary one ) I get:

```
[   67.109585] int3472-discrete INT3472:01: power-enable \_SB.GPI0 pin number mismatch _DSM 227 resource 131
[   67.110136] int3472-discrete INT3472:01: privacy-led \_SB.GPI0 pin number mismatch _DSM 101 resource 357
[   67.110639] int3472-discrete INT3472:01: clk-enable \_SB.GPI0 pin number mismatch _DSM 246 resource 150
```

On inserting intel-ipu6, I get:

```
[   67.122114] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[   67.122363] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[   67.122468] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[   67.122471] intel-ipu6 0000:00:05.0: mapped as: 0x00000000668f70c9
[   67.122535] intel-ipu6 0000:00:05.0: Unable to set secure mode
[   67.122537] intel-ipu6 0000:00:05.0: IPU in non-secure mode
[   67.122538] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[   67.122541] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[   67.122547] intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
[   67.122549] intel-ipu6 0000:00:05.0: IPC reset done
[   67.122550] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[   67.126118] intel-ipu6 0000:00:05.0: FW version: 20230102
[   67.127348] intel-ipu6 0000:00:05.0: Found supported sensor OVTI02C1:00
[   67.127577] intel-ipu6 0000:00:05.0: Connected 1 cameras
[   67.128992] intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
```

This is without the ov02c10 module loaded - but note that it seems to find the sensor anyway. When I insert ov02c10 and intel-ipu6-isys, I get :

```
[  464.694770] intel-ipu6-isys intel-ipu6-isys0: bind ov02c10 15-0036 nlanes is 2 port is 0
[  464.695135] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
```

So *something* is happening. Any ideas?

ItzDerock, 2023-11-16:
I am also having issues with ipu6. I have a raptor lake chip and the OV02C10 sensor (also from the galaxy book lineup -- different model though)

I was successfully able to install and load ipu6 on `6.6.1-arch1.1`.
```
[    5.214279] intel-ipu6 0000:00:05.0: FW version: 20230102
[    5.214810] intel-ipu6 0000:00:05.0: Found supported sensor OVTI02C1:00
[    5.214923] intel-ipu6 0000:00:05.0: Connected 1 cameras
[    5.215622] intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[    5.215729] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[    5.215810] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
```

but when I try to use the webcam,
```
[11-15 21:26:44.389] CamHAL[ERR] invalid media format, default value used.
[11-15 21:26:44.389] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[11-15 21:26:44.389] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[11-15 21:26:44.389] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[11-15 21:26:44.389] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[11-15 21:26:44.390] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[11-15 21:26:44.395] CamHAL[ERR] Get entity fail for calling getEntityById
[11-15 21:26:44.395] CamHAL[ERR] Get entity fail for calling getEntityById
[11-15 21:26:44.395] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[11-15 21:26:44.395] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[11-15 21:26:44.395] CamHAL[ERR] set up mediaCtl failed
[11-15 21:26:44.395] CamHAL[ERR] @configure Device Configure failed
[11-15 21:26:44.395] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3153): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.003198734
Setting pipeline to NULL ...
Freeing pipeline ...
```

The LED light on my webcam never turns on either. Let me know if there's any way I can help out with this.

sidepipe, 2023-11-16:
One thing I forgot to mention was that initially when I made the changes there was an error "external clock 26000000 is not supported" so I had to change OV02C10_MCLK from 19200000 to 26000000. However, clearly the module doesn't expect the value to be 26M and the difference is significant enough to mean that some of the other constants might be off for this config ( or that the value is read incorrectly. ) As I mentioned, I really don't know enough about this stuff nor do I have access to the appropriate design docs, so I'm poking about in the dark really. Just hoping that this might trigger something for someone else that might be able to give some pointers.

hao-yao, 2023-11-20:
@sidepipe The register settings document of ov02c10 on my hand is rather limited. I calculated the link frequency according to the document but can't figure out the correct answer (19.2M -> 400M). I think we need help from OmniVision.

sidepipe, 2023-11-20:
> That's nice! Would you mind submit a PR to master branch?

@hao-yao The problem there is that as I mentioned, I don't really know _anything_ about this stuff and don't really have the time to look at the docs, so in reality there could be something glaringly stupid in what I've done! If we can get to the point where it works, then I'll certainly submit a PR :smile: 

I think we ARE getting closer... after adding ov02c10-uf-0 to the profile, gstreamer starts, and the activity light comes on for the camera. Capturing still isn't working though, with gstreamer reporting:

```
Setting` pipeline to PAUSED ...
[11-20 09:35:28.607] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[11-20 09:35:28.607] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[11-20 09:35:28.607] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[11-20 09:35:28.607] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[11-20 09:35:28.607] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[11-20 09:35:28.607] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[11-20 09:35:28.648] CamHAL[ERR] Failed to calc fragment desc.
[11-20 09:35:28.648] CamHAL[ERR] prepare, prepare p2p fail

(gst-launch-1.0:17988): GStreamer-CRITICAL **: 09:35:28.653: Registering meta implementation 'GstCamerasrcMeta' without init function
[11-20 09:35:28.747] CamHAL[INF] threadLoop: mCameraStreams[0] == 0x7fb2e00038a0
[11-20 09:35:28.747] CamHAL[INF] threadLoop: getPrivacyBuffer returned nullptr
[11-20 09:35:28.814] CamHAL[INF] threadLoop: mCameraStreams[0] == 0x7fb2e00038a0
[11-20 09:35:28.814] CamHAL[INF] threadLoop: getPrivacyBuffer returned nullptr
[11-20 09:35:28.823] CamHAL[ERR] Can't encode due to null pg.
[11-20 09:35:28.823] CamHAL[ERR] isa_lb_video_bayer, prepareTerminalBuffers fail with -38
[11-20 09:35:28.823] CamHAL[ERR] ipu6_lb_video_bayer: pipe iteration error -38
[11-20 09:35:28.823] CamHAL[ERR] @processNewFrame: failed to run pipe
[11-20 09:35:28.881] CamHAL[INF] threadLoop: mCameraStreams[0] == 0x7fb2e00038a0
[11-20 09:35:28.881] CamHAL[INF] threadLoop: getPrivacyBuffer returned nullptr
[11-20 09:35:28.948] CamHAL[INF] threadLoop: mCameraStreams[0] == 0x7fb2e00038a0
[11-20 09:35:28.948] CamHAL[INF] threadLoop: getPrivacyBuffer returned nullptr
[11-20 09:35:29.16] CamHAL[INF] threadLoop: mCameraStreams[0] == 0x7fb2e00038a0
[11-20 09:35:29.16] CamHAL[INF] threadLoop: getPrivacyBuffer returned nullptr
[11-20 09:35:29.83] CamHAL[INF] threadLoop: mCameraStreams[0] == 0x7fb2e00038a0
[11-20 09:35:29.83] CamHAL[INF] threadLoop: getPrivacyBuffer returned nullptr
[11-20 09:35:29.119] CamHAL[ERR] Poll: Device node fd 14 poll timeout.
```

The kernel log also mentions a frame sync error, though I'm sorry but I didn't get the exact message and now can't reproduce ( see next paragraph. )

Those last three lines are then repeated. Is this likely to be something I've missed when making the code changes, or could it be that the clocks are wrong? fwnode_property_read_u32 for "clock-frequency" reads 26000000 - is that likely to be the correct value? I've tried removing the OV02C10_SCLK define and calculating as per the hi556.c file, and also changing OV02C10_DATA_LANES to 2 ( which seems to be correct from ov02c10_read_mipi_lanes? ) but since then I get a segfault when loading intel-ipu6-isys. Interestingly, this same segfault always happened when I unloaded the sensor and isys modules, then tried to reinsert them - however, now it seems to happen the first time. #147 seems to refer to this also but only after the reload - I don't know why this now seems to be happening when I load the module the first time, or if it's some sort of coincidence! No idea if I'm even on the right lines with the clock changes?

sidepipe, 2023-11-22:
There's a register setting array defined, mipi_data_rate_960mbps[], which is empty. Its intended purpose seems to be, on the face of it, to configure the MIPI clock appropriately, probably amongst other things. Should there be anything in there? Why does it say "960mbps" anyway.... where does that figure come from, and if it does nothing, why is it even there?!

PowerKiKi, 2024-07-09:
https://github.com/intel/ipu6-drivers/pull/230 added support for OV02C10, so this issue should be closed as solved, doesn't it ?

closed 2024-07-29 by hao-yao