intel/ipu6-drivers #172: MIPI Camera Drivers Compatibility for Lenovo X1 Carbon Gen 11 on Ubuntu 22.04

rmasad, 2023-08-24:
**Issue Description**:
The MIPI camera drivers for the Lenovo X1 Carbon Gen 11 are currently not working as expected on the Ubuntu 22.04 operating system. This has led to a lack of camera functionality on this particular laptop model when running Ubuntu 22.04.

**Expected Behavior**:
The MIPI camera drivers for the Lenovo X1 Carbon Gen 11 should be compatible with Ubuntu 22.04, enabling users to utilize the built-in camera for various applications such as video conferencing, streaming, and capturing media.

**Steps to Reproduce:**

- Install Ubuntu 22.04 on a Lenovo X1 Carbon Gen 11 laptop.
- Attempt to use the built-in camera with native camera applications or third-party software.
- Notice that the camera is not detected or not functioning as expected.

Current Observations:
- The camera is not recognized by the system, and no video output is received from the camera when attempting to use it in applications such as Cheese.

**Environment:**

Laptop Model: Lenovo X1 Carbon Gen 11
Operating System: Ubuntu 22.04
Kernel Version: 6.2.0-26-generic

gouyelliot, 2023-08-24:
I'm in the same situation, [a post on AskUbuntu](https://askubuntu.com/questions/1476147/mipi-webcam-on-lenovo-thinkpad-x1-carbon-gen-11-on-ubuntu-22-04-not-fully-workin) give more information about the situation (I have the exact same logs as the one in the post).

I can provide more info or run commands if needed, just ping me !

ChenYuYeh, 2023-08-25:
@rmasad 

As the prominent method shared by Hans for the [romfusion package for Fedora](https://hansdegoede.dreamwidth.org/27235.html), it implies the driver delivered by Intel on this github can properly support the webcam on Lenovo ThinkPad X1. (Although there might be regression sometimes, it could be [fixed](https://hansdegoede.dreamwidth.org/27603.html) soon. )

So it should be suspicious that Canonical is not willing to provide adb package for Lenovo ThinkPad X1 even it is a Ubuntu certified system though - https://ubuntu.com/certified/202301-31165.

We should follow up the progress of [a post on AskUbuntu](https://askubuntu.com/questions/1476147/mipi-webcam-on-lenovo-thinkpad-x1-carbon-gen-11-on-ubuntu-22-04-not-fully-workin), and influence Canonical to see your expectations for providing some packages to bring your webcam back to life.

rmasad, 2023-08-25:
Independent of an official support from Canonical (that would be great!) the instructions in this repository do not work. The same thing happens to me that happened to @gouyelliot (and it is described in this link). Any help to get it working?

Happy to help/test.

ChenYuYeh, 2023-08-26:
If I guess correctly, this GitHub might be meant for OS vendors (Ubuntu, Fedora) to generate their ppa/rpmfusion packages. Due to the complexity, includes loopback module, userspace, kernel, firmware and some closed binaries, is not quite friendly for ordinary user like you or me. And it makes me feel this GitHub is temporary rather permanent purposes.

prochac, 2023-08-27:
> And it makes me feel this GitHub is for temporary rather permanent purposes.
It's temporary so far for 3 years, so I hope you're right.

yorik, 2023-08-28:
The camera on Lenovo X1 Carbon Gen 11 works if you follow instructions in this repo and all related + apply patch https://gitlab.com/vicamo/v4l2-relayd/-/merge_requests/13/diffs on `v4l2-relayd`.

It's a lot of compiling, but at least it's possible.

rmasad, 2023-08-28:
It doesn't work, it gives me the same error:
- icamerasrc: [08-28 19:06:13.865] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery
- Cheese: Device wants 2:0:0:0 colorimetry

twouters, 2023-08-29:
@rmasad did you try it on https://webrtc.github.io/samples/src/content/devices/input-output/ with a browser too? Cheese doesn't seem to work with this driver/framework, I don't have the details on why it doesn't.

rpayanm, 2023-08-29:
> The camera on Lenovo X1 Carbon Gen 11 works if you follow instructions in this repo and all related + apply patch https://gitlab.com/vicamo/v4l2-relayd/-/merge_requests/13/diffs on `v4l2-relayd`.
> 
> It's a lot of compiling, but at least it's possible.

@yorik Oh! Impressive, my friend!
Would you mind making a video installing all the thing? Please. You would help us a lot :heart:

rmasad, 2023-08-29:
@yorik, when executing what you mention, this is what it returns:

```~ ❯ sudo modprobe v4l2loopback card_label="Intel MIPI Camera" exclusive_caps=1                                                                          ✘ INT 10:04:32
export GST_PLUGIN_PATH=/usr/lib/gstreamer-1.0/ && rm -rf ~/.cache/gstreamer1.0
DEVICE=$(v4l2-ctl --list-devices | grep platform:v4l2loopback -A 1 | tail -n 1 | xargs echo)
sudo -E gst-launch-1.0 icamerasrc buffer-count=7 device-name=ov2740-uf ! video/x-raw,format=NV12,width=1280,height=720 ! v4l2sink device=${DEVICE}

[sudo] contraseña para rmasad: 
[08-29 10:04:36.853] CamHAL[INF] load_camera_hal_library, the library name: /usr/lib/ipu_adl/libcamhal.so
[08-29 10:04:36.910] CamHAL[INF] load_camera_hal_library, the library name: /usr/lib/ipu_adl/libcamhal.so
Estableciendo el conducto a PAUSA …
[08-29 10:04:37.184] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[08-29 10:04:37.185] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[08-29 10:04:37.185] CamHAL[ERR] invalid media format, default value used.
[08-29 10:04:37.186] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev16 IOCTL VIDIOC_G_EXT_CTRLS error: Argumento inválido
El conducto está vivo y no necesita PREPARARSE …
El conducto está PREPARADO …
Estableciendo el conducto a REPRODUCIENDO …
New clock: GstSystemClock

(gst-launch-1.0:6509): GStreamer-CRITICAL **: 10:04:37.269: Registering meta implementation 'GstCamerasrcMeta' without init function
[08-29 10:04:37.269] CamHAL[ERR] sensor output sub device is not set
[08-29 10:04:39.692] CamHAL[ERR] Start: Device node /dev/video1 IOCTL VIDIOC_STREAMON error: Error de entrada/salida
[08-29 10:04:39.692] CamHAL[ERR] Device:Generic stream on failed.
[08-29 10:04:39.692] CamHAL[ERR] Devices stream on failed:-38
[08-29 10:04:39.692] CamHAL[ERR] Start capture unit failed with ret:-38.
[08-29 10:04:39.692] CamHAL[ERR] Camera device starts failed.
[08-29 10:04:39.692] CamHAL[WAR] @stop: device not started
[08-29 10:04:44.692] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery
[08-29 10:04:49.693] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery
[08-29 10:04:54.177] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery
[08-29 10:04:59.177] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery
[08-29 10:05:04.178] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery
[08-29 10:05:09.178] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery
[08-29 10:05:14.178] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery
[08-29 10:05:19.178] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery
[08-29 10:05:24.179] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery
[08-29 10:05:29.179] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery
[08-29 10:05:34.179] CamHAL[WAR] <id0>@waitFrame, time out happens, wait recovery```

In other terminal:

```~ ❯ journalctl -k | grep ipu                                                                                                                                  10:05:13
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: mapped as: 0x000000008a4d479b
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPC reset done
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: FW version: 20220510
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: mapped as: 0x000000008a4d479b
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPC reset done
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: FW version: 20220510
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: mapped as: 0x000000008a4d479b
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
ago 29 10:04:21 rmasad-X1-Gen-11 kernel: intel-ipu6 0000:00:05.0: IPC reset done
[... cut ...]
ago 29 10:04:39 rmasad-X1-Gen-11 kernel: intel-ipu6-isys intel-ipu6-isys0: stream stop time out
~ ❯ gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! xvimagesink                                     10:05:15

(gst-plugin-scanner:9456): GStreamer-WARNING **: 10:05:36.821: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsticamerasrc.so': /lib/x86_64-linux-gnu/libcamhal.so.0: undefined symbol: _ZN3cca8IntelCCA6getCMCERNS_7cca_cmcE
ADVERTENCIA: conducto erróneo: no hay un elemento «icamerasrc»```

yorik, 2023-08-30:
> @yorik Oh! Impressive, my friend! Would you mind making a video installing all the thing? Please. You would help us a lot ❤️

Thank you! Sorry, I don't have time now to re-install everything and learn how to do videos. Also I'm using Debian, which , looks like, isn't popular now. So it wouldn't be very helpful for most people.

yorik, 2023-08-30:
@rmasad looks like `icamerasrc` was build incorrectly and there is some problem with the driver. I guess you need to clean up everything and build/install it again.
Are there any chance that `/lib/x86_64-linux-gnu/libcamhal.so.0`  is a leftover from previous attempts?

gouyelliot, 2023-08-31:
Thanks @yorik for the info, could you run `uname -a` and share the output here please ?

I spent some time re-compiling/re-installing everything, and now I get this error when trying to start the pipeline

```
$ export GST_PLUGIN_PATH=/usr/lib/gstreamer-1.0/ && rm -rf ~/.cache/gstreamer1.0
$ sudo -E gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! xvimagesink
```

```
août 31 13:12:15 Elliot-ThinkPad kernel: ================================================================================
août 31 13:12:15 Elliot-ThinkPad kernel: UBSAN: invalid-load in /build/linux-hwe-6.2-GoDE7J/linux-hwe-6.2-6.2.0/include/linux/iosys-map.h:267:9
août 31 13:12:15 Elliot-ThinkPad kernel: load of value 68 is not a valid value for type '_Bool'
août 31 13:12:15 Elliot-ThinkPad kernel: CPU: 0 PID: 6100 Comm: camerasrc0:src Tainted: G           OE      6.2.0-31-generic #31~22.04.1-Ubuntu
août 31 13:12:15 Elliot-ThinkPad kernel: Hardware name: LENOVO 21HMCTO1WW/21HMCTO1WW, BIOS N3XET41W (1.16 ) 07/06/2023
août 31 13:12:15 Elliot-ThinkPad kernel: Call Trace:
août 31 13:12:15 Elliot-ThinkPad kernel:  <TASK>
août 31 13:12:15 Elliot-ThinkPad kernel:  dump_stack_lvl+0x48/0x70
août 31 13:12:15 Elliot-ThinkPad kernel:  dump_stack+0x10/0x20
août 31 13:12:15 Elliot-ThinkPad kernel:  __ubsan_handle_load_invalid_value+0xa2/0x100
août 31 13:12:15 Elliot-ThinkPad kernel:  ? dma_buf_map_attachment+0x84/0xc0
août 31 13:12:15 Elliot-ThinkPad kernel:  dma_buf_vmap_unlocked.cold+0x10/0x15
août 31 13:12:15 Elliot-ThinkPad kernel:  ipu_psys_mapbuf_locked+0x17d/0x460 [intel_ipu6_psys]
août 31 13:12:15 Elliot-ThinkPad kernel:  ipu_psys_ioctl+0x182/0x550 [intel_ipu6_psys]
août 31 13:12:15 Elliot-ThinkPad kernel:  ? ipu_psys_ioctl+0x2fa/0x550 [intel_ipu6_psys]
août 31 13:12:15 Elliot-ThinkPad kernel:  ? _copy_to_user+0x25/0x60
août 31 13:12:15 Elliot-ThinkPad kernel:  ? ipu_psys_ioctl+0x2fa/0x550 [intel_ipu6_psys]
août 31 13:12:15 Elliot-ThinkPad kernel:  ? __fget_light+0xb5/0x160
août 31 13:12:15 Elliot-ThinkPad kernel:  __x64_sys_ioctl+0x9a/0xe0
août 31 13:12:15 Elliot-ThinkPad kernel:  do_syscall_64+0x59/0x90
août 31 13:12:15 Elliot-ThinkPad kernel:  ? do_syscall_64+0x69/0x90
août 31 13:12:15 Elliot-ThinkPad kernel:  ? exit_to_user_mode_prepare+0x3b/0xd0
août 31 13:12:15 Elliot-ThinkPad kernel:  ? syscall_exit_to_user_mode+0x2a/0x50
août 31 13:12:15 Elliot-ThinkPad kernel:  ? do_syscall_64+0x69/0x90
août 31 13:12:15 Elliot-ThinkPad kernel:  ? exit_to_user_mode_prepare+0x3b/0xd0
août 31 13:12:15 Elliot-ThinkPad kernel:  ? irqentry_exit_to_user_mode+0x9/0x20
août 31 13:12:15 Elliot-ThinkPad kernel:  ? irqentry_exit+0x43/0x50
août 31 13:12:15 Elliot-ThinkPad kernel:  ? exc_page_fault+0x92/0x1b0
août 31 13:12:15 Elliot-ThinkPad kernel:  entry_SYSCALL_64_after_hwframe+0x72/0xdc
août 31 13:12:15 Elliot-ThinkPad kernel: RIP: 0033:0x7fe238d1aaff
août 31 13:12:15 Elliot-ThinkPad kernel: Code: 00 48 89 44 24 18 31 c0 48 8d 44 24 60 c7 04 24 10 00 00 00 48 89 44 24 08 48 8d 44 24 20 48 89 44 24 10 b8 10 00 00 00 0f 05 <41> 89 c0 3d 00 f0 ff ff 77 1f 48 8b 44 24 18 64 48 2b 04 25 28 00
août 31 13:12:15 Elliot-ThinkPad kernel: RSP: 002b:00007fe234b24110 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
août 31 13:12:15 Elliot-ThinkPad kernel: RAX: ffffffffffffffda RBX: 00007fe215380ed0 RCX: 00007fe238d1aaff
août 31 13:12:15 Elliot-ThinkPad kernel: RDX: 0000000000000013 RSI: 00000000c0044102 RDI: 0000000000000012
août 31 13:12:15 Elliot-ThinkPad kernel: RBP: 00007fe215382210 R08: 0000000000000000 R09: 00007fe215382210
août 31 13:12:15 Elliot-ThinkPad kernel: R10: 00007fe228000690 R11: 0000000000000246 R12: 0000000000000000
août 31 13:12:15 Elliot-ThinkPad kernel: R13: 00007fe21537caf0 R14: 00007fe215380ed0 R15: 00007fe234b2423c
août 31 13:12:15 Elliot-ThinkPad kernel:  </TASK>
août 31 13:12:15 Elliot-ThinkPad kernel: ================================================================================
```

gouyelliot, 2023-09-01:
Thanks again for the details @yorik. Looks like I would need to upgrade my kernel, it's possible on Ubuntu using the [intermediate kernels repository ](https://launchpad.net/~canonical-hwe-team/+archive/ubuntu/intermediate-kernel) if anyone wants to give it a try.

I'm not going to do it as this is my workstation that I use daily and I'd rather stay on a stable kernel. Let's wait for the update I guess...

fourdollars, 2023-09-08:
There is a known issue on Cheese that it can not support Intel MIPI camera via the v4l2loopback workaround.
It is better to use Google Chrome or Mozilla Firefox to visit https://webcamtests.com/ to check your MIPI camera.

rmasad, 2023-09-08:
Re-install everything, and run the command: `gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! xvimagesink`

It opens a window (the camera light does not turn on) and shows me this:

[image]

fourdollars, 2023-09-11:
FYI, I usually executed `sudo gst-launch-1.0 icamerasrc ! autovideosink` to check the MIPI camera on Ubuntu.

rmasad, 2023-09-11:
When I run `sudo -E gst-launch-1.0 icamerasrc ! autovideosink`, it response:

```
~ ❯ sudo -E gst-launch-1.0 icamerasrc ! autovideosink                                                                                                                                                                                                                                                                                                                                                                             10:08:25
[09-11 10:08:29.236] CamHAL[INF] load_camera_hal_library, the library name: /usr/lib/ipu_adl/libcamhal.so
[09-11 10:08:29.244] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.244] CamHAL[ERR] Parse AE eExposure time range failed
[09-11 10:08:29.244] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.244] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.244] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.244] CamHAL[ERR] Parse AE eExposure time range failed
[09-11 10:08:29.244] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.244] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE eExposure time range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE eExposure time range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE eExposure time range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE eExposure time range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE eExposure time range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE eExposure time range failed
[09-11 10:08:29.245] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.245] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.246] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.246] CamHAL[ERR] Parse AE eExposure time range failed
[09-11 10:08:29.246] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.246] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.246] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.246] CamHAL[ERR] Parse AE eExposure time range failed
[09-11 10:08:29.246] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.246] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.246] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.246] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.246] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.246] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.246] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.246] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.246] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.246] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.246] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.246] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.247] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.247] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.247] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.247] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.247] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.247] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.247] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.247] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.247] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.247] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.247] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.247] CamHAL[ERR] Parse AE gain range failed
[09-11 10:08:29.247] CamHAL[ERR] Malformed ET range in exposure time range configuration
[09-11 10:08:29.247] CamHAL[ERR] Parse AE gain range failed
[... cut ...]
[09-11 10:08:29.911] CamHAL[ERR] @configure Device Configure failed
[09-11 10:08:29.911] CamHAL[ERR] failed to config streams.
ERROR: del elemento /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Información adicional de depuración:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.006924328
Estableciendo el conducto a NULL …
Liberando el conducto…```

fourdollars, 2023-09-13:
@rmasad Could you execute `sudo -E oem-getlogs` on Ubuntu 22.04 and attach the log?

rmasad, 2023-09-13:
It is a 62~ mb file. Any subset of the file (to grep it)?

fourdollars, 2023-09-14:
@rmasad [Lenovo ThinkPad X1 Carbon Gen 11](https://ubuntu.com/certified/202301-31165) is an Ubuntu certified platform.
Could you open a bug against https://bugs.launchpad.net/oem-priority and then attach your log there?

rmasad, 2023-09-15:
I try installing gstreamer1.0-icamera from oem-solutions-group/intel-ipu6 repository and now I am getting this error: https://pastebin.com/3XevZ83z

rmasad, 2023-09-22:
I try with 6.1.0-1022-oem, not working yet 

```~ ❯ uname -r
6.1.0-1022-oem

~ ❯ sudo gst-launch-1.0 v4l2src ! glimagesink
[sudo] contraseña para rmasad: 
Estableciendo el conducto a PAUSA …
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
ERROR: del elemento /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: El dispositivo «/dev/video0» no es un dispositivo de captura.
Información adicional de depuración:
../sys/v4l2/v4l2_calls.c(629): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
Capabilities: 0x5200002
ERROR: el conducto no quiere prepararse.
Falló al establecer el conducto a PAUSA …
Estableciendo el conducto a NULL …
Liberando el conducto…

~ ❯ sudo gst-launch-1.0 icamerasrc
Estableciendo el conducto a PAUSA …
[09-22 10:16:31.895] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[09-22 10:16:31.896] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[09-22 10:16:31.896] CamHAL[ERR] invalid media format, default value used.
[09-22 10:16:31.896] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev9 IOCTL VIDIOC_G_EXT_CTRLS error: Argumento inválido
El conducto está vivo y no necesita PREPARARSE …
El conducto está PREPARADO …
Estableciendo el conducto a REPRODUCIENDO …
New clock: GstSystemClock

(gst-launch-1.0:14552): GStreamer-CRITICAL **: 10:16:31.943: Registering meta implementation 'GstCamerasrcMeta' without init function
[09-22 10:16:31.943] CamHAL[ERR] sensor output sub device is not set
ERROR: del elemento /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Información adicional de depuración:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-linked (-1)
Execution ended after 0:00:00.568868400
Estableciendo el conducto a NULL …
Enter ia_aiq_get_aiqd_data()
Out-aiqd data size: 41480
[09-22 10:16:32.599] CamHAL[WAR] Failed to open file /run/camera/ov2740-uf_VIDEO.aiqd, error No existe el archivo o el directorio
Liberando el conducto…
 ```

Trollgeir, 2023-09-28:
Can confirm after numerous attempts I managed to get my camera working too. 

Ubuntu 22.04, kernel 6.2.0-33-generic

After several failed attempts I tried to uninstall everything and go again from scratch, removing everything related to gst-launch/gstreamer. And then;

```
sudo add-apt-repository ppa:oem-solutions-group/intel-ipu6
sudo apt install linux-modules-ipu6-generic-hwe-22.04 linux-modules-ivsc-generic-hwe-22.04
```
After doing this I selected the additional drivers from the ppa:
[image]

After a reboot the camera worked in google meet (although seems to be broken for cheese, but I don't really care).

gouyelliot, 2023-09-29:
That's good news @Trollgeir ! I tried following your instructions, but now I don't have any `/dev/video` file on boot. Did you configure anything else after installing the packages ? I feel I'm missing something but can't figure out what

Trollgeir, 2023-09-29:
> That's good news @Trollgeir ! I tried following your instructions, but now I don't have any `/dev/video` file on boot. Did you configure anything else after installing the packages ? I feel I'm missing something but can't figure out what

I don't think so. I might have rebooted once before running the last line, and then a second reboot. I'll be honest I'm too scared to try to reproduce it because I've spent so many hours to fix this 😞

lukemarsden, 2023-09-29:
Running ov2740 on X1 Carbon Gen 11 here.

Both https://github.com/intel/ipu6-drivers/issues/172#issuecomment-1739853484 on 6.2 kernel and https://github.com/intel/ipu6-drivers/issues/22#issuecomment-1705198032 on 6.5 kernel are working for me, although _intermittently_ (in the same way on both kernels). It's very strange, if I load https://meet.new for example in Chrome, my `dmesg -w` will look like this:

```
[33900.129696] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 1
[33900.129703] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 0
[33900.231053] intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 12-0036
[33907.585989] intel-ipu6-isys intel-ipu6-isys0: stream stop time out
[33907.586009] intel-ipu6-isys intel-ipu6-isys0: stream off ov2740 12-0036
[33909.634053] intel-ipu6-isys intel-ipu6-isys0: stream close time out
[33909.634095] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 1
[33909.634103] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 0
[33909.759273] intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 12-0036
```

Sometimes the video stream shows up immediately, sometimes it takes a few seconds, and sometimes it takes a few minutes. During that time, the light on the camera will flick off and on again in time with the `stream stop/close time out` messages above. In Chrome, it always seems to work _eventually_, in Zoom it seems like sometimes it permafails.

It seems like the intel-ipu6-isys driver is retrying on some internal failure. I'd love to get to the bottom of this and figure out how to make it work reliably all the time :) Anyone interested in debugging with me on my system?

Logs from one time turning the camera on in google meet worked first time:

`dmesg -w`
```
[34405.387505] intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 12-0036
[34446.455300] intel-ipu6-isys intel-ipu6-isys0: stream off ov2740 12-0036
```

`sudo journalctl -u v4l2-relayd.service -f`: no logs

One time it took two retries (during which time the video stream was a blank black square for a short time):

```
[34451.128818] intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 12-0036
[34458.513640] intel-ipu6-isys intel-ipu6-isys0: stream stop time out
[34458.513657] intel-ipu6-isys intel-ipu6-isys0: stream off ov2740 12-0036
[34460.557312] intel-ipu6-isys intel-ipu6-isys0: stream close time out
[34460.557351] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 1
[34460.557359] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 0
[34460.728742] intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 12-0036
[34461.220302] intel-ipu6-isys intel-ipu6-isys0: stream off ov2740 12-0036
[34461.327925] intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 12-0036
[34468.687606] intel-ipu6-isys intel-ipu6-isys0: stream stop time out
[34468.687627] intel-ipu6-isys intel-ipu6-isys0: stream off ov2740 12-0036
[34470.731202] intel-ipu6-isys intel-ipu6-isys0: stream close time out
[34470.731239] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 1
[34470.731246] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 0
[34470.846949] intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 12-0036
[34471.327767] intel-ipu6-isys intel-ipu6-isys0: stream off ov2740 12-0036
[34471.446117] intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 12-0036
[34478.701532] intel-ipu6-isys intel-ipu6-isys0: stream stop time out
[34478.701545] intel-ipu6-isys intel-ipu6-isys0: stream off ov2740 12-0036
[34480.745107] intel-ipu6-isys intel-ipu6-isys0: stream close time out
[34480.745140] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 1
[34480.745147] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 0
[34480.856014] intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 12-0036
[34488.231597] intel-ipu6-isys intel-ipu6-isys0: stream stop time out
[34488.231611] intel-ipu6-isys intel-ipu6-isys0: stream off ov2740 12-0036
[34490.283232] intel-ipu6-isys intel-ipu6-isys0: stream close time out
[34490.283256] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 1
[34490.283261] intel-ipu6-isys intel-ipu6-isys0: Intel IPU6 BE SOC capture 0: cleaning active queue 0
[34490.412371] intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 12-0036
```

`sudo journalctl -u v4l2-relayd.service -f`:
```
Sep 29 09:35:48 rocket sh[20815]: Out-aiqd data size: 41480
Sep 29 09:35:48 rocket sh[20815]: [09-29 09:26:15.475] CamHAL[WAR] Failed to open file /run/camera/ov2740-uf_VIDEO.aiqd, error No such file or directory
Sep 29 09:35:48 rocket sh[20815]: [09-29 09:26:15.529] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
Sep 29 09:35:48 rocket sh[20815]: [09-29 09:26:15.529] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[... 30 lines cut ...]
Sep 29 09:35:48 rocket sh[20815]: [09-29 09:26:35.240] CamHAL[ERR] sensor output sub device is not set
Sep 29 09:35:48 rocket sh[20815]: Enter ia_aiq_get_aiqd_data()
Sep 29 09:35:48 rocket sh[20815]: Out-aiqd data size: 41480
Sep 29 09:35:48 rocket sh[20815]: [09-29 09:26:45.385] CamHAL[WAR] Failed to open file /run/camera/ov2740-uf_VIDEO.aiqd, error No such file or directory
[... cut ...]
Sep 29 09:35:48 rocket sh[20815]: [09-29 09:35:44.822] CamHAL[INF] Time out or thread is not running, ret = 0
Sep 29 09:35:48 rocket sh[20815]: Enter ia_aiq_get_aiqd_data()
Sep 29 09:35:48 rocket sh[20815]: Out-aiqd data size: 41480
Sep 29 09:35:48 rocket sh[20815]: [09-29 09:35:48.900] CamHAL[WAR] Failed to open file /run/camera/ov2740-uf_VIDEO.aiqd, error No such file or directory
```

Is the `/run/camera/ov2740-uf_VIDEO.aiqd` a red herring or do I need to install some firmware in that location? (If so, why does it work occasionally? :sweat_smile:)

yorik, 2023-09-29:
Just for the record: I'm running ov2740 on X1 Carbon Gen 11 here. Debian sid, 6.5.0-1-amd64 stock kernel.

It works all the time, in dmesg I can see only:
```
intel-ipu6-isys intel-ipu6-isys0: stream on ov2740 12-0036
```
and when close meeting tab:
```
intel-ipu6-isys intel-ipu6-isys0: stream off ov2740 12-0036
```

lukemarsden, 2023-09-29:
oh, interesting @yorik! What steps exactly did you follow to set it up?

yorik, 2023-09-29:
All by the README
* ipu6-drivers dkms way
* ipu6-camera-bins
* ipu6-camera-hal
* icamerasrc
* v4l2loopback via dkms from https://github.com/umlaeute/v4l2loopback
* v4l2-relayd with https://gitlab.com/vicamo/v4l2-relayd/-/commit/35a06d89747ff31fbaabf744b7dae585bfa86723 patch

lukemarsden, 2023-09-30:
Thanks @yorik! The instructions for https://github.com/intel/icamerasrc/tree/icamerasrc_slim_api are unclear to me, with either of the `export PKG_CONFIG_PATH` lines or with that env var unset, I get:
```
checking for
  gstreamer-1.0 >= 1.0.0
  gstreamer-base-1.0 >= 1.0.0
  gstreamer-controller-1.0 >= 1.0.0
... yes
Building with libcamhal
checking for libcamhal... no
configure: error: Cannot find libcamhal pkgconfig
configure failed
```
This is on a brand new debian sid install, following the items in your list.
I can't find a debian package for libcamhal, was it supposed to come from `ipu6-camera-hal`? That [installation](https://github.com/intel/ipu6-camera-hal) completed successfully, including copying the results into /usr.

lukemarsden, 2023-09-30:
Oh, `export PKG_CONFIG_PATH="$HOME/webcam/ipu6-camera-hal"` unblocked me

lukemarsden, 2023-09-30:
Oh, but then `make` doesn't actually build anything
```
luke@oldschool:~/webcam/icamerasrc$ make
make  all-recursive
make[1]: Entering directory '/home/luke/webcam/icamerasrc'
Making all in src
make[2]: Entering directory '/home/luke/webcam/icamerasrc/src'
Making all in interfaces
make[3]: Entering directory '/home/luke/webcam/icamerasrc/src/interfaces'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/luke/webcam/icamerasrc/src/interfaces'
make[3]: Entering directory '/home/luke/webcam/icamerasrc/src'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/home/luke/webcam/icamerasrc/src'
make[2]: Leaving directory '/home/luke/webcam/icamerasrc/src'
make[2]: Entering directory '/home/luke/webcam/icamerasrc'
make[2]: Leaving directory '/home/luke/webcam/icamerasrc'
make[1]: Leaving directory '/home/luke/webcam/icamerasrc'
luke@oldschool:~/webcam/icamerasrc$ sudo make install
Making install in src
make[1]: Entering directory '/home/luke/webcam/icamerasrc/src'
Making install in interfaces
make[2]: Entering directory '/home/luke/webcam/icamerasrc/src/interfaces'
make[3]: Entering directory '/home/luke/webcam/icamerasrc/src/interfaces'
 /bin/mkdir -p '/usr/local/lib'
 /bin/bash ../../libtool   --mode=install /bin/install -c   libgsticamerainterface-1.0.la '/usr/local/lib'
libtool: install: /bin/install -c .libs/libgsticamerainterface-1.0.so.1.0.0 /usr/local/lib/libgsticamerainterface-1.0.so.1.0.0
libtool: install: (cd /usr/local/lib && { ln -s -f libgsticamerainterface-1.0.so.1.0.0 libgsticamerainterface-1.0.so.1 || { rm -f libgsticamerainterface-1.0.so.1 && ln -s libgsticamerainterface-1.0.so.1.0.0 libgsticamerainterface-1.0.so.1; }; })
libtool: install: (cd /usr/local/lib && { ln -s -f libgsticamerainterface-1.0.so.1.0.0 libgsticamerainterface-1.0.so || { rm -f libgsticamerainterface-1.0.so && ln -s libgsticamerainterface-1.0.so.1.0.0 libgsticamerainterface-1.0.so; }; })
libtool: install: /bin/install -c .libs/libgsticamerainterface-1.0.lai /usr/local/lib/libgsticamerainterface-1.0.la
libtool: warning: remember to run 'libtool --finish /home/luke/webcam/out/install/lib'
 /bin/mkdir -p '/usr/local/include/gstreamer-1.0/gst/icamera'
 /bin/install -c -m 644 gstcamera3ainterface.h gstcameraispinterface.h gstcameradewarpinginterface.h gstcamerawfovinterface.h '/usr/local/include/gstreamer-1.0/gst/icamera'
make[3]: Leaving directory '/home/luke/webcam/icamerasrc/src/interfaces'
make[2]: Leaving directory '/home/luke/webcam/icamerasrc/src/interfaces'
make[2]: Entering directory '/home/luke/webcam/icamerasrc/src'
make[3]: Entering directory '/home/luke/webcam/icamerasrc/src'
make[3]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/local/lib/gstreamer-1.0'
 /bin/bash ../libtool   --mode=install /bin/install -c   libgsticamerasrc.la '/usr/local/lib/gstreamer-1.0'
libtool:   error: error: cannot install 'libgsticamerasrc.la' to a directory not ending in /home/luke/webcam/out/install/lib/gstreamer-1.0
make[3]: *** [Makefile:546: install-pluginLTLIBRARIES] Error 1
make[3]: Leaving directory '/home/luke/webcam/icamerasrc/src'
make[2]: *** [Makefile:848: install-am] Error 2
make[2]: Leaving directory '/home/luke/webcam/icamerasrc/src'
make[1]: *** [Makefile:685: install-recursive] Error 1
make[1]: Leaving directory '/home/luke/webcam/icamerasrc/src'
make: *** [Makefile:520: install-recursive] Error 1
```
What am I doing wrong?

lukemarsden, 2023-09-30:
Hmm, maybe the `build.sh` from -hal also installed icamerasrc? Anyway, guessing `./autogen.sh && ./configure && make && sudo make install` for https://gitlab.com/vicamo/v4l2-relayd (the commit you linked to seemed to be in main already), and rebooting, I'm now just getting a lot of this in `dmesg -w`:

```
[    6.371669] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[    6.371823] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[    8.215767] intel-ipu6-isys intel-ipu6-isys0: isys port open ready failed -16
[    8.215775] intel-ipu6-isys intel-ipu6-isys0: Device close failure: -16
[   10.215709] intel-ipu6-isys intel-ipu6-isys0: Device release time out -16
[   10.240410] intel-ipu6-isys intel-ipu6-isys0: Clearing old context
[   11.312568] intel-ipu6-isys intel-ipu6-isys0: isys port open ready failed -16
[   11.312584] intel-ipu6-isys intel-ipu6-isys0: Device close failure: -16
[   13.315721] intel-ipu6-isys intel-ipu6-isys0: Device release time out -16
```

GinaMuuss, 2023-10-02:
Does the camera work for you if v4l2-relayd is not running, and you run the manual command yorik proposed before?
`gst-launch-1.0 -v icamerasrc ! video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink device=/dev/video0`
With this command it works for me even though I get the same errors as you.
v4l2-relayd does not work though, whether it is running or not does not change that I just get a black image.

lukemarsden, 2023-10-03:
Got the ov2740 in my X1 Yoga Gen 8 working reliably using Fedora 38 on kernel 6.5.5-200.fc38.x86_64 following the instructions here: https://hansdegoede.livejournal.com/27276.html

It seemed to be broken at first but I think I just needed to reboot twice!?

I've turned the camera off and on maybe 50 times - only one time I got the dreaded black square and had to `sudo systemctl restart v4l2-relayd.service`

I'll come back to the X1 Carbon Gen 11 another day. Despite having, I think, an idential sensor it seems to be much flakier, per https://github.com/intel/ipu6-drivers/issues/172#issuecomment-1740521437.

thekidconnect, 2025-04-22:
Did we ever resolve the `Cannot find libcamhal pkgconfig` error? Running `export PKG_CONFIG_PATH="$HOME/webcam/ipu6-camera-hal"` does not unblock me. I'm on a Lenovo ThinkPad X1 Carbon.