intel/ipu6-drivers #176: Installation using dkms  - dkms_packages.py No such file

vagi8, 2023-08-29:
I have tried to install using dkms method and I am getting the following error 

> Cleaning build area...
make -j12 KERNELRELEASE=5.15.0-79-generic KERNELRELEASE=5.15.0-79-generic KERNEL_SRC=/lib/modules/5.15.0-79-generic/build...(bad exit status: 2)
python3: can't open file '/usr/share/apport/package-hooks/dkms_packages.py': [Errno 2] No such file or directory
Error! Bad return status for module build on kernel: 5.15.0-79-generic (x86_64)
Consult /var/lib/dkms/ipu6-drivers/0.0.0/build/make.log for more information.
dkms autoinstall on 5.15.0-79-generic/x86_64 failed for ipu6-drivers(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.

Kernel version -
> Linux XPS-9315 5.15.0-79-generic #86~20.04.2-Ubuntu SMP Mon Jul 17 23:27:17 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

lscpu - XPS 13 9315

> Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   39 bits physical, 48 bits virtual
CPU(s):                          12
On-line CPU(s) list:             0-11
Thread(s) per core:              1
Core(s) per socket:              10
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           154
Model name:                      12th Gen Intel(R) Core(TM) i7-1250U

Any help is appreciated. I need to make it work via dkms

yorik, 2023-08-30:
I don't think dkms way would work with kernel 5.15. You need to apply https://github.com/intel/ipu6-drivers/blob/master/patch/int3472-v5.15/int3472-support-independent-clock-and-LED-gpios.patch to it and rebuild, so it's easier just to use the first way.

yorik, 2023-08-30:
I think, you'd better upgrade kernel to 6.4.11, it requires no custom patches, you only need to have needed modules enabled, especially `CONFIG_INTEL_SKL_INT3472=m`.

vagi8, 2023-08-30:
So I came across this canonical hardware compatibility for my laptop. https://ubuntu.com/certified/202203-30061 . Hence I was saying 5.14. FYI - The issues I am trying to solve is that my camera is not working in ubuntu but it works in windows.

fourdollars, 2023-09-08:
`Dell XPS 9315` is certified on Ubuntu 20.04 and the system should install oem-somerville-psyduck-meta automatically by `Software Updater` and it will enable the usage of OEM archive and install the MIPI camera userspace stack.

All OEM kernels on Ubuntu 20.04 have been transitted to generic-hwe 5.15 kernel.

You should be able to use the MIPI camera on Google Chrome and Mozilla Firefox.

BTW, oem-somerville-psyduck-meta should also be listed in the `Additional Drivers` of `Software & Updates`.

It may break the MIPI camera if you used packages from some PPA or some hacking in your system.

vagi8, 2023-09-08:
@fourdollars . Thank you for the information. But when I installed the ubuntu it did not install the said oem version rather it had 5.15 generic kernel. Also I tried changing my kernel to 5.14 No luck. Could you suggest me some solutions how I can proceed with this ? reinstall ubuntu or just try to apply patches ?
PS - I got window with this XPS. I created a bootable drive from the image downloaded from canonical (20.04.6) kernel 5.15.

fourdollars, 2023-09-11:
@vagi8 You should be able to get the MIPI camera work after you installed Ubuntu 20.04.6 and upgraded the system by `Software Updater`.
If it doesn't work for you on Ubuntu 20.04.6, please open a bug against https://bugs.launchpad.net/ubuntu/+source/oem-somerville-psyduck-meta.

closed 2023-09-11 by vagi8