intel/ipu6-drivers #177: Black screen on XPS 9320

pepa-cz, 2023-08-30:
Hello,

When trying to get the camera working I am getting a black screen (does not matter if on Cheese or in the browser). Driver initialization seems to be correct, the camera lights up for a second after boot and nothing suspicious in the log:

```
# dmesg|grep ipu
[    7.491721] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    7.492664] intel-ipu6 0000:00:05.0: Device 0x465d (rev: 0x2)
[    7.492693] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    7.492695] intel-ipu6 0000:00:05.0: mapped as: 0x00000000e80cd2b3
[    7.536646] intel-ipu6 0000:00:05.0: Unable to set secure mode
[    7.536654] intel-ipu6 0000:00:05.0: IPU in non-secure mode
[    7.536656] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    7.536659] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    7.536669] intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
[    7.536671] intel-ipu6 0000:00:05.0: IPC reset done
[    7.536672] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    7.538444] intel-ipu6 0000:00:05.0: FW version: 20220510
[    7.541179] intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[    7.597018] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[    7.597611] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[   17.254118] intel-ipu6-isys intel-ipu6-isys0: bind ov01a10 3-0036 nlanes is 1 port is 2
[   17.254449] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
```
When testing with Cheese application (`cheese -d "Intel MIPI Camera"`), I always get black screen and following in the log:

```
[  415.856919] intel-ipu6-isys intel-ipu6-isys0: removed
[  415.857177] intel-ipu6-isys intel-ipu6-isys0: unbind ov01a10 3-0036
[  415.922524] intel-ipu6-isys intel-ipu6-isys0: bind ov01a10 3-0036 nlanes is 1 port is 2
[  415.926305] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
[  421.278119] intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
[  421.305927] intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
[  421.334000] intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
[  421.361894] intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
[  421.390070] intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
[  421.417908] intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
[  421.445900] intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
[  421.473908] intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
```

What may be wrong in this case? Driver is compiled from source, how can I debug driver more deeply?

yorik, 2023-08-30:
Try to run:
```
gst-launch-1.0 -v icamerasrc ! video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink device=/dev/video0
```
and then use cheese, if it will work your v4l2-relayd is missing patch https://gitlab.com/vicamo/v4l2-relayd/-/merge_requests/13/diffs .

gabri94, 2023-09-01:
Hi, I have the same issue on Debian 12. I installed the drivers via dkms, disabled secure boot but cheese wont work and I have the same logs,.

yukim, 2023-09-01:
I also have xps 9320 running Ubuntu 22.04, and the webcam had been working until recently (I think around the beginning of August webcam stop working.)

```sh
uname -a
Linux xps 6.2.0-26-generic #26~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jul 13 16:27:29 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

In my case, `dmesg` does not show anything (the same as OP).
What I found in my machine is that gstreamer blacklisted icamerasrc.
Here is an output from `gst-inspect-1.0`:

```sh
gst-inspect-1.0 /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsticamerasrc.so

(gst-inspect-1.0:181819): GStreamer-WARNING **: 12:58:02.292: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsticamerasrc.so': /lib/x86_64-linux-gnu/libcamhal.so.0: undefined symbol: _ZN3cca8IntelCCA6getCMCERNS_7cca_cmcE
Could not load plugin file: Opening module failed: /lib/x86_64-linux-gnu/libcamhal.so.0: undefined symbol: _ZN3cca8IntelCCA6getCMCERNS_7cca_cmcE
```

pepa-cz, 2023-09-01:
In my case, the problem is solved. `icamerasrc` plugin was blacklisted. I compiled it [from source](https://github.com/intel/icamerasrc/tree/icamerasrc_slim_api), moved it to the same location `/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsticamerasrc.so` and loaded again (as root):

```
# rm -rf ~/.cache/gstreamer-1.0 && gst-inspect-1.0 -b
[09-01 06:57:59.168] CamHAL[INF] load_camera_hal_library, the library name: /usr/lib/ipu_adl/libcamhal.so
```

Then I got working following gst pipeline and at the same time I was able to see the video stream in Google Meet.

```
# sudo gst-launch-1.0 -v icamerasrc ! video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink device=/dev/video0
```

The next step was to get to work `v4l2-relayd` service. I modified options for v4l2loopback module so the newly created virtual camera has the label `MYCamera`.

```
# modprobe -r v4l2loopback   && modprobe  v4l2loopback  video_nr=0 card_label=MYCamera exclusive_caps=1
# cat /sys/devices/virtual/video4linux/video0/name
MYCamera
```
For persistent change
```
# cat /etc/modprobe.d/v4l2loopback.conf 
options v4l2loopback card_label=MYCamera exclusive_caps=1 video_nr=0
```
And finally 
```
# cat /etc/default/v4l2-relayd
# GStreamer source element name:
#VIDEOSRC="videotestsrc"
VIDEOSRC="icamerasrc"
#SPLASHSRC="filesrc location=/.../splash.png ! pngdec ! imagefreeze num-buffers=4 ! videoscale ! videoconvert"

# Output format, width, height, and frame rate:
#FORMAT=YUY2
FORMAT=NV12
WIDTH=1280
HEIGHT=720
FRAMERATE=30/1

# Virtual video device name:
#CARD_LABEL="Virtual Camera"
CARD_LABEL="MYCamera"

# Extra options to pass to v4l2-relayd:
#EXTRA_OPTS=-d
```

Service `systemctl status v4l2-relayd` was running well and camera is working with `sudo cheese -d "MYCamera"`.

@yukim I suppose you installed drivers from the public repository (I don't remember the name of the repository now). I did the same before but the camera stopped working after some time. Then I uninstalled the appropriate packages and compiled everything from the source.

pepa-cz, 2023-09-06:
@hakimkac99 I have no idea what is happenning in your case I did not have errors like this.

However the first error is `load_camera_hal_library, failed to open PCI device. error: (null)` so I would say that is something wrong with `/dev/video0`.

fourdollars, 2023-09-08:
`Dell XPS 13 9320` is officially supported on both Ubuntu 20.04 and Ubuntu 22.04.

> The first device to be certified for Ubuntu Desktop 22.04 LTS is the [Dell XPS 13 Plus](https://www.dell.com/en-uk/shop/laptops/xps-13-plus-laptop/spd/xps-13-9320-laptop) Developer Edition, Dell’s most powerful 13-inch laptop. https://ubuntu.com/blog/upgrade-your-desktop-ubuntu-22-04-lts,

You should be able to use the MIPI camera on `Dell XPS 13 9320` after installing Ubuntu 20.04 or Ubuntu 22.04 and upgrading the system to the latest.

It may break the MIPI camera if you used packages from some PPA or other hacking in your system.

rickschippers, 2023-09-18:
> You should be able to use the MIPI camera in Google Chrome and Mozilla Firefox on `Dell XPS 13 9320` just after installing Ubuntu 20.04 or Ubuntu 22.04 and upgrading the system to the latest.

I can confirm this works. But only if you stick with the default (5.19) kernel. I have ubuntu 22.04 with the 6.1 oem kernel, and the camera does not work then.

XargonWan, 2024-02-16:
Greetings, is there any advancement on this issue?

johndevs, 2025-03-24:
I can confirm this is not working on vanilla Ubuntu 24:

```
~ $ sudo gst-launch-1.0 icamerasrc ! autovideosink
[03-24 09:28:39.784] CamHAL[ERR] load_camera_hal_library, failed to open PCI device. error: (null)
[03-24 09:28:39.784] CamHAL[ERR] get_number_of_cameras, function call is nullptr
[03-24 09:28:39.784] CamHAL[ERR] get_number_of_cameras, function call is nullptr

(gst-launch-1.0:8659): GLib-GObject-CRITICAL **: 09:28:39.784: g_param_spec_enum: assertion 'g_enum_get_value (enum_class, default_value) != NULL' failed

(gst-launch-1.0:8659): GLib-GObject-CRITICAL **: 09:28:39.784: validate_pspec_to_install: assertion 'G_IS_PARAM_SPEC (pspec)' failed

(gst-launch-1.0:8659): GLib-GObject-CRITICAL **: 09:28:39.784: g_param_spec_ref_sink: assertion 'G_IS_PARAM_SPEC (pspec)' failed

(gst-launch-1.0:8659): GLib-GObject-CRITICAL **: 09:28:39.784: g_param_spec_unref: assertion 'G_IS_PARAM_SPEC (pspec)' failed
[03-24 09:28:39.784] CamHAL[ERR] get_number_of_cameras, function call is nullptr
WARNING: erroneous pipeline: could not link camerasrc0 to autovideosink0
```

Camera is found both on web browsers and Cheese but the output is black.

thekidconnect, 2025-04-22:
Hi. Was this issue ever resolved? I get the same `CamHAL[ERR] load_camera_hal_library, failed to open PCI device. error: (null)` error on my Lenovo Thinkpad X1 Carbon on Ubuntu 22.04 :)

huyhoang8398, 2025-04-28:
same on dell xps 9320/ Ubuntu 24.04

mathstuf, 2025-09-14:
Hmm. I'm getting no content, but I don't see any `ipu` modules logged in `dmesg` (the core `ov08x40 driver seems to be loaded though). No messages when loading it either. I have an uv08x40 (see #386).

mathstuf, 2025-10-04:
I've been informed on my Fedora bug report that my camera is actually ipu7, not ipu6 which explains why this driver doesn't help at all. Is there a reliable way to tell the difference? A table of hardware names that get reported by the kernel to expected driver repo perhaps?