intel/ipu6-drivers #181: Debian testing with Dell XPS 13 plus (9320)

gabri94, 2023-09-07:
Hi everyone, I've followed the guide to install the webcam drivers on this machine. 

I have used the dkms version of the kernel driver and disabled secure boot. The kernel modules are, in fact, being correctly loaded:

``` lsmod | grep ipu
intel_ipu6_isys       155648  0
videobuf2_dma_contig    24576  1 intel_ipu6_isys
videobuf2_v4l2         36864  2 intel_ipu6_isys,uvcvideo
videobuf2_common       73728  6 videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,intel_ipu6_isys,uvcvideo,videobuf2_memops
intel_ipu6_psys       110592  0
v4l2_fwnode            32768  2 ov01a10,intel_ipu6_isys
intel_ipu6            118784  2 intel_ipu6_isys,intel_ipu6_psys
v4l2_async             28672  3 v4l2_fwnode,ov01a10,intel_ipu6_isys
videodev              327680  6 v4l2_async,v4l2_fwnode,videobuf2_v4l2,ov01a10,intel_ipu6_isys,uvcvideo
```

I've then installed the IPU firmware and proprietary image processing libraries and both the HAL and icamerasrc plugin for gstreamer. Then seeing that I had some errors regarding ivsc-firmware, i have manually installed it in the right folders so that i got loaded by the system.

The camera is seen by `v4l2-ctl`:

```
Driver Info:                                                              
	Driver name      : intel-ipu6-isys
	Card type        : ipu6
	Bus info         : PCI:pci:pci0000:00
	Driver version   : 6.4.11
	Capabilities     : 0x84201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (camera: ok)
Format Video Capture Multiplanar:
	Width/Height      : 1/1
	Pixel Format      : 'Y10 ' (10-bit Greyscale)
	Field             : None
	Number of planes  : 1
	Flags             : 
	Colorspace        : Raw
	Transfer Function : Default
	YCbCr/HSV Encoding: Default
	Quantization      : Default
	Plane 0           :
	   Bytes per Line : 64
	   Size Image     : 1088

User Controls

        isys_be_soc_compression 0x00981983 (bool)   : default=0 value=0
```

However,  by using either cheese or gstreamer I am not able to see any image from the webcam.

This are the logs from journalctl:
```
 sudo journalctl -k --grep=ipu6
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: physical base address 0x603e000000
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: mapped as: 0x00000000bd14bdd6
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: Unable to set secure mode
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: IPU in non-secure mode
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: IPC reset done
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: firmware: direct-loading firmware intel/ipu6ep_fw.bin
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: FW version: 20230102
Sep 06 01:13:35 zeniba kernel: intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
Sep 06 01:13:35 zeniba kernel: intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
Sep 06 01:13:35 zeniba kernel: intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
Sep 06 01:13:45 zeniba kernel: intel-ipu6-isys intel-ipu6-isys0: bind ov01a10 18-0036 nlanes is 1 port is 2
Sep 06 01:13:45 zeniba kernel: intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
Sep 06 08:14:57 zeniba kernel: intel-ipu6 0000:00:05.0: IPU in non-secure mode
Sep 06 08:14:57 zeniba kernel: intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
Sep 06 09:50:54 zeniba kernel: intel-ipu6 0000:00:05.0: IPU in non-secure mode
Sep 06 09:50:54 zeniba kernel: intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
Sep 06 10:14:52 zeniba kernel: intel-ipu6 0000:00:05.0: IPU in non-secure mode
Sep 06 10:14:52 zeniba kernel: intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
Sep 06 11:35:55 zeniba kernel: intel-ipu6 0000:00:05.0: IPU in non-secure mode
Sep 06 11:35:55 zeniba kernel: intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
Sep 06 13:04:43 zeniba kernel: intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
Sep 06 13:04:43 zeniba kernel: intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
Sep 06 13:04:43 zeniba kernel: intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
Sep 06 13:04:43 zeniba kernel: intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
Sep 06 13:04:43 zeniba kernel: intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
Sep 06 13:04:43 zeniba kernel: intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
Sep 06 13:04:43 zeniba kernel: intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
Sep 06 13:04:43 zeniba kernel: intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
```

These are the logs when running a bunch of different gstreamer commands:
```
~# sudo -E gst-launch-1.0 icamerasrc buffer-count=7 device-name=ov01a10-uf ! ximagesink                                        
[09-07 16:28:43.613] CamHAL[INF] load_camera_hal_library, the library name: /usr/lib/ipu_adl/libcamhal.so
Setting pipeline to PAUSED ...
[09-07 16:28:43.665] CamHAL[INF] aiqb file name ov01a10.aiqb
[09-07 16:28:43.666] CamHAL[INF] aiqb file name ov01a10.aiqb
[09-07 16:28:43.667] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[09-07 16:28:43.668] CamHAL[ERR] Open: Failed to open device node /dev/v4l-subdev16 No such device
[09-07 16:28:43.668] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev16 is not opened! No such device
Pipeline is live and does not need PREROLL ...
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Redistribute latency...
Execution ended after 0:00:00.000148672
Setting pipeline to NULL ...
[09-07 16:28:43.677] CamHAL[ERR] Close: Device node /dev/v4l-subdev16 is not opened! No such file or directory
[09-07 16:28:43.677] CamHAL[ERR] Close: Device node /dev/v4l-subdev16 is not opened! No such file or directory
Freeing pipeline ...
```

```
~# sudo -E gst-launch-1.0 icamerasrc buffer-count=7 device-name=ov01a10-uf ! ximagesink                                        
[09-07 16:28:43.613] CamHAL[INF] load_camera_hal_library, the library name: /usr/lib/ipu_adl/libcamhal.so
Setting pipeline to PAUSED ...
[09-07 16:28:43.665] CamHAL[INF] aiqb file name ov01a10.aiqb
[09-07 16:28:43.666] CamHAL[INF] aiqb file name ov01a10.aiqb
[09-07 16:28:43.667] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[09-07 16:28:43.668] CamHAL[ERR] Open: Failed to open device node /dev/v4l-subdev16 No such device
[09-07 16:28:43.668] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev16 is not opened! No such device
Pipeline is live and does not need PREROLL ...
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Redistribute latency...
Execution ended after 0:00:00.000148672
Setting pipeline to NULL ...
[09-07 16:28:43.677] CamHAL[ERR] Close: Device node /dev/v4l-subdev16 is not opened! No such file or directory
[09-07 16:28:43.677] CamHAL[ERR] Close: Device node /dev/v4l-subdev16 is not opened! No such file or directory
Freeing pipeline ...
```

Am I using the wrong gst-launch parameters? 

Thanks

Jorilx, 2023-11-02:
I'm facing the same problem, kernel version 6.4.4-3~bpo12+1

Jorilx, 2023-11-02:
Installed kernel version 6.5.3-1~bpo12+1, now the gst-launch call works!
I think this is because the 6.4.0 kernel packaged by Debian is missing the `CONFIG_VIDEO_V4L2_SUBDEV_API=y` flag

gabri94, 2023-11-03:
Thanks for the tip. I just updated the kernel to the latest version in debian testing, and now it works!
`6.5.0-3-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.5.8-1 (2023-10-22) x86_64 GNU/Linux`

closed 2023-11-03 by gabri94