intel/ipu6-drivers #182: Video not working on  redrix (HP Elite Dragonfly Chromebook)

ChocolateLoverRaj, 2023-09-13:
OS: Fedora 38
Kernel: `6.4.14-200.fc38.x86_64`
Output of `sudo -E gst-launch-1.0 icamerasrc device-name=ov13b10-wf af-mode=2 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink`:
```
Setting pipeline to PAUSED ...
[09-13 12:55:17.663] CamHAL[INF] aiqb file name ov13b10.aiqb
[09-13 12:55:17.663] CamHAL[ERR] there is no aiqb file:ov13b10
[09-13 12:55:17.663] CamHAL[INF] aiqb file name ov13b10.aiqb
[09-13 12:55:17.663] CamHAL[ERR] there is no aiqb file:ov13b10
[09-13 12:55:17.663] CamHAL[INF] aiqb file name ov8856.aiqb
[09-13 12:55:17.663] CamHAL[ERR] there is no aiqb file:ov8856
[09-13 12:55:17.663] CamHAL[INF] aiqb file name ov8856.aiqb
[09-13 12:55:17.663] CamHAL[ERR] there is no aiqb file:ov8856
[09-13 12:55:17.663] CamHAL[INF] aiqb file name ov01a10.aiqb
[09-13 12:55:17.663] CamHAL[INF] aiqb file name ov01a10.aiqb
[09-13 12:55:17.663] CamHAL[INF] aiqb file name ov01a10.aiqb
[09-13 12:55:17.663] CamHAL[INF] aiqb file name ov01a10.aiqb
[09-13 12:55:17.663] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[09-13 12:55:17.663] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[09-13 12:55:17.663] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[09-13 12:55:17.663] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[09-13 12:55:17.663] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[09-13 12:55:17.664] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[09-13 12:55:17.665] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[09-13 12:55:17.665] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[09-13 12:55:17.665] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[09-13 12:55:17.665] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[09-13 12:55:17.665] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[09-13 12:55:17.665] CamHAL[INF] aiqb file name ov01a1s.aiqb
[09-13 12:55:17.665] CamHAL[INF] aiqb file name ov01a1s.aiqb
[09-13 12:55:17.665] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[09-13 12:55:17.672] CamHAL[ERR] Get entity fail for calling getEntityById
[09-13 12:55:17.672] CamHAL[ERR] Get entity fail for calling getEntityById
[09-13 12:55:17.672] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[09-13 12:55:17.672] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[09-13 12:55:17.672] CamHAL[ERR] set up mediaCtl failed
[09-13 12:55:17.672] CamHAL[ERR] @configure Device Configure failed
[09-13 12:55:17.672] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.003726475
Setting pipeline to NULL ...
Freeing pipeline ...
```
Error in `dmesg`:
```
[18054.850533] intel-ipu6-isys intel-ipu6-isys0: Format not found in mapping table.
```
Output of `/sys/bus/i2c/devices`:
```
i2c-0  i2c-10  i2c-12  i2c-14  i2c-16  i2c-18  i2c-2  i2c-4  i2c-6  i2c-8  i2c-ELAN2513:00  i2c-GOOG0005:00  i2c-MX98390:00  i2c-MX98390:02  i2c-PRP0001:00
i2c-1  i2c-11  i2c-13  i2c-15  i2c-17  i2c-19  i2c-3  i2c-5  i2c-7  i2c-9  i2c-ELAN2703:00  i2c-INT3537:00   i2c-MX98390:01  i2c-MX98390:03  i2c-RTL5682:00
```

How I installed Intel ipu6 drivers:
```
sudo dnf update \
  --enablerepo=updates-testing \
  --enablerepo=rpmfusion-free-updates-testing \
  --enablerepo=rpmfusion-nonfree-updates-testing \
  'kernel*' '*v4l2loopback' -y

sudo dnf install \
  --enablerepo=updates-testing \
  --enablerepo=rpmfusion-free-updates-testing \
  --enablerepo=rpmfusion-nonfree-updates-testing \
  akmod-intel-ipu6 -y
  ```

hao-yao, 2023-11-08:
> i2c-INT3537:00

This is hi556 camera module. There seems no ov13b10 cam on your I2C devices list.

Maybe this comment is useful for you: https://github.com/intel/ipu6-drivers/commit/3f813580116cc66099fb26e9a7ccf7e8f72424ec#comments

jwrdegoede, 2024-01-21:
@ChocolateLoverRaj 

As mentioned in issue #202 there currently is an issue where the mainline version of the hi556 driver conflicts with the one from the kmod-intel-ipu6 package. A workaround for this is to run the following command:

```
sudo rm "/lib/modules/$(uname -r)/kernel/drivers/media/i2c/hi556.ko.xz"
```

after running the rm command reboot. After rebooting first run the rm command again, it should now fail with a "No such file or directory" error. If it does not fail you have rebooted into another kernel (likely due to a kernel update) in this case reboot again and run the rm command again until it does fail.

Once the rm command fails after a reboot, check dmesg for messages about the hi556 driver:

```
sudo dmesg | grep hi556
```

This should now no longer show any errors.

The next problem is that your gst-launch commandline is wrong, specifically the device-name parameter points to another sensor then you have.

Please try this commandline instead:

```
gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink
```

If that works, then you can try the v4l2relayd/v4l2loopback /dev/video0 compatibility by opening https://mozilla.github.io/webrtc-landing/gum_test.html in firefox. If the gst-launch command works, but firefox does not then try running:

```
sudo systemctl restart v4l2-relayd
```

and then try the gum_test page in firefox again.

ChocolateLoverRaj, 2024-01-21:
@jwrdegoede I no longer have that laptop so I cannot test it. I am currently using a laptop with a USB built in webcam so I don't really have anything to do with ipu anymore.

jwrdegoede, 2024-01-22:
> @jwrdegoede I no longer have that laptop so I cannot test it. I am currently using a laptop with a USB built in webcam so I don't really have anything to do with ipu anymore.

Ok, can you close this issue then please ?

ChocolateLoverRaj, 2024-03-06:
I got a redrix again, so I will be trying out https://github.com/intel/ipu6-drivers/issues/182#issuecomment-1902773307 again. I'll reopen the issue if it still doesn't fix the problem.

ChocolateLoverRaj, 2024-03-07:
@jwrdegoede I followed your instructions in https://github.com/intel/ipu6-drivers/issues/182#issuecomment-1902773307 and it's still not working. Here is the output of the relevant commands (I did reboot after removing the kernel's file):
```
rajas@Redrix:~$ fastfetch 
             .',;::::;,'.                 rajas@Redrix
         .';:cccccccccccc:;,.             ------------
      .;cccccccccccccccccccccc;.          OS: Fedora Linux 39 (KDE Plasma) x86_64
    .:cccccccccccccccccccccccccc:.        Host: Redrix (rev3)
  .;ccccccccccccc;.:dddl:.;ccccccc;.      Kernel: 6.7.7-200.fc39.x86_64
 .:ccccccccccccc;OWMKOOXMWd;ccccccc:.     Uptime: 3 mins
.:ccccccccccccc;KMMc;cc;xMMc;ccccccc:.    Packages: 2171 (rpm), 9 (flatpak)
,cccccccccccccc;MMM.;cc;;WW:;cccccccc,    Shell: bash 5.2.26
:cccccccccccccc;MMM.;cccccccccccccccc:    Display (BOE0A76): 2256x1504 @ 60Hz (as 1128x752) [Built-in]
:ccccccc;oxOOOo;MMM000k.;cccccccccccc:    DE: KDE Plasma 5.27.10
cccccc;0MMKxdd:;MMMkddc.;cccccccccccc;    WM: KWin (Wayland)
ccccc;XMO';cccc;MMM.;cccccccccccccccc'    WM Theme: Breeze
ccccc;MMo;ccccc;MMW.;ccccccccccccccc;     Theme: Breeze (Dark) [QT], Breeze [GTK3]
ccccc;0MNc.ccc.xMMd;ccccccccccccccc;      Icons: breeze-dark [QT], breeze-dark [GTK3/4]
cccccc;dNMWXXXWM0:;cccccccccccccc:,       Font: Noto Sans (10pt) [QT], Noto Sans (10pt) [GTK3/4]
cccccccc;.:odl:.;cccccccccccccc:,.        Cursor: breeze (24px)
ccccccccccccccccccccccccccccc:'.          Terminal: konsole 23.8.5
:ccccccccccccccccccccccc:;,..             CPU: 12th Gen Intel(R) Core(TM) i7-1265U (12) @ 4.80 GHz
 ':cccccccccccccccc::;,.                  GPU: Intel Iris Xe Graphics
                                          Memory: 3.17 GiB / 31.19 GiB (10%)
                                          Swap: 0 B / 8.00 GiB (0%)
                                          Disk (/): 12.31 GiB / 245.02 GiB (5%) - ext4
                                          Battery: 94% [Charging]
                                          Locale: en_US.UTF-8

                                          ████████████████████████
                                          ████████████████████████
rajas@Redrix:~$ sudo rm "/lib/modules/$(uname -r)/kernel/drivers/media/i2c/hi556.ko.xz"
[sudo] password for rajas: 
rm: cannot remove '/lib/modules/6.7.7-200.fc39.x86_64/kernel/drivers/media/i2c/hi556.ko.xz': No such file or directory
rajas@Redrix:~$ sudo dmesg | grep hi556
[    6.365329] Modules linked in: kvm_intel(+) intel_ipu6_isys(OE+) snd_soc_acpi_intel_match videobuf2_dma_contig videobuf2_memops mac80211 cros_ec_sensorhub snd_soc_acpi cros_ec_sysfs videobuf2_v4l2 kvm soundwire_generic_allocation soundwire_bus spi_nor cros_usbpd_logger videobuf2_common cros_usbpd_charger cros_peripheral_charger cros_ec_chardev vfat pmt_telemetry libarc4 mtd mei_hdcp mei_pxp intel_rapl_msr pmt_class cros_ec_dev fat intel_ipu6_psys(OE) irqbypass snd_soc_rt5682s snd_soc_max98390 snd_hda_intel rapl btusb iwlwifi snd_intel_dspcfg snd_soc_core dm_thin_pool snd_intel_sdw_acpi intel_cstate btrtl snd_hda_codec btintel dm_persistent_data btbcm dm_bio_prison btmtk intel_uncore snd_compress processor_thermal_device_pci ac97_bus pcspkr cfg80211 processor_thermal_device spi_intel_pci snd_hda_core bluetooth processor_thermal_wt_hint snd_pcm_dmaengine snd_hwdep spi_intel processor_thermal_rfim snd_seq processor_thermal_rapl cros_ec_typec snd_seq_device hi556(OE) mei_csi(OE) snd_pcm intel_rapl_common mei_ace(OE)
rajas@Redrix:~$ gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
[03-06 16:55:31.10] CamHAL[INF] aiqb file name ov13b10.aiqb
[03-06 16:55:31.10] CamHAL[ERR] there is no aiqb file:ov13b10
[03-06 16:55:31.10] CamHAL[INF] aiqb file name ov13b10.aiqb
[03-06 16:55:31.10] CamHAL[ERR] there is no aiqb file:ov13b10
[03-06 16:55:31.10] CamHAL[INF] aiqb file name ov8856.aiqb
[03-06 16:55:31.10] CamHAL[ERR] there is no aiqb file:ov8856
[03-06 16:55:31.10] CamHAL[INF] aiqb file name ov8856.aiqb
[03-06 16:55:31.10] CamHAL[ERR] there is no aiqb file:ov8856
[03-06 16:55:31.10] CamHAL[INF] aiqb file name ov01a10.aiqb
[03-06 16:55:31.10] CamHAL[INF] aiqb file name ov01a10.aiqb
[03-06 16:55:31.10] CamHAL[INF] aiqb file name ov01a10.aiqb
[03-06 16:55:31.10] CamHAL[INF] aiqb file name ov01a10.aiqb
[03-06 16:55:31.10] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[03-06 16:55:31.10] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[03-06 16:55:31.10] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[03-06 16:55:31.11] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[... cut ...]
[03-06 16:55:45.16] CamHAL[WAR] wait event time out, 0 requests processing, 0 requests in HAL
[03-06 16:55:47.16] CamHAL[WAR] wait event time out, 0 requests processing, 0 requests in HAL
^C
rajas@Redrix:~$

```
The Camera does show up on Firefox but the camera light doesn't turn on and it's just a black rectangle (or maybe some other error).
[image]

Also I suspect that installing these ipu6 drivers stopped suspend and shutdown from working. It could be caused by something else though, I'm not sure.

ChenYuYeh, 2024-03-07:
there are multiple camera used on Redrix as far as I know. 
can you share the kernel log when you book up the system under ChromeOS?
And due to Google is using coreboot instead of UEFI BIOS.  It is interesting to know whether Google will provide technical support to port Linux on Chromebook.

ChocolateLoverRaj, 2024-03-07:
> there are multiple camera used on Redrix as far as I know.

@ChenYuYeh my Chromebook has the `hi556` camera. 

> can you share the kernel log when you book up the system under ChromeOS?

These are the Chrome os logs: [messages.txt](https://github.com/intel/ipu6-drivers/files/14519253/messages.txt). Let me know if these were the wrong kind of logs.

> And due to Google is using coreboot instead of UEFI BIOS.

I am booting Linux from from altfw, aka RW_LEGACY. It boots Coreboot with edk2. [More info](https://doc.coreboot.org/getting_started/faq.html?highlight=faq#what-s-the-difference-between-coreboot-and-uefi). I'm pretty sure it's technically correct to say that I boot fedora with UEFI.

> It is interesting to know whether Google will provide technical support to port Linux on Chromebook.

As far as I know, Google doesn't provide technical support for using anything other than ChromeOS on Chromebooks. Some Chromebook developers sometimes help out the Chromebook modding community, but there is no "support" from Google.

Btw, on Fedora on this Chromebook, the camera does work when using the [open source MIPI stack](https://hansdegoede.dreamwidth.org/28153.html) instead of this proprietary one, except it has a beige tint to it.

nickcipriani, 2024-06-11:
Did you get this working? I am having the same no camera issue on my HP Elite Dragonfly Chromebook.

ChocolateLoverRaj, 2024-06-11:
@nickcipriani it works with the open source MIPI stack. I never got it to work with the Intel way.