intel/ipu6-drivers #187: Sensor never starts streaming data 70% of the time on X1 Carbon Gen 11

lukemarsden, 2023-10-04:
I have an X1 Carbon Gen 11 21HMCTO1WW with an ov2740 sensor.

I am running the latest 6.5.5 kernel with the Fedora driver stack from https://hansdegoede.livejournal.com/27276.html
```
Linux fedora 6.5.5-200.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Sep 24 15:52:44 UTC 2023 x86_64 GNU/Linux
```

Very often the stream just fails to start streaming. I can run `sudo gst-launch-1.0 icamerasrc ! autovideosink` 10 times in a row and on average, I will only get a video stream 3/10 times.

`dmesg -w` while doing this shows the following pattern:

[image]

Here you can see when it worked (on, off with no red errors), and when it didn't (timeouts), from `dmesg -w`.

Here's the output from running sudo gst-launch-1.0 icamerasrc ! autovideosink > gst-1.txt 2>&1:
https://gist.github.com/lukemarsden/d51e5f438d72f3211c402ea4e9af7cf6

And the journal from that time: sudo journalctl -f > journal-1.txt 2>&1:
https://gist.github.com/lukemarsden/a5022bebdadaa5ae34aab822aff636b9

The only log lines that seem interesting are the timeouts from gstreamer and the:
> Oct 04 17:46:12 fedora kernel: intel-ipu6-isys intel-ipu6-isys0: stream stop time out
line from the ipu6 driver. That "stream stop time out" only happens _after_ I hit ctrl+c (twice) on the gstreamer stream because I'm seeing a black video feed.

dmesg |grep ipu6:
https://gist.github.com/lukemarsden/01beb5c00d37e57f71706922aa713832

I followed the instructions at https://hansdegoede.livejournal.com/27276.html

Versions:
```
akmod-intel-ipu6.x86_64                                                                  0.0-7.20230622git8e41080.fc38                                    @rpmfusion-nonfree-updates                    
ipu6-camera-bins.x86_64                                                                  0.0-7.20230208git276859f.fc38                                    @rpmfusion-nonfree-updates                    
ipu6-camera-bins-firmware.x86_64                                                         0.0-7.20230208git276859f.fc38                                    @rpmfusion-nonfree-updates                    
ipu6-camera-hal.x86_64                                                                   0.0-14.20230208git884b81a.fc38                                   @rpmfusion-nonfree-updates                    
kmod-intel-ipu6-6.4.15-200.fc38.x86_64.x86_64                                            0.0-7.20230622git8e41080.fc38                                    @@commandline                                 
kmod-intel-ipu6-6.5.5-200.fc38.x86_64.x86_64                                             0.0-7.20230622git8e41080.fc38                                    @@commandline                                 
kmod-intel-ipu6-6.5.5-200.fc38.x86_64+debug.x86_64                                       0.0-7.20230622git8e41080.fc38                                    @@commandline                                 
ipu6-camera-bins-devel.x86_64                                                            0.0-7.20230208git276859f.fc38                                    rpmfusion-nonfree-updates                     
ipu6-camera-hal-devel.x86_64                                                             0.0-14.20230208git884b81a.fc38                                   rpmfusion-nonfree-updates                     
kmod-intel-ipu6.x86_64                                                                   0.0-7.20230622git8e41080.fc38                                    rpmfusion-nonfree-updates                     
```

omriasta, 2023-10-16:
+1 on this, same symptoms x1 carbon Gen 10 i7-1270P Let me know if you need me to add any other info about my hardware.

jwrdegoede, 2023-11-07:
@lukemarsden so I was doing some IPU6 work today and this made me think back to this bug. The ov2740 driver has 2 sets of register settings one for Lenovo devices and one for other devices.

Which set of register settings to use is decided by making a special ACPI call which gets a camera-sensor-module id string. I wonder if the X1 Gen 10 / Gen 11 maybe has a different sensor-module (different sensor module ID) causing the ov2740 to use the non Lenovo register settings which in turn then causes this problem.

Here is an updated akmod with a patch to try this:
[akmod-intel-ipu6-0.0-11.20230622git8e41080.fc38.x86_64.rpm.gz](https://github.com/intel/ipu6-drivers/files/13279525/akmod-intel-ipu6-0.0-11.20230622git8e41080.fc38.x86_64.rpm.gz)

After downloading run the following commands to install it:

```
gzip -d akmod-intel-ipu6-0.0-11.20230622git8e41080.fc38.x86_64.rpm.gz
sudo rpm -Uvh akmod-intel-ipu6-0.0-11.20230622git8e41080.fc38.x86_64.rpm
```

After this wait for the new kmod to be build and then run, e.g.:

rpm -qa | grep kmod-intel-ipu6

This should show the kmod-intel-ipu6 release field for your currently running kernel is now `11.20230622git8e41080` this will be part of the full string for the package.

After this reboot and do:

dmesg | grep "Sensor module id"

This should now show your sensor module id. If your sensor-module id is `CJFLE23` then this patched akmod pkg will not make a difference.  But if it is a different id, then this patch will hopefully help to make the camera work better.

lukemarsden, 2023-11-07:
Thanks for thinking of this bug!

Unfortunately:
[    6.854496] ov2740 i2c-INT3474:01: Sensor module id: 'CJFLE23'

Any other thoughts about what might be causing this? Happy to run commands
and report back :-)

Cheers,
Luke

jwrdegoede, 2023-11-13:
> Any other thoughts about what might be causing this?

Sorry, but I don't have any other ideas how to fix this.

hao-yao, 2023-11-15:
@lukemarsden  Could you try adding "options intel-ipu6-isys dyndbg=+p" to /etc/modprobe.d/intel-ipu6.conf then restart to check dmesg log again? We can check if IPU received any frames.

lukemarsden, 2023-11-18:
Thank you @hao-yao 

Here's a video of what happens when I set that option, reboot, then run `dmesg -w` in one terminal and `sudo gst-launch-1.0 icamerasrc ! autovideosink` in another

https://github.com/intel/ipu6-drivers/assets/264658/ee14514a-ace8-454f-951d-a735efe57f32

I've also uploaded it to YouTube here: https://www.youtube.com/watch?v=OQpBTUpM4sk

lukemarsden, 2023-11-18:
> By the way, the settings used for module CJFLE23 should be 180M link frequency and I fixed that on master branch. It might helps but need a try.

@jwrdegoede is this change something you might be able to conveniently package for me to try please? :smile:

hao-yao, 2023-11-20:
> I've also uploaded it to YouTube here: https://www.youtube.com/watch?v=OQpBTUpM4sk

@lukemarsden This shows that IPU didn't receive CSI-2 packages. My experience is that the issue should between camera sensor module and IPU CSI-2 port, either register settings or link frequency wrong, or they don't match.

jwrdegoede, 2023-11-23:
@lukemarsden

> > By the way, the settings used for module CJFLE23 should be 180M link frequency and I fixed that on master branch. It might helps but need a try.
> 
> @jwrdegoede is this change something you might be able to conveniently package for me to try please? 😄

Here is an updated akmod:

https://fedorapeople.org/~jwrdegoede/akmod-intel-ipu6-0.0-11.20231115git067270f.fc39.x86_64.rpm

Download it and then run `sudo rpm -Uvh akmod-intel-ipu6-....rpm` after this wait for the akmod to be build for your currently running kernel (e.g. check top) and then reboot.

Hopefully this will resolve the X1 Carbon gen 10 / gen 11 issues.

lukemarsden, 2023-11-23:
Thanks! Unfortunately I'm now getting this:
```
luke@fixed:~$ sudo gst-launch-1.0 icamerasrc ! autovideosink
Setting pipeline to PAUSED ...
[11-23 17:02:46.291] CamHAL[INF] aiqb file name ov13b10.aiqb
[11-23 17:02:46.291] CamHAL[ERR] there is no aiqb file:ov13b10
[11-23 17:02:46.291] CamHAL[INF] aiqb file name ov13b10.aiqb
[11-23 17:02:46.291] CamHAL[ERR] there is no aiqb file:ov13b10
[11-23 17:02:46.291] CamHAL[INF] aiqb file name ov8856.aiqb
[11-23 17:02:46.291] CamHAL[ERR] there is no aiqb file:ov8856
[11-23 17:02:46.291] CamHAL[INF] aiqb file name ov8856.aiqb
[11-23 17:02:46.291] CamHAL[ERR] there is no aiqb file:ov8856
[11-23 17:02:46.291] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-23 17:02:46.291] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-23 17:02:46.291] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-23 17:02:46.291] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-23 17:02:46.291] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[11-23 17:02:46.291] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[11-23 17:02:46.291] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[11-23 17:02:46.292] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[11-23 17:02:46.293] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[11-23 17:02:46.293] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[11-23 17:02:46.293] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[11-23 17:02:46.293] CamHAL[INF] aiqb file name ov01a1s.aiqb
[11-23 17:02:46.293] CamHAL[INF] aiqb file name ov01a1s.aiqb
[11-23 17:02:46.293] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[11-23 17:02:46.299] CamHAL[ERR] Get entity fail for calling getEntityById
[11-23 17:02:46.299] CamHAL[ERR] Get entity fail for calling getEntityById
[11-23 17:02:46.299] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[11-23 17:02:46.299] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[11-23 17:02:46.299] CamHAL[ERR] set up mediaCtl failed
[11-23 17:02:46.299] CamHAL[ERR] @configure Device Configure failed
[11-23 17:02:46.299] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.004130346
Setting pipeline to NULL ...
Freeing pipeline ...
```

Is it because I'm still on Fedora 38? I noticed the rpm was fc39.

dmesg: https://gist.github.com/lukemarsden/7fa4f4e720c01a1452a604fc0a00b08c

Linux fixed 6.5.10-200.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Nov  2 19:59:55 UTC 2023 x86_64 GNU/Linux

lukemarsden, 2023-11-23:
Oh, after rebooting twice, it's now working!

At least, it's working MORE reliably than it was before, I think, but still not 100%...

lukemarsden, 2023-11-23:
Running `sudo gst-launch-1.0 icamerasrc ! autovideosink` 11 times:

1 - worked straight away
2 - black screen, LED goes on, `[11-23 17:10:23.715] CamHAL[ERR] Poll: Device node fd 14 poll timeout.`
3 - black screen, LED goes on, `[11-23 17:10:23.715] CamHAL[ERR] Poll: Device node fd 14 poll timeout.`
4 - black screen, LED goes on, `[11-23 17:10:23.715] CamHAL[ERR] Poll: Device node fd 14 poll timeout.`
5 - black screen, LED goes on, `[11-23 17:10:23.715] CamHAL[ERR] Poll: Device node fd 14 poll timeout.`
6 - black screen, LED goes on, `[11-23 17:10:23.715] CamHAL[ERR] Poll: Device node fd 14 poll timeout.`
7 - worked straight away
8 - black screen, LED goes on, `[11-23 17:10:23.715] CamHAL[ERR] Poll: Device node fd 14 poll timeout.`
9 - black screen, LED goes on, `[11-23 17:10:23.715] CamHAL[ERR] Poll: Device node fd 14 poll timeout.`
10 - black screen, LED goes on, `[11-23 17:10:23.715] CamHAL[ERR] Poll: Device node fd 14 poll timeout.`
11 - worked straight away

So yeah we're still at about 30% - I'd say this change unfortunately hasn't made a difference.

Any other logs etc I can share?

lukemarsden, 2023-11-23:
Some more dmesg after all those tries:

https://gist.github.com/lukemarsden/f8c21affa6f0a0ffd5e3d6550208583f

jwrdegoede, 2023-11-23:
That is unfortunate, can you run:

```
rpm -qf "/lib/modules/$(uname -r)/extra/intel-ipu6/drivers/media/i2c/ov2740.ko.xz"
```

And check the printed kmod version ends with "0.0-11.20231115git067270f.fc39" ?

This checks that you are indeed running the new module.

lukemarsden, 2023-11-26:
Yup,
```
kmod-intel-ipu6-6.5.10-200.fc38.x86_64-0.0-11.20231115git067270f.fc38.x86_64
```

lukemarsden, 2023-11-26:
I tested upgrading my X1 Carbon Gen 11 to Fedora 39 and even upgraded to 6.6.2-201.fc39.x86_64 kernel and still getting the same symptoms.

Strangely enough, my X1 Yoga Gen 8 running Fedora 38 seems to have regressed to the same behavior described in this issue - I had the kernel pinned to 6.5.5 but the userland was upgrading. I wonder if something in the userland has changed?

jwrdegoede, 2023-11-27:
> Strangely enough, my X1 Yoga Gen 8 running Fedora 38 seems to have regressed to the same behavior described in this issue - I had the kernel pinned to 6.5.5 but the userland was upgrading. I wonder if something in the userland has changed?

Hmm, I have a X1 Yoga Gen 8 myself and I just fully updated the F39 on it and then did:

gst-launch-1.0 v4l2src ! xvimagesink

10 times and the stream started correctly 10 / 10 times.

So I wonder is anything special on your setup(s) ?

Are you testing with a charger or thunderbolt dock connected maybe?

lukemarsden, 2023-11-27:
Interesting. No charger or dock plugged in. What kernel are you running?
What sensor, and what does `sudo dmidecode -s system-product-name` return
out of interest?

jwrdegoede, 2023-11-27:
kernel: 6.5.8-300.fc39.x86_64
dmidecode: 21HQSIT025

jwrdegoede, 2023-11-27:
I'm using the new akmod which I linked to above:

kmod-intel-ipu6-6.5.8-300.fc39.x86_64-0.0-11.20231115git067270f.fc39.x86_64

jwrdegoede, 2023-11-27:
Are you perhaps using the laptop(s) near some electrical appliance which may be giving of electro-magnetic interference?  Have you tried using the laptop in some other place, e.g. another room ?

lukemarsden, 2023-11-27:
Interesting, my dmidecode is 21HQCTO1WW on the Yoga. I wonder what the
difference entails...

lukemarsden, 2023-11-27:
Yeah I've tested it in various different locations. It also works fine on
Windows.

lukemarsden, 2023-11-27:
Super weird - after rebooting the X1 Yoga Gen 8 to install some updates,
the camera's now working 100% of the time.

The X1 Carbon Gen 11 this issue initially relates to is still exhibiting
the described issues though.

lukemarsden, 2023-11-29:
Just for the record, back on the X1 Yoga Gen 8 (ov2740), running:
```
kmod-intel-ipu6-6.5.5-200.fc38.x86_64-0.0-10.20230622git8e41080.fc38.x86_64
```
I am getting <30% success rate on launching the camera again today, even after rebooting.

lukemarsden, 2023-11-29:
Same issue with kernel 6.6.2-101.fc38.x86_64 and 0.0-10.20230622git8e41080.fc38.x86_64
Trying 0.0-11.20231115git067270f.fc38.x86_64 now...

lukemarsden, 2023-11-29:
On 6.6.2, the newer rpm you shared above fails every time now (after three reboots) the same way:
```
luke@dynamic:~$ sudo gst-launch-1.0 icamerasrc ! autovideosink
Setting pipeline to PAUSED ...
[11-29 12:32:00.447] CamHAL[INF] aiqb file name ov13b10.aiqb
[11-29 12:32:00.447] CamHAL[ERR] there is no aiqb file:ov13b10
[11-29 12:32:00.447] CamHAL[INF] aiqb file name ov13b10.aiqb
[11-29 12:32:00.447] CamHAL[ERR] there is no aiqb file:ov13b10
[11-29 12:32:00.447] CamHAL[INF] aiqb file name ov8856.aiqb
[11-29 12:32:00.447] CamHAL[ERR] there is no aiqb file:ov8856
[11-29 12:32:00.447] CamHAL[INF] aiqb file name ov8856.aiqb
[11-29 12:32:00.447] CamHAL[ERR] there is no aiqb file:ov8856
[11-29 12:32:00.447] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-29 12:32:00.447] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-29 12:32:00.447] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-29 12:32:00.447] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-29 12:32:00.447] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[11-29 12:32:00.447] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[11-29 12:32:00.447] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[11-29 12:32:00.447] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[11-29 12:32:00.447] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[11-29 12:32:00.448] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[11-29 12:32:00.449] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[11-29 12:32:00.449] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[11-29 12:32:00.449] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[11-29 12:32:00.449] CamHAL[INF] aiqb file name ov01a1s.aiqb
[11-29 12:32:00.449] CamHAL[INF] aiqb file name ov01a1s.aiqb
[11-29 12:32:00.449] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[11-29 12:32:00.458] CamHAL[ERR] Get entity fail for calling getEntityById
[11-29 12:32:00.458] CamHAL[ERR] Get entity fail for calling getEntityById
[11-29 12:32:00.458] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[11-29 12:32:00.458] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[11-29 12:32:00.458] CamHAL[ERR] set up mediaCtl failed
[11-29 12:32:00.458] CamHAL[ERR] @configure Device Configure failed
[11-29 12:32:00.458] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.006819703
Setting pipeline to NULL ...
Freeing pipeline ...
```

wille, 2023-12-20:
Seeing same problems but I'm running arch on 6.6.7

SmokinCaterpillar, 2024-01-19:
I think I have a similar issue: https://github.com/intel/ipu6-drivers/issues/204

lukemarsden, 2024-01-20:
Yeah it's frustrating isn't it. My X1 Yoga Gen 8 that I bought because the
ipu6 webcam seemed to work more reliably on Linux than the X1 Carbon Gen 11
(both UK versions) isn't working either now, on Fedora, Ubuntu or Manjaro
(Arch) - they are both just expensive paperweights at this point and I've
gone back to using my old X1CG9

lukemarsden.net
***@***.***
@lmarsden <https://twitter.com/lmarsden>

lukemarsden, 2024-01-21:
OMG, on a hunch, installing latest nightly Ubuntu 24.04 https://cdimage.ubuntu.com/daily-live/current/ from today, then following the instructions at https://bugs.launchpad.net/ubuntu/+source/ipu6-drivers/+bug/2044991 (I know, different machine) has brought my X1 Yoga Gen 8 webcam back to life.

```
Linux fresh 6.6.0-14-generic #14-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 30 10:27:29 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

lukemarsden, 2024-01-21:
The same steps on my X1 Carbon Gen 11 result in a sensor that still works about 30% of the time, like the original title & description of this issue.

lukemarsden, 2024-01-21:
Hmm, the webcam on the Yoga on some boots works 100% of the time, on other boots, only 30% or so. What could cause this per-boot flakiness?

lukemarsden, 2024-01-21:
Actually it seems that it's per `sudo systemctl restart v4l2-relayd.service`. So if the webcam's being flaky, restarting v4l2-relayd once or twice will normally be sufficient to nudge the system into reliable webcam.

Borderline usable, will start using this machine again.

SmokinCaterpillar, 2024-01-23:
Honestly I bought an external webcam, and now I have to carry this clunky thing around. :sob:

anil-e, 2024-01-25:
Wow, actually `sudo systemctl restart v4l2-relayd.service` seems to help me at the moment, on TP X1 Gen 11, Fedora 39, Kernel 6.6.13.

I usually have to try it more than 1-2 times, but after a few more attempts I actually get an image.
Finally at least some kind of quick fix that I can use when i'm out of the house or at the office and need my camera. Many thanks @lukemarsden

jwrdegoede, 2024-01-27:
@lukemarsden I have just announced an alternative IPU6 stack for Fedora which does not use v4l2-relayd.service but instead relies on the pipewire camera support in the latest Firefox release, see:
https://hansdegoede.dreamwidth.org/28153.html

It would be interesting to see if this also suffers from the issue of the stream often not starting properly on your X1 Carbon Gen 11. Please give this a try when you have some time.

kupietz, 2024-01-27:
> @lukemarsden I have just announced an alternative IPU6 stack for Fedora which does not use v4l2-relayd.service but instead relies on the pipewire camera support in the latest Firefox release, see: [...]

I followed the installation steps here https://copr.fedorainfracloud.org/coprs/jwrdegoede/ipu6-softisp/ on my TP X1 Gen 11 with Fedora 39, but there is no camera I can select with qcam and also no camera on the gUM test page with Firefox (`NotFoundError: The object can not be found here.`) – also after rebooting.

Please let me know, how I can help debugging.

jwrdegoede, 2024-01-27:
> I followed the installation steps here https://copr.fedorainfracloud.org/coprs/jwrdegoede/ipu6-softisp/ on my TP X1 Gen 11 with Fedora 39, but there is no camera I can select with qcam and also no [...]

What is the output of the following commands:

1. `dmesg | grep -E 'ov2740|ipu6'`
2. `uname -a`
3. `qcam`

when run from a terminal?

kupietz, 2024-01-27:
>     1. `dmesg | grep -E 'ov2740|ipu6'`
>
```
[    0.000000] Linux version 6.7.1-200.1.ipu6.fc39.x86_64 (mockbuild@ec318dcebdd54443ad88e7a1e85dfd0a) (gcc (GCC) 13.2.1 20231205 (Red Hat 13.2.1-6), GNU ld version 2.40-13.fc39) #1 SMP PREEMPT_DYNAMIC Wed Jan 24 21:55:31 UTC 2024
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt6)/vmlinuz-6.7.1-200.1.ipu6.fc39.x86_64 root=UUID=47740b5c-b8f4-4e4e-ad07-337b86ea4c7d ro rootflags=subvol=root rhgb quiet
[    0.065113] Kernel command line: BOOT_IMAGE=(hd0,gpt6)/vmlinuz-6.7.1-200.1.ipu6.fc39.x86_64 root=UUID=47740b5c-b8f4-4e4e-ad07-337b86ea4c7d ro rootflags=subvol=root rhgb quiet
[    0.065177] Unknown kernel command line parameters "rhgb BOOT_IMAGE=(hd0,gpt6)/vmlinuz-6.7.1-200.1.ipu6.fc39.x86_64", will be passed to user space.
[    1.971435] usb usb1: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    1.972083] usb usb2: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    1.976328] usb usb3: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    1.978783] usb usb4: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    2.237790]     BOOT_IMAGE=(hd0,gpt6)/vmlinuz-6.7.1-200.1.ipu6.fc39.x86_64
[    7.431124] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    7.431758] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    7.431771] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    7.431773] intel-ipu6 0000:00:05.0: mapped as: 0x0000000014c24a7c
[    7.432217] intel-ipu6 0000:00:05.0: IPU in secure mode
[    7.432220] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    7.432221] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    7.449642] intel-ipu6 0000:00:05.0: IPC reset done
[    7.449645] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    7.452558] intel-ipu6 0000:00:05.0: FW version: 20220510
[    7.460471] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    7.460490] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    7.460492] intel-ipu6 0000:00:05.0: mapped as: 0x0000000014c24a7c
[    7.460533] intel-ipu6 0000:00:05.0: IPU in secure mode
[    7.460535] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    7.460537] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    7.463674] intel-ipu6 0000:00:05.0: IPC reset done
[    7.463679] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    7.464153] intel-ipu6 0000:00:05.0: FW version: 20220510
[    7.490167] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    7.490200] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    7.490202] intel-ipu6 0000:00:05.0: mapped as: 0x0000000014c24a7c
[    7.490961] intel-ipu6 0000:00:05.0: IPU in secure mode
[    7.490963] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    7.490965] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    7.493034] intel-ipu6 0000:00:05.0: IPC reset done
[    7.493037] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    7.493474] intel-ipu6 0000:00:05.0: FW version: 20220510
[    7.518600] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    7.518629] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    7.518631] intel-ipu6 0000:00:05.0: mapped as: 0x0000000014c24a7c
[    7.518684] intel-ipu6 0000:00:05.0: IPU in secure mode
[    7.518687] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    7.518690] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    7.522335] intel-ipu6 0000:00:05.0: IPC reset done
[    7.522340] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    7.522795] intel-ipu6 0000:00:05.0: FW version: 20220510
[    7.541921] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    7.541945] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    7.541947] intel-ipu6 0000:00:05.0: mapped as: 0x0000000014c24a7c
[    7.541995] intel-ipu6 0000:00:05.0: IPU in secure mode
[    7.541998] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    7.542000] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    7.544393] intel-ipu6 0000:00:05.0: IPC reset done
[    7.544397] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    7.544847] intel-ipu6 0000:00:05.0: FW version: 20220510
[    7.561224] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    7.561251] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    7.561253] intel-ipu6 0000:00:05.0: mapped as: 0x0000000014c24a7c
[    7.561298] intel-ipu6 0000:00:05.0: IPU in secure mode
[    7.561300] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    7.561301] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    7.563826] intel-ipu6 0000:00:05.0: IPC reset done
[    7.563828] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    7.564190] intel-ipu6 0000:00:05.0: FW version: 20220510
[    7.692664] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    7.692686] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    7.692688] intel-ipu6 0000:00:05.0: mapped as: 0x0000000014c24a7c
[... 137 lines cut ...]
[   47.234548] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[   47.253564] intel-ipu6 0000:00:05.0: IPC reset done
[   47.253572] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[   47.254656] intel-ipu6 0000:00:05.0: FW version: 20220510
[  284.294394] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[  284.294550] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[  284.294552] intel-ipu6 0000:00:05.0: mapped as: 0x00000000af698b65
[... cut ...]
[  284.294823] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[  284.314597] intel-ipu6 0000:00:05.0: IPC reset done
[  284.314608] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[  284.315911] intel-ipu6 0000:00:05.0: FW version: 20220510
```
>     2. `uname -a`

``` 
Linux nbmk 6.7.1-200.1.ipu6.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Jan 24 21:55:31 UTC 2024 x86_64 GNU/Linux
```
>     3. `qcam`

```
[0:06:48.744222625] [8542]  INFO Camera camera_manager.cpp:284 libcamera v0.0.0+3-4ea14140
```
[image]

kupietz, 2024-01-28:
The actual error message from dmesg is:

```
int3472-discrete INT3472:05: cannot find GPIO chip INTC1096:00, deferring
```

jwrdegoede, 2024-01-28:
@kupietz that is a somewhat normal message, but I agree that it might be related.

Can you please reboot (to ensure the dmesg ringbuffer has not overflowed) and then collect a full dmesg: `sudo dmesg > dmesg.txt` and attach dmesg.txt here?

Also please run: `sudo cat /sys/kernel/debug/gpio` and provide the output of that command.

kupietz, 2024-01-28:
> @kupietz that is a somewhat normal message, but I agree that it might be related.
> 
> Can you please reboot (to ensure the dmesg ringbuffer has not overflowed) and then collect a full dmesg: `sudo dmesg > dmesg.txt` and attach dmesg.txt here?

OK, attached.
> 
> Also please run: `sudo cat /sys/kernel/debug/gpio` and provide the output of that command.

```
gpiochip0: GPIOs 512-871, parent: platform/INTC1055:00, INTC1055:00:
 gpio-834 (                    |firmware            ) out lo 
 gpio-837 (                    |enable              ) out lo 
```
[dmesg.txt](https://github.com/intel/ipu6-drivers/files/14077634/dmesg.txt)

jwrdegoede, 2024-01-28:
@kupietz ok this seems to be the culprit:

```
[    8.985068] ljca: exports duplicate symbol ljca_register_event_cb (owned by usb_ljca)
```

I think that if you run:

```
ls /lib/modules/6.7.1-200.1.ipu6.fc39.x86_64/{extra,updates,weak-updates}
```

You will find that something has installed a copy of the out of tree intel-ipu6 drivers in one of those dirs. Maybe you installed the proprietary drivers with dkms or something?  The intel-ipu6 akmod should not have been build since the 6.7.1-200.1.ipu6.fc39.x86_64 kernel already comes with an `intel-ipu6` module.

Either way please try moving (`sudo mv ...`) any copy of the out of tree intel-ipu6 drivers to some place outside `/lib/modules/6.7.1-200.1.ipu6.fc39.x86_64/` and then run `sudo depmod -a` and reboot.

Hopefully after that `sudo cat /sys/kernel/debug/gpio` will show the INTC1096:00 gpio chip and things will work.

kupietz, 2024-01-28:
Thanks, that did the trick! After all the trials and errors, I had the modules installed via akmod and also dkms, sorry.

qcam and gUM do work now. Please let me know how else I can help.

jwrdegoede, 2024-01-29:
> qcam and gUM do work now. Please let me know how else I can help.

This issue is about the camera often (7/10 attempts) failing to start streaming. Can you test if this still happens with the libcamera software ISP stack? E.g. start qcam 10 times and check how often the camera successfully streams out of those 10 times.

kupietz, 2024-01-29:
> This issue is about the camera often (7/10 attempts) failing to start streaming. Can you test if this still happens with the libcamera software ISP stack? E.g. start qcam 10 times and check how [...]

Unfortunately, no progress concerning reliability: `for i in $(seq 1 10); do echo $i; qcam; done`  yields successful camera streams in ~4 of 10 cases. The camera light turns on each time.

Messages for the unsuccessful case:
```
[11:10:39.099218500] [85506]  INFO Camera camera_manager.cpp:284 libcamera v0.0.0+3-4ea14140
[11:10:39.119228317] [85518]  WARN CameraSensor camera_sensor.cpp:251 'ov2740 18-0036': Recommended V4L2 control 0x009a0922 not supported
[11:10:39.119276416] [85518] ERROR V4L2 v4l2_subdevice.cpp:394 'ov2740 18-0036': Unable to get rectangle 2 on pad 0: Inappropriate ioctl for device
[11:10:39.119295435] [85518]  WARN CameraSensor camera_sensor.cpp:293 'ov2740 18-0036': The PixelArraySize property has been defaulted to 1932x1092
[11:10:39.119306861] [85518] ERROR V4L2 v4l2_subdevice.cpp:394 'ov2740 18-0036': Unable to get rectangle 1 on pad 0: Inappropriate ioctl for device
[11:10:39.119321333] [85518]  WARN CameraSensor camera_sensor.cpp:304 'ov2740 18-0036': The PixelArrayActiveAreas property has been defaulted to (0, 0)/1932x1092
[11:10:39.119336125] [85518] ERROR V4L2 v4l2_subdevice.cpp:394 'ov2740 18-0036': Unable to get rectangle 0 on pad 0: Inappropriate ioctl for device
[11:10:39.119346367] [85518]  WARN CameraSensor camera_sensor.cpp:312 'ov2740 18-0036': Failed to retrieve the sensor crop rectangle
[11:10:39.119354453] [85518]  WARN CameraSensor camera_sensor.cpp:318 'ov2740 18-0036': The sensor kernel driver needs to be fixed
[11:10:39.119361941] [85518]  WARN CameraSensor camera_sensor.cpp:320 'ov2740 18-0036': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[11:10:39.119856536] [85518]  WARN CameraSensor camera_sensor.cpp:469 'ov2740 18-0036': Failed to retrieve the camera location
[11:10:39.119875020] [85518]  WARN CameraSensor camera_sensor.cpp:491 'ov2740 18-0036': Rotation control not available, default to 0 degrees
[11:10:46.398639932] [85506]  INFO Camera camera.cpp:1183 configuring streams: (0) 1924x1088-BGR888
[11:10:46.398862116] [85518]  INFO IPASoft soft_simple.cpp:115 Exposure 4-1102, gain 128-1983
Zero-copy enabled
```

Often repeated in dmesg:
```
[78676.371874] intel_ipu6_isys.isys intel_ipu6.isys.40: stream stop time out
[78678.419859] intel_ipu6_isys.isys intel_ipu6.isys.40: stream close time out
```

Messages for the successful case:
```
[11:09:51.479868863] [85450]  INFO Camera camera_manager.cpp:284 libcamera v0.0.0+3-4ea14140
[11:09:51.493158977] [85462]  WARN CameraSensor camera_sensor.cpp:251 'ov2740 18-0036': Recommended V4L2 control 0x009a0922 not supported
[11:09:51.493195348] [85462] ERROR V4L2 v4l2_subdevice.cpp:394 'ov2740 18-0036': Unable to get rectangle 2 on pad 0: Inappropriate ioctl for device
[11:09:51.493204643] [85462]  WARN CameraSensor camera_sensor.cpp:293 'ov2740 18-0036': The PixelArraySize property has been defaulted to 1932x1092
[11:09:51.493210062] [85462] ERROR V4L2 v4l2_subdevice.cpp:394 'ov2740 18-0036': Unable to get rectangle 1 on pad 0: Inappropriate ioctl for device
[11:09:51.493214278] [85462]  WARN CameraSensor camera_sensor.cpp:304 'ov2740 18-0036': The PixelArrayActiveAreas property has been defaulted to (0, 0)/1932x1092
[11:09:51.493220123] [85462] ERROR V4L2 v4l2_subdevice.cpp:394 'ov2740 18-0036': Unable to get rectangle 0 on pad 0: Inappropriate ioctl for device
[11:09:51.493224067] [85462]  WARN CameraSensor camera_sensor.cpp:312 'ov2740 18-0036': Failed to retrieve the sensor crop rectangle
[11:09:51.493227939] [85462]  WARN CameraSensor camera_sensor.cpp:318 'ov2740 18-0036': The sensor kernel driver needs to be fixed
[11:09:51.493231383] [85462]  WARN CameraSensor camera_sensor.cpp:320 'ov2740 18-0036': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[11:09:51.493490904] [85462]  WARN CameraSensor camera_sensor.cpp:469 'ov2740 18-0036': Failed to retrieve the camera location
[11:09:51.493496876] [85462]  WARN CameraSensor camera_sensor.cpp:491 'ov2740 18-0036': Rotation control not available, default to 0 degrees
[11:09:55.023035821] [85450]  INFO Camera camera.cpp:1183 configuring streams: (0) 1924x1088-BGR888
[11:09:55.023240237] [85462]  INFO IPASoft soft_simple.cpp:115 Exposure 4-1102, gain 128-1983
Zero-copy enabled
[11:09:57.233326594] [85469]  INFO Debayer debayer_cpu.cpp:935 Processed 30 frames in 270157us, 9005 us/frame
```

kupietz, 2024-01-29:
Also, reboot doesn't help. After one reboot, no camera at all, `dmesg | grep -E 'ov2740|ipu6'`:

```
[    2.006265] usb usb1: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    2.007424] usb usb2: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    2.012133] usb usb3: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    2.016288] usb usb4: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    2.278501]     BOOT_IMAGE=(hd0,gpt6)/vmlinuz-6.7.1-200.1.ipu6.fc39.x86_64
[    7.679762] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    7.680497] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x0 mask 0xff
[    7.680959] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[    7.681467] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[    7.681959] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[    7.682469] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[    7.682964] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[    7.683501] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[    7.683959] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[    7.684477] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[    7.685000] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[    7.685509] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[    7.700673] intel-ipu6 0000:00:05.0: FW version: 20220510
[    7.704469] intel-ipu6 0000:00:05.0: Found supported sensor INT3474:01
[    7.704528] intel-ipu6 0000:00:05.0: Connected 1 cameras
[    7.705855] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[    7.723483] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[    7.792733] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[    7.792741] intel-ipu6 0000:00:05.0: IPU6-v3[a75d] hardware version 5
[    7.993712] intel_ipu6_isys: unknown parameter 'video_nr' ignored
[    9.396858] ov2740 i2c-INT3474:01: chip id mismatch: 2740 != 0
[    9.396870] ov2740 i2c-INT3474:01: error -ENXIO: failed to find sensor
```

jwrdegoede, 2024-01-30:
> Unfortunately, no progress concerning reliability: for i in $(seq 1 10); do echo $i; qcam; done yields successful camera streams in ~4 of 10 cases. The camera light turns on each time.

Ok, so this really is an issue with ov2740 and/or ipu6-isys drivers. This means that someone from Intel really needs to take a closer look at this issue.

kupietz, 2024-02-12:
With kernel 6.7 on Fedora 39, success rate with my Lenovo X11 Gen 11 has dropped to 0%. Also, booting Windows in between does not help anymore. 

```
# dmesg | grep -E 'ov2740|ipu6'
[    7.510170] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    7.510470] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    7.510488] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    7.510490] intel-ipu6 0000:00:05.0: mapped as: 0x0000000077df4fb4
[    7.510565] intel-ipu6 0000:00:05.0: IPU in secure mode
[    7.510569] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    7.510571] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    7.530341] intel-ipu6 0000:00:05.0: IPC reset done
[    7.530345] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    7.534974] intel-ipu6 0000:00:05.0: FW version: 20230102
[    7.586383] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    7.586426] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    7.586428] intel-ipu6 0000:00:05.0: mapped as: 0x0000000077df4fb4
[    7.587498] intel-ipu6 0000:00:05.0: IPU in secure mode
[    7.587502] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    7.587504] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    7.590365] intel-ipu6 0000:00:05.0: IPC reset done
[    7.590368] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    7.590835] intel-ipu6 0000:00:05.0: FW version: 20230102
```
Has anyone had any better luck?

lukemarsden, 2024-03-17:
@jwrdegoede great progress with the open source driver.

Unfortunately still only 30% reliable on my X1 Gen Carbon 11 and about 70% reliable on the X1 Yoga Gen 8.

Also seeing more washed out colour, but it's exciting to see it working without the v4l-relay.

Any ideas who we can ping at Intel to get some movement on the more fundamental reliability issue on these machines? What further device details or instrumentation can I provide?

lukemarsden, 2024-03-17:
Following https://copr.fedorainfracloud.org/coprs/jwrdegoede/ipu6-softisp/, seeing the same

```
[ 1257.974947] intel_ipu6_isys.isys intel_ipu6.isys.40: stream stop time out
[ 1260.152009] intel_ipu6_isys.isys intel_ipu6.isys.40: stream close time out
[ 1267.447995] intel_ipu6_isys.isys intel_ipu6.isys.40: stream stop time out
[ 1269.495069] intel_ipu6_isys.isys intel_ipu6.isys.40: stream close time out
```

```
System Information
        Manufacturer: LENOVO
        Product Name: 21HMCTO1WW
        Version: ThinkPad X1 Carbon Gen 11
[...]
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_21HM_BU_Think_FM_ThinkPad X1 Carbon Gen 11
        Family: ThinkPad X1 Carbon Gen 11
```

lukemarsden, 2024-03-18:
@jwrdegoede just wanted to add a thank-you, your COPR https://copr.fedorainfracloud.org/coprs/jwrdegoede/ipu6-softisp/ is the only stable way to run the webcam on my X1 Yoga Gen 8 right now. Ubuntu 24.04 nightly was working with the webcam, but broke itself (and emojis stopped working :joy:), and the original https://hansdegoede.livejournal.com/27276.html is no longer working on latest 6.7.9 kernel on fc39.

Even though the colour is a bit washed out, I'm able to use my newer hardware again, so thank you! I also love the fact that the kernel is pinned, knowing that a kernel update can break your webcam any day is annoying :)

Is there a near-term plan to improve the washed out colours?

andaag, 2024-03-18:
Not succeeding in fedora 39 on a dell xps 9320, but I feel like I'm missing something basic. It's not it's not trying to connect to he camera at all. Not finding any relevant errors either.

```
# dmesg | grep -E 'ov2740|ipu6'
[    0.000000] Linux version 6.7.1-200.1.ipu6.fc39.x86_64 (mockbuild@ec318dcebdd54443ad88e7a1e85dfd0a) (gcc (GCC) 13.2.1 20231205 (Red Hat 13.2.1-6), GNU ld version 2.40-13.fc39) #1 SMP PREEMPT_DYNAMIC Wed Jan 24 21:55:31 UTC 2024
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt7)/vmlinuz-6.7.1-200.1.ipu6.fc39.x86_64 root=UUID=363ed805-a453-4c64-b849-f8ae575f439f ro rootflags=subvol=root rd.luks.uuid=luks-625a417b-ab29-4568-983c-c40770ed3480 rhgb quiet mem_sleep_default=s2idle i915.enable_psr=0 resume=UUID=363ed805-a453-4c64-b849-f8ae575f439f resume_offset=75549882 rd.driver.blacklist=nouveau modprobe.blacklist=nouveau
[    0.055212] Kernel command line: BOOT_IMAGE=(hd0,gpt7)/vmlinuz-6.7.1-200.1.ipu6.fc39.x86_64 root=UUID=363ed805-a453-4c64-b849-f8ae575f439f ro rootflags=subvol=root rd.luks.uuid=luks-625a417b-ab29-4568-983c-c40770ed3480 rhgb quiet mem_sleep_default=s2idle i915.enable_psr=0 resume=UUID=363ed805-a453-4c64-b849-f8ae575f439f resume_offset=75549882 rd.driver.blacklist=nouveau modprobe.blacklist=nouveau
[    0.055321] Unknown kernel command line parameters "rhgb BOOT_IMAGE=(hd0,gpt7)/vmlinuz-6.7.1-200.1.ipu6.fc39.x86_64", will be passed to user space.
[    1.606878] usb usb1: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    1.607141] usb usb2: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    1.609344] usb usb3: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    1.610639] usb usb4: Manufacturer: Linux 6.7.1-200.1.ipu6.fc39.x86_64 xhci-hcd
[    2.399849]     BOOT_IMAGE=(hd0,gpt7)/vmlinuz-6.7.1-200.1.ipu6.fc39.x86_64
[   15.424259] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[   15.424504] intel-ipu6 0000:00:05.0: IPU6 in non-secure mode touch 0x0 mask 0xff
[   15.428447] intel-ipu6 0000:00:05.0: FW version: 20220510
[   15.432657] intel-ipu6 0000:00:05.0: IPU6-v3[465d] hardware version 5
# uname -a

Linux 6.7.1-200.1.ipu6.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Jan 24 21:55:31 UTC 2024 x86_64 GNU/Linux

# qcam
[0:01:09.812481594] [5024]  INFO Camera camera_manager.cpp:284 libcamera v0.0.0+3-70b69666
[0:01:09.820633664] [5040] ERROR SimplePipeline simple.cpp:1496 No sensor found

# cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 512-871, parent: platform/INTC1055:00, INTC1055:00:

gpiochip1: GPIOs 872-935, parent: auxiliary/usb_ljca.ljca-gpio.0, INTC1096:00, can sleep:
```

Any ideas?

lukemarsden, 2024-03-18:
@jwrdegoede one more bit of feedback on the copr, when the webcam has to do a "retry" before it starts, sometimes the video comes up "juddery". It feels like there are maybe two simultaneous streams from the webcam that are slightly out of sync with eachother. Maybe this has to do with the first attempt to connect appearing to fail, but actually starting streaming frames, and then the second attempt also ends up streaming them? I'm totally guessing there. Just FYI.

Restarting firefox is normally sufficient to fix it when it does that.

s-ailus, 2024-03-28:
I think I came across this issue while testing the embedded data support with Lenovo X1 Carbon Gen 10. I don't know what the actual reason is but I can start streaming reliably (tried about 200 times) after disabling runtime PM on the ov2740 sensor: [https://git.linuxtv.org/sailus/media_tree.git/commit/?h=ipu6&id=af377d7fa423f371a3a9b8288b8d65da8a90f47e](url). This is indeed just a hack, a proper fix would likely end up being somewhere else than this driver. Without the patch in about 50 % of the attempts streaming is supposed to start but no buffers (or anything else) are returned by the ISYS driver.

jwrdegoede, 2024-03-28:
> I think I came across this issue while testing the embedded data support with Lenovo X1 Carbon Gen 10. I don't know what the actual reason is but I can start streaming reliably (tried about 200 [...]

@s-ailus good catch. Can you try extending the msleep in ov2740_resume() and also try putting a msleep between the clk_prepare_enable() and the gpiod_set_value_cansleep() ?

Hopefully that will fix it. If not then I guess that the reset pin may not be wired up, and that the ACPI PS0 / PS3 methods do actually turn the sensor on/off and without a proper reset it then gets stuck.

Which would suggest Windows never powers down the sensor ...   Or maybe we are missing a sw-reset as part of our stream on commands and/or the sleep after the swreset is too short ?

lukemarsden, 2024-03-29:
Great news, thanks @s-ailus !

@jwrdegoede any chance you could bake this workaround into a build of your copr kernel please? I'd love to test it on my X1 Carbon Gen 11 where I couldn't get more than 30% success rate with the camera. Also making the Yoga webcam more reliable would be a huge quality of life improvement. I don't mind a slight reduction in battery life.

kupietz, 2024-04-27:
> I think I came across this issue while testing the embedded data support with Lenovo X1 Carbon Gen 10. I don't know what the actual reason is but I can start streaming reliably (tried about 200 [...]

This solves the issue for me, too, on Fedora 39 and Fedora 40, thanks! Camera indeed seems to work 100%, now.

Patch for current tree, here:

```patch
iff --git a/drivers/media/i2c/ov2740.c b/drivers/media/i2c/ov2740.c
index 2b2fbf2d9..f00b95f79 100644
--- a/drivers/media/i2c/ov2740.c
+++ b/drivers/media/i2c/ov2740.c
@@ -1585,7 +1585,7 @@ static int ov2740_probe(struct i2c_client *client)
         */
        pm_runtime_set_active(&client->dev);
        pm_runtime_enable(&client->dev);
-       pm_runtime_idle(&client->dev);
+       pm_runtime_get(&client->dev);
 
        return 0;
```

lukemarsden, 2024-04-27:
Do you mind sharing a kernel rpm please?

lukemarsden.net
***@***.***
@lmarsden <https://twitter.com/lmarsden>

[55 more replies]