intel/ipu6-drivers #188: **Title:** Samsung Galaxy Book3 Ultra and Fedora 39

gutopardini, 2023-10-13:
[dmesg.txt](https://github.com/intel/ipu6-drivers/files/12897613/dmesg.txt)
[i2c-devices.txt](https://github.com/intel/ipu6-drivers/files/12897614/i2c-devices.txt)
[ipu6-rpms.txt](https://github.com/intel/ipu6-drivers/files/12897615/ipu6-rpms.txt)
[lsmod.txt](https://github.com/intel/ipu6-drivers/files/12897616/lsmod.txt)
[video-devices.txt](https://github.com/intel/ipu6-drivers/files/12897617/video-devices.txt)
**Issue Description:**
The camera displays a black screen.

**Steps to Reproduce:**
1. Install Fedora 39.
2. Run the following commands to install the drivers:
    ```bash
    sudo dnf update \
      --enablerepo=updates-testing \
      --enablerepo=rpmfusion-free-updates-testing \
      --enablerepo=rpmfusion-nonfree-updates-testing \
      'kernel*' '*v4l2loopback'
    ```
    And
    ```bash
    sudo dnf install \
      --enablerepo=updates-testing \
      --enablerepo=rpmfusion-free-updates-testing \
      --enablerepo=rpmfusion-nonfree-updates-testing \
      akmod-intel-ipu6
    ```

**Expected Behavior:**
The camera should work as intended.

**Current Behavior:**
The camera is recognized but displays a black screen when tested.

**Additional Information:**
Attached files: `dmesg.txt`, `12c-devices.txt`, `ipu6-rpms.txt`, `lsmod.txt`, `video-devices.txt`.

If any other output or logs are needed, please let me know.

gutopardini, 2023-10-13:
@jwrdegoede,

I've encountered an issue with the camera on the Samsung Galaxy Book3 Ultra while using Fedora 39. Despite my attempts to resolve it and extensive research, I've been unable to find a solution. Considering your expertise and significant contributions to this project, I was hoping you might be able to provide some insight or guidance on this matter when you have a moment. I appreciate any help or direction you can provide and understand completely if your schedule doesn't allow it. Thank you in advance for your time and consideration.

jwrdegoede, 2023-10-24:
@gutopardini unfortunately just having IPU6 support is not enough to make IPU6 attached cameras work.

There also needs to be a driver for the camera sensor, which tends to differ per laptop model / series. E.g. Lenovo laptops use the ov2740, where as Dell laptops use the ov01a1s, ov01a10 or ov02c10.

Each of these sensors needs to have a driver specifically for that sensor written and there also may be other involved chips, like IO-expanders, sensor-PMICs or Intel's VSC chip involved too, each of which also needs a driver written.

And writing these drivers requires documentation, hardware access and a lot of time.  And on top of this special tools are necessary to create a configuration for the proprietary userspace stack provided by Intel.

Only Intel has access to the necessary documentation and tools. AFAIK Intel only works on Linux support for laptop models where the laptop vendors specifically asks them to enable the camera under Linux because they are offering the laptop model with an OEM Linux pre-install.

So unless you can get Samsung to ask Intel to add support for your laptop model I'm afraid that you are out of luck.

closed 2023-10-24 by gutopardini