intel/ipu6-drivers #198: Cannot pass build using kernel 6.5.7

yaoengine, 2023-11-16:
get compile error
```
drivers/media/i2c/power_ctrl_logic.c:121:27: error: initialization of ‘void (*)(struct acpi_device *)’ from incompatible pointer type ‘int (*)(struct acpi_device *)’ [-Werror=incompatible-pointer-types]
  121 |                 .remove = power_ctrl_logic_remove,
      |                           ^~~~~~~~~~~~~~~~~~~~~~~
drivers/media/i2c/power_ctrl_logic.c:121:27: note: (near initialization for ‘_driver.ops.remove’)
cc1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:243: drivers/media/i2c/power_ctrl_logic.o] Error 1
make[8]: *** [scripts/Makefile.build:480: drivers/media/i2c] Error 2
```

While check the code, found that power_ctrl_logic_remove() function is ‘int (*)(struct acpi_device *)’
```C++
static int power_ctrl_logic_remove(struct acpi_device *adev)
{
	dev_dbg(&adev->dev, "@%s, enter\n", __func__);
	mutex_lock(&pcl.status_lock);
	pcl.gpio_ready = false;
	gpiod_set_value_cansleep(pcl.reset_gpio, 0);
	gpiod_put(pcl.reset_gpio);
	gpiod_set_value_cansleep(pcl.powerdn_gpio, 0);
	gpiod_put(pcl.powerdn_gpio);
	gpiod_set_value_cansleep(pcl.clocken_gpio, 0);
	gpiod_put(pcl.clocken_gpio);
	gpiod_set_value_cansleep(pcl.indled_gpio, 0);
	gpiod_put(pcl.indled_gpio);
	mutex_unlock(&pcl.status_lock);
	dev_dbg(&adev->dev, "@%s, exit\n", __func__);
	return 0;
}
```
and the function pointer is:
```C++
typedef void (*acpi_op_remove) (struct acpi_device *device);
```

So is this repo not compatible with kernel 6.5.7?

Thanks!

hao-yao, 2023-11-24:
@yaoengine power_ctrl_logic.c is deprecated after kernel v5.15. There is an official INT3472 power control logic driver since v5.15.

closed 2023-11-24 by hao-yao

yaoengine, 2023-11-24:
@hao-yao 
I cannot get your idea. Do you mean that I should not use the master branch? Since https://github.com/intel/ipu6-drivers/blob/master/drivers/media/i2c/power_ctrl_logic.c is here in your repo code in master. And as it is deprecated, so how can I use it correctly?

hao-yao, 2023-11-26:
@yaoengine I can't get your idea either. Do you know what I mean by "deprecated after kernel v5.15"?

Please go to README and find this:

 > If your kernel < 5.15 or not set CONFIG_INTEL_SKL_INT3472, please set this
 > CONFIG_POWER_CTRL_LOGIC=m

yaoengine, 2023-11-26:
My question is I use master branch for the build. Should I change to iotg_ipu6 branch?
I did not see the explaination of which branch should be used.

hao-yao, 2023-11-27:
@yaoengine You should always use the "master" branch. Other branches are for some special customers.

However it's not an issue about "branch".

> @yaoengine power_ctrl_logic.c is deprecated after kernel v5.15. There is an official INT3472 power control logic driver since v5.15.

That means you should use CONFIG_INTEL_SKL_INT3472 instead of CONFIG_POWER_CTRL_LOGIC after kernel v5.15. So don't touch power_ctrl_logic.c after v5.15 unless you know what you are doing.

yaoengine, 2023-11-27:
OK, thanks, I got it.

But I cannot find set CONFIG_INTEL_SKL_INT3472 in master branch README. But found to enable CONFIG_INTEL_SKL_INT3472 in iotg_ipu6 branch README. I just follow all the things in master README and cannot pass the build, and follow iotg_ipu6 branch README, the build at least passed. So it make me confuse, which one to follow.
```
	c.	Check and enable kernel config list in below
		Must be enabled:
	                CONFIG_VIDEO_INTEL_IPU6=m 
	                CONFIG_INTEL_SKL_INT3472=m
	                CONFIG_INTEL_IPU6_ACPI=m
	                CONFIG_VIDEO_INTEL_IPU_SOC=y
	                CONFIG_VIDEO_INTEL_IPU_USE_PLATFORMDATA=y
```

yaoengine, 2023-11-27:
I found INTEL_SKL_INT3472 should be enabled finally in the Dependencies of master branch README, I will try again the master branch.
Thanks!