intel/ipu6-drivers #202: HP Spectre x360 (2023): webcam INT3537 does not work

MrTomRod, 2023-11-27:
I'm having trouble to get my webcam to work on my new laptop running Fedora 39.

- Laptop: 13.5" HP x360 [14-ef2759nz](https://support.hp.com/us-en/drivers/hp-spectre-x360-13.5-inch-2-in-1-laptop-pc-14-ef2000/model/2101549652?sku=7P401EA&serialnumber=CND3290HZY) 
- Camera description: "HP True Vision 5MP IR camera with camera shutter, temporal noise reduction and integrated dual array digital microphones".
- Model `INT3537` (According to `dmesg`.)
- This model has been mentioned in these issues: #182 #48  (maybe #105)

<details>
  <summary>I'm using Fedora 39 and installed the latest kernel and ipu6 as explained <a href="https://hansdegoede.livejournal.com/27276.html">here</a>.</summary>

```bash
sudo dnf update \
  --enablerepo=updates-testing \
  --enablerepo=rpmfusion-free-updates-testing \
  --enablerepo=rpmfusion-nonfree-updates-testing \
  'kernel*' '*v4l2loopback'

# And now things are ready to install the IPU6 driver stack:

sudo dnf install \
  --enablerepo=updates-testing \
  --enablerepo=rpmfusion-free-updates-testing \
  --enablerepo=rpmfusion-nonfree-updates-testing \
  akmod-intel-ipu6

# Secure boot must be disabled.
# Reboot.
```

</details>

Installed packages:

- kernel-6.6.2-201.fc39.x86_64
- kmod-intel-ipu6-6.6.2-201.fc39.x86_64-0.0-10.20230622git8e41080.fc39.x86_64
- kmod-v4l2loopback-6.6.2-201.fc39.x86_64-0.12.7^20230503g2c9b670-2.fc39.x86_64

The webcam seems to be detected:

<details>
  <summary>Output of <code>v4l2-ctl</code></summary>
  
```bash
$ v4l2-ctl --list-devices
ipu6 (PCI:pci:pci0000:00):
	/dev/video7

ipu6 (pci:pci0000:00):
	/dev/media0

Intel MIPI Camera (platform:v4l2loopback-000):
	/dev/video0
$ v4l2-ctl --device=/dev/video0 --all
Driver Info:
	Driver name      : v4l2 loopback
	Card type        : Intel MIPI Camera
	Bus info         : platform:v4l2loopback-000
	Driver version   : 6.6.2
	Capabilities     : 0x85200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x05200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
Priority: 2
[... 130 lines cut ...]
[11-27 21:18:06.671] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.004439587
Setting pipeline to NULL ...
Freeing pipeline ...
```

</details>

<details>
  <summary>Output of <code>ls /sys/bus/i2c/devices</code></summary>

```text
$ ls /sys/bus/i2c/devices
i2c-0   i2c-11  i2c-14  i2c-17  i2c-2  i2c-5  i2c-8            i2c-INT3537:00   i2c-TXNW3643:01
i2c-1   i2c-12  i2c-15  i2c-18  i2c-3  i2c-6  i2c-9            i2c-OVTI00AB:00
i2c-10  i2c-13  i2c-16  i2c-19  i2c-4  i2c-7  i2c-ELAN2513:00  i2c-SYNA32E9:00
```

</details>

Can you please help me get it to work?

odanoburu, 2023-12-07:
I had it working on Fedora 38, but I just updated to 39 and it doesn't work anymore. Maybe it's the Fedora update per se, just the new kernel. The driver has been stable lately (until this upgrade, that is), but I've been using it for some time now and it was overall pretty unstable, with kernel/package updates often breaking it, to the point that I can't rely on the camera to be working if/when I need it.

So maybe it will just fix itself in a few updates? My hardware is different, however, so it's not exactly the same sensor; maybe you could downgrade to 38 and see if it works?

odanoburu, 2023-12-07:
You could try posting [here](https://bugzilla.rpmfusion.org/show_bug.cgi?id=6721), or opening another bug report. For me, restarting the `v4l2-relayd` service with `systemctl restart v4l2-relayd` restored the camera functionality on kernel 6.6.4-200 (Fedora 39).

MrTomRod, 2024-01-10:
Thanks for the tips. Unfortunately, restarting `systemctl restart v4l2-relayd` doesn't help and the [alternative package](https://thias.marmotte.net/2024/01/fedora-ipu6-camera-support/) didn't solve the problem either.

jeffreyfq, 2024-01-11:
Hi,
I have a 16" HP spectre x360 ( 16-f1xxx ) with the same camera.
I installed the driver recently (yesterday) from github via DKMS and I am on kernel version "6.6.4-060604-generic"

Same problem, and same `intel-ipu6` error messages!

Further `dmesg | grep hi556` returns:
```
[    5.322990] hi556 i2c-INT3537:00: failed to find sensor: -5
[    5.323694] hi556: probe of i2c-INT3537:00 failed with error -5
```
The driver can not find the sensor, I assume this causes later errors of the IPU6 module.

I am not sure if the following entries in dmesg are related to the problem,
`dmesg | grep mei` returns:
```
[    5.269611] mei_me 0000:00:16.0: enabling device (0000 -> 0002)
[    5.328964] mei_gsc i915.mei-gscfi.768: FW not ready: resetting: dev_state = 2 pxp = 0
[    5.328994] mei_gsc i915.mei-gscfi.768: unexpected reset: dev_state = ENABLED fw status = 00000345 84670000 00000000 00000000 E0020002 00000000
[    5.329012] mei i915.mei-gscfi.768-46e0c1fb-a546-414f-9170-b7f46d57b4ad: Could not read FW version ret = -19
[    5.329013] mei i915.mei-gscfi.768-46e0c1fb-a546-414f-9170-b7f46d57b4ad: FW version command failed -5
[    5.329759] mei_gsc i915.mei-gsc.768: FW not ready: resetting: dev_state = 2 pxp = 2
[    5.329797] mei_gsc i915.mei-gsc.768: unexpected reset: dev_state = ENABLED fw status = 00000345 84670000 00000000 00000000 E0020002 00000000
[    5.389717] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    5.390552] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])

```

paulmenzel, 2024-01-12:
To my knowledge, for full support specific knowledge about the hardware design is needed, that only HP has. Did you contact their support?

PS: As HP already made money by selling the device, in my opinion, you should return the devices to HP and buy a device with full GNU/Linux support.

MrTomRod, 2024-01-21:
@jeffreyfq I get different dmesg errors for `mei` and `hi556`:

```text
$ dmesg | grep hi556
[   11.089283] hi556 i2c-INT3537:00: can't get clock frequency
[   11.089288] hi556 i2c-INT3537:00: failed to check HW configuration: -22
[   11.089334] hi556: probe of i2c-INT3537:00 failed with error -22
$ dmesg | grep mei
[   11.146932] mei_me 0000:00:16.0: enabling device (0000 -> 0002)
[   11.255533] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[   11.260689] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
```

jwrdegoede, 2024-01-21:
For Fedora users with this issue, please try running:

```
sudo rm "/lib/modules/$(uname -r)/kernel/drivers/media/i2c/hi556.ko.xz"
```

This will cause the kmod-intel-ipu6-... version of the hi556 driver to get used instead of the one from the mainline kernel, hopefully this will help.

After running the rm command reboot. After rebooting first run the rm command again, it should not fail with a "No such file or directory" error. If it does not fail you have rebooted into another kernel (likely due to a kernel update) in this case reboot again and run the rm command again until it does fail.

Once the rm command fails after a reboot, check dmesg for messages about the hi556 driver:

```
sudo dmesg | grep hi556
```

This should no longer show any errors and now your camera will hopefully work.

jwrdegoede, 2024-01-21:
@MrTomRod ok so it looks like we are getting somewhere.

Cheese is known to not work with the proprietary userspace stack from Intel, not even with the v4l2loopback /dev/video0 compat support.

The best test to just test the camera stack and not v4l2-relayd is using gst-launch as you did. But your commandline is wrong, specifically the device-name parameter points to another sensor then you have.

Please try this commandline instead:

```
gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink
```

If that works, then you can try the v4l2relayd/v4l2loopback /dev/video0 compatibility by  opening https://mozilla.github.io/webrtc-landing/gum_test.html in firefox. If the gst-launch command works, but firefox does not then try running:

```
sudo systemctl restart v4l2-relayd
```

and then try the gum_test page in firefox again.

MrTomRod, 2024-01-21:
This is the output, seems like the device is not found (`Failed to find DevName for cameraId: 0`):

```text
$ gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
[01-21 22:42:33.151] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-21 22:42:33.151] CamHAL[ERR] there is no aiqb file:ov13b10
[01-21 22:42:33.151] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-21 22:42:33.151] CamHAL[ERR] there is no aiqb file:ov13b10
[01-21 22:42:33.151] CamHAL[INF] aiqb file name ov8856.aiqb
[01-21 22:42:33.151] CamHAL[ERR] there is no aiqb file:ov8856
[01-21 22:42:33.151] CamHAL[INF] aiqb file name ov8856.aiqb
[01-21 22:42:33.151] CamHAL[ERR] there is no aiqb file:ov8856
[01-21 22:42:33.151] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-21 22:42:33.151] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-21 22:42:33.151] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-21 22:42:33.151] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-21 22:42:33.151] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-21 22:42:33.151] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-21 22:42:33.151] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-21 22:42:33.151] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-21 22:42:33.151] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[01-21 22:42:33.152] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[01-21 22:42:33.153] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[01-21 22:42:33.153] CamHAL[INF] aiqb file name ov01a1s.aiqb
[01-21 22:42:33.153] CamHAL[INF] aiqb file name ov01a1s.aiqb
[01-21 22:42:33.153] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[01-21 22:42:33.162] CamHAL[ERR] Get entity fail for calling getEntityById
[01-21 22:42:33.162] CamHAL[ERR] Get entity fail for calling getEntityById
[01-21 22:42:33.162] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[01-21 22:42:33.162] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[01-21 22:42:33.162] CamHAL[ERR] set up mediaCtl failed
[01-21 22:42:33.162] CamHAL[ERR] @configure Device Configure failed
[01-21 22:42:33.162] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.007305264
Setting pipeline to NULL ...
Freeing pipeline ...
```

jwrdegoede, 2024-01-22:
@MrTomRod ok so it looks like the proprietary userspace stack does not recognize your specific camera module (sensor + lens module) even though there is support for some hi556 modules, that is unfortunate.

In that case there is nothing we can do.

MrTomRod, 2024-01-22:
Thanks again for your efforts.

If I wanted to escalate this, whom should I talk to, HP or Intel? Maybe [ipu6-camera-bins](https://github.com/intel/ipu6-camera-bins)? Do you think the issue will not be fixed and I should return my laptop?

MrTomRod, 2024-01-26:
Thanks to @jwrdegoede, the camera is now working thanks to a new, fully open source software stack! Here's the link and the installation instructions: https://copr.fedorainfracloud.org/coprs/jwrdegoede/ipu6-softisp

Here's @jwrdegoede 's blog on the subject: https://hansdegoede.dreamwidth.org/28153.html

Without the proprietary AI magic, the image quality isn't fantastic, but at least there _is_ an image! I still have trouble with Firefox pipewire camera support, but I'm confident this will be fixed soon.

sterlingdcs-damian, 2024-02-05:
Where might i find simple steps for getting this working on ubuntu?

ChocolateLoverRaj, 2024-03-07:
@MrTomRod I have a Redrix #182 which also has a `hi556` sensor. WHen I use the open source MIPI stack I also get that beige tint to the image. It's nice to know that this is normal, and it seems like there are plans / possibility of further improvements to the image.

jwrdegoede, 2025-11-25:
@MrTomRod do you still have your HP Spectre x360 (2023) ?

And if yes can you try running the following commands with a recent Fedora? : 

"qcam -s width=2560,height=1920"
"qcam -s width=2560,height=1440"
"qcam -s width=1280,height=960"
"qcam -s width=1280,height=720"

And let me know if all 4 commands work? And if they do not work can you let me know which commands work and which ones do not work ?

MrTomRod, 2025-11-25:
Hey Hans

Happy to help out!

```bash
$ cat /proc/cmdline
BOOT_IMAGE=(hd0,gpt5)/vmlinuz-6.17.8-300.fc43.x86_64 root=UUID=628826cc-3b2e-4035-995e-84d535857e43 ro rootflags=subvol=root rhgb quiet mem_sleep_default=deep intel_skl_int3472_discrete.dyndbg intel_skl_int3472_common.dyndbg
$ journalctl -b 0 -k > kernel.txt
$ lsusb > lsusb.txt
$ ls -l /sys/bus/i2c/devices/ > i2c-devices.txt
$ ls -l /sys/bus/spi/devices/ > spi-devices.txt
```

- [kernel.txt](https://github.com/user-attachments/files/23753501/kernel.txt)
- [lsusb.txt](https://github.com/user-attachments/files/23753502/lsusb.txt)
- [i2c-devices.txt](https://github.com/user-attachments/files/23753505/i2c-devices.txt)
- [dmesg-dyndbg.txt](https://github.com/user-attachments/files/23753510/dmesg-dyndbg.txt)

My camera works with the different commands:

<details>
<summary>Testing qcam output</summary>

```
$ qcam -s width=2560,height=1920
[0:07:13.225616106] [18032]  INFO Camera camera_manager.cpp:330 libcamera v0.5.2
[0:07:13.233887969] [18045]  WARN CameraSensor camera_sensor_legacy.cpp:354 'hi556 4-0028': Recommended V4L2 control 0x009a0922 not supported
[0:07:13.233906850] [18045]  WARN CameraSensor camera_sensor_legacy.cpp:426 'hi556 4-0028': The sensor kernel driver needs to be fixed
[0:07:13.233909106] [18045]  WARN CameraSensor camera_sensor_legacy.cpp:428 'hi556 4-0028': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[0:07:13.234063881] [18045]  WARN CameraSensorProperties camera_sensor_properties.cpp:484 No static properties available for 'hi556'
[0:07:13.234068063] [18045]  WARN CameraSensorProperties camera_sensor_properties.cpp:486 Please consider updating the camera sensor properties database
[0:07:13.234070487] [18045]  WARN CameraSensor camera_sensor_legacy.cpp:594 'hi556 4-0028': Failed to retrieve the camera location
[0:07:13.234072458] [18045]  WARN CameraSensor camera_sensor_legacy.cpp:616 'hi556 4-0028': Rotation control not available, default to 0 degrees
[0:07:13.234078957] [18045]  WARN CameraSensor camera_sensor_legacy.cpp:501 'hi556 4-0028': No sensor delays found in static properties. Assuming unverified defaults.
[0:07:13.236209623] [18045]  WARN IPAProxy ipa_proxy.cpp:177 Configuration file 'hi556.yaml' not found for IPA module 'simple', falling back to 'uncalibrated.yaml'
[0:07:13.243786900] [18046]  INFO IPAProxySoftWorker soft_ipa_proxy_worker.cpp:448 Starting worker for IPA module /usr/lib64/libcamera/ipa/ipa_soft_simple.so with IPC fd = 35
[0:07:13.244143838] [18046]  WARN IPASoft soft_simple.cpp:103 IPASoft: Failed to create camera sensor helper for hi556
[0:07:13.244594759] [18045]  INFO Camera camera_manager.cpp:220 Adding camera '\_SB_.PC00.LNK0' for pipeline handler simple
[0:07:13.304745763] [18032]  INFO Camera camera.cpp:1215 configuring streams: (0) 2560x1920-ABGR8888/Unset
[0:07:13.304975392] [18046]  WARN IPASoft soft_simple.cpp:255 IPASoft: Minimum gain is zero, that can't be linear
[0:07:13.305110529] [18046]  INFO IPASoft soft_simple.cpp:269 IPASoft: Exposure 6-2066, gain 100-240 (1)
Zero-copy enabled
^CExiting

$ qcam -s width=2560,height=1440
[0:07:19.319504954] [19204]  INFO Camera camera_manager.cpp:330 libcamera v0.5.2
[0:07:19.330697988] [19216]  WARN CameraSensor camera_sensor_legacy.cpp:354 'hi556 4-0028': Recommended V4L2 control 0x009a0922 not supported
[0:07:19.330709902] [19216]  WARN CameraSensor camera_sensor_legacy.cpp:426 'hi556 4-0028': The sensor kernel driver needs to be fixed
[0:07:19.330712048] [19216]  WARN CameraSensor camera_sensor_legacy.cpp:428 'hi556 4-0028': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
[0:07:19.330841568] [19216]  WARN CameraSensorProperties camera_sensor_properties.cpp:484 No static properties available for 'hi556'
[0:07:19.330844763] [19216]  WARN CameraSensorProperties camera_sensor_properties.cpp:486 Please consider updating the camera sensor properties database
[0:07:19.330848823] [19216]  WARN CameraSensor camera_sensor_legacy.cpp:594 'hi556 4-0028': Failed to retrieve the camera location
[0:07:19.330851264] [19216]  WARN CameraSensor camera_sensor_legacy.cpp:616 'hi556 4-0028': Rotation control not available, default to 0 degrees
[0:07:19.330854933] [19216]  WARN CameraSensor camera_sensor_legacy.cpp:501 'hi556 4-0028': No sensor delays found in static properties. Assuming unverified defaults.
[0:07:19.332609822] [19216]  WARN IPAProxy ipa_proxy.cpp:177 Configuration file 'hi556.yaml' not found for IPA module 'simple', falling back to 'uncalibrated.yaml'
[0:07:19.340235105] [19217]  INFO IPAProxySoftWorker soft_ipa_proxy_worker.cpp:448 Starting worker for IPA module /usr/lib64/libcamera/ipa/ipa_soft_simple.so with IPC fd = 35
[0:07:19.340532751] [19217]  WARN IPASoft soft_simple.cpp:103 IPASoft: Failed to create camera sensor helper for hi556
[0:07:19.340859943] [19216]  INFO Camera camera_manager.cpp:220 Adding camera '\_SB_.PC00.LNK0' for pipeline handler simple
[0:07:19.362364866] [19204]  INFO Camera camera.cpp:1215 configuring streams: (0) 2560x1440-ABGR8888/Unset
[0:07:19.362693396] [19217]  WARN IPASoft soft_simple.cpp:255 IPASoft: Minimum gain is zero, that can't be linear
[0:07:19.362823417] [19217]  INFO IPASoft soft_simple.cpp:269 IPASoft: Exposure 6-2079, gain 100-240 (1)
Zero-copy enabled
^CExiting

$ qcam -s width=1280,height=960
[0:07:26.104757521] [20061]  INFO Camera camera_manager.cpp:330 libcamera v0.5.2
[... cut ...]
[0:07:32.679592570] [20998]  WARN IPASoft soft_simple.cpp:103 IPASoft: Failed to create camera sensor helper for hi556
[0:07:32.680141223] [20997]  INFO Camera camera_manager.cpp:220 Adding camera '\_SB_.PC00.LNK0' for pipeline handler simple
[0:07:32.704531065] [20985]  INFO Camera camera.cpp:1215 configuring streams: (0) 1280x720-ABGR8888/Unset
[0:07:32.704728974] [20998]  WARN IPASoft soft_simple.cpp:255 IPASoft: Minimum gain is zero, that can't be linear
[0:07:32.704784744] [20998]  INFO IPASoft soft_simple.cpp:269 IPASoft: Exposure 6-2066, gain 100-240 (1)
Zero-copy enabled
^CExiting
```

</details>

jwrdegoede, 2025-11-26:
Great, thank you.

jwrdegoede, 2025-11-28:
Thank you for the quick response.

AlexCK-STFC, 2026-09-03:
@fingertap Resolved using this patch for my INT3537, GPIO now binds: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/2154140

Previously, I had indefinite `int3472-discrete INT3472:01: cannot find GPIO chip INTC10D1:00, deferring` 

See:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/2162099

MrTomRod, 2026-09-03:
Hi @ChocolateLoverRaj 

> [@MrTomRod](https://github.com/MrTomRod) I have a Redrix [#182](https://github.com/intel/ipu6-drivers/issues/182) which also has a `hi556` sensor. WHen I use the open source MIPI stack I also get [...]

I was able to improve the tint like this:

`/usr/share/libcamera/ipa/simple/hi556.yaml`:

```ca
# SPDX-License-Identifier: CC0-1.0
%YAML 1.1
---
version: 1
algorithms:
  - BlackLevel:
  - Awb:
  - Agc:
  - Ccm:
      ccms:
        - ct: 2860
          ccm: [ 2.1, -0.8, -0.3, -0.4, 0.9, -0.1, -0.2, -0.8, 2.0 ]
        - ct: 5000
          ccm: [ 1.9, -0.7, -0.2, -0.3, 0.8, -0.1, -0.1, -0.7, 1.8 ]
```

Not perfect, but I don't have time to fine-tune this.