intel/ipu6-drivers #204: MiPi Camera and Ubuntu: New kernel 6.5 no longer working

SmokinCaterpillar, 2024-01-14:
I followed the steps here (https://wiki.ubuntu.com/Lenovo) to get my MiPi camera working on my Thinkpad X1Gen10 with Ubuntu 22.04.

These steps work under 6.2:
```
$ sudo apt install linux-modules-ipu6-generic-hwe-22.04 linux-modules-ivsc-generic-hwe-22.04 (For generic kernel, like 6.2.0-32-generic)
$ sudo apt install linux-modules-ipu6-oem-22.04c linux-modules-ivsc-oem-22.04c (For oem kernel, like 6.1.0-1021-oem)
$ sudo add-apt-repository ppa:oem-solutions-group/intel-ipu6 (This ppa is still in development, might be not stable, in future you don't need this step)
$ sudo apt install libcamhal0
$ sudo add-apt-repository -r ppa:oem-solutions-group/intel-ipu6 (the ppa will keep updating packages from upstream, might be broken)
$ sudo reboot
```
This worked fine with kernel 6.2, but now fails when booting with the newest kernel with version 6.5.0.14.14. Camera is not starting anymore and I get a black screen instead of the camera image in any application. Booting back to 6.2 does resolve the issue.

Is there a fix for 6.5 on the way?

Thank You!

SmokinCaterpillar, 2024-01-19:
Hey, is there any update on this?
running `sudo gst-launch-1.0 icamerasrc ! autovideosink`
outputs
```
[01-19 23:06:16.237] CamHAL[ERR] Malformed ET range in exposure time range configuration
[01-19 23:06:16.237] CamHAL[ERR] Parse AE eExposure time range failed
[01-19 23:06:16.237] CamHAL[ERR] Malformed ET range in exposure time range configuration
[01-19 23:06:16.237] CamHAL[ERR] Parse AE gain range failed
[01-19 23:06:16.238] CamHAL[ERR] Malformed ET range in exposure time range configuration
[01-19 23:06:16.238] CamHAL[ERR] Parse AE eExposure time range failed
[01-19 23:06:16.238] CamHAL[ERR] Malformed ET range in exposure time range configuration
[01-19 23:06:16.238] CamHAL[ERR] Parse AE gain range failed
[01-19 23:06:16.239] CamHAL[ERR] Malformed ET range in exposure time range configuration
[01-19 23:06:16.239] CamHAL[ERR] Parse AE eExposure time range failed
```
...
```
Setting pipeline to PAUSED ...
[01-19 23:06:18.506] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-19 23:06:18.509] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-19 23:06:18.511] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-19 23:06:18.511] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-19 23:06:18.511] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-19 23:06:18.512] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-19 23:06:18.512] CamHAL[INF] aiqb file name ov8856.aiqb
[01-19 23:06:18.514] CamHAL[INF] aiqb file name ov8856.aiqb
[01-19 23:06:18.514] CamHAL[ERR] invalid media format, default value used.
[01-19 23:06:18.514] CamHAL[INF] aiqb file name ov8856.aiqb
[01-19 23:06:18.514] CamHAL[INF] aiqb file name ov8856.aiqb
[01-19 23:06:18.515] CamHAL[ERR] invalid media format, default value used.
[01-19 23:06:18.515] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-19 23:06:18.517] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-19 23:06:18.518] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-19 23:06:18.518] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-19 23:06:18.518] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-19 23:06:18.519] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-19 23:06:18.519] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-19 23:06:18.521] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-19 23:06:18.521] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-19 23:06:18.521] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-19 23:06:18.523] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-19 23:06:18.523] CamHAL[ERR] invalid media format, default value used.
[01-19 23:06:18.523] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-19 23:06:18.525] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-19 23:06:18.525] CamHAL[ERR] invalid media format, default value used.
[01-19 23:06:18.525] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-19 23:06:18.525] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-19 23:06:18.526] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-19 23:06:18.526] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-19 23:06:18.526] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-19 23:06:18.526] CamHAL[ERR] invalid media format, default value used.
[01-19 23:06:18.526] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-19 23:06:18.526] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-19 23:06:18.527] CamHAL[ERR] invalid media format, default value used.
[01-19 23:06:18.527] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[01-19 23:06:18.529] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[01-19 23:06:18.529] CamHAL[ERR] invalid media format, default value used.
[01-19 23:06:18.529] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-19 23:06:18.531] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-19 23:06:18.532] CamHAL[ERR] invalid media format, default value used.
[01-19 23:06:18.532] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-19 23:06:18.534] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-19 23:06:18.534] CamHAL[ERR] invalid media format, default value used.
[01-19 23:06:18.534] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-19 23:06:18.534] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-19 23:06:18.535] CamHAL[ERR] invalid media format, default value used.
[01-19 23:06:18.535] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-19 23:06:18.535] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-19 23:06:18.535] CamHAL[ERR] invalid media format, default value used.
[01-19 23:06:18.535] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[... 75 lines cut ...]
[01-19 23:06:18.558] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-19 23:06:18.558] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-19 23:06:18.558] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-19 23:06:18.559] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[01-19 23:06:18.586] CamHAL[ERR] Get entity fail for calling getEntityById
[01-19 23:06:18.586] CamHAL[ERR] Get entity fail for calling getEntityById
[... cut ...]
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.016210934
Setting pipeline to NULL ...
Freeing pipeline ...
```

What can I do about this? I am growing more and more frustrated by the day. Why on earth Intel did you create these annoying useless MiPi cameras?

igor-semenov, 2024-01-20:
I have the same issue: fix which worked for 6.2 kernel is not working on 6.5 anymore.
In `dmesg -T` there are following errors:
`[23:14:10 2024] ov2740 i2c-INT3474:01: error while getting pled gpio: -2
[23:14:10 2024] ov2740: probe of i2c-INT3474:01 failed with error -22`

SmokinCaterpillar, 2024-02-11:
Any update, maybe?

deathsfaction, 2024-02-21:
Also have the same issue but with 6.7

djice, 2024-03-07:
I have the issue too... Since one month many people of my entreprise cannot use the camera for videoconference meeting...

These type of camera are on many dell (7440 series for us). I'm wondering how is it possible in 2024 and why nobody of IBM move/communicate on this issue.

odontomachus, 2024-03-08:
Same with the linux (6.7) and linux-lts (6.6) kernel on archlinux, it broke around 1 month ago, I haven't been able to get it to work again.

kop7, 2024-03-17:
same with  6.6.10-76060610-generic

strophy, 2024-03-19:
Still not working with 6.8.1 installed through mainline with Ubuntu 23.10 on a Dell XPS 9315. This computer was purchased with Linux pre-installed to "guarantee" working hardware, very disappointing to see such poor support.

fingertap, 2024-03-26:
I am working on Archlinux, with kernel 6.8.1. The package seems not working with dkms.

dh-harald, 2024-03-27:
Have you tried this repo: https://github.com/jwrdegoede/ipu6-drivers?
I'm running ubuntu lts 22.04 with 6.5-hwe kernel on Dell Precision 5680. My problem has been solved with that module

kupietz, 2024-03-27:
> Have you tried this repo: https://github.com/jwrdegoede/ipu6-drivers? I'm running ubuntu lts 22.04 with 6.5-hwe kernel on Dell Precision 5680. My problem has been solved with that module

Thanks! This works for me also with Fedora 39 (kernel 6.7.10-200.fc39.x86_64) on a Lenovo Carbon X11 Gen 11, after deleting the conflicting old module versions with:

```bash
for m in /usr/lib/modules/$(uname -r)/extra/*.ko.xz; do find /usr/lib/modules/$(uname -r) -name $(basename $m) -not -path $m -delete; done; 
find /usr/lib/modules/$(uname -r) -name usb-ljca.ko.xz -delete
depmod -a
```

Shashankti, 2024-04-02:
@dh-harald hey can you please explain which method you used to install the drivers?

dh-harald, 2024-04-02:
> @dh-harald hey can you please explain which method you used to install the drivers?

I used previously a dkms module, so first, I've removed the old one, then I compiled it dkms. I think, the most important part to remove the old modules as @kupietz suggested before. I had a small problem with the backported libraries from `ivsc-drivers`, as the command not exactly did that I've excepted, but fixed it with investigated the previous driver, and put them into the right directory.

djice, 2024-04-25:
Solve today with

`sudo dnf update --enablerepo=rpmfusion-nonfree-updates* '*ipu6*'`

And reboot

deathsfaction, 2024-04-25:
> Solve today with
> 
> `sudo dnf update --enablerepo=rpmfusion-nonfree-updates* '*ipu6*'`
> 
> And reboot

Makes no difference for me.

nphilbrook, 2024-08-19:
Was _working_ for me on 6.5, then broke on upgrade to 6.8. I had previously run these steps:
https://answers.launchpad.net/ubuntu-certification/+question/708392
(I am on a Dell Precision 5680)

ngiann, 2024-09-24:
> Was _working_ for me on 6.5, then broke on upgrade to 6.8. I had previously run these steps: https://answers.launchpad.net/ubuntu-certification/+question/708392 (I am on a Dell Precision 5680)

I followed the link, applied the instructions and my camera is working now. 

I have a precision 5680, running ubuntu 22.04.5 LTS. In case this is useful, here is the output of uname -a:

` Linux nikos-Precision-5680 6.8.0-45-generic #45~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Sep 11 15:25:05 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`

nphilbrook, 2024-09-24:
> I followed the link, applied the instructions and my camera is working now.
>
> I have a precision 5680, running ubuntu 22.04.5 LTS. In case this is useful, here is the output of uname -a:
>
> ` Linux [...]

Yeah, I'm on the same exact kernel version and Ubuntu version as you, and my webcam is now working. Kernel version 6.8.0-40 was experience the issues, although I can't guarantee it wasn't another apt package getting upgraded in the mix that actually caused and/or remediated the issue.

For now, just cross fingers and pray. My employer mandates that all available apt upgrades are applied every day, which throws a wrench into things :grimacing:

toorandom, 2024-12-13:
Hello. Are there any news here ? 

I am trying to make this work and have passed from 24.04 to 24.10 but nothing. Libhal does not appear to be available in 24.10, only the opensource kernel drivers but nothing... 

Do I really need to go back to 24.04-1 ?

$ uname -a
Linux pythagoras 6.11.0-12-generic #13-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 21 20:03:13 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux