intel/ipu6-drivers #205: kernel panic on 6.7.0-9.gaedda80-default

alinelena, 2024-01-17:
using opensuse tumbleweed and 6.7.0-9.gaedda80-default trying to get the drivers loading with out of tree builds and failing.

building all the stack like this

```bash

#!/usr/bin/env bash

pushd ipu6-drivers

git clone https://github.com/intel/ivsc-driver.git
cp -r ivsc-driver/backport-include ivsc-driver/drivers ivsc-driver/include .
rm -rf ivsc-driver

make -j3
sudo make modules_install
sudo depmod -a

popd

for i in ipu_adl  ipu_mtl  ipu_tgl; do 
  sudo cp -r ipu6-camera-bins/include/$i /usr/include/
done

sudo mkdir -p /usr/lib/firmware/intel/
sudo mkdir -p /usr/lib/firmware/intel/unsigned

for i in ipu6epadln_fw.bin  ipu6ep_fw.bin  ipu6epmtl_fw.bin  ipu6_fw.bin; do 
  sudo cp -r ipu6-camera-bins/lib/firmware/intel/$i  /usr/lib/firmware/intel/
  sudo cp -r ipu6-camera-bins/lib/firmware/intel/$i  /usr/lib/firmware/intel/unsigned/
done 

for i in ipu_adl  ipu_mtl  ipu_tgl; do
  sudo cp -r ipu6-camera-bins/lib/$i  /usr/lib
done

sh ./build.sh

sudo cp -r ./out/install/lib/* /usr/lib/
sudo cp -r ./out/install/lib64/* /usr/lib/
sudo cp -r ./out/install/share/* /usr/share/

```

main error is 

```
[   17.212136] input: Dell Privacy Driver as /devices/platform/PNP0C14:02/wmi_bus/wmi_bus-PNP0C14:02/6932965F-1671-4CEB-B988-D3AB0A901919/input/input21
[   17.212755] spi-nor spi1.0: w25q512jvq (65536 Kbytes)
[   17.220358] input: Dell WMI hotkeys as /devices/platform/PNP0C14:02/wmi_bus/wmi_bus-PNP0C14:02/9DBB5994-A997-11DA-B012-B622A1EF5492/input/input22
[   17.220407] Creating 1 MTD partitions on "0000:00:1f.5":
[   17.220410] 0x000000000000-0x000004000000 : "BIOS"
[   17.221399] intel_tcc_cooling: TCC Offset locked
[   17.221446] BUG: kernel NULL pointer dereference, address: 0000000000000000
[   17.221453] #PF: supervisor read access in kernel mode
[   17.221456] #PF: error_code(0x0000) - not-present page
[   17.221458] PGD 0 P4D 0 
[   17.221462] Oops: 0000 [#1] PREEMPT SMP NOPTI
[   17.221467] CPU: 5 PID: 1028 Comm: (udev-worker) Tainted: G           OE      6.7.0-9.gaedda80-default #1 openSUSE Tumbleweed (unreleased) efa6eaac34d992d2676b5a6b9d2124424bc6ccc9
[   17.221472] Hardware name: Dell Inc. XPS 9320/0CW9KM, BIOS 2.5.0 07/07/2023
[   17.221474] RIP: 0010:__wake_up_common+0x2a/0xa0
[   17.221481] Code: 0f 1f 44 00 00 41 57 41 56 41 55 41 54 55 53 48 89 fb 48 83 c3 08 48 83 ec 08 48 8b 47 08 89 54 24 04 48 39 c3 74 55 4d 89 c7 <4c> 8b 00 41 89 f5 41 89 ce 48 8d 78 e8 4d 8d 60 e8 eb 27 74 0c 83
[   17.221485] RSP: 0018:ffffabb580e6fa20 EFLAGS: 00010082
[   17.221488] RAX: 0000000000000000 RBX: ffff9e7cc1991f20 RCX: 0000000000000000
[   17.221491] RDX: 0000000000000000 RSI: 0000000000000003 RDI: ffff9e7cc1991f18
[   17.221492] RBP: ffff9e7cc1991f18 R08: 0000000000000000 R09: 0000000000000004
[   17.221494] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000003
[   17.221496] R13: 0000000000000246 R14: 0000000000000000 R15: 0000000000000000
[   17.221499] FS:  00007f3bbbf81d40(0000) GS:ffff9e83ef480000(0000) knlGS:0000000000000000
[   17.221502] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[... 85 lines cut ...]
[   17.417227] debugfs: File '\_SB.PC00.HDAS.SNDW' in directory 'domains' already present!
[   17.427885] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 4
[   17.428208] iwlwifi 0000:00:14.3: Detected RF GF, rfid=0x2010d000
[   17.431039] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[   17.431048] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[   17.431056] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[   17.514504] iwlwifi 0000:00:14.3: base HW address: f0:d4:15:b2:91:e3
[   17.533924] iwlwifi 0000:00:14.3 wlp0s20f3: renamed from wlan0
[   17.552847] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[   17.552857] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[   17.584768] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:0
[   17.584935] sof_sdw sof_sdw: ASoC: Parent card not yet available, widget card binding deferred
[   17.608014] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
[   17.633615] input: sof-soundwire HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input23
[   17.633723] input: sof-soundwire HDMI/DP,pcm=6 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input24
[   17.633851] input: sof-soundwire HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input25
[   18.246154] usbcore: registered new interface driver ljca
[   18.250244] ljca: exports duplicate symbol ljca_register_event_cb (owned by usb_ljca)

```

full log is here 
[full.log](https://github.com/intel/ipu6-drivers/files/13961999/full.log)

in the full log i see few interesting things that make me think this is really porcelain territory, missing configs for the default kernel I load, but I am not expert in these things.

bverhagen, 2024-01-26:
I seem to have the same issue with OpenSUSE Tumbleweed 6.7.1 ( 4959dd8).

alinelena, 2024-01-27:
@bverhagen thanks for confirming, did it ever worked on opensuse?

bverhagen, 2024-01-27:
> @bverhagen thanks for confirming, did it ever worked on opensuse?

Not for me. The furthest I got (in the past) was where my browser recognizes the camera, tries to read it, the 'active' led flashes on for half a second, after which it all errors out.

Every now and then, I pull the latest version for all the repositories and retry. Currently, I do not get this far and seem to have the same issue as you do.

During my debugging, I found one post of an openSUSE user (no idea whether this user used TW or Leap) who seemingly had it working, but I would not call that solid evidence.

benjiman, 2024-03-01:
I'm having the same issue. Any pointers for helping debug?

jwrdegoede, 2024-03-05:
This is fixed by: https://github.com/jwrdegoede/ipu6-drivers/commit/da1fb08fe045a83fbdc585490a3ce3bcc4c658e6 I will submit a pull-request for this soon.

madsl, 2024-04-17:
I'm not able to get it to work with or after this commit though: 92fb37ebd8d7d210e9e1ea5213b848b87da9619a

I tried building the kernel with 0004 and 0005 6.8 kernel patches, and it builds just fine, but icamerasrc stops working . There's so many moving parts here that I don't know how to debug it, so I just revert and build as before.

Dell XPS 9320 has a ov01a10 sensor.

PR #213 (2024-05-09): Ipu6 kernel 6.7 and 6.8 fixes
Here is a set of 3 fixes to make ipu6-drivers work with the 6.7 and 6.8 kernels

1. The first commit fixes the ipu6-isys driver crashing (due to a bug in a cleanup on error paths) after a failure in isys_probe() (this fixes #205)
2. The second commit fixes the drivers not compiling against kernel 6.8
3. The third commit fixes the kernel crashing with kernel 6.8 due to the cio2-bridge code making the .name attribute of 2 long-lived software_nodes point to on stack memory

Cc: @vicamo @smallorange
closed 2024-05-09 by hao-yao