intel/ipu6-drivers #207: Internal webcam not detected on Samsung Galaxy Book 3 PRO running Ubuntu 23.04

Beor18, 2024-01-22:
Hello, 

I would like to know if there is a possibility of making the webcam work in Samsung Galaxy Book 3 PRO with Ubuntu 23.04?

Thanks you!

Beor18, 2024-06-07:
Help me! Please

gustavohfelixs, 2024-06-11:
Do your sound works man? I have a Galaxy book 4 pro and having issues too

AdityaMitra5102, 2024-07-12:
@gustavohfelixs Open your terminal as root and run the following. I have tested it on my Book 3 Pro, I had similar issues but now its better. The right speaker is still very faint but the left one is working. Run this script on boot with root priviledges.

`
#!/bin/bash

hda-verb /dev/snd/hwC0D0 0x20 0x500 0x0
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x0
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x99
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x99
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x99
hda-verb /dev/snd/hwC0D0 0x20 0x480 0x0
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x99
hda-verb /dev/snd/hwC0D0 0x20 0x480 0x0
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x99
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x99
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x0
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x0
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x1
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x1
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x2
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x2
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x3
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x3
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x4
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x4
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x5
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x5
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x6
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x6
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x7
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x7
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x8
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x8
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x9
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x9
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xa
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xa
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xb
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xb
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xc
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xc
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xd
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xd
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xe
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xe
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xf
hda-verb /dev/snd/hwC0D0 0x20 0x500 0xf
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x10
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x10
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x11
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x11
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x12
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x12
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x13
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x13
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x14
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x14
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x15
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x15
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x16
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x16
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x17
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x17
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x18
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x18
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x19
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x19
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x1a
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x1a
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x1b
[... 3780 lines cut ...]
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x23
hda-verb /dev/snd/hwC0D0 0x20 0x420 0x3a
hda-verb /dev/snd/hwC0D0 0x20 0x400 0x0
hda-verb /dev/snd/hwC0D0 0x20 0x400 0x81
hda-verb /dev/snd/hwC0D0 0x20 0x4b0 0x11
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x26
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x26
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x23
hda-verb /dev/snd/hwC0D0 0x20 0x423 0xff
hda-verb /dev/snd/hwC0D0 0x20 0x400 0x0
hda-verb /dev/snd/hwC0D0 0x20 0x400 0x1
hda-verb /dev/snd/hwC0D0 0x20 0x4b0 0x11
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x23
hda-verb /dev/snd/hwC0D0 0x20 0x423 0xff
hda-verb /dev/snd/hwC0D0 0x20 0x400 0x0
hda-verb /dev/snd/hwC0D0 0x20 0x400 0x1
hda-verb /dev/snd/hwC0D0 0x20 0x4b0 0x11
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x22
hda-verb /dev/snd/hwC0D0 0x20 0x400 0x38
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x22
hda-verb /dev/snd/hwC0D0 0x20 0x400 0x38
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x26
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x26
hda-verb /dev/snd/hwC0D0 0x20 0x500 0x23
`