intel/ipu6-drivers #223: Driver fails to load for Meteor Lake on Linux 6.9

moritz89, 2024-04-18:
Tested on Dell XPS 16 9640 with Ubuntu 24.04 (beta) and Linux kernel 6.9-rc4

```
[    2.958449] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    2.958540] intel-ipu6 0000:00:05.0: Device 0x7d19 (rev: 0x4)
[    2.958550] intel-ipu6 0000:00:05.0: physical base address 0x522c000000
[    2.958551] intel-ipu6 0000:00:05.0: mapped as: 0x000000004e245cb9
[    2.958614] intel-ipu6 0000:00:05.0: IPU in secure mode
[    2.958615] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[    2.958616] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
...
[    2.970260] intel-ipu6 0000:00:05.0: IPC reset done
[    2.970262] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6epmtl_fw.bin
[    2.970367] intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6epmtl_fw.bin failed with error -2
[    2.970370] intel-ipu6 0000:00:05.0: Requesting signed firmware failed
[    2.970376] intel-ipu6 0000:00:05.0: probe with driver intel-ipu6 failed with error -2
```

Also, copying the `ipu6epmtl_fw.bin` file up one directory where it is expected leads to further actions, but no functioning camera.

ilgrosso, 2024-04-30:
Same error here, again on Dell XPS 16 9640 and kernel 6.9.0-rc6.
Went also through https://bugs.launchpad.net/ubuntu/+source/ipu6-drivers/+bug/2031412 with no luck.

cstickel, 2024-05-09:
same with the Dell XPS 14 9440. would be great to get the camera working. camera + IR sensors are the only thing that aren't supported.
@jwrdegoede is there any chance that you guys are working on that issue soon? it's really a shame that the 2024 dell notebooks don't have working webcam support.

AntiNSA, 2024-05-21:
same with ubuntu 24.04 and dell precision 5680

MarjanHJ, 2024-05-29:
I have the same issue and tried different things , here is the output of dmesg for camera, appreciate your help:

   2.265001] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[    2.269101] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[    2.269104] pci 0000:00:14.2: enabling device (0000 -> 0002)
[    2.269410] intel_vpu 0000:00:0b.0: enabling device (0000 -> 0002)
[    2.269413] intel_pmc_core INT33A1:00: Assuming a default substate order for this platform
[    2.269507] intel_pmc_core INT33A1:00:  initialized
[    2.271083] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    2.271193] intel-ipu6 0000:00:05.0: Device 0x7d19 (rev: 0x4)
[    2.271205] intel-ipu6 0000:00:05.0: physical base address 0x5229000000
[    2.271206] intel-ipu6 0000:00:05.0: mapped as: 0x0000000092cb4a15
[    2.271257] intel-ipu6 0000:00:05.0: IPU in secure mode
[    2.271262] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[    2.271263] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    2.280095] intel_vpu 0000:00:0b.0: [drm] Firmware: intel/vpu/vpu_37xx_v0.0.bin, version: 20230726*MTL_CLIENT_SILICON-release*2101*ci_tag_mtl_pv_vpu_rc_20230726_2101*648a666b8b9
[    2.283324] intel-ipu6 0000:00:05.0: IPC reset done
[    2.283345] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6epmtl_fw.bin
[    2.283492] intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6epmtl_fw.bin failed with error -2
[    2.283505] intel-ipu6 0000:00:05.0: Requesting signed firmware failed
[    2.283526] intel-ipu6 0000:00:05.0: probe with driver intel-ipu6 failed with error -2
[    2.288784] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[    2.293333] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[    2.297487] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring

mbojoly, 2024-06-14:
Still got the problem on Dell XPS 16 9640 and `Linux 6.9.0-060900-generic #202405122134 SMP PREEMPT_DYNAMIC Sun May 12 21:47:18 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux` even with `ac347ffb9 - (HEAD, tag: Release_20240409_1)`

dmesg output is
```
[    0.000000] Linux version 6.9.0-060900-generic (kernel@sita) (x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #202405122134 SMP PREEMPT_DYNAMIC Sun May 12 21:47:18 UTC 2024
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-6.9.0-060900-generic root=/dev/mapper/ubuntu--vg-ubuntu--lv ro quiet splash vt.handoff=7
...
[   33.444885] intel_ipu6: loading out-of-tree module taints kernel.
[   33.444890] intel_ipu6: module verification failed: signature and/or required key missing - tainting kernel
[   33.445044] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.447227] Consider using thermal netlink events interface
[   33.448542] input: Intel HID events as /devices/platform/INTC1077:00/input/input12
[   33.448662] EDAC MC0: Giving out device to module igen6_edac controller Intel_client_SoC MC#0: DEV 0000:00:00.0 (INTERRUPT)
[   33.448916] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[   33.449072] intel-ipu6 0000:00:05.0: Device 0x7d19 (rev: 0x4)
[   33.449086] intel-ipu6 0000:00:05.0: physical base address 0x581c000000
[   33.449088] intel-ipu6 0000:00:05.0: mapped as: 0x00000000d1d36f24
[   33.449215] intel-ipu6 0000:00:05.0: IPU in secure mode
[   33.449217] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[   33.449218] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[   33.449332] intel_vsec 0000:00:0a.0: enabling device (0000 -> 0002)
[   33.449447] intel-hid INTC1077:00: platform supports 5 button array
[   33.449560] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.450931] input: Intel HID 5 button array as /devices/platform/INTC1077:00/input/input13
[   33.452383] EDAC MC1: Giving out device to module igen6_edac controller Intel_client_SoC MC#1: DEV 0000:00:00.0 (INTERRUPT)
[   33.452436] EDAC igen6 MC1: HANDLING IBECC MEMORY ERROR
[   33.452438] EDAC igen6 MC1: ADDR 0x7fffffffe0 
[   33.452439] EDAC igen6 MC0: HANDLING IBECC MEMORY ERROR
[   33.452440] EDAC igen6 MC0: ADDR 0x7fffffffe0 
[   33.453155] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.453163] EDAC igen6: v2.5.1
[   33.454926] mc: Linux media interface: v0.10
[   33.455559] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.457279] pci 0000:00:08.0: Setting to D3hot
[   33.458017] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.462095] intel-ipu6 0000:00:05.0: IPC reset done
[   33.463619] intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6epmtl_fw.bin failed with error -2
[   33.466269] intel-ipu6 0000:00:05.0: FW version: 20230925
[   33.466538] videodev: Linux video capture interface: v2.00
[   33.469016] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   33.469335] pci 0000:00:0b.0: Setting to D3hot
[   33.470766] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.481268] pci 0000:00:14.2: enabling device (0000 -> 0002)
[   33.485257] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.487123] proc_thermal_pci 0000:00:04.0: enabling device (0000 -> 0002)
[   33.487567] intel_vpu 0000:00:0b.0: enabling device (0000 -> 0002)
[   33.488133] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.488383] i801_smbus 0000:00:1f.4: SPD Write Disable is set
[   33.488442] i801_smbus 0000:00:1f.4: SMBus using PCI interrupt
[   33.488583] intel_rapl_common: Found RAPL domain package
[   33.491788] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.492413] intel_pmc_core INT33A1:00: Assuming a default substate order for this platform
[   33.492524] intel_pmc_core INT33A1:00:  initialized
[   33.494586] intel_vpu 0000:00:0b.0: [drm] Firmware: intel/vpu/vpu_37xx_v0.0.bin, version: 20230726*MTL_CLIENT_SILICON-release*2101*ci_tag_mtl_pv_vpu_rc_20230726_2101*648a666b8b9
[   33.502256] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.505756] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   33.506565] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.510999] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.515381] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.516808] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   33.517009] ACPI: battery: new extension: Dell DDV Battery Extension
[   33.517057] usbcore: registered new interface driver ljca
[   33.517078] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   33.517206] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[   33.522034] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   33.525501] RAPL PMU: API unit is 2^-32 Joules, 4 fixed counters, 655360 ms ovfl timer
[   33.525506] RAPL PMU: hw unit of domain pp0-core 2^-14 Joules
[   33.525507] RAPL PMU: hw unit of domain package 2^-14 Joules
[   33.525508] RAPL PMU: hw unit of domain pp1-gpu 2^-14 Joules
[   33.525509] RAPL PMU: hw unit of domain psys 2^-14 Joules
[... 67 lines cut ...]
[... cut ...]
[   50.601945] intel_vsc intel_vsc: error -ENODEV: init hw failed
[   50.642579] usbcore: registered new interface driver btusb
[   50.644193] Bluetooth: hci0: Firmware timestamp 2023.42 buildtype 1 build 45894
[   50.644200] Bluetooth: hci0: Firmware SHA1: 0x2e575f2a
[   50.644210] Bluetooth: hci0: DSM reset method type: 0x01
[   50.644306] ACPI Warning: \_SB.PC00.XHCI.RHUB.HS10._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20230628/nsarguments-61)
[   50.648488] intel-ipu6-isys intel-ipu6-isys0: no subdev found in graph

```

mbojoly, 2024-06-16:
After activating logs with `sudo modprobe intel-ipu6 debug=0xffffffff` I got

```
[    0.000000] Linux version 6.9.0-060900-generic (kernel@sita) (x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #202405122134 SMP PREEMPT_DYNAMIC Sun May 12 21:47:18 UTC 2024
...
[   18.985992] intel_ipu6: loading out-of-tree module taints kernel.
[   18.986001] intel_ipu6: module verification failed: signature and/or required key missing - tainting kernel
[   18.989733] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   18.991290] proc_thermal_pci 0000:00:04.0: enabling device (0000 -> 0002)
[   18.991663] intel_rapl_common: Found RAPL domain package
[   18.992268] pci 0000:00:14.2: enabling device (0000 -> 0002)
[   18.992462] intel_vpu 0000:00:0b.0: enabling device (0000 -> 0002)
[   19.002383] intel_pmc_core INT33A1:00: Assuming a default substate order for this platform
[   19.002519] intel_pmc_core INT33A1:00:  initialized
[   19.002571] intel_vpu 0000:00:0b.0: [drm] Firmware: intel/vpu/vpu_37xx_v0.0.bin, version: 20230726*MTL_CLIENT_SILICON-release*2101*ci_tag_mtl_pv_vpu_rc_20230726_2101*648a666b8b9
[   19.003258] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[   19.003668] intel-ipu6 0000:00:05.0: Device 0x7d19 (rev: 0x4)
[   19.003691] intel-ipu6 0000:00:05.0: physical base address 0x581c000000
[   19.003692] intel-ipu6 0000:00:05.0: mapped as: 0x00000000eeca132a
[   19.003780] intel-ipu6 0000:00:05.0: IPU in secure mode
[   19.003781] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[   19.003783] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[   19.005554] intel-ipu6 0000:00:05.0: IPC reset done
[   19.005757] intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6epmtl_fw.bin failed with error -2
[   19.007631] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.007824] intel-ipu6 0000:00:05.0: FW version: 20230925
[   19.009256] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   19.010173] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.012881] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.017512] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.024146] ACPI: battery: new extension: Dell DDV Battery Extension
[   19.024354] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   19.024543] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   19.024543] i801_smbus 0000:00:1f.4: SPD Write Disable is set
[   19.024599] i801_smbus 0000:00:1f.4: SMBus using PCI interrupt
[   19.024636] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[   19.025520] usbcore: registered new interface driver ljca
[   19.033889] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.048809] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.055251] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   19.062028] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.098064] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.098251] RAPL PMU: API unit is 2^-32 Joules, 3 fixed counters, 655360 ms ovfl timer
[   19.098255] RAPL PMU: hw unit of domain pp0-core 2^-14 Joules
[   19.098258] RAPL PMU: hw unit of domain package 2^-14 Joules
[   19.098260] RAPL PMU: hw unit of domain psys 2^-14 Joules
[   19.102799] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   19.102857] intel-ipu6 0000:00:05.0: IPU6-v4 driver version 1.0
[   19.105867] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.165871] [drm] Initialized intel_vpu 1.0.0 20230117 for 0000:00:0b.0 on minor 0
[   19.170849] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.212347] i2c i2c-4: Memory type 0x23 not supported yet, not instantiating SPD
[   19.216485] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.258871] int3472-discrete INT3472:0c: cannot find GPIO chip INTC10D1:00, deferring
[   19.328291] intel_rapl_msr: PL4 support detected.
[   19.328552] intel_rapl_common: Found RAPL domain package
[   19.328555] intel_rapl_common: Found RAPL domain core
[   19.328557] intel_rapl_common: Found RAPL domain psys
[   19.328604] Intel(R) Wireless WiFi driver for Linux
[   19.328607] Bluetooth: Core ver 2.22
[   19.328627] NET: Registered PF_BLUETOOTH protocol family
[   19.328628] Bluetooth: HCI device and connection manager initialized
[   19.328630] Bluetooth: HCI socket layer initialized
[   19.328632] Bluetooth: L2CAP socket layer initialized
[   19.328634] Bluetooth: SCO socket layer initialized
[   19.329072] iwlwifi 0000:01:00.0: enabling device (0000 -> 0002)
[   19.340622] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[   19.340756] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[   19.341669] iwlwifi 0000:01:00.0: Detected crf-id 0x2001910, cnv-id 0x2001910 wfpm id 0x80000000
[   19.341677] iwlwifi 0000:01:00.0: PCI dev 272b/40f0, rev=0x472, rfid=0x112200
[... 25 lines cut ...]
[   20.493371] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[   20.639765] i915 0000:00:02.0: [drm] GT1: Loaded GSC firmware i915/mtl_gsc_1.bin (cv1.0, r102.0.0.1655, svn 1)
[   20.660014] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for all workloads
[... cut ...]
[   29.764949] Bluetooth: hci0: Found device firmware: intel/ibt-0291-0291.sfi
[   29.764961] Bluetooth: hci0: Boot Address: 0x100800
[   29.764962] Bluetooth: hci0: Firmware Version: 70-42.23
[   29.764962] Bluetooth: hci0: Firmware already loaded
[   29.769209] Bluetooth: hci0: Fseq status: Success (0x00)
[   29.769212] Bluetooth: hci0: Fseq executed: 00.00.04.125
[   29.769213] Bluetooth: hci0: Fseq BT Top: 00.00.04.125
[   29.770492] intel-ipu6-isys intel-ipu6-isys0: no subdev found in graph
```

plskeggs, 2024-09-18:
Same with Dell Latitude 7650.

stratism, 2024-09-27:
Same with Dell Latidude 7450 and 6.8.0-45 Ubuntu latest

Alexander-Aghili, 2024-10-10:
Same with Dell XPS 16 9640, Ubuntu 24.04.1, and 6.8.0-1013-oem.

ManuelB, 2024-10-23:
I have the following problem with this set up:

```
sudo dmesg -T | grep -C 20 error
...
[Mi Okt 23 09:44:45 2024] intel-ipu6 0000:00:05.0: FW version: 20230925
[Mi Okt 23 09:44:45 2024] intel-ipu6 0000:00:05.0: Device 0x7d19 (rev: 0x4)
[Mi Okt 23 09:44:45 2024] intel-ipu6 0000:00:05.0: physical base address 0x5019000000
[Mi Okt 23 09:44:45 2024] intel-ipu6 0000:00:05.0: mapped as: 0x00000000a523fb81
[Mi Okt 23 09:44:45 2024] intel-ipu6 0000:00:05.0: IPU in secure mode
[Mi Okt 23 09:44:45 2024] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[Mi Okt 23 09:44:45 2024] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[Mi Okt 23 09:44:45 2024] intel-ipu6 0000:00:05.0: IPC reset done
[Mi Okt 23 09:44:45 2024] intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6epmtl_fw.bin failed with error -2
...
```

```
uname -ar
Linux manuel-XPS-13-9340 6.11.0-9-generic #9-Ubuntu SMP PREEMPT_DYNAMIC Mon Oct 14 13:19:59 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

```
cat /etc/*-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.10
DISTRIB_CODENAME=oracular
DISTRIB_DESCRIPTION="Ubuntu 24.10"
PRETTY_NAME="Ubuntu 24.10"
NAME="Ubuntu"
VERSION_ID="24.10"
VERSION="24.10 (Oracular Oriole)"
VERSION_CODENAME=oracular
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=oracular
LOGO=ubuntu-logo
```

Hardware:

# System Details Report
---

## Report details
- **Date generated:**                              2024-10-23 09:54:27

## Hardware Information:
- **Hardware Model:**                              Dell Inc. XPS 13 9340
- **Memory:**                                      64.0 GiB
- **Processor:**                                   Intel® Core™ Ultra 7 155H × 22
- **Graphics:**                                    Intel® Arc™ Graphics (MTL)
- **Disk Capacity:**                               512.1 GB

## Software Information:
- **Firmware Version:**                            1.10.0
- **OS Name:**                                     Ubuntu 24.10
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               47
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.11.0-9-generic

ManuelB, 2024-10-23:
I was able to get the camera running by installing nightly ubuntu builds from:

https://launchpad.net/~oem-solutions-group/+archive/ubuntu/intel-ipu6

```
sudo add-apt-repository ppa:oem-solutions-group/intel-ipu6
sudo apt update
sudo apt upgrade
```

[image]

almereyda, 2024-10-23:
Looking forward for Meteor Lake support to hit Fedora 41 as well, which was recently communicated to be yet incomplete. Your message suggests this will be available soon'ish.

Reference:

- [hansdegoede | IPU6 camera support in Fedora 41](https://hansdegoede.dreamwidth.org/28841.html)

romayiii, 2024-10-23:
> I was able to get the camera running by installing nightly ubuntu builds from:
> 
> https://launchpad.net/~oem-solutions-group/+archive/ubuntu/intel-ipu6
> 
> ```
> sudo add-apt-repository ppa:oem-solutions-group/intel-ipu6
> sudo apt update
> sudo apt upgrade
> ```

I'm not sure if I'm missing something, I did add the ppa and update/upgrade, but still nothing. IS there anything that needs to be explicitly "sudo apt install ..."?

Dell XPS 14 9440
OS Name: Ubuntu 24.10

ManuelB, 2024-10-23:
@romayiii it is only working for me in chrome e.g. cheese is not working.

[image]

For me the packages were installed automatically. You could try to install the debs manually:

```
sudo apt install gst-plugins-icamera  ipu6-camera-bins ipu6-camera-hal v4l2-relayd
```

cstickel, 2024-10-23:
@ManuelB @romayiii i don't think just because it's working on XPS 13 9340 it necessarily also works on XPS 14 9440.
IPU6 seems just to be extremely hardware specific. at least it sounds like in the post from @jwrdegoede which @almereyda shared.

do you know which sensor is in the XPS 13 9340 @ManuelB?
XPS 14 9440 seems to be OV02C10, if i get it right.

for me it's not working with Ubuntu 24.10 and latest from ppa:oem-solutions-group/intel-ipu6 and ppa:oem-solutions-group/intel-ipu7 as suggested on https://wiki.ubuntu.com/IntelMIPICamera

ManuelB, 2024-10-23:
@cstickel this is what I see:

```
$ v4l2-ctl --list-devices
ipu6 (PCI:pci:pci0000:00):
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8

ipu6 (pci:pci0000:00):
	/dev/media0

Intel MIPI Camera (platform:v4l2loopback-000):
	/dev/video0
```

romayiii, 2024-10-23:
Thanks @cstickel and @ManuelB , indeed seems not to be working on the XPS 14 9440 . I found some options here, but also did not work for this one, but maybe for others is useful: https://wiki.ubuntu.com/Dell

cstickel, 2024-10-23:
@ManuelB can you send the output of `ls /sys/bus/i2c/devices` as written in the post @almereyda mentioned above?

stratism, 2024-10-23:
For me 6.11.0-9-generic Ubuntu 24.10 and on a DELL Latitude 7450 doesn't work either. Tried all the above but no luck

ManuelB, 2024-10-23:
@cstickel 
```
ls /sys/bus/i2c/devices
i2c-0  i2c-10  i2c-12  i2c-14  i2c-16  i2c-18  i2c-3  i2c-5  i2c-7  i2c-9            i2c-OVTI02C1:00
i2c-1  i2c-11  i2c-13  i2c-15  i2c-17  i2c-2   i2c-4  i2c-6  i2c-8  i2c-HIMX1092:00  i2c-VEN_04F3:00
```

cstickel, 2024-10-28:
ah, thx, @ManuelB...
as it seems it's also OV02C10... so looks like dell is using quite the same hardware. i would expect, that it should then also work on xps 14 9400 if it's working for you.

romayiii, 2025-01-21:
So, after googling again I found this thread which made an improvement for me, I can now view the camera stream in Chrome (but not yet in guvcview for example. same as @ManuelB) https://www.reddit.com/r/DellXPS/comments/1b9ddhz/comment/lrwq1aa/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1 

The thread is for Ubuntu 24.10 (answer by https://www.reddit.com/user/Logicwax/), so I had to figure out some changes for my Ubuntu 24.04 

In my case the commands that helped are: 

1. sudo apt install linux-modules-ipu6-6.11.0-1012-oem
2. sudo add-apt-repository ppa:oem-solutions-group/intel-ipu6
3. sudo apt update
4. sudo apt install libcamhal0 libcamhal-ipu6ep0 gstreamer1.0-icamera v4l2-relayd
5. Go to "Additional drivers" → under "Intel corporation" → check / enable "Use Dynamic loading plugin for IPU6 camera..."
6. Reboot
7. goto [webcamtests.com/check](http://webcamtests.com/check) to verify your webcam works.
8. If not, for some reason... reboot again... This is also mentioned in one of the answers...

mkabac-ikano, 2025-03-14:
Thank you for the solution @romayiii. It works like a charm on 24.04 ! 2 reboots necessary in my case.

fagalliu, 2025-03-23:
@romayiii, thanks a lot. It worked for me as well. Cheese and Camera apps do not work but I use google meet and it works.