intel/ipu6-drivers #226: Fedora 40 issues with sensor ov02c10 and custom repository to track master branches

scaronni, 2024-05-08:
Dell Inspiron 5680 with Alder/Raptor Lake PCI devices and `ov02c10` sensor:
```
$ lspci | grep "Multimedia controller"
00:05.0 Multimedia controller: Intel Corporation Device a75d
$ lsmod | grep ^ov02c10
ov02c10                28672  0
```
I'm at the point where I reach this state, but can't figure out what is missing to get the camera working:
```
$ GST_DEBUG=1 gst-launch-1.0 icamerasrc device-name=ov02c10-uf ! video/x-raw,format=NV12,width=1280,height=720 ! autovideoconvert ! ximagesink
[...]
[05-08 09:40:15.389] CamHAL[INF] updateMediaFormat, media format in test repositorytuning: 0, media format for aic 0.
[05-08 09:40:15.390] CamHAL[ERR] Failed to find DevName for cameraId: 6, get video node: ov02c10 , devname: /dev/v4l-subdev1
[05-08 09:40:15.392] CamHAL[ERR] Failed to find DevName for cameraId: 6, get video node: ov02c10 , devname: /dev/v4l-subdev1
[05-08 09:40:15.392] CamHAL[ERR] init: Init privacy checker falied
0:00:02.643895227 14783 0x5556bf3385e0 ERROR             icamerasrc gstcamerasrc.cpp:2874:gst_camerasrc_start: CameraId=6 failed to open libcamhal device.
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
[05-08 09:40:15.757] CamHAL[WAR] Seems camera device 6 is not closed properly (pid 14783).
```

I've run out of ideas to make this work, so if anyone has any idea I would be grateful to hear something.

After trying with the RPMFusion packages, the repackage of those from [Mathias Sou](https://thias.marmotte.net/2024/01/fedora-ipu6-camera-support/) and nothing was working, I've decided to try the master branch of all components, which meant repackaging everything due to differences. I took it as an exercise to get a deep understanding of this pile of... "work".

I've created a [test repository](https://negativo17.org/repos/ipu6-f40-x86_64/) for Fedora 40 x86_64 that containes `ipu6-drivers`, `ipu6-camera-bins`, `ipu6-camera-hal` and `icamerasrc` all built from the master branches plus the various patches from @jwrdegoede. I've been following his awesome work with `libcamera`, upstream kernel patches and integrations and all the stuff that is going here, and without him we would not be where we stand now. These packages contain only the packages mentioned above, not the `v4l2loopback/v4lrelayd` stuff as I'm still stuck to get the basic GStreamer stuff to work for the moment.

To install and test:
```
$ sudo wget -c https://negativo17.org/repos/ipu6-f40-x86_64.repo -O /etc/yum.repos.d/ipu6-f40-x86_64.repo
$ sudo dnf install ipu6-driver gstreamer1-plugin-icamerasrc
Last metadata expiration check: 0:01:12 ago on Wed 08 May 2024 09:52:05 AM CEST.
Dependencies resolved.
============================================================================================
 Package                       Arch    Version                       Repository        Size
============================================================================================
Installing:
 gstreamer1-plugin-icamerasrc  x86_64  0-1.20240411git9b2f7e3.fc40   ipu6-f40-x86_64   97 k
 ipu6-camera-hal               x86_64  0-1.20240411gitf073cb6.fc40   ipu6-f40-x86_64  7.7 M
Installing dependencies:
 akmod-ipu6                    x86_64  0-1.20240418git71e0c69.fc40   ipu6-f40-x86_64  305 k
 akmod-ivsc                    x86_64  0-1.20240416gita6dccbb.fc40   ipu6-f40-x86_64   63 k
 intel-vsc-firmware            noarch  20240410-1.fc40               updates          7.5 M
 ipu6-camera-bins              x86_64  0-1.20240411git814c869.fc40   ipu6-f40-x86_64  2.0 M

Transaction Summary
============================================================================================
Install  6 Packages

Total download size: 18 M
Installed size: 73 M
Is this ok [y/N]: 
```
If you prefer DKMS:
```
$ sudo dnf install dkms-ipu6 dkms-ivsc gstreamer1-plugin-icamerasrc
Last metadata expiration check: 0:02:02 ago on Wed 08 May 2024 09:52:05 AM CEST.
Dependencies resolved.
==============================================================================================
 Package                       Arch    Version                         Repository        Size
==============================================================================================
Installing:
 dkms-ipu6                     noarch  0-1.20240418git71e0c69.fc40     ipu6-f40-x86_64  227 k
 dkms-ivsc                     noarch  0-1.20240416gita6dccbb.fc40     ipu6-f40-x86_64   57 k
 gstreamer1-plugin-icamerasrc  x86_64  0-1.20240411git9b2f7e3.fc40     ipu6-f40-x86_64   97 k
Installing dependencies:
 dkms                          noarch  3.0.13-1.fc40                   fedora            87 k
 intel-vsc-firmware            noarch  20240410-1.fc40                 updates          7.5 M
 ipu6-camera-bins              x86_64  0-1.20240411git814c869.fc40     ipu6-f40-x86_64  2.0 M
 ipu6-camera-hal               x86_64  0-1.20240411gitf073cb6.fc40     ipu6-f40-x86_64  7.7 M

Transaction Summary
==============================================================================================
Install  7 Packages

Total download size: 18 M
Installed size: 75 M
Is this ok [y/N]: 
```

Then just reboot, everything module and binary wise should be properly installed.

Main differences from @jwrdegoede / RPMFusion packages:
- Based on master branch, `libcamhal` is now per-architecture (i.e. different between `ipu_adl`, `ipu_mtl`, etc.) and some more small building difference.
- There are both DKMS and akmod variations of the kernel module packages, up to the user what to install.
- I've used a simple `ldconfig` configuration file to find the libraries in the `udev` [rule](https://github.com/negativo17/ipu6-camera-hal/blob/master/60-intel-ipu6.rules), I found the `udev` rule + RPATH too complicated for my taste.
- It's using Fedora's own IVSC firmware (upstream) files from the main repositories.

Source code is here:
- https://github.com/negativo17/ipu6-camera-bins
- https://github.com/negativo17/ipu6-camera-hal
- https://github.com/negativo17/gstreamer1-plugin-icamerasrc
- https://github.com/negativo17/dkms-ivsc
- https://github.com/negativo17/dkms-ipu6
- https://github.com/negativo17/ivsc-kmod
- https://github.com/negativo17/ipu6-kmod

scaronni, 2024-05-08:
Side question, how does one know the PCI ids of the IPU6 "Multimedia Controller" devices without having the devices at hand? I see for example that we don't know the PCI ID for Meteor Lake. Is there a list somewhere?

bigerl, 2024-05-24:
Thx for the work. I am facing a similar issue with my DELL Precision 5490. 

The exact module is 
```
lspci | grep "Multimedia controller"
0000:00:05.0 Multimedia controller: Intel Corporation Device 7d19 (rev 04)
```

It should be the relevant portion of lsmod:
```
lsmod | grep ov01a10
ov01a10                28672  1
intel_vsc              16384  3 mei_ace,mei_csi,ov01a10
v4l2_fwnode            36864  2 ov01a10,intel_ipu6_isys
v4l2_async             32768  3 v4l2_fwnode,ov01a10,intel_ipu6_isys
videodev              417792  6 v4l2_async,v4l2_fwnode,videobuf2_v4l2,ov01a10,v4l2loopback,intel_ipu6_isys
mc                     94208  7 v4l2_async,videodev,snd_usb_audio,videobuf2_v4l2,ov01a10,intel_ipu6_isys,videobuf2_common
```

Let me know if I can help you with testing something.

scaronni, 2024-05-25:
Well, if you can tell me what's the magic sauce to make it work :)

You have the `v4l2` modules already installed on your system, so I guess you are using the RPMFusion packages. I'm not there yet, as I can't even make the camera work in the previous step (i.e. before `v4l2loopback`).

bigerl, 2024-05-25:
Oh, that's a misunderstanding. I got these devices from following the instructions there: https://hansdegoede.livejournal.com/27276.html So yes, RPMFusion. 

Didn't help me anything, though. Only the status light of the cam was enabled randomly on during startup. Cam was still not working. 

I just removed the following packages:
```
  akmod-v4l2loopback-0.13.1-1.fc40.x86_64                                       gstreamer1-plugins-icamerasrc-0.0-9.20231023git528a6f1.fc40.x86_64                                 
  ipu6-camera-bins-0.0-10.20231027gitaf5ba0c.fc40.x86_64                        ipu6-camera-bins-devel-0.0-10.20231027gitaf5ba0c.fc40.x86_64                                       
  ipu6-camera-hal-0.0-17.20240226gitda2e282.fc40.x86_64                         ipu6-camera-hal-devel-0.0-17.20240226gitda2e282.fc40.x86_64                                        
  ivsc-firmware-0.0-7.20230811git10c214f.fc40.x86_64                            kmod-intel-ipu6-6.8.10-300.fc40.x86_64-0.0-13.20240226gitfb4c170.fc40.x86_64                       
  kmod-v4l2loopback-0.13.1-1.fc40.x86_64                                        kmod-v4l2loopback-6.8.10-300.fc40.x86_64-0.13.1-1.fc40.x86_64                                      
  v4l2-relayd-0.1.2-12.20220126git2e4d5c9.fc40.x86_64                           v4l2loopback-0.13.1-1.fc40.x86_64   
```

Now, its gone from lsmod:
```
lsmod | grep ov01a10
```

Before, cheese found a couple of icu6 devices that did not work. No it finds none.

Gren-95, 2024-08-23:
anyone found a solution?

scaronni, 2024-08-24:
Still nothing. Rebased constantly to their upstream master  + whatever patches from Hans, went from kernel 6.8 to 6.10 and i still get a kernel panic when the modules load.

bigerl, 2024-08-25:
For fedora 41, support is planned: https://discussion.fedoraproject.org/t/f41-change-proposal-ipu6-camera-support-self-contained/124329

At least for applications that access the camera via pipewire.

scaronni, 2024-08-25:
Yes, but the situation is not so simple, as you can see from the change page "Different IPU6 laptop models use different sensors, hw-enablement needs to be done on a laptop by laptop basis". My laptop uses the `ov02c10` module, which at the moment is still out of tree (so inside `ipu6-drivers`) and using the out of tree `ipu6`. There are plenty of camera modules still out of tree.

Long story short: I'm not expecting this to work anytime soon and loading the external modules I guess will result in instability for a very long time...

scaronni, 2024-08-25:
"Fedora 41 has added support for IPU6 cameras on laptops using ov2740, ov01a10 and hi556 sensors. "

These are the currently out of tree modules:
```
export CONFIG_VIDEO_HM11B1 = m
export CONFIG_VIDEO_OV01A1S = m
export CONFIG_VIDEO_OV02C10 = m
export CONFIG_VIDEO_OV02E10 = m
export CONFIG_VIDEO_HM2170 = m
export CONFIG_VIDEO_HM2172 = m
export CONFIG_VIDEO_GC5035 = m
export CONFIG_ICAMERA_OV2740 = m
export CONFIG_ICAMERA_OV01A10 = m
export CONFIG_ICAMERA_HI556 = m
export CONFIG_VIDEO_OV05C10 = m
```