intel/ipu6-drivers #228: Not working on Ubuntu 24.04 with kernel 6.8

rmasad, 2024-05-16:
Previously, I managed to get the camera working with this guide on Ubuntu 23.04: https://dshedd.com/2023/07/18/fixing-lenovos-mipi-camera-problems-on-ubuntu-22-04/

Today, I updated to Ubuntu 24.04 (kernel 6.8.0-31-generic) and my camera stopped working. I repeated the steps and still have problems.

AntiNSA, 2024-05-21:
I have a dell precision 5680 and it is now working in 24.04 too...

AntiNSA, 2024-05-21:
> What kernel version do you have?
> Any extra step do you do that fix the camara?

I've tried 6.8 and 6.9 and 6.91 and it doesn't work no matter what I do.

rmasad, 2024-05-21:
What's more strange is that I go back to kernel 6.5.0-35-generic (where it worked before) and it doesn't work anymore.

yorik, 2024-05-22:
I was silly enough to upgrade kernel to 6.8 and the driver the latest and my camera doesn't work anymore:
```
int3472-discrete INT3472:06: cannot find GPIO chip INTC1096:00, deferring
```

Before I had kernel 6.6 for 3 months and it worked fine. Unfortunately rolling back to kernel 6.6 and even all 4 ipu6 repos to ~February did help.
The laptop: X1 Carbon Gen 11.
Debian sid with patched kernel.

The9Cat, 2024-05-22:
Ubuntu 24.04 with kernel and packages from the 6.8.0-32 series (currently in noble-proposed) do work out of the box.   To get the noble-proposed packages,  one need to select Software & Updates->Developer Options->pre-release updates, apt update, and apt install the 6.8.0-32 linux image, module, and dkms packages. These image and module packages contain the upstream kernel patches and intel-ipu6 drivers. 

For example, it just works with zoom.  

It is not perfect, however. Sometimes the video hangs and needs to be reset.  Sometimes these hangs prevent the system from suspending which requires a reboot.  But it's usable.  Hoping that this will soon be reliable?

heredia21, 2024-05-23:
> Ubuntu 24.04 with kernel and packages from the 6.8.0-32 series (currently in noble-proposed) do work out of the box. To get the noble-proposed packages, one need to select Software & Updates->Developer Options->pre-release updates, apt update, and apt install the 6.8.0-32 linux image, module, and dkms packages. These image and module packages contain the upstream kernel patches and intel-ipu6 drivers.
> 
> For example, it just works with zoom.
> 
> It is not perfect, however. Sometimes the video hangs and needs to be reset. Sometimes these hangs prevent the system from suspending which requires a reboot. But it's usable. Hoping that this will soon be reliable?
> 
> I should say: I have a Carbon X11 Gen 11. For some reason, this hardware is particularly finicky by combining experiences of others.

Hmmm. No updates available after toggling pre-release updates then apt update.

The9Cat, 2024-05-23:
I think you need to `apt install` the packages directly.  Not an `apt upgrade`.   So stuff like:
```
apt install linux-generic-hwe-24.04/noble-proposed linux-headers-generic-hwe-24.04/noble-proposed
apt install linux-modules-ipu6-6.8.0-32-generic/noble-proposed
```

Hopefully, I didn't miss one (this is by memory).  I also seemed to need:
```
apt reinstall intel-ipu6-dkms

```
Not 100% sure that this is right.  But it worked.

AntiNSA, 2024-05-23:
What if you are using mainline 6.9.1 ?

rmasad, 2024-05-23:
@The9Cat I tried what you proposed and nothing.

What catches my attention is that when doing v4l2-ctl --list-devices I get only one device (and not the 3 that appear in the blog post I sent before):

``` bash
$ v4l2-ctl --list-devices

Intel MIPI Camera (platform:v4l2loopback-000):
	/dev/video0
```

heredia21, 2024-05-23:
> I think you need to `apt install` the packages directly. Not an `apt upgrade`. So stuff like:
>
> ```
> apt install linux-generic-hwe-24.04/noble-proposed linux-headers-generic-hwe-24.04/noble-proposed [...]

Thanks. But this caused my trackpad to not work and a few issues with the screen itself. Rolled back to 6.8.0-31.

rmasad, 2024-05-26:
@The9Cat the output of `sudo grep ipu6 /var/log/dmesg` is:

```
[    4.015205] kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    4.015602] kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    4.015627] kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    4.015629] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x000000002c53b4a4
[    4.015698] kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.015700] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.015702] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    4.017795] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    4.017800] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    4.018682] kernel: intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6ep_fw.bin failed with error -2
[    4.018689] kernel: intel-ipu6 0000:00:05.0: Requesting signed firmware failed
[    4.018970] kernel: intel-ipu6: probe of 0000:00:05.0 failed with error -2
```

The9Cat, 2024-05-26:
That's strange.  I thought that issue was fixed in the latest source but maybe not.  I had to copy `/lib/firmware/intel/ipu/ipu6ep_fw.bin.zst` to `/lib/firmware/intel` .  Maybe do that and try again?

rmasad, 2024-05-26:
It works on kernel 6.5.0-35 but not on 6.8.0-32.

I leave the logs of `sudo grep ipu6 /var/log/dmesg`:

```
[    3.833691] kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    3.833857] kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    3.833878] kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    3.833880] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x00000000adfa0952
[    3.834075] kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.834078] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.834080] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.852020] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    3.852024] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    3.857344] kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.860740] kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    3.860756] kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    3.860758] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x00000000adfa0952
[    3.860804] kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.860807] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.860809] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.865971] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    3.865975] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    3.869159] kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.871462] kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    3.871479] kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    3.871481] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x0000000052e114ab
[    3.871567] kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.871569] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.871570] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.874609] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    3.874629] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    3.878413] kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.901518] kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    3.901540] kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    3.901543] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x00000000adfa0952
[    3.901600] kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.901604] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.901606] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.903949] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    3.903953] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    3.908547] kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.914126] kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    3.914140] kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    3.914141] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x0000000052e114ab
[    3.914189] kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.914194] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.914196] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.916960] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    3.916964] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    3.919755] kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.921958] kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    3.921977] kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    3.921979] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x0000000052e114ab
[    3.922033] kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.922036] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.922038] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.923825] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    3.923828] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    3.928858] kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.942759] kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    3.942781] kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    3.942783] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x0000000052e114ab
[    3.942873] kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.942875] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.942876] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.944701] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    3.944705] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    3.948861] kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.956970] kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[... 44 lines cut ...]
[    4.090575] kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    4.090606] kernel: intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[    4.090607] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x0000000052e114ab
[    4.090692] kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.090694] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.090695] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    4.092508] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    4.092511] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[... cut ...] ass joydev input_leds mac_hid serio_raw v4l2loopback(O) msr snd_aloop snd_pcm snd_timer videodev mc parport_pc ppdev lp parport efi_pstore nfnetlink dmi_sysfs ip_tables x_tables autofs4 intel_ishtp_hid hid_multitouch hid_generic nvme ucsi_acpi intel_lpss_pci snd nvme_core typec_ucsi xhci_pci intel_ish_ipc intel_lpss psmouse i2c_hid_acpi crc32_pclmul soundcore idma64 intel_ishtp xhci_pci_renesas nvme_auth typec thunderbolt i2c_hid ledtrig_audio video hid platform_profile wmi pinctrl_tigerlake
```

yger, 2024-05-26:
Same here, noble 24.04 and no camera :-(

However I don't have the same logs

[    2.978150] kernel: intel_ipu6: loading out-of-tree module taints kernel.
[    2.978157] kernel: intel_ipu6: module verification failed: signature and/or required key missing - tainting kernel
[    2.982148] kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    2.982299] kernel: intel-ipu6 0000:00:05.0: Device 0x465d (rev: 0x2)
[    2.982325] kernel: intel-ipu6 0000:00:05.0: physical base address 0x618e000000
[    2.982327] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x00000000e6481492
[    2.982381] kernel: intel-ipu6 0000:00:05.0: Unable to set secure mode
[    2.982385] kernel: intel-ipu6 0000:00:05.0: IPU in non-secure mode
[    2.982387] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    2.982388] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    2.982727] kernel: intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
[    2.982730] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    2.982732] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    2.984033] kernel: intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6ep_fw.bin failed with error -2
[    2.984040] kernel: intel-ipu6 0000:00:05.0: Requesting signed firmware failed

The package was install via apt reinstall intel-ipu6-dkm

The9Cat, 2024-05-26:
@yger I believe that you need the following steps:

1. Enable proposed updates in your Software & Updates->Developer Options->pre-release updates
2. Then `apt update`, and `apt install` the 6.8.0-35 linux image and modules like this:
```
sudo apt install linux-image-6.8.0-35-generic linux-headers-6.8.0-35-generic linux-modules-6.8.0-35-generic  linux-modules-extra-6.8.0-35-generic linux-modules-ipu6-6.8.0-35-generic  linux-tools-6.8.0-35 
sudo apt reinstall intel-ipu6-dkms
```
3. Also note: there is a minor issue with the driver and the firmware location.  The `linux-firmware` package is putting the firmware in `/lib/firmware/intel/ipu` but the driver is expecting it in `/lib/firmware/intel`.  So copy `/lib/firmware/intel/ipu/ipu6ep_fw.bin.zst` to `/lib/firmware/intel`.  This directory issue has been noted by a develper and is supposed to be fixed upstream.  So eventually, this workaround will not be necessary.

Also, you man or may not need all of these, depending on your use case, and I needed the `apt reinstall`.

Also, just so everyone knows: I'm just a user trying to provide info to the developers to get this thing fixed.  So I'm sharing my experience to help anyone who's willing to try but I'm not in a position to really fix this myself.

The9Cat, 2024-05-26:
@rmasad I've also seen stuff like this.  You might try rebooting again, just to test.  I've found that most of the time it boots correctly but not every time.  Some sort of hardware timing issue?  Not sure.  I'm currently finding it to be mostly reliable but not 100%.

yger, 2024-05-26:
Thanks a lot for the help, really appreciated. Sadly however, I followed all your suggestion and the problem remains. I now have kernel 6.8.0.35,  but sill the following

[    2.984544] kernel: intel_ipu6: loading out-of-tree module taints kernel.
[    2.984553] kernel: intel_ipu6: module verification failed: signature and/or required key missing - tainting kernel
[    2.991590] kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    2.992194] kernel: intel-ipu6 0000:00:05.0: Device 0x465d (rev: 0x2)
[    2.992218] kernel: intel-ipu6 0000:00:05.0: physical base address 0x618e000000
[    2.992219] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x00000000a751386a
[    2.992267] kernel: intel-ipu6 0000:00:05.0: Unable to set secure mode
[    2.992268] kernel: intel-ipu6 0000:00:05.0: IPU in non-secure mode
[    2.992269] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    2.992271] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    2.992439] kernel: intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
[    2.992442] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    2.992443] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    2.997230] kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.000422] kernel: intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[    3.043934] kernel: intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[    3.044071] kernel: intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[    3.661015] kernel:  ipu_isys_queue_cleanup+0x12/0x20 [intel_ipu6_isys]
[    3.661030] kernel:  ipu_isys_video_cleanup+0x2e/0x40 [intel_ipu6_isys]
[    3.661042] kernel:  ipu_isys_csi2_be_soc_cleanup+0x37/0x50 [intel_ipu6_isys]
[    3.661055] kernel:  isys_unregister_subdevices+0x40/0xa0 [intel_ipu6_isys]
[    3.661067] kernel:  isys_probe+0x5c5/0xb80 [intel_ipu6_isys]
[    3.661082] kernel:  ipu_bus_probe+0x72/0x130 [intel_ipu6]
[    3.661115] kernel:  ? __pfx_isys_driver_init+0x10/0x10 [intel_ipu6_isys]
[    3.661128] kernel:  ipu_bus_register_driver+0x16/0x20 [intel_ipu6]
[    3.661137] kernel:  isys_driver_init+0x15/0xff0 [intel_ipu6_isys]
[    3.661214] kernel: Modules linked in: btintel(+) crct10dif_pclmul snd_rawmidi intel_ipu6_isys(OE+) polyval_clmulni libarc4 polyval_generic btbcm ghash_clmulni_intel videobuf2_dma_contig btmtk hid_sensor_custom_intel_hinge hid_sensor_als hid_sensor_gyro_3d nvidia_drm(POE+) sha256_ssse3 snd_seq dell_wmi(+) hid_sensor_trigger videobuf2_memops sha1_ssse3 snd_seq_device bluetooth nvidia_modeset(POE) videobuf2_v4l2 industrialio_triggered_buffer drm_buddy processor_thermal_device_pci(+) iwlwifi aesni_intel dell_smbios snd_timer kfifo_buf processor_thermal_device ttm videobuf2_common dell_wmi_sysman(+) ecdh_generic hid_sensor_iio_common processor_thermal_wt_hint crypto_simd dcdbas nvidia(POE) cryptd rapl intel_cstate processor_thermal_rfim ecc usb_ljca dell_smm_hwmon dell_wmi_ddv ledtrig_audio intel_ipu6_psys(OE) dell_wmi_descriptor spi_intel_pci firmware_attributes_class wmi_bmof industrialio snd drm_display_helper processor_thermal_rapl i2c_i801 ov01a10(OE) cfg80211 spi_intel mei_me i2c_smbus intel_rapl_common soundcore cec
[    3.661250] kernel:  v4l2_fwnode intel_skl_int3472_tps68470 processor_thermal_wt_req processor_thermal_power_floor v4l2_async mei tps68470_regulator rc_core int3403_thermal processor_thermal_mbox clk_tps68470 i2c_algo_bit intel_ipu6(OE) mei_vsc_hw int340x_thermal_zone igen6_edac videodev intel_pmc_core mc intel_skl_int3472_discrete intel_hid input_leds joydev intel_vsec int3400_thermal pmt_telemetry sparse_keymap pmt_class acpi_tad acpi_pad acpi_thermal_rel mac_hid serio_raw msr parport_pc ppdev lp parport efi_pstore nfnetlink dmi_sysfs ip_tables x_tables autofs4 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 hid_sensor_custom hid_sensor_hub intel_ishtp_hid nvme nvme_core hid_multitouch nvme_auth hid_generic spi_pxa2xx_platform ahci dw_dmac dw_dmac_core libahci 8250_dw rtsx_pci_sdmmc video crc32_pclmul ucsi_acpi psmouse thunderbolt typec_ucsi intel_lpss_pci intel_ish_ipc xhci_pci intel_lpss rtsx_pci intel_ishtp vmd xhci_pci_renesas idma64 typec i2c_hid_acpi i2c_hid

I'm using a Dell Precision 5470, and loooking forward for a possible fix!

The9Cat, 2024-05-26:
Hm.  I don't have a good answer for this.  A few things/questions come to mind:
1. I wonder what's up with the 'out-of-tree' module message.   Which module package did you install?  Is it possible that some previous out-of-tree module package is in your dkms config?
2. Your Dell has a different sensor than my machine (Lenovo X1C11).   Not sure that's significant here, but maybe.

MarjanHJ, 2024-05-30:
did you manage to solve this , I have the same issue and tried many mehtods but no lucks here is the output of my sudo grep ipu6 /var/log/dmesg:

[    4.842580] kernel: intel_ipu6: loading out-of-tree module taints kernel.
[    4.842585] kernel: intel_ipu6: module verification failed: signature and/or required key missing - tainting kernel
[    4.866168] kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    4.866643] kernel: intel-ipu6 0000:00:05.0: Device 0x7d19 (rev: 0x4)
[    4.866652] kernel: intel-ipu6 0000:00:05.0: physical base address 0x5229000000
[    4.866654] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x0000000000d78e90
[    4.866726] kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.866728] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[    4.866729] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    4.878670] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    4.878677] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6epmtl_fw.bin
[    4.878738] kernel: intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6epmtl_fw.bin failed with error -2
[    4.878741] kernel: intel-ipu6 0000:00:05.0: Requesting signed firmware failed
[    4.878754] kernel: intel-ipu6 0000:00:05.0: probe with driver intel-ipu6 failed with error -2

The9Cat, 2024-05-31:
@hao-yao I believe that the Ubuntu package maintainer has integrated your kernel patches, the in-tree ivsc driver, and ipu6 drivers into the 6.8.0-35 series now in noble-proposed.   Is this any way for you to check if this is enough?   It does seem to work, although it occasionally times out.  Restarting the v4l2-relayd often gets it working, but not always. 
 In some cases, the system requires a reboot to allow a suspend.  The suspend problem is occasional; many times it just works as expected.

The9Cat, 2024-05-31:
@MarjanHJ As in my e-mails above, you might try the 6.8.0-35 series now in noble-proposed.  It does work for me (most of the time) but as you can see by reading this thread, it does not work for everyone.  Also note: there is some confusion about the fw location.  You might try, as I suggested [here](https://github.com/intel/ipu6-drivers/issues/228#issuecomment-2132043858), copying the fw file that you need for your sensor, in this case `/lib/firmware/intel/ipu/ipu6epmtl_fw.bin.zst` up to `/lib/firware/intel`.

MarjanHJ, 2024-06-01:
@The9Cat thanks for your response, I have already updated my kernel to 6.9 because my audio was not working with kernel 6.8 as well , fortunately audio is working now with 6.9 mainline but camera is still not working, I have also copied the firmware file and didn't work, I hope someone can suggest a way to solve this. thanks

rmasad, 2024-06-02:
6.8.0-35 still not working

rawler, 2024-06-03:
Also struggling with getting ipu6 camera working on noble. My HW is Dell Precision 5680. Related to the claim by @The9Cat that it should "work out of the box", I fail to understand _how_ exactly?

I thought these cameras requires the gstreamer-icamera package to work, and I cannot find any package containing this in standard noble repositories?

What packages should I need, from where?

MarjanHJ, 2024-06-03:
I also still haven't managed to fix this issue, my laptop is xps 16 , ubuntu 24.04, I updated the kernel to 6.9 and that fixed my audio but tried many things including copying firmware etc but camera is still not working.

luigimassa, 2024-06-03:
I've same problem but after upgrade in ubuntu 24 I've problem with `gstreamer1.0-icamera` (not exists) and v4l2loopback 
Sob!

rawler, 2024-06-03:
>  Rather I meant: the recent proposed Noble updates provide integrated packages with all that you need
>  ... You still need the `v4l2-relayd` package which uses Gstreamer support to map the device with a v4l2-loopback. This will install `gstreamer1.0-icamera` as a dependency.

How? As far as I can tell, `v4l2-relayd` only depends on `gstreamer1.0-plugins-base`, which does _not_ depend on `gstreamer1.0-icamera`? In fact, I cannot find any packaging of `gstreamer1.0-icamera` in the main archives? (Only in PPA)

So, if I'm understanding this right, the _kernel components_ from noble-proposed is expected to work out of the box (replacing ipu6-* and ivsc-*), but I still need the [oem-solutions PPA](https://launchpad.net/~oem-solutions-group/+archive/ubuntu/intel-ipu6) for the userland components?

The9Cat, 2024-06-04:
@rawler I did some digging and I see your point.  I do not have the intel-ipu6 PPA.  With apt, I find:
```
$ apt-cache rdepends v4l2-relayd 
v4l2-relayd
Reverse Depends:
  libcamhal0
$ apt-cache rdepends libcamhal0
libcamhal0
Reverse Depends:
  gstreamer1.0-icamera
$ apt policy gstreamer1.0-icamera 
gstreamer1.0-icamera:
  Installed: 0~git202404150111.3d9cd3a~ubuntu24.04.2
  Candidate: 0~git202404150111.3d9cd3a~ubuntu24.04.2
  Version table:
 *** 0~git202404150111.3d9cd3a~ubuntu24.04.2 100
        100 /var/lib/dpkg/status
```
This was a fresh install of 24.04, so no cruft from an previous versions can be there.  I note that this version gstreamer1.0-icamera is the one from the oem-solutions PPA, but I am *not* updating from that PPA.   At one point, I tried the noble-proposed oem kernel 6.8.0-32 for testing, but ended up using the generic flavor instead.  Wondering if that might have pulled those packages?

Give it a shot, maybe?

yger, 2024-06-04:
I'm a bit lost, and don't know really what to do. I still use my 6.8.0.35 kernel, but so far nothing regarding the camera side. Now, I have

sudo grep ipu6 /var/log/dmesg

[    4.401210] kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    4.401409] kernel: intel-ipu6 0000:00:05.0: Device 0x465d (rev: 0x2)
[    4.401444] kernel: intel-ipu6 0000:00:05.0: physical base address 0x618e000000
[    4.401446] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x000000003f50496d
[    4.401508] kernel: intel-ipu6 0000:00:05.0: Unable to set secure mode
[    4.401512] kernel: intel-ipu6 0000:00:05.0: IPU in non-secure mode
[    4.401514] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.401516] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    4.402279] kernel: intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
[    4.402284] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    4.402286] kernel: intel-ipu6 0000:00:05.0: cpd file name: ipu6ep_fw.bin
[    4.408430] kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
[    4.411200] kernel: intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[    4.473965] kernel: intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[    4.474115] kernel: intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0

So no more tainted kernel or anything. But no camera either. Should I go for an install of the module here on github via dkms, will it work? Or should I try a linux repo via apt (such as intel-ipu6-dkms). Will such an ipu6 module solve my camera issue?

The9Cat, 2024-06-04:
@rawler @yger It is possible that I had the oem solutions group PPA enabled at one point just after my initial installation, following the suggestions on the Ubuntu 22.04 Lenovo wiki web page.  Then I realized that the 6.8.0-31 kernel did not support the intel-ipu6 patches and modules.   So then I moved to the noble-proposed kernel which were supposed to include that new work.   However, I did not uninstall the `libcamhal0` package.  So if you want to experiment, perhaps try installing the `libcamhal0` package after enabling the PPA?   Just a thought.  It *is* working for me.  That's all I can say.  It seems that I stumbled up this solution accidentally.

hao-yao, 2024-06-05:
Hello folks,

Checked that the package intel-ipu6-dkms on Ubuntu 24.04 is `0~git202311240921.07f0612e-0ubuntu2`, which is outdated and don't work with kernel v6.8. Please clone the master branch directly and add to dkms manually:

```bash
git clone https://github.com/intel/ipu6-drivers.git
cd ipu6-drivers
sudo dkms add .
sudo dkms install -m ipu6-drivers/0.0.0
# Then reboot and check by sudo dmesg | grep -i ipu
# To remove:
# sudo dkms remove -m ipu6-drivers/0.0.0
```

I validated it works on Ubuntu 6.8.0-35-generic kernel.

luigimassa, 2024-06-05:
hello
I compiled and install v4l2loopback ( I had problems)
I cloned and installed ipu6-drivers.git
But I have this problem

```
 λ sudo grep ipu6 /var/log/dmesg                                           
[    5.560213] kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    5.560496] kernel: intel-ipu6 0000:00:05.0: Device 0x465d (rev: 0x6)
[    5.560517] kernel: intel-ipu6 0000:00:05.0: physical base address 0x6076000000
[    5.560519] kernel: intel-ipu6 0000:00:05.0: mapped as: 0x00000000e269010a
[    5.560568] kernel: intel-ipu6 0000:00:05.0: Unable to set secure mode
[    5.560572] kernel: intel-ipu6 0000:00:05.0: IPU in non-secure mode
[    5.560573] kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    5.560575] kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    5.560583] kernel: intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
[    5.560584] kernel: intel-ipu6 0000:00:05.0: IPC reset done
[    5.560585] kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    5.561609] kernel: intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6ep_fw.bin failed with error -2
[    5.561612] kernel: intel-ipu6 0000:00:05.0: Requesting signed firmware failed
[    5.561915] kernel: intel-ipu6: probe of 0000:00:05.0 failed with error -2
```
Thanks in advance

heredia21, 2024-06-05:
> Hello folks,
> 
> Checked that the package intel-ipu6-dkms on Ubuntu 24.04 is `0~git202311240921.07f0612e-0ubuntu2`, which is outdated and don't work with kernel v6.8. Please clone the master branch directly and add to dkms manually:
> 
> ```shell
> # Remove Ubuntu package intel-ipu6-dkms
> sudo apt purge intel-ipu6-dkms
> git clone https://github.com/intel/ipu6-drivers.git
> cd ipu6-drivers
> sudo dkms add .
> sudo dkms install -m ipu6-drivers/0.0.0
> # Then reboot and check by sudo dmesg | grep -i ipu
> # To remove:
> # sudo dkms remove -m ipu6-drivers/0.0.0
> ```
> 
> I validated it works on Ubuntu 6.8.0-35-generic kernel.

Thanks for these steps. My camera now works on Ubuntu 6.8.0-35-generic kernel following this.

heredia21, 2024-06-05:
> @heredia21 Did you need to do anything else besides these steps? Other packages perhaps? I have no idea what I've screwed up, but there are no `ipu6` kernel messages at all at this point in dmesg. I'm going to have to give up on this for now. My own fault for trying this change when I had a solution that worked. So disappointing.

I was on 6.8.0-31-generic, so changed it to 6.8.0-35-generic. That was the only difference.

The9Cat, 2024-06-05:
It's weird.  I worked for me earlier on 6.8.0-35-generic even though @hao-yao confirmed that it shouldn't.  And now the kernel seems to ignore the entire ipu6 driver.  It's probably something that is confused somewhere but I'm not seeing it.

heredia21, 2024-06-05:
I would remove everything and start over.

heredia21, 2024-06-05:
@The9Cat What do you get when you run dkms status?

The9Cat, 2024-06-05:
@heredia21 I get:
```
ipu6-drivers/0.0.0, 6.8.0-35-generic, x86_64: installed
```

heredia21, 2024-06-05:
This is what I get:
```

$ dkms status
ipu6-drivers/0.0.0, 6.5.0-14-generic, x86_64: built
ipu6-drivers/0.0.0, 6.8.0-31-generic, x86_64: installed
ipu6-drivers/0.0.0, 6.8.0-35-generic, x86_64: installed
ipu6-drivers/0.0.1: added
v4l2loopback/0.12.7, 6.8.0-31-generic, x86_64: installed (WARNING! Diff between built and installed module!)
v4l2loopback/0.12.7, 6.8.0-35-generic, x86_64: installed (WARNING! Diff between built and installed module!)
```

The9Cat, 2024-06-05:
@heredia21 Ah. so you have v4l2loopback-dkms installed?  Is that right?

The9Cat, 2024-06-05:
I tried using the dkms v4l2loopback and it tells me that it's the same version of the module as in 6.8.0-35.  Which is the same as your dkms status.  What do you find for `sudo dmesg | grep -i ipu`?

The9Cat, 2024-06-05:
I purged and reinstalled the kernel.  Then the camera started to work again.  Yay!  @heredia21: thanks for your suggestions and help!

It is still a bit flaky and needs occasional restarts of the v4l2-relayd as before.  But that's progress.  I'm not sure what was wrong before.  Some cruft perhaps?

MarjanHJ, 2024-06-06:
@hao-yao I followed this and it is still not working, though my kernel is 6.9 as my audio was not working with kernel 6.8, my laptop is xps 16 and Ubuntu 24.04, kernel 6.9, here is the out put of sudo dmesg | grep -i ipu , can you please advice on how I can fix this, thanks

[    0.006567] ACPI: SSDT 0x000000006DFDE000 0000F9 (v02 INTEL  IpuSsdt  00001000 INTL 20210930)
[    2.348319] intel_ipu6: loading out-of-tree module taints kernel.
[    2.348325] intel_ipu6: module verification failed: signature and/or required key missing - tainting kernel
[    2.368570] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    2.368974] intel-ipu6 0000:00:05.0: Device 0x7d19 (rev: 0x4)
[    2.368997] intel-ipu6 0000:00:05.0: physical base address 0x5229000000
[    2.368999] intel-ipu6 0000:00:05.0: mapped as: 0x00000000d4b6d3b0
[    2.369127] intel-ipu6 0000:00:05.0: IPU in secure mode
[    2.369129] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[    2.369131] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    2.381378] intel-ipu6 0000:00:05.0: IPC reset done
[    2.381685] intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6epmtl_fw.bin failed with error -2
[    2.386450] intel-ipu6 0000:00:05.0: FW version: 20230925
[    2.388857] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[    2.424884] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[    2.471875] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[    2.471949] intel-ipu6 0000:00:05.0: IPU6-v4 driver version 1.0
[    2.920402] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[    2.920567] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[    2.994906] intel-ipu6-isys intel-ipu6-isys0: no subdev found in graph

The9Cat, 2024-06-06:
@MarjanHJ Are you patching the kernel?  My understanding is that you would need the to patch the 6.9 kernel source with the patches and build the drivers as described [https://github.com/intel/ipu6-drivers](url).  I suspect that the patches are not yet in mainline but you could check.

rawler, 2024-06-06:
@hao-yao I purged all the existing "ipu6"-packages from my system and followed the instructions to install DKMS from git HEAD.

`modinfo intel_ipu6_isys` now says its loaded from `/lib/modules/6.8.0-35-generic/updates/dkms/intel-ipu6-isys.ko.zst`, which whose mtime is now 15 minutes old. (Should be the new DKMS version)

My problem seems similar to that of @yorik. Intermingled with ipu6-initialization are repeated rows from `int3472-discrete`:

```
[   31.743876] i801_smbus 0000:00:1f.4: SMBus using PCI interrupt
[   31.744342] intel-ipu6 0000:00:05.0: IPC reset done
[   31.746181] intel-ipu6 0000:00:05.0: FW version: 20230925
[   31.750813] int3472-discrete INT3472:06: cannot find GPIO chip INTC100B:00, deferring
[   31.751078] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[   31.751097] intel-ipu6 0000:00:05.0: physical base address 0x644e000000
[   31.751098] intel-ipu6 0000:00:05.0: mapped as: 0x00000000bec249fb
[   31.751135] intel-ipu6 0000:00:05.0: IPU in secure mode
[   31.751135] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[   31.751136] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[   31.752979] intel-ipu6 0000:00:05.0: IPC reset done
[   31.754103] intel-ipu6 0000:00:05.0: FW version: 20230925
[   31.757521] mei_me 0000:00:16.0: enabling device (0000 -> 0002)
[   31.760352] int3472-discrete INT3472:06: cannot find GPIO chip INTC100B:00, deferring
[   31.760958] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[   31.760975] intel-ipu6 0000:00:05.0: physical base address 0x644e000000
[   31.760977] intel-ipu6 0000:00:05.0: mapped as: 0x00000000bec249fb
[   31.761020] intel-ipu6 0000:00:05.0: IPU in secure mode
```

Eventually, int3472-discrete seems to give up

```
[   32.092349] intel-ipu6 0000:00:05.0: mapped as: 0x00000000239780ce
[   32.092445] intel-ipu6 0000:00:05.0: IPU in secure mode
[   32.092447] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[   32.092448] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[   32.094276] intel-ipu6 0000:00:05.0: IPC reset done
[   32.097579] intel-ipu6 0000:00:05.0: FW version: 20230925
[   32.145271] int3472-discrete INT3472:06: GPIO type 0x02 unknown; the sensor may not work
```

The ipu6 modules finally finish with

```
[   44.914306] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[   44.914382] intel-ipu6 0000:00:05.0: physical base address 0x644e000000
[   44.914387] intel-ipu6 0000:00:05.0: mapped as: 0x00000000b373d712
[   44.914615] intel-ipu6 0000:00:05.0: IPU in secure mode
[   44.914623] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[   44.914629] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[   44.932844] intel-ipu6 0000:00:05.0: IPC reset done
[   44.934479] intel-ipu6 0000:00:05.0: FW version: 20230925
[   44.935607] intel-ipu6 0000:00:05.0: Found supported sensor OVTI02C1:00
[   44.935636] intel-ipu6 0000:00:05.0: Connected 1 cameras
[   44.936470] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   44.946197] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   44.967930] ACPI Warning: \_SB.NPCF._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20230628/nsarguments-61)
[   44.968020] ACPI Warning: \_SB.PC00.PEG1.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20230628/nsarguments-61)
[   45.015143] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   45.015160] intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[   45.071644] intel-ipu6-isys intel-ipu6-isys0: bind Intel IVSC CSI nlanes is 2 port is 2
[   45.074478] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[   45.074768] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[   45.118364] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
```

`sudo v4l2-ctl --list-devices` outputs a big list of devices
```
ipu6 (PCI:pci:pci0000:00):
	/dev/video0
	/dev/video1
	/dev/video2
[... 54 lines cut ...]
[06-06 20:40:16.945] CamHAL[ERR] invalid media format, default value used.
[06-06 20:40:16.945] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev17 IOCTL VIDIOC_G_EXT_CTRLS error: Invalid argument
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[06-06 20:40:16.954] CamHAL[ERR] setup Link Intel IVSC CSI [225:0] ==> Intel IPU6 CSI-2 2 [7x0] enable 1 failed.
[06-06 20:40:16.954] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[06-06 20:40:16.954] CamHAL[ERR] set up mediaCtl failed
[06-06 20:40:16.954] CamHAL[ERR] @configure Device Configure failed
[06-06 20:40:16.954] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.005514349
Setting pipeline to NULL ...
[06-06 20:40:17.15] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev17 IOCTL VIDIOC_G_EXT_CTRLS error: Invalid argument
[06-06 20:40:17.15] CamHAL[ERR] Couldn't get V4L2_CID_PRIVACY, status:-1
[06-06 20:40:17.15] CamHAL[INF] threadLoop: mCameraStreams[0] == 0x77d230003ae0
[06-06 20:40:17.15] CamHAL[INF] threadLoop: getPrivacyBuffer returned nullptr
Freeing pipeline ...
```

Anything obvious that is wrong here?

hao-yao, 2024-06-07:
@MarjanHJ I didn't test the compatibility on kernel v6.9, but as @The9Cat commented you need at lease patch your v6.9 kernel following [README.md](https://github.com/intel/ipu6-drivers/blob/master/README.md).

> [ 2.994906] intel-ipu6-isys intel-ipu6-isys0: no subdev found in graph

This means ipu-bridge.c didn't include the changes under patch/v6.8/*.patch.

hao-yao, 2024-06-07:
@rawler For ipu6-drivers, when you see intel-ipu6-isys prints:
> intel-ipu6-isys intel-ipu6-isys0: bind Intel IVSC CSI nlanes is 2 port is 2
> intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.

It is loaded correctly, and no need to care other warnings of INT3472.

For this error:
> CamHAL[ERR] Couldn't get V4L2_CID_PRIVACY, status:-1

Please try to hack the configuration at maybe /etc/camera/ipu6ep/sensors/ov02c10-uf.xml, by changing
```xml
<supportPrivacy value="1"/>
```
to 
```xml
<supportPrivacy value="0"/>
```
This actually need a fix to ipu6-camera-hal to make it work with upstream IVSC driver + privacy mode support. I have fixed it internally, but release to Github and integration won't be so soon so please try this hack first.

juletx, 2024-06-07:
I have followed the instructions here but camera is still not working. I have installed ipu6-drivers and v4l2loopback.

 `dkms status`

```
ipu6-drivers/0.0.0, 6.8.0-35-generic, x86_64: installed
v4l2loopback/0.12.7, 6.8.0-35-generic, x86_64: installed (WARNING! Diff between built and installed module!)
```

`sudo v4l2-ctl --list-devices`
```
ipu6 (PCI:pci:pci0000:00):
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8

ipu6 (pci:pci0000:00):
	/dev/media0

Intel MIPI Camera (platform:v4l2loopback-000):
	/dev/video0
```

`sudo gst-launch-1.0 icamerasrc ! autovideosink`

```
[06-07 21:12:07.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.23] CamHAL[ERR] Parse AE eExposure time range failed
[06-07 21:12:07.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.23] CamHAL[ERR] Parse AE gain range failed
[06-07 21:12:07.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.23] CamHAL[ERR] Parse AE eExposure time range failed
[06-07 21:12:07.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.23] CamHAL[ERR] Parse AE gain range failed
[06-07 21:12:07.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.23] CamHAL[ERR] Parse AE eExposure time range failed
[06-07 21:12:07.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.23] CamHAL[ERR] Parse AE gain range failed
[06-07 21:12:07.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.24] CamHAL[ERR] Parse AE eExposure time range failed
[06-07 21:12:07.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.24] CamHAL[ERR] Parse AE gain range failed
[06-07 21:12:07.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.24] CamHAL[ERR] Parse AE eExposure time range failed
[06-07 21:12:07.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.24] CamHAL[ERR] Parse AE gain range failed
[06-07 21:12:07.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.24] CamHAL[ERR] Parse AE eExposure time range failed
[06-07 21:12:07.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.24] CamHAL[ERR] Parse AE gain range failed
[06-07 21:12:07.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.24] CamHAL[ERR] Parse AE eExposure time range failed
[06-07 21:12:07.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.24] CamHAL[ERR] Parse AE gain range failed
[06-07 21:12:07.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.24] CamHAL[ERR] Parse AE eExposure time range failed
[06-07 21:12:07.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.24] CamHAL[ERR] Parse AE gain range failed
[06-07 21:12:07.25] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.25] CamHAL[ERR] Parse AE eExposure time range failed
[06-07 21:12:07.25] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.25] CamHAL[ERR] Parse AE gain range failed
[06-07 21:12:07.25] CamHAL[ERR] Malformed ET range in exposure time range configuration
[06-07 21:12:07.25] CamHAL[ERR] Parse AE eExposure time range failed
[06-07 21:12:07.25] CamHAL[ERR] Malformed ET range in exposure time range configuration
[... 167 lines cut ...]
[06-07 21:12:07.727] CamHAL[ERR] invalid media format, default value used.
[06-07 21:12:07.727] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[06-07 21:12:07.727] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[06-07 21:12:07.727] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[06-07 21:12:07.727] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[06-07 21:12:07.727] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[06-07 21:12:07.737] CamHAL[ERR] Get entity fail for calling getEntityById
[06-07 21:12:07.737] CamHAL[ERR] Get entity fail for calling getEntityById
[06-07 21:12:07.737] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[06-07 21:12:07.737] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[06-07 21:12:07.737] CamHAL[ERR] set up mediaCtl failed
[06-07 21:12:07.737] CamHAL[ERR] @configure Device Configure failed
[06-07 21:12:07.737] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.007706088
Setting pipeline to NULL ...
Freeing pipeline ...
```

rawler, 2024-06-07:
> > CamHAL[ERR] Couldn't get V4L2_CID_PRIVACY, status:-1
> 
> Please try to hack the configuration at maybe /etc/camera/ipu6ep/sensors/ov02c10-uf.xml, by changing
> 
> to
> 
> `<supportPrivacy value="0"/>`
Tried this. It got rid of this specific error, but it still no working camera. It seems something critical crashes even before reaching this stage;

```
[06-06 20:40:16.945] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[06-06 20:40:16.945] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[06-06 20:40:16.945] CamHAL[ERR] invalid media format, default value used.
[06-06 20:40:16.945] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev17 IOCTL VIDIOC_G_EXT_CTRLS error: Invalid argument
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[06-06 20:40:16.954] CamHAL[ERR] setup Link Intel IVSC CSI [225:0] ==> Intel IPU6 CSI-2 2 [7x0] enable 1 failed.
[06-06 20:40:16.954] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[06-06 20:40:16.954] CamHAL[ERR] set up mediaCtl failed
[06-06 20:40:16.954] CamHAL[ERR] @configure Device Configure failed
[06-06 20:40:16.954] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.005514349
Setting pipeline to NULL ...
[06-06 20:40:17.15] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev17 IOCTL VIDIOC_G_EXT_CTRLS error: Invalid argument
[06-06 20:40:17.15] CamHAL[ERR] Couldn't get V4L2_CID_PRIVACY, status:-1
```

Like @juletx, I'm also seeing _many_ different v4l2 devices detected, which I found kindof strange, though I don't know enough about the v4l2 model to understand if that's sign of a problem or not:

`sudo v4l2-ctl --list-devices`
```
ipu6 (PCI:pci:pci0000:00):
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8

ipu6 (pci:pci0000:00):
	/dev/media0
```

The9Cat, 2024-06-09:
@rawler I don't think those multiple devnodes are a problem; they are the v4l2 metadata interfaces.   I have them as well, and the ipu6 camera works (most of the time) on my hardware.  In addition to those, I have the v4l2-loopback device which you do not seem to have.  Maybe check whether the v4l2loopback device is configured (e.g. check `dmesg` it should be reported as `kernel: v4l2loopback driver version 0.12.7 loaded`)?   I believe that it's in the kernel modules by default for 6.8.0-35, although some earlier versions required  `v4l2loopback-dkms`.

linuxcf, 2024-06-09:
Running 6.8.0-35 on Lenovo Yoga Gen 8 (64GB RAM, Raptor Lake CPU).   Initially, on stock 24.04LTS, I got only /dev/video0.  I followed @hao-yao 's  advice, unistalled Ubuntu's stock antiquated 'intel-ipu6-dkms' and installed the current one from github.  Now, I get the full gamut of devices, and /dev/video0 is listed as 'Virtual Camera (platform:v4l2loopback-000)'.  However, I can find no directory '/etc/camera' @hao-yao refers to, above.  And the camera does the exact weird thing it was doing before:  It outputs rectangles of different colors and sizes, and, on the bottom right, a small rectangle with "snow," like when the old analog TVs had no signal.  Programs recognize the camera: vlc does not complain, skype outputs the same thing (but mirrored, as it should in a real situation), and the web site testing the camera recognizes it, outputs the same thing, but displays NaN for the frame rate.  Prior to doing a clean install of 24.04LTS, I was running 22.04LTS, with kernel 6.5, and the camera was working, although I had to run 'v4l2-ctl -d 0 -D' and then restart service v4l2-relayd a couple of times to get it to work right (I found out by accident that only restarting v4l2-relayd was not enough; I had to run 'v4l2-ctl -d 0 -D' first for some unknown reason).  Doing this in my 24.04LTS now does not get the camera working.

rmasad, 2024-06-09:
I had to reinstall Ubuntu on this machine and since I installed Ubuntu 24.04 directly, the 6.5.0-generic kernel is no longer available. I tried to follow this blog, but it didn't work: https://dshedd.com/2023/07/18/fixing-lenovos-mipi-camera-problems-on-ubuntu-22-04/

An element that caught my attention is that unlike the 6.5 kernel, with 6.8 it cannot be installed (the package is not there): linux-modules-ivsc-<your-kernel-version>

Does anyone know how to install the 6.5.0-generic kernel on Ubuntu 24.04?

linuxcf, 2024-06-09:
> Does anyone know how to install the 6.5.0-generic kernel on Ubuntu 24.04?

I have read in other forums that after people switched back to kernel 6.5 (because previously their camera worked on 6.5), their camera did not work on 6.5 either!

The9Cat, 2024-06-10:
@rmasad I'm guessing that the problem with using 6.5.0 is incompatibility with the dkms packages in 24.04.  In other words, that kernel needs different versions of the modules and drivers than in the 24.04 repo.  If you want to make it work, I'd start with the same commits from the intel-ipu6 repo used by the Ubuntu packages from 22.04 in that list of steps from your previous post.

That said, this is strange.  I also have a Carbon X1 Gen 11 (same sensor device as in your dmesg output earlier in the post).  It mostly works for me.  Is yours also a Gen 11?  I wonder what's different here?

linuxcf, 2024-06-10:
> That said, this is strange. I also have a Carbon X1 Gen 11 (same sensor device as in your dmesg output earlier in the post). It mostly works for me. Is yours also a Gen 11? I wonder what's different here?

@The9Cat, the plot thickens.  My X1 Yoga Gen 8 should be identical to your X1 Carbon gen 11, except for the touch screen, which I enjoy.  What is your Carbon's CPU, if you don't mine?  Mine is listed as "Intel 13th Gen Core i7-1370P," the most powerful CPU offered for this box at the time of order.

The9Cat, 2024-06-10:
@linuxcf Yes, my X1C has the same CPU.  A Lenovo person on the Lenovo users' site made the same comment about X1 Yoga Gen8: it's the same chassis.

I also have to `systemctl restart v4l2-relayd.service`  before or after using the camera about 20% of the time.  Not every time.  I suspect that the driver still needs some timing work to support the diversity of different MIPI sensor devices.  But it mostly works for me.  

I'm not knowledgeable enough about the hardware and driver details to help fix this, alas.  Curious situation.  I do appreciate all of the work that the developers are doing and suspect that wrinkles will get ironed out.

The9Cat, 2024-06-10:
Just to add: I have not seen the 'test pattern' or 'snow' failure mode.  My failure mode, when it fails, is no video at all often with the camera in use indicator stuck on.

linuxcf, 2024-06-10:
> Just to add: I have not seen the 'test pattern' or 'snow' failure mode. My failure mode, when it fails, is no video at all often with the camera in use indicator stuck on.

@The9Cat, thank you for taking the time to discuss this problem here while this is not too much of a problem for you.  We have the same CPU and, presumably, the same camera.  Yet yours works, at least most of the time, while mine doesn't.  I think some program must be installed in my system.   In the interest of everybody monitoring this thread, I am attaching a snapshot of vlc "showing" video from my camera.  The "snow" on the bottom right is dynamic: the black dots move around randomly.  Skype shows the same, but like mirrored over a vertical mirror.  I, too, hope the developers iron out the wrinkles because this laptop I am using is supposed to be Ubuntu certified.
[image]

The9Cat, 2024-06-10:
Crazy.  This is from icamerasrc or some other application?  Have you tried the gUM test page or vlc?  Is the behavior the same?

linuxcf, 2024-06-10:
> Crazy. This is from icamerasrc or some other application? Have you tried the gUM test page or vlc? Is the behavior the same?

The above picture is a snapshot from vlc.  The gUM test page gives me the same video, declares a success(!!!!!!!!), but as for fps gives me this underneath the "live" video: FPS now: `NaN , entire capture: NaN`.  That is why I think I am either missing a module (I see no `libcamhal0` available for installation) or I have a wrong version of a module installed.  If, on synaptic (apt GUI), I pick Settings → Repostiories → Additional Drivers, it lets me pick "Continue using a manually installed driver," but the "Apply Changes" button is not selectable, so I cannot activate my choice, as per the attached snapshot. (When my camera was working, with quirks, in kernel 6.5 in 22.04LTS, the top option was marked and selectable.) So, the choice there (obviously referring to the MIPI camera), remains stuck at "Do not use this device."
[image]

[495 more replies]