intel/ipu6-drivers #229: Dell Latitude 7450 hm2172 report failed to check hwcfg: -517

xanthein, 2024-05-23:
Install Ubuntu 24.04 on Dell Latitude 7440 (Intel Core Ultra 5 135H). Upgrade kernel to 6.8.0-31-generic, clone ipu6-drivers (sha: 6fcc4c5) and build against the kernel. 
After system reboot, camera sensor report error `hm2172 i2v-HIMX272:00 failed to check hwcfg: -517`.

vicamo, 2024-05-27:
> Install Ubuntu 24.04 on Dell Latitude 7440 (Intel Core Ultra 5 135H). Upgrade kernel to 6.8.0-31-generic,

It takes 6.8.0-34 or newer to have basic IPU6 support on 6.8 kernel on Ubuntu.

> clone ipu6-drivers (sha: [6fcc4c5](https://github.com/intel/ipu6-drivers/commit/6fcc4c5955bbfd011aa3023d6c03b9d1faaa367b)) and build against the kernel. After system reboot, camera sensor report error `hm2172 i2v-HIMX272:00 failed to check hwcfg: -517`.

This is a separate issue for ipu6-drivers upstream repo. Fixed in Ubuntu kernel already.

arter97, 2024-05-27:
> 
> This is a separate issue for ipu6-drivers upstream repo. Fixed in Ubuntu kernel already.

May I know what this is? I'm running a custom kernel on my Ubuntu setup.

vicamo, 2024-05-27:
> > This is a separate issue for ipu6-drivers upstream repo. Fixed in Ubuntu kernel already.
> 
> May I know what this is? I'm running a custom kernel on my Ubuntu setup.

Everything Ubuntu kernel related are open, so sure, it's [0016-UBUNTU-SAUCE-don-t-fail-probing-at-hwcfg-checks.patch](https://git.launchpad.net/ubuntu/+source/ipu6-drivers/tree/debian/patches/0016-UBUNTU-SAUCE-don-t-fail-probing-at-hwcfg-checks.patch?h=applied/ubuntu/devel).

arter97, 2024-05-27:
Oh, so this was an optional check... Thanks!

hao-yao, 2024-05-28:
Return value -517 should be -EPROBE_DEFER and the module should be probed again later. The check is not optional because it blocks sensor driver probing before cio2-bridge or ipu-bridge creating fwnode, which will be used by V4L2 to bind sensors and other devices as IPU subdevice. After that, the check will pass and sensor driver loading should finally pass. If no, I think it is caused by missing of ipu-bridge or cio2-bridge is not built, or they don't include the sensor's HID. However [0003-media-ipu-bridge-Support-more-sensors.patch](https://github.com/intel/ipu6-drivers/blob/master/patch/v6.8/0003-media-ipu-bridge-Support-more-sensors.patch) has already added all sensors IPU6 supports.

@vicamo , please let me know if removing [0016-UBUNTU-SAUCE-don-t-fail-probing-at-hwcfg-checks.patch](https://git.launchpad.net/ubuntu/+source/ipu6-drivers/tree/debian/patches/0016-UBUNTU-SAUCE-don-t-fail-probing-at-hwcfg-checks.patch?h=applied/ubuntu/devel) can cause sensor driver not loaded properly.