intel/ipu6-drivers #235: Failed to probe camera sensor on Dell XPS 9315: probe of i2c-OVTI01A0:00 failed with error -22

vicamo, 2024-06-05:
Log: [dmesg.6.5.0-20240409-oem.failed-to-check-hwcfg.txt](https://github.com/user-attachments/files/15581712/dmesg.6.5.0-20240409-oem.failed-to-check-hwcfg.txt)
```
[    5.142286] ov01a10 i2c-OVTI01A0:00: can't get clock frequency
[    5.142291] ov01a10 i2c-OVTI01A0:00: failed to check hwcfg: -22
[    5.142436] ov01a10: probe of i2c-OVTI01A0:00 failed with error -22
```

dmidecode:
```
Handle 0x0001, DMI type 0, 26 bytes
BIOS Information
        Vendor: Dell Inc.
        Version: 1.12.0
        Release Date: 06/12/2023
        ...

Handle 0x0100, DMI type 1, 27 bytes
System Information
        Manufacturer: Dell Inc.
        Product Name: XPS 9315
        ...
```

hao-yao, 2024-06-05:
Duplicated to #232.

vicamo, 2024-06-05:
Have you checked the actual symptoms and cause for #232? They're different.

vicamo, 2024-06-05:
`ipu_isys_check_fwnode_graph` returns 0, and therefore `cio2_bridge_init` is never called, device properties never defined, and hwcfg check returns -22.

hao-yao, 2024-06-06:
You are right. But I still don't understand why because I remembered on XPS 9315 we added the connection of sensor, and it should filled clock-frequency. Could you help to dump the DSDT and attach here?
```bash
sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.aml
iasl -d dsdt.aml
```

vicamo, 2024-06-06:
> You are right. But I still don't understand why because I remembered on XPS 9315 we added the connection of sensor, and it should filled clock-frequency. Could you help to dump the DSDT and attach here?
> 
> ```shell
> sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.aml
> iasl -d dsdt.aml
> ```

Attached in bug description.