intel/ipu6-drivers #237: Internal webcam not detected on Samsung Galaxy Book 3 PRO running Ubuntu 23.04

Beor18, 2024-06-07:
Hello,

I would like to know if there is a possibility of making the webcam work in Samsung Galaxy Book 3 PRO with Ubuntu 23.04?

Thanks you!

gustavohfelixs, 2024-06-11:
I'm having the same issue in my galaxy book 4 pro, did you any answer?

gutopardini, 2024-07-04:
> I'm having the same issue in my galaxy book 4 pro, did you any answer?

I have the GB3U and I'm sorry to tell you but Samsung is working to ensure that you cannot use Linux

gustavohfelixs, 2024-07-04:
> > I'm having the same issue in my galaxy book 4 pro, did you any answer?
> 
> I have the GB3U and I'm sorry to tell you but Samsung is working to ensure that you cannot use Linux
> 

This is sad, btw thanks for answer

semyon-gordeev, 2024-08-13:
@Beor18 @gustavohfelixs @gutopardini try this https://github.com/intel/ipu6-drivers/issues/228#issuecomment-2286137675

Beor18, 2024-08-13:
Thanks you @semyon-gordeev it's worked

On ubuntu 22.04 samsung book 3 pro

gutopardini, 2024-08-14:
> @Beor18 @gustavohfelixs @gutopardini try this [#228 (comment)](https://github.com/intel/ipu6-drivers/issues/228#issuecomment-2286137675)

I can´t believe, it worked.
Thank you!

gustavohfelixs, 2024-08-26:
> @Beor18 @gustavohfelixs @gutopardini try this [#228 (comment)](https://github.com/intel/ipu6-drivers/issues/228#issuecomment-2286137675)

Thanks man it works, after follow the steps I thought my camera wasn't working because have no image, then i've notive that i was in my room with 0 light hahahah.

gutopardini, 2024-12-11:
> @Beor18 @gustavohfelixs @gutopardini try this [#228 (comment)](https://github.com/intel/ipu6-drivers/issues/228#issuecomment-2286137675)

Last week an IPU6 update arrived on Ubuntu and now it doesn't work

:(

Beor18, 2025-04-18:
Hi everyone,

I don't know what happened but now my webcam doesn't work.

I mean it detects but it goes black