intel/ipu6-drivers #238: MIPI Camera Drivers Compatibility for ASUS VivoBook Slate 13

rautyrauty, 2024-06-10:
The CPU of this laptop has MIPI camera control, and neither of the two cameras work on Linux. But they work on Windows.
What can I do to solve the problem? Are intel interested in the solution?
The dmesg logs look like in this issue: https://github.com/intel/ipu6-drivers/issues/228#issuecomment-2132039519

Tested on ubuntu with https://launchpad.net/ubuntu/+source/ipu6-drivers, sosreport: [sosreport-kubuntu-2024-06-08-fdavxlx.zip](https://github.com/user-attachments/files/15764215/sosreport-kubuntu-2024-06-08-fdavxlx.zip). Also tested on ALT linux with kernel 6.8.12.

ChenYuYeh, 2024-06-10:
Hi,

Same as #202 for HP and #237 for Samsung, we have observed that even those designs with IPU6 Camera but has no Ubuntu cert granted. As you may know that there are camera, lens, and circuit hardware dependency, it takes effort from PC vendor and Intel to enable the function together.  

Not sure Asus customer service will give you the response, but there is No other way to make it happen on your Asus device.

> The CPU of this laptop has MIPI camera control, and neither of the two cameras work on Linux. But they work on Windows.
> What can I do to solve the problem? Are intel interested in the solution?
> The [...]

rautyrauty, 2024-06-24:
I wrote to the ASUS technical support service asking for help to solve this problem.

rautyrauty, 2024-06-24:
@ChenYuYeh What kind of information do we want to get from ASUS to add support?

ChenYuYeh, 2024-06-24:
@rautyrauty 
Supposedly Asus can reply you whether they are seeking for obtaining the certificate by Canonical? If just be mute then it means there is no plan yet.

rautyrauty, 2024-06-25:
Well, ASUS replied that they were not going to support linux. Is there only one way now - reverse engineering?

rautyrauty, 2024-07-25:
Can't we get the information we need through reverse engineering of drivers? I can disassemble the laptop if necessary.

ChenYuYeh, 2024-07-29:
@rautyrauty  I don't think it is HW related. So far the integration should be involved with BIOS configuration (I have seen in some cases BIOS config is wrong but works in Windows but failed in Linux!), upstream kernel driver implementation, and few xml files required to be modified in userspace. PC vendor would have to drive to make the camera work both on Windows and Linux.

closed 2024-07-29 by hao-yao