intel/ipu6-drivers #252: ov2740 i2c-INT3474:01: failed to find sensor: -121

henriqueslaporta, 2024-07-10:
> System
```
Acer Swift SF314-71
12th Gen Intel® Core™ i7-12700H

Ubuntu 24.04 LTS
6.8.0-36-generic
```

> sudo dmesg -T
```
intel-ipu6 0000:00:05.0: Device 0x465d (rev: 0x2)
intel-ipu6 0000:00:05.0: physical base address 0x601e000000
intel-ipu6 0000:00:05.0: mapped as: 0x000000003b09940f
intel-ipu6 0000:00:05.0: IPU in secure mode
intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
intel-ipu6 0000:00:05.0: IPC reset done
intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
intel-ipu6 0000:00:05.0: FW version: 20230925
intel-ipu6 0000:00:05.0: Found supported sensor INT3474:01
intel-ipu6 0000:00:05.0: Connected 1 cameras
intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
ov2740 i2c-INT3474:01: failed to find sensor: -121
```
Full log: [dmesg_file.txt](https://github.com/user-attachments/files/16159061/dmesg_file.txt)

> v4l2-ctl --list-devices
```
ipu6 (pci:pci0000:00):
	/dev/media0
```

I already tried different solutions mentioned here #228 with no success.
I had this problem since older versions o the kernel and other versions of Ubuntu

andyyeh75, 2024-07-29:
Hi,

Same as https://github.com/intel/ipu6-drivers/issues/237 for Samsung, we have observed that even those designs with IPU6 Camera but has no Ubuntu cert granted. Hence the integration should be involved with BIOS configuration (I have seen in some cases BIOS config is wrong; but the camera still works in Windows. Suspects there are hacks in Windows driver), upstream kernel driver implementation, and few xml files required to be modified in user space. PC vendor would have to step in to make the camera work both on Windows and Linux.

Not sure Acer customer service will give you the response, but there is no easy way to obtain an installation package on your Acer device. Thanks.

Cyborgscode, 2024-09-22:
Vendor: Wortmann AG
Device: TERRAPAD 1263v2

Same Bug with Fedora 40:

Kernel 6.10.11 
libcamera-0.3.1-3
intel-ipu6 kernel module via akmods

```
# v4l2-ctl --all
Driver Info:
	Driver name      : isys
	Card type        : ipu6
	Bus info         : 
	Driver version   : 6.10.11
	Capabilities     : 0xa4a00001
		Video Capture
		Metadata Capture
		I/O MC
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x24a00001
		Video Capture
		Metadata Capture
		I/O MC
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Intel IPU6 ISYS Capture 0: ok)
Format Video Capture:
	Width/Height      : 1920/1080
	Pixel Format      : 'BG12' (12-bit Bayer BGBG/GRGR)
	Field             : None
	Bytes per Line    : 3840
	Size Image        : 4151040
	Colorspace        : Raw
	Transfer Function : Default (maps to None)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Format Metadata Capture:
	Sample Format   : 'MET8' (8-bit Generic Metadata)
	Buffer Size     : 9600
```
```

[   23.151136] i2c i2c-INT3474:02: deferred probe pending: ov2740: failed to check HW configuration
[   23.151148] i2c i2c-INT3474:01: deferred probe pending: ov2740: failed to check HW configuration
[ 1113.606464] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[ 1113.606787] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x0 mask 0x0
[ 1113.607353] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[ 1113.607870] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[ 1113.608323] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[ 1113.608835] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[ 1113.609322] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[ 1113.609846] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[ 1113.610315] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[ 1113.610834] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[ 1113.611317] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[ 1113.611837] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[ 1113.656989] intel-ipu6 0000:00:05.0: FW version: 20230925
[ 1113.658229] intel-ipu6 0000:00:05.0: Found supported sensor INT3474:01
[ 1113.658758] intel-ipu6 0000:00:05.0: Found supported sensor INT3474:02
[ 1113.658973] intel-ipu6 0000:00:05.0: Connected 2 cameras
[ 1113.660489] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[ 1113.670134] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[ 1113.739069] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[ 1113.739100] intel-ipu6 0000:00:05.0: IPU6-v3[465d] hardware version 5
**[ 1113.766134] intel_ipu6_isys: unknown parameter 'video_nr' ignored**
[ 1113.798585] ov2740 i2c-INT3474:02: error -EREMOTEIO: failed to find sensor
[ 1113.799688] ov2740 i2c-INT3474:02: probe with driver ov2740 failed with error -121
[ 1113.860529] ov2740 i2c-INT3474:01: error -EREMOTEIO: failed to find sensor
[ 1113.862717] ov2740 i2c-INT3474:01: probe with driver ov2740 failed with error -121
[ 1144.441286] ov2740 i2c-INT3474:02: error -EREMOTEIO: failed to find sensor
[ 1144.443319] ov2740 i2c-INT3474:02: probe with driver ov2740 failed with error -121
[ 1144.503227] ov2740 i2c-INT3474:01: error -EREMOTEIO: failed to find sensor
[ 1144.505116] ov2740 i2c-INT3474:01: probe with driver ov2740 failed with error -121

```

Cyborgscode, 2024-09-22:
Debug so far i could get without recompiling the kernel with debug messages:

ov2740 driver calls stack as far as it matters for the issue:

Line 628 ov2740_identify_module(ov2740) 

Line 637 ov2740_read_reg(ov2740, OV2740_REG_CHIP_ID, 3, &val);

Line 579 ret=i2c_transfer(client->adapter, msgs, ARRAY_SIZE(msgs));

ret must be != ARRAY_SIZE(MSGS), because otherwise 0 or -EIO would be returned, but it is EREMOTE_IO ( -121 )

(this is a stupid check anyway, because you can get any return value by accident that you expect and than your code continues and will crash sooner or later. A defined return value with 0 => worked and all other values mean fail + a errorcode attribute set in the structure, would be much more fault tolerant, but i think you can't expect good code from c coder at all ;) )

So, the i2c sys could not talk to the chip at all, otherwise the "ret" would > 0 and still != ARRAY_SIZE(msgs), in which case, it would have received more/less msgs than it expected to receive.

If one can give advises to debug the TRACE event nightmare that comes next, feel free.

henriqueslaporta, 2024-10-11:
Just to keep the issue updated. I tried to upgrade the kernel from 6.8 to 6.11 and also made a fresh install of Ubuntu 24.10 (kernel 6.11).

Every time still the same issue and same error message.

alaincou, 2024-11-11:
Hello,
I have the same problem on HP Spectre 14-eu0xxx (metor lake i7) and the ov08x40 driver under debian SID with kernel 6.11 with the patch here.
Already I notice at boot that the module tries to load too early and gives the errors:
ov08x40 i2c-OVTI08F4:00: failed to check hwcfg: -517
ov08x40 i2c-OVTI08F4:00: failed to check hwcfg: -517
...
Then I modified a single line in the code in the power_on function in ov08x40.c
- usleep_range(5000, 5500);
+ msleep(250);
And there by miracle, the module works, but with other problems to use correctly on linux.

I think there are connection issues between  ipu6,  i2c modules and cameras...

Ah, just I would also like to add the og0va1b IR camera which is on many laptops, but I don't have the technical specifications to do so...

moozhub, 2024-11-23:
This is what I'm currently getting on Fedora 41, qcam shows no available sensors to use. I forced the usage of the ov08x40 kernel module and blacklisted ov2740 since it seemed to get automatically loaded for some reason, but didn't make any difference. 

This is on a Lenovo X1C G12

v4l2-ctl --list-devices

```
ipu6 ():
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8
	/dev/video9
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video17
	/dev/video18
	/dev/video19
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/video24
	/dev/video25
	/dev/video26
	/dev/video27
	/dev/video28
	/dev/video29
	/dev/video30
	/dev/video31
	/dev/video32
	/dev/video33
	/dev/video34
	/dev/video35
	/dev/video36
	/dev/video37
	/dev/video38
	/dev/video39
	/dev/video40
	/dev/video41
	/dev/video42
	/dev/video43
	/dev/video44
	/dev/video45
	/dev/video46
	/dev/video47

ipu6 (PCI:0000:00:05.0):
	/dev/media0
```

dmesg

```[    6.015026] int3472-discrete INT3472:0b: GPIO type 0x12 unknown; the sensor may not work
[    6.015418] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[    6.018610] int3472-discrete INT3472:0b: GPIO type 0x12 unknown; the sensor may not work
[    6.018639] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[    6.019412] int3472-discrete INT3472:0c: cannot find GPIO chip INTC1007:00, deferring
[    6.028520] int3472-discrete INT3472:0b: GPIO type 0x12 unknown; the sensor may not work
[    6.028544] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[... 67 lines cut ...]
[    6.518770] int3472-discrete INT3472:0b: GPIO type 0x12 unknown; the sensor may not work
[    6.519095] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[    6.521853] int3472-discrete INT3472:0c: cannot find GPIO chip INTC1007:00, deferring
[    6.523991] int3472-discrete INT3472:0b: GPIO type 0x12 unknown; the sensor may not work
[    6.524378] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[    6.542964] int3472-discrete INT3472:0c: cannot find GPIO chip INTC1007:00, deferring
[    6.544309] int3472-discrete INT3472:0b: GPIO type 0x12 unknown; the sensor may not work
[    6.544599] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[    6.548337] int3472-discrete INT3472:0c: cannot find GPIO chip INTC1007:00, deferring
[    6.549570] int3472-discrete INT3472:0b: GPIO type 0x12 unknown; the sensor may not work
[    6.549856] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[    6.865025] int3472-discrete INT3472:0c: cannot find GPIO chip INTC1007:00, deferring
[    6.866654] int3472-discrete INT3472:0b: GPIO type 0x12 unknown; the sensor may not work
[    6.866923] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[    7.055045] int3472-discrete INT3472:0c: cannot find GPIO chip INTC1007:00, deferring
[    7.055768] int3472-discrete INT3472:0b: GPIO type 0x12 unknown; the sensor may not work
[    7.055797] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[   14.427982] int3472-discrete INT3472:0c: cannot find GPIO chip INTC1007:00, deferring
[   14.428809] int3472-discrete INT3472:0b: GPIO type 0x12 unknown; the sensor may not work
[   14.428841] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[   17.533445] int3472-discrete INT3472:0c: cannot find GPIO chip INTC1007:00, deferring
[   17.534376] int3472-discrete INT3472:0b: GPIO type 0x12 unknown; the sensor may not work
[   17.534448] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[   17.536525] platform INT3472:0c: deferred probe pending: int3472-discrete: Failed to get GPIO
[   17.536528] platform INT3472:0b: deferred probe pending: int3472-discrete: Failed to get GPIO```

Cyborgscode, 2024-11-23:
News from the war front:    Terrapad 1262v3 ov2740 sensor

the chips do not propagate theire pins correctly AND the manufactorer had used some hardwiring tricks, so both sensors share the wires of only one sensor on the board for powering and reseting. Sensor#2 does not react to it's own power/reset lines. 

If initialized correctly, the sensors react to startup, stop and reset , which i have seen with my own eyes, so there is hope to end this soon, as one of Fedora's Best gpio specialists is on it on my Pad to get us all a fixed driver and upstream the hack, no other term is correct here, into the kernel. I will try to get a detection algorithm from him, so there is a testtool for all the others, which then can send theire manufacturer id to the kernel team to add it to the necessary database.

Cyborgscode, 2024-11-25:
See more details here: 

https://github.com/intel/ipu6-drivers/issues/292#issuecomment-2497614331

alaincou, 2024-11-25:
I don't know if it's useful but I'll allow myself to make my remarks (I'm French I use a translator).
So I'm on HP Spectre 14-eu0xxx
I notice that /sys/bus/acpi/devices lists devices that do not exist on my laptop such as OVTI01AS or OVTI13B1.
But /sys/bus/i2c/devices seems more correct to me and I have 3 devices linked to camera.
OVTI00AB which is the IR Camera, OG0VA or OGVA1B without Linux device
OVTI08F4 which is the 9MP Camera, og08x40 with Linux device but which does not work or not well.
TXNW3643 which is the Led Torch for the IR Camera which is lm3643 that we find in old versions for Android. According to the Windows .inf it is ACPI\TXNW3643 or ACPI\INT347D

I tested by making a fake diver OVTI00AB based on og01a1b.c and the driver loads well, without working, of course, because I can't find the details of the sensor.

I also notice on my laptop that when I only put the OVTI08F4 module, so without the other modules at startup, v4l2-ctl -all gives me a resolution of 640x480 which is impossible with this sensor but which is the default value of OVTI00AB.

However, the image I got by following the https://github.com/aigilea/hp_spectre_x360_14_eu0xxx is not good and it may be the IR image and not the Camera that I got?

Which makes me also think that the 2 cameras are on the same circuit.

Kind regards, let me know if I can help

debackerl, 2025-02-26:
@Cyborgscode , I can try your patch if it helps. I got an Acer Swift 5 (2022) SF514-56T with an ov2740.

I posted my dmesg here: https://github.com/intel/ipu6-drivers/issues/292#issuecomment-2684263970

Just need to know the few command lines to match and recompile the kernel, I can use the distro of your choice (I'm a system dev, but not kernel dev).

Cyborgscode, 2025-02-26:
The Patch for the TerraPad may not work on your device, because the TerraPad devs have made some strange decisions on how to hardwire the sensors together. The Best you can do is to contact @jwrdegoede and let him on your device to check it.
with the knowledge he received from the TerraPad, it should be done in a few minutes.

The "Patches" are already in the 6.13 kernel branch, but not yet approved afaik. Also: Suspend und Resume are a problem, as those sensors, to my current knowledge, do not get reinited proper :(